espressif/esp-hosted #492: [ESP32-C6][linux] Can ESP32 and Linux have network communication

zephyr-zhu, 2024-09-18:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

I have a usage scenario now, considering waiting for NG or using FG. The host side runs Linux connected to ESP32 through SDIO and runs in softAP mode. At the same time, there is a need for communication between the host side and ESP32 through a network interface. Can this be met?
I don't know if ESP32 can obtain data packets at the application layer through the IP address of the gateway?

### Use cases

no case

mantriyogesh, 2024-09-18:
> I don't know if ESP32 can obtain data packets at the application layer through the IP address of the gateway

You meant dual IP at esp and host. which is not yet supported.

mantriyogesh, 2024-09-18:
sorry, I did not understand clearly.

If you are setting up softap and host has ethap0 set up, you can run dhcp server on host on ethap0 network interface.
Now your esp act as softap, but the network interface is exposed at host. if you assign 192.16.43.1 on your host and allow station like your mobile or laptop to connect to esp softap, once they are connected, they would request dhcp request. it would be transparently received at host.

basically, your mobile(station) and host (fake softap) would communicate on network interface.
imp thing to remember, esp doesn't have network interface. it only has layer 2. transparent to host.

Is this what you wanted?
If yes, if you wish you can try to do 'start_softap' using python in FG.
python app : https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_path_python.gif
[Python demo](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/python_demo.md)

zephyr-zhu, 2024-09-18:
> sorry, I did not understand clearly.对不起，我没有明白。
> 
> If you are setting up softap and host has ethap0 set up, you can run dhcp server on host on ethap0 network interface.如果要设置 softap 并且 host 设置了 ethap0，则可以在 ethap0 网络接口上的主机上运行 dhcp 服务器。 Now your esp act as softap, but the network interface is exposed at host. if you assign 192.16.43.1 on your host and allow station like your mobile or laptop to connect to esp softap, once they are connected, they would request dhcp request. it would be transparently received at host.现在，您的 esp 充当 softap，但网络接口在 host 上公开。如果您在主机上分配 192.16.43.1 并允许您的手机或笔记本电脑等工作站连接到 ESP SoftAP，则一旦它们连接，它们将请求 DHCP 请求。它将在 Host 中透明地接收。
> 
> basically, your mobile(station) and host (fake softap) would communicate on network interface.基本上，您的 mobile（station）和主机（假 SoftAP）将在网络接口上通信。 imp thing to remember, esp doesn't have network interface. it only has layer 2. transparent to host.请记住，ESP 没有网络接口。它只有第 2 层。对主机透明。
> 
> Is this what you wanted?这是你想要的吗？ If yes, if you wish you can try to do 'start_softap' using python in FG.如果是，如果您愿意，您可以尝试在 FG 中使用 python 进行 'start_softap。 python app : https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_path_python.gifPython 应用程序 ： https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_path_python.gif [Python demo Python 演示](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/python_demo.md)

Sorry, I may not have described it very clearly. Can I understand it this way.

Based on FG or NG(in the future), in softAP mode.

1. Host (Linux) can be assigned an IP address,  such as 192.168.43.1.

2. ESP32 does not have any network interface for communication and can only communicate with the host (Linux) through SDIO protocol.

3. My mobile (station) can be assigned an IP address, such as 192.168.43.2

4. My mobile and host(linux) can communicate through a network interface.

mantriyogesh, 2024-09-18:
all 1 to 4 are correct.

in (2) , there is already header for communication in between esp and host. you can use ESP_PRIV_IF with TLV (Tag - length- value) format to communicate (endineness need to handle on your own).

example: 
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_driver/esp32/main.c#L495

which ends at : https://github.com/espressif/esp-hosted/blob/ce00ca53189ffd52455dd5426d54f9be6fae1871/esp_hosted_fg/host/linux/host_driver/esp32/main.c#L388

at slave:

https://github.com/espressif/esp-hosted/blob/ce00ca53189ffd52455dd5426d54f9be6fae1871/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L128

For more details of header used while communicating in between slave and host:
https://github.com/espressif/esp-hosted/blob/ce00ca53189ffd52455dd5426d54f9be6fae1871/esp_hosted_fg/common/include/adapter.h#L23-L50

zephyr-zhu, 2024-09-18:
Thank you very much for your answer. I understand now. By the way, When can NG's softap mode be released?

closed 2024-09-18 by zephyr-zhu