espressif/esp-hosted #495: esp_hosted_ng - esp32c3 - RSSI/level/signal strengh in WiFi scans is always 0

caipiblack, 2024-09-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

When I execute a WiFi scan, I am supposed to see correct signal level

### Actual behavior (suspected bug)

The RSSI/signal level/signal strengh is always seen to 0.

```
root@board:~# wpa_cli -i espsta0 scan_r
bssid / frequency / signal level / flags / ssid
60:38:e0:cd:92:69	2447	0	[WPA2-PSK-CCMP][ESS][UTF-8]	Linksys-DEA9
50:6f:0c:93:ea:90	2412	0	[WPA2-PSK-CCMP][WPS][ESS]	Livebox-EA90
50:6f:0c:93:ea:91	2412	0	[WPA2-PSK-CCMP][ESS]	Livebox-EA90_wifi_invite

root@board:~# wpa_cli scan
Selected interface 'esps[ 1570.865503] esp_cfg80211_scan
ta0'
OK
root@board:~# wpa_cli scan_results
Selected interface 'espsta0'
bssid / frequency / signal level / flags / ssid
60:38:e0:cd:92:69	2447	0	[WPA2-PSK-CCMP][ESS][UTF-8]	Linksys-DEA9
50:6f:0c:93:ea:90	2412	0	[WPA2-PSK-CCMP][WPS][ESS]	Livebox-EA90
50:6f:0c:93:ea:91	2412	0	[WPA2-PSK-CCMP][ESS]	Livebox-EA90_wifi_invite
70:fc:8f:ef:62:60	2437	0	[WPA2-PSK-CCMP][WPS][ESS]	Freebox-51C2B0_5GHz
50:6f:0c:94:64:80	2462	0	[WPA2-PSK-CCMP][WPS][ESS]	Livebox-6480

root@board:~# iwlist scan
lo        Interf[ 1638.323724] esp_cfg80211_scan
ace doesn't support scanning.

sit0      Interface doesn't support scanning.

eth0      Interface doesn't support scanning.

espsta0   Scan completed :
          Cell 01 - Address: 50:6F:0C:93:EA:90
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=0 dBm  
                    Encryption key:on
                    ESSID:"Livebox-EA90"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=000000d9f7d7b48b
                    Extra: Last beacon: 1624ms ago
                    IE: Unknown: 000C4C697665626F782D45413930
                    IE: Unknown: 010882848B962430486C
                    IE: Unknown: 030101
                    IE: Unknown: 0706465220010D14
                    IE: Unknown: 200100
                    IE: Unknown: 23021100
                    IE: Unknown: 2A0100
                    IE: Unknown: 32040C121860
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 0B050100180000
                    IE: Unknown: 42020000
                    IE: Unknown: 4605334F010000
                    IE: Unknown: 2D1AAC091BFFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D1601080400000000000000000000000000000000000000
                    IE: Unknown: 7F080400080000000040
                    IE: Unknown: DD9A0050F204104A0001101044000102103B00010310470010353036463043F933F54139303530364610210008536167656D636F6D10230014536167656D636F6D46617374353635365F4F46521024000C53475F4C42465F312E322E31104200094C4B323132383146461054000800060050F20400011011000D4C697665626F78204669627265100800020006103C0001011049000600372A000120
                    IE: Unknown: DD090010180201001C0000
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
          Cell 02 - Address: 50:6F:0C:93:EA:91
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=0 dBm  
                    Encryption key:on
                    ESSID:"Livebox-EA90_wifi_invite"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=000000d9f7d7cc3d
                    Extra: Last beacon: 1624ms ago
                    IE: Unknown: 00184C697665626F782D454139305F776966695F696E76697465
                    IE: Unknown: 010882848B962430486C
                    IE: Unknown: 030101
                    IE: Unknown: 0706465220010D14
                    IE: Unknown: 200100
                    IE: Unknown: 23021100
                    IE: Unknown: 2A0100
                    IE: Unknown: 32040C121860
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 0B050000180000
                    IE: Unknown: 42020000
                    IE: Unknown: 2D1AAC091BFFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D1601080400000000000000000000000000000000000000
                    IE: Unknown: 7F080400080000000040
                    IE: Unknown: DD090010180200001C0000
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
          Cell 03 - Address: 60:38:E0:CD:92:69
                    Channel:8
                    Frequency:2.447 GHz (Channel 8)
                    Quality=70/70  Signal level=0 dBm  
                    Encryption key:on
                    ESSID:"Linksys-DEA9"
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=000002d86b1448ea
                    Extra: Last beacon: 1752ms ago
                    IE: Unknown: 000C4C696E6B7379732D44454139
                    IE: Unknown: 01088C129824B048606C
                    IE: Unknown: 030108
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 7F080400000001000140
                    IE: Unknown: DD180050F2020101810003A4000027A4000042435E0062322F00
          Cell 04 - Address: 70:97:41:C1:6C:75
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=0 dBm  
                    Encryption key:on
                    ESSID:"Livebox-6C75"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=000000d9f7c8ef11
                    Extra: Last beacon: 1632ms ago
                    IE: Unknown: 000C4C697665626F782D36433735
                    IE: Unknown: 010882848B962430486C
                    IE: Unknown: 030101
                    IE: Unknown: 0706465220010D14
                    IE: Unknown: 200100
                    IE: Unknown: 23021100
                    IE: Unknown: 2A0100
                    IE: Unknown: 32040C121860
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 0B0501000F0000
                    IE: Unknown: 42020000
                    IE: Unknown: 4605334F010000
                    IE: Unknown: 2D1AAC091BFFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D1601080400000000000000000000000000000000000000
                    IE: Unknown: 7F080400080000000040
                    IE: Unknown: DD990050F204104A0001101044000102103B00010310470010373039373431F331F64337353730393710210008417263616479616E10230013417263616479616E4C4246494252455F4F46521024000C41525F4C42465F312E322E32104200094A41323131373946461054000800060050F20400011011000D4C697665626F78204669627265100800020006103C0001011049000600372A000120
                    IE: Unknown: DD090010180201001C0000
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00

wpan0     Interface doesn't support scanning.
```

### Steps to reproduce the behavior

We are using the release 1.0.2, (ESP-Hosted-NG_release_v1.0.2.bin)

### Project release version

1.0.2

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

Linux

### Operating system version

Yocto - Hardknott

### Shell

Bash

mantriyogesh, 2024-09-23:
Fixed in c13c46a ?

caipiblack, 2024-09-23:
> Fixed in [c13c46a](https://github.com/espressif/esp-hosted/commit/c13c46a9329b539b09c22738ff758901a06923ba) ?

I don't know, we are using releases versions only.

If it's fixed, then are you going to provide a new release version soon ?

mantriyogesh, 2024-09-23:
Please test with master, at both sides.

When the new release will happen, it will be included.

caipiblack, 2024-09-23:
Currently I am not able to run the built firmware (impossible to connect to an AP), but I didn't change the driver yet!

This is what I do:

# Create the firmware using the master branch

1. Clone the repository:
```
git clone https://github.com/espressif/esp-hosted.git
```
2. Generate the firmware ([documentation](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng/esp/esp_driver)):
```
cd esp-hosted/esp_hosted_ng/esp/esp_driver
cmake .
cd esp-idf
. ./export.sh
cd ../network_adapter
idf.py set-target esp32c3
idf.py build
```
3. Create the binary file:
```
esptool.py --chip esp32c3 merge_bin -o ESP-Hosted-NG_dirty_v1.0.3.bin --flash_mode dio --flash_size 4MB 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0xd000 build/ota_data_initial.bin 0x10000 build/network_adapter.bin
```

Previously I was using the firmwares from `esp32c3/spi_only`, so is it building this firmware when we build with idf.py build ? 

I am going to update the kernel module now

caipiblack, 2024-09-23:
Nice, same issue as here now: https://github.com/espressif/esp-hosted/issues/426

mantriyogesh, 2024-09-23:
Can you please attach the log from esp as well? 

What is your Linux based system, also please attach the photo of your setup

caipiblack, 2024-09-23:
> Can you please attach the log from esp as well?
> 
> What is your Linux based system, also please attach the photo of your setup

Thanks for your help,

- We are running on a Yocto (Hardknott) board
- This is a custom IOT board using a RZG2UL + ESP32C3 for WiFi
- The esp32-c3 is connected using SPI
- It was working good with the last release (1.0.2) except the RSSI level that was always returned to "0 Dbm" in the scans. And we need to get is returning correct values for PCBA tests.

Currently I am not able to have traces on my esp32, I think this is firmware related because with old FW (pre-built in the release) it was OK.

Did we have to configure something in menuconfig ? 

For example:
```
[ ] Debug Serial driver data path
[ ] Debug Wlan driver data path
[ ] Debug Bluetooth driver data path
```

caipiblack, 2024-09-23:
Also I see something strange. With the most recent version (https://github.com/espressif/esp-hosted/commit/9f809fff86ce19cea79011df7c2de92524d56a36) I see that the driver is not able to stop:
- The command `rmmod {driver name}` is blocking infinitely

In our usecase it's a problem, because the "reset" pin of the ESP32 is "taken" by the driver.
We need to keep the possibility to control it from userspace (to update/flash the ESP32 using UART).

Note: Currently I don't know if the driver refuse to stop due to a bug or if it's a new feature that comes with recent commits.

caipiblack, 2024-09-23:
I just realized that with the recent changes of the driver, the interface `espsta0` become `wlan0` so I had to change it in the `wpa_supplicant` service!

So at this point, i can connect to access-point!

```
root@board:~# dmesg | grep esp
[    5.624660] esp_hosted_ng: loading out-of-tree module taints kernel.
[    5.877084] esp_hosted_ng: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
[    7.326568] esp_hosted_ng: process_esp_bootup_event: Received ESP bootup event
[    7.334007] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 3
[    7.341364] esp_hosted_ng: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[    7.349618] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 2
[    7.356343] esp_hosted_ng: adjust_spi_clock: ESP Reconfigure SPI CLK to 30 MHz
[    7.363590] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 0
[    7.370856] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 1
[    7.377648] esp_hosted_ng: process_fw_data: ESP chipset's last reset cause:
[    7.384670] esp_hosted_ng: print_reset_reason: POWERON_RESET
[    7.390356] esp_hosted_ng: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.0
[    7.403637] esp_hosted_ng: esp_reg_notifier: Driver init is ongoing
[    7.675208] esp_hosted_ng: init_bt: ESP Bluetooth init
[    7.680937] esp_hosted_ng: print_capabilities: Capabilities: 0xe8. Features supported are:
[    7.689858] esp_hosted_ng: print_capabilities: 	 * WLAN on SPI
[    7.695803] esp_hosted_ng: print_capabilities: 	 * BT/BLE
[    7.701203] esp_hosted_ng: print_capabilities: 	   - HCI over SPI
[    7.707286] esp_hosted_ng: print_capabilities: 	   - BLE only
[    7.838538] esp_hosted_ng: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=\x00\x00
[   13.138398] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface lo ip changed to  127.000.000.001
[   13.191811] esp_hosted_ng: cmd_set_ip_address: Invalid argument
[   14.956141] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface eth0 ip changed to  192.168.100.001
[   15.006509] esp_hosted_ng: cmd_set_ip_address: Invalid argument
[   46.083498] esp_hosted_ng: cmd_set_ip_address: Invalid argument
[   46.089596] esp_hosted_ng: esp_inetaddr_event: Interface Down: 0
[   57.735371] esp_hosted_ng: cmd_auth_request: Authentication request: 60:38:e0:cd:92:69 8 0 0 0
[   57.980747] esp_hosted_ng: cmd_assoc_request: Association request: 60:38:e0:cd:92:69 8 35
[   58.043154] esp_hosted_ng: process_assoc_event: Connection status: 0
[   58.208411] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.003.243
[   58.271931] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface lo ip changed to  127.000.000.001
[   58.316936] esp_hosted_ng: cmd_set_ip_address: Invalid argument
[   89.275583] esp_hosted_ng: cmd_set_ip_address: Invalid argument
[   89.282336] esp_hosted_ng: esp_inetaddr_event: Interface Down: 0

root@board:~# journalctl -u wpa_supplicant
Sep 23 14:26:55 board systemd[1]: Starting WPA supplicant...
Sep 23 14:26:55 board systemd[1]: Started WPA supplicant.
Sep 23 14:26:55 board wpa_supplicant[888]: Successfully initialized wpa_supplicant
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
Sep 23 14:26:55 board wpa_supplicant[888]: nl80211: kernel reports: Registration to specific type not supported
[... cut ...]
    inet6 2a01:cb00:879d:bd0:8265:99ff:fe85:2d8/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 86214sec preferred_lft 414sec
    inet6 fe80::8265:99ff:fe85:2d8/64 scope link 
       valid_lft forever preferred_lft forever
# ping 192.168.100.2
PING 192.168.100.2 (192.168.100.2): 56 data bytes

^C
--- 192.168.100.2 ping statistics ---
6 packets transmitted, 0 packets received, 100% packet loss

```

2. This is currently impossible to stop the driver from the userspace (as root), using `rmmod` the command is blocking.

mantriyogesh, 2024-09-23:
Old firmware and new host driver might behave strange , @caipiblack .

Please flash the same git commit both sides..

caipiblack, 2024-09-24:
Hello,

I have the host driver + ESP32C3 firmware using the same commit: https://github.com/espressif/esp-hosted/commit/9f809fff86ce19cea79011df7c2de92524d56a36

This was the most recent commit of yesterday.

Can someone try to reproduce to check if he see the same issues:

1. Impossible to stop the driver using `rmmod` (was possible in release 1.0.2)
2. Errors appear when we configure a second IP (the interface have DHCP IPv4, and you add a fixed ip) `ip addr add 192.168.100.1/24 dev wlan0`

kapilkedawat, 2024-09-24:
@caipiblack could you please share your kernel version number?

caipiblack, 2024-09-24:
> @caipiblack could you please share your kernel version number?

Hello,

```
Linux hubmz-rzg2ul 5.10.201-cip41-yocto-standard #1 SMP PREEMPT Sat Feb 27 02:21:18 UTC 2021 aarch64 GNU/Linux
````

This is `5.10.201-cip41` is it what you are looking for ?

About the version, note that in `esp_kernel_port.h` I had to remove somes lines to fix an error as said in a previous comment (https://github.com/espressif/esp-hosted/issues/495#issuecomment-2368158924):

```
#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 15, 0))
static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
{
	ether_addr_copy(dev->dev_addr, addr);
}
#endif
```

My version is less than 5.15.0, (5.10.201) but this function is defined in my side so I don't need this.

kapilkedawat, 2024-09-24:
@caipiblack thanks for the detail, we will try to check the behavior on nearest available kernel version with us. Compilation error should be resolved by https://github.com/espressif/esp-hosted/commit/382dc37deca815727bc61d829362297dd4accaab 

Also you should be able to unload the driver now, please let us know if that doesn't work. We will check IP issue further.

caipiblack, 2024-09-24:
Hello,

I take your commit for the ESP32C3 firmware + Host driver.

1. I am now able to stop the driver (like in 1.0.2, with rmmod) (Thanks!)
2. I still have issues with IP addresses
3. I see that you put a ifndef around `eth_hw_addr_set` in esp_kernel_ports.h, this is not enought to resolve the build issue. Note that with my kernel this function is defined in `include/linux/etherdevice.h` also I don't know if in C we can do "ifndef" with function names (I never do it and will learn something if possible).

etherdevice.h:
[image]

esp_kernel_ports.h:
[image]

Logs from dmesg, grepped with 'esp':

```
[    5.695347] esp_hosted_ng: loading out-of-tree module taints kernel.
[    5.932601] esp_hosted_ng: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
[    7.375873] esp_hosted_ng: process_esp_bootup_event: Received ESP bootup event
[    7.384162] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 3
[    7.391592] esp_hosted_ng: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[    7.400028] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 2
[    7.406811] esp_hosted_ng: adjust_spi_clock: ESP Reconfigure SPI CLK to 30 MHz
[    7.414121] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 0
[    7.421510] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 1
[    7.428364] esp_hosted_ng: process_fw_data: ESP chipset's last reset cause:
[    7.435433] esp_hosted_ng: print_reset_reason: POWERON_RESET
[    7.441205] esp_hosted_ng: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.0
[    7.453281] esp_hosted_ng: esp_reg_notifier: Driver init is ongoing
[    7.703366] esp_hosted_ng: init_bt: ESP Bluetooth init
[    7.709060] esp_hosted_ng: print_capabilities: Capabilities: 0xe8. Features supported are:
[    7.717957] esp_hosted_ng: print_capabilities: 	 * WLAN on SPI
[    7.723846] esp_hosted_ng: print_capabilities: 	 * BT/BLE
[    7.729244] esp_hosted_ng: print_capabilities: 	   - HCI over SPI
[    7.735337] esp_hosted_ng: print_capabilities: 	   - BLE only
[    7.918385] esp_hosted_ng: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=\x00\x00
[   12.424388] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface lo ip changed to  127.000.000.001
[   12.469216] esp_hosted_ng: cmd_set_ip_address: Invalid argument
[   13.903620] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface eth0 ip changed to  192.168.100.001
[   13.945486] esp_hosted_ng: cmd_set_ip_address: Invalid argument
[   14.701285] esp_hosted_ng: cmd_auth_request: Authentication request: 60:38:e0:cd:92:69 8 0 0 0
[   14.997467] esp_hosted_ng: cmd_assoc_request: Association request: 60:38:e0:cd:92:69 8 35
[   15.119346] esp_hosted_ng: process_assoc_event: Connection status: 0
[   15.415790] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.003.243
[   45.479829] esp_hosted_ng: cmd_set_ip_address: Invalid argument
[   45.485837] esp_hosted_ng: esp_inetaddr_event: Interface Down: 0
[   91.655447] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.100.001
```

But your response give me some ideas, here you can see what I change in the driver to make it working on my board:

```
diff --git a/esp_hosted_ng/host/include/esp_kernel_port.h b/esp_hosted_ng/host/include/esp_kernel_port.h
index 028daeb..5f0be21 100644
--- a/esp_hosted_ng/host/include/esp_kernel_port.h
+++ b/esp_hosted_ng/host/include/esp_kernel_port.h
@@ -217,12 +217,14 @@ static inline bool wireless_dev_current_bss_exists(struct wireless_dev *wdev)
 #define ZERO_LINK_ID
 #endif
 
+#if 0
 #ifndef eth_hw_addr_set
 static inline void eth_hw_addr_set(struct net_device *dev, const u8 *addr)
 {
 	ether_addr_copy(dev->dev_addr, addr);
 }
 #endif
+#endif
 
[... 45 lines cut ...]
And i don't really configure the parameters of the driver as the driver starts automatically on the first run.

So these parameters keep their defaults values:

```
module_param(resetpin, int, S_IRUSR | S_IWUSR | S_IRGRP | S_IROTH);
MODULE_PARM_DESC(resetpin, "Host's GPIO pin number which is connected to ESP32's EN to reset ESP32 device");

module_param(clockspeed, uint, S_IRUSR | S_IWUSR | S_IRGRP | S_IROTH);
MODULE_PARM_DESC(clockspeed, "Hosts clock speed in MHz");

module_param(raw_tp_mode, uint, S_IRUSR | S_IWUSR | S_IRGRP | S_IROTH);
MODULE_PARM_DESC(raw_tp_mode, "Mode choosed to test raw throughput");
```

In my case:
```
resetpin = 506; // Modified by me from my patch
clockspeed = 0; // Default value, so the driver use default SPI CLK value, maybe OK
raw_tp_mode = 0; // Default value, maybe OK
```

But if these parameters are not "set" when the driver is started, can it produce `Invalid argument` because driver thinks he is not "initialized" ?

This is just an idea, I don't know if the solution is around this.

kapilkedawat, 2024-09-24:
Yes, my bad, I'll correct the function later. It seems there are some missing checks in `esp_inetaddr_event`, which is causing it to process the IP for other interfaces as well. However, that shouldn't cause any issues, since the IP is only used for SDIO wakeup in the firmware.

You can skip the event processing using this patch(will be committed later):

```
diff --git a/esp_hosted_ng/host/esp_cfg80211.c b/esp_hosted_ng/host/esp_cfg80211.c
index 09380d4d..a2b9af9a 100644
--- a/esp_hosted_ng/host/esp_cfg80211.c
+++ b/esp_hosted_ng/host/esp_cfg80211.c
@@ -158,9 +158,18 @@ static int esp_inetaddr_event(struct notifier_block *nb,
 {
        struct in_ifaddr *ifa = data;
        struct net_device *netdev = ifa->ifa_dev ? ifa->ifa_dev->dev : NULL;
-       struct esp_wifi_device *priv = netdev_priv(netdev);
+       struct esp_wifi_device *priv;
+       struct esp_adapter *adapter = esp_get_adapter();
+       struct esp_wifi_device *esp_priv = adapter->priv[0];
+
+       if (!netdev)
+               return 0;
 
-       esp_verbose("------- IP event -------\n");
+       esp_info("------- IP event for interface %s -------\n", netdev->name);
+
+       priv = netdev_priv(netdev);
+       if (esp_priv != priv)
+               return 0;
 
        switch (event) {
 
@@ -174,7 +183,7 @@ static int esp_inetaddr_event(struct notifier_block *nb,
        case NETDEV_DOWN:
                if (priv && (priv->if_type == ESP_STA_IF)) {
                        cmd_set_ip_address(priv, 0);
-                       esp_info("Interface Down: %d\n", priv->if_type);
+                       esp_info("Interface %s Down: %d\n", netdev->name, priv->if_type);
                }
                break;
        }

```

Could you please share your system's route after you give two IPs, also does it work as expected when you add two IPs on Ethernet interface?

caipiblack, 2024-09-25:
Hello, your patch solved the invalid parameters errors!

```
[    5.487006] esp_hosted_ng: loading out-of-tree module taints kernel.
[    5.742435] esp_hosted_ng: spi_dev_init: ESP32 peripheral is registered to SPI bus [1],chip select [0], SPI Clock [10]
[    7.182184] esp_hosted_ng: process_esp_bootup_event: Received ESP bootup event
[    7.189572] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 3
[    7.196599] esp_hosted_ng: esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[    7.204832] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 2
[    7.211567] esp_hosted_ng: adjust_spi_clock: ESP Reconfigure SPI CLK to 30 MHz
[    7.218819] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 0
[    7.225548] esp_hosted_ng: process_event_esp_bootup: Bootup Event tag: 1
[    7.232281] esp_hosted_ng: process_fw_data: ESP chipset's last reset cause:
[    7.239281] esp_hosted_ng: print_reset_reason: POWERON_RESET
[    7.244939] esp_hosted_ng: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.0
[    7.256683] esp_hosted_ng: esp_reg_notifier: Driver init is ongoing
[    7.507384] esp_hosted_ng: init_bt: ESP Bluetooth init
[    7.513157] esp_hosted_ng: print_capabilities: Capabilities: 0xe8. Features supported are:
[    7.521932] esp_hosted_ng: print_capabilities: 	 * WLAN on SPI
[    7.527972] esp_hosted_ng: print_capabilities: 	 * BT/BLE
[    7.533377] esp_hosted_ng: print_capabilities: 	   - HCI over SPI
[    7.539527] esp_hosted_ng: print_capabilities: 	   - BLE only
[    9.143485] esp_hosted_ng: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=\x00\x00
[   13.356542] esp_hosted_ng: esp_inetaddr_event: ------- IP event for interface lo -------
[   15.492317] esp_hosted_ng: cmd_auth_request: Authentication request: 60:38:e0:cd:92:69 8 0 0 0
[   15.861018] esp_hosted_ng: cmd_assoc_request: Association request: 60:38:e0:cd:92:69 8 35
[   15.949929] esp_hosted_ng: process_assoc_event: Connection status: 0
[   16.300224] esp_hosted_ng: esp_inetaddr_event: ------- IP event for interface eth0 -------
[   16.418064] esp_hosted_ng: esp_inetaddr_event: ------- IP event for interface wlan0 -------
[   16.493508] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.003.243
[   47.111989] esp_hosted_ng: esp_inetaddr_event: ------- IP event for interface lo -------
```

For the IP address it's also working, i found out that the IP address was also set to the ETH interface.
This board does not have eth interface, but it is enabled in DTS, for unknown reason the interface appear as up/connected (very strange)

So at this point the main issues are solved!

There are still minor issues that I fix by my side:
- Are you going to push your recent changes that "ignore" the network confgiuration not for the driver ?
- Are you going to fix the issue with `eth_hw_addr_set` ? (not with ifndef) (I don't know how)

This is just to know if I wait a little to update the commits (after you fix these issues) or if I fix them by my side.

Also, is there a 1.0.3 release coming soon ? We always prefer to use "released versions" in our platform instead of "the commit of a day"

kapilkedawat, 2024-09-25:
@caipiblack thanks for the confirmation, changes will be pushed shortly. Yes, will plan a release in some time, but it may take a week or two.

caipiblack, 2024-09-25:
> @caipiblack thanks for the confirmation, changes will be pushed shortly. Yes, will plan a release in some time, but it may take a week or two.

If it just take one week or two, I think I'm going to wait for the release!

I keep the issue open if someone want to say something / to test something,

You can close if it's all good.

caipiblack, 2024-10-18:
> @caipiblack thanks for the confirmation, changes will be pushed shortly. Yes, will plan a release in some time, but it may take a week or two.

Hello, What is the status of the future new release ?

Next week we need to know if we are going to use "the commit of the day" (last commit of the master branch), or a release version

If you are not working on a release 1.0.3 then I'm going to use "the last commit", we don't really like to use "un-released versions" but if we have no choice..

closed 2024-11-20 by caipiblack