espressif/esp-hosted #496: Not able to compile the driver source for arm64

obulesuveggidi, 2024-09-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

should be compiled and create .ko

### Actual behavior (suspected bug)

compilation error

### Error logs or terminal output

```shell
mantra_banglore@mantra:~/Downloads/esp-hosted-master/esp_hosted_ng/host$ ./rpi_init.sh btuart
Configure Host BT UART with 4 pins, RX, TX, CTS, RTS
Building for sdio protocol
rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=arm64 CROSS_COMPILE=/opt/fsl-imx-xwayland/6.1-mickledore/sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux/aarch64-poky-linux- -C /opt/fsl-imx-xwayland/6.1-mickledore/sysroots/armv8a-poky-linux/lib/modules/6.1.36+g04b05c5527e9 M=/home/mantra_banglore/Downloads/imx8mp_docs/esp-hosted-master/esp_hosted_ng/host clean
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/opt/fsl-imx-xwayland/6.1-mickledore/sysroots/armv8a-poky-linux/lib/modules/6.1.36+g04b05c5527e9'
make[1]: *** No rule to make target 'clean'.  Stop.
make[1]: Leaving directory '/opt/fsl-imx-xwayland/6.1-mickledore/sysroots/armv8a-poky-linux/lib/modules/6.1.36+g04b05c5527e9'
make: *** [Makefile:61: clean] Error 2
insmod: ERROR: could not load module esp32_sdio.ko: No such file or directory
sudo: pinctrl: command not found
sudo: pinctrl: command not found
sudo: pinctrl: command not found
sudo: pinctrl: command not found
```

### Steps to reproduce the behavior

1. change the make command and compile

### Project release version

esp-hosted-master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu 20

### Shell

sh

mantriyogesh, 2024-09-23:
Please try:

https://github.com/espressif/esp-hosted/blob/9f809fff86ce19cea79011df7c2de92524d56a36/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L105-L109

obulesuveggidi, 2024-09-24:
The above lines are already available in "esp-hosted/esp_hosted_fg/host/linux/host_control/rpi_init.sh"

obulesuveggidi, 2024-09-24:
Our requirement is,
1. We interfaced ESP32-C3 with imx8mp through UART
2. We need to compile the driver for the same.
Kindly help us to compile the driver source and also the dts configuration

mantriyogesh, 2024-09-24:
> The above lines are already available in "esp-hosted/esp_hosted_fg/host/linux/host_control/rpi_init.sh"

it is in 'FG' and not in 'NG'. Anyway, it needs porting if you use non-Raspberry Pi SoC as Linux. See [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#adjust-spi-timings)

> We interfaced ESP32-C3 with imx8mp through UART

Uart driver is not yet available. We plan to addbut can't commit right now, when would it be available. right now, spi and sdio driver are available. if you only need bluetooth (and nothing else), uart driver is available. For Wi-Fi or control path, uart driver is not available for Linux SoC.

> Kindly help us to compile the driver source and also the dts configuration

If you face issues in SPI or SDIO driver building, we can help there.
dts config is something dependent upon the SoC. you might have to figure it out. if you are stuck , we can guide but it is more of 'help' than 'support'.