espressif/esp-hosted #497: [ESP32 FG mode, stm32mp131 host, sdio mode] WiFi transmission rate

renliangli, 2024-09-24:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

WiFi transmission speed can reach the expected level

### Actual behavior (suspected bug)

The actual transmission rate of WiFi downlink TCP is 1.18 Mbits/sec
[image]

[raw transport throughput](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md)
[image]

### Error logs or terminal output

```shell
root@sofar:/# dmesg | grep mmc
[    2.651651] mmci-pl18x 58007000.mmc: mmc0: PL180 manf 53 rev2 at 0x58007000 irq 45,0 (pio)
[    2.716092] mmc0: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[    2.732790] mmc0: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[    2.742361] mmc0: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[    2.759702] mmc0: new SDIO card at address 0001
root@sofar:/#
root@sofar:/#
root@sofar:/# cat /sys/kernel/debug/mmc0/ios
clock:          25000000 Hz
actual clock:   25000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    0 (legacy)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)
```

### Steps to reproduce the behavior

1、insmod esp32_sdio.ko
2、iperf3 -s -i 1 -p 2222
3、iperf3 -c 192.168.2.144 -p 2222 -i 1 -t 30 -P 5 -R

### Project release version

master

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

Linux version 5.15.67

### Shell

sh

mantriyogesh, 2024-09-24:
before running iperf, check issues in raw throughput and stabilise them.

Please send photo captured of your setup.

Also, please ensure that the [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md) is followed.

renliangli, 2024-09-24:
[raw transport throughput](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md)
[image]

[image]

We tested it on the PCB board, and SDIO has pull-up capabilities

mantriyogesh, 2024-09-24:
Something is wrong yet.
Also, did you test raw throughput in opposite direction?

setup photo please?

renliangli, 2024-09-24:
Host --> ESP
[image]

[image]

ESP32

The pull-up resistance value is 4.7K

renliangli, 2024-09-24:
Also, when will esp_hosted_ng release a version that supports SoftAP ?

mantriyogesh, 2024-09-24:
4.7K is not acceptable value, btw. check the porting guide. Try to stick with the values we use.
Minimal recommended is 10K, we use 51K. Expected on lines: CMD, DAT0,DAT1,DAT2,DAT3.

For ESP32-WROOM32, the [efuse_burning](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#conflicts-between-bootstrap-and-sdio-on-dat2) is mandatory.

From above log, host tx seems good (still there is chance to be higher, but at is fair enough).
however rx not.

Can you let us know what (blue and yellow) extra wires added beside esp32?

mantriyogesh, 2024-09-24:
[NG supports softAP](https://github.com/espressif/esp-hosted/blob/f86bfbc8e73827988d16060d75a70095b48eae32/esp_hosted_ng/host/Makefile#L16) already.

renliangli, 2024-09-24:
> [NG supports softAP](https://github.com/espressif/esp-hosted/blob/f86bfbc8e73827988d16060d75a70095b48eae32/esp_hosted_ng/host/Makefile#L16) already.

Does the latest NG code on the master branch already support SoftAp? When compiling ko, just add CONFIG_AP_SUPPORT，
Is there anything that needs to be modified on the ESP side ？

mantriyogesh, 2024-09-24:
Just add 
CONFIG_AP_SUPPORT = y 
at top.

Latest ng firmware to be used from master. It should already support softap.

renliangli, 2024-09-25:
> Just add CONFIG_AP_SUPPORT = y at top.
> 
> Latest ng firmware to be used from master. It should already support softap.

NG no longer supports softap+station ?

mantriyogesh, 2024-09-25:
NG doesn't yet support softap+station.
FG supports it.

renliangli, 2024-09-25:
> NG doesn't yet support softap+station. FG supports it.

How long does it take for NG support softap+station

mantriyogesh, 2024-09-25:
It is not currently planned. so can't commit.
If your use case is specific to station + softap, I would recommend using FG. it is supported there.

mantriyogesh, 2024-09-25:
1. Did the raw throughput test both sides worked fine in your case @renliangli ?
2. Did you manage to change the pull-ups?

renliangli, 2024-09-27:
> 1. Did the raw throughput test both sides worked fine in your case @renliangli ?
yeah, The raw throughput test both sides worked fine, Above 20Mb.

how to enable wifi atuo reconnect after disconnected in FG station wifi mode ?

mantriyogesh, 2024-09-27:
## Short answer
Customise disconnected: Add hook of `test_station_mode_connect()` [here](https://github.com/espressif/esp-hosted/blob/250e29294a740a45dd853115a25b62f1b2cfc51b/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L126)

## Long answer
1. Doc: [Control path understanding](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/contrl_path.md)
2. Doc: You would anyway arrive here through above link, just to point, [Event Handler APIs](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#2-control-path-events)
3. Doc: [C demo](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/c_demo.md) Code: [C demo](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg/host/linux/host_control/c_support) -> test.c I think you are already past this.
4. Code: Event callbacks registered [here](https://github.com/espressif/esp-hosted/blob/250e29294a740a45dd853115a25b62f1b2cfc51b/esp_hosted_fg/host/linux/host_control/c_support/test.c#L87)
5. Code: Event callbacks hooks: https://github.com/espressif/esp-hosted/blob/250e29294a740a45dd853115a25b62f1b2cfc51b/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L109-L147
6. Customise disconnected: Add hook of [test_station_mode_connect()](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L745C5-L745C30) [here](https://github.com/espressif/esp-hosted/blob/250e29294a740a45dd853115a25b62f1b2cfc51b/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L126)

> [!NOTE]
> 
> 1. Try to use **async** event handlers when using in event handlers.
> We faced some issues in past for 'sync' handlers when used in event handlers.
> Except event handler hooks, in general, preference should always to 'sync' APIs.
>
>  2. Diff between sync and async:
> In Async, you register response callback. In sync, you do not register response callback.

Let us know if you face any difficulty.

renliangli, 2024-09-29:
> ## Short answer
> Customise disconnected: Add hook of `test_station_mode_connect()` [...]

Your method is to listen for the disconnection message and then restart the connection on the host side.

What I want is for the ESP32 chip station to be able to reconnect to the previously set AP hotspot by itself when rescanning the hotspot.

mantriyogesh, 2024-09-29:
> What I want is for the ESP32 chip station to be able to reconnect to the previously set AP hotspot by itself when rescanning the hotspot.

Just to understand this clearly, so do you mean that one every reboot, esp should auto connect to the prior AP (or router) it was connected to.
This should happen even when host is **not** triggering connect explicitly.

If host triggered connect with new credentials, these new credentials should become default credentials.

Is this the complete context? Anything I missed?

renliangli, 2024-09-29:
> Just to understand this clearly, so do you mean that one every reboot, esp should auto connect to the prior AP (or router) it was connected to. This should happen even when host is **not** [...]

In addition to reboot, there is another scenario where the station disconnects when the signal is poor, and the ESP32 chip can automatically reconnect when the signal is restored

mantriyogesh, 2024-09-29:
The main intention is that host should be notified on disconnect as soon as possible.

May be we retry 1-2 times and then send the disconnect to host.

The main problem here is that host wouldn't even know that the network is disconnected.

This is a typical ownership issue. In hosted ideology, we intend to control slave from host and host should be **always** and **ASAP** consistent with esp state.
In general, host should auto-connect on receiving the disconnect. ESP cannot send disconnect notification to host and at the same time, retry connecting without host knowing about it. Host would also trigger to connect (basically both parties start to issue connect) at the same time as that of esp, then the state machines would go for toss in esp_wifi.

As workaround,
What we can do, is introduce some MACRO, AUTO_DISCONNECT_AT_SLAVE in slave_control.c to auto connect on disconnect, **by default disabled**. AUTO_DISCONNECT_AT_SLAVE_TIMEOUT and AUTO_DISCONNECT_AT_SLAVE_RETRY

When this macro is enabled and flashed, the slave would try to reconnect with above config, before sending the disconnect notification to host.

> [!NOTE]
> 
> Word of caution: Unless this disconnect is received at host, host still thinks that it is connected (which is or should be transient state, but still invalid state)

Please let me know if this covers the scenario?

renliangli, 2024-09-29:
right

mantriyogesh, 2024-09-29:
We will make this change and notify you, once fixed.

renliangli, 2024-10-15:
[image]
The Station connect SoftAp was successful, why didn't receive the STA-Connected event ?

mantriyogesh, 2024-10-15:
>  The Station connect SoftAp was successful, why didn't receive the STA-Connected event ?

This is little confusing, are your using ESP softap (softap mode) ? or using external AP (station mode) ?

WRT **station mode**
if you are triggering connect, it is going to send back the response (in both sync and async)

In sync way of connection, if the APIs is not returned till the time AP is either connected / failed to connect.
In async way of connection, your response callback would be called with appropriate response (connected or failed).

As in manual way of connect, you are getting the connection successful or failure, the event was not triggered.

The 'Connected' event is treated for non manual (auto disconnect->connect) case.

mantriyogesh, 2024-10-15:
You can check 
https://github.com/espressif/esp-hosted/blob/a4a5f00923b45f11d833a3631768f8f7ad1df868/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_control.c#L150-%3EL162

see the `WIFI_HOST_REQUEST_BIT`

renliangli, 2024-10-15:
> This is little confusing, are your using ESP softap (softap mode) ? or using external AP (station mode) ?
>
> WRT **station mode** if you are triggering connect, it is going to send back the response [...]

station mode. 
(auto disconnect->connect):  Is this supported now ?

renliangli, 2024-11-25:
[image]

When ethap0 is in up state, ethsta0 can ping the internet, but when ethap0 is down, ethsta0 can no longer ping the internet.

but why?

mantriyogesh, 2024-11-25:
Check the routing rules if need to be added?

mantriyogesh, 2024-11-25:
To cross-check bring up ethsta0 and ethap0. ping using: 
```
ping -I ethsta0 8.8.8.8
```

if it still doesn't work, let the ethap0 be up. but kill dnsmasq:
```
sudo killall dnsmasq
```

dnsmasq or dhcp might be controlling or conflicting with your `resolv.conf` or `systemd-resolved` service.

in any case, just easy way is to check if static ip works, without running any dhcp server like dnsmasq and try if pingable from both.

but essentially this issue looks to be DNS/ network manager issue.

renliangli, 2024-11-25:
[image]

The routing and DNS are both normal, but the gateway cannot ping

renliangli, 2024-11-25:
[image]

Everything was normal after ethap0 be up

mantriyogesh, 2024-11-25:
Okay. So it's no more issues?

renliangli, 2024-11-25:
> Okay. So it's no more issue?

My question is why the down of ethap0 will affect ethsta0.  They should be independent

mantriyogesh, 2024-11-25:
The softap and station are actually implemented independent of each other. So one might be down and other might be up.

So there could be two possible alternatives:
1. The network configuration and the network tools are fighting somehow to have incorrect effect (we can give pointers, but you might need to localise on this on your Linux environment)
2. There is bug just below the network interfaces

For (2) , can you please provide exact steps I can reproduce? Also please ensure that you are testing with purely static ip for both, to localise and provide the logs (strictly textual and as attachment)

if in case it works fine with static ip, you can add rest blocks, like DHCP server (ap) , DHCP client(sta),
Full network manager, routes etc.

We limit ourselves till network interface getting up. Beyond network interface getting up, networking tools differ very much from one platform to other.
There is no generic way to handle networking for all Linux platforms. But till static ip it should get to work together.

While doing static ip based test, ensure,

1. The ethsta0 and ethap0 interfaces are not hooked to network manager 
2. Existing if any wlan0 (native) is disabled
3. Routes are clean
4. Possibly, No ethernet or alternative connectivity present, which would confuse the packet
5. Dnsmasq , hostapd and wpa supplicants should be killed.

Once above is done, you can just assign the network ip manually.

Example assignment:
With assumptions, router provides ip in range 192.168.1.20-192.168.1.100,

And there is no other node with 192.168.1.55, 
```
ifconfig ethsta0 192.168.1.55 up
```

Similarly, killall dnsmasq, and hostapd
```
ifconfig ethap0 192.168.4.1 up
```
Connect station to ap and assign static ip again to 192.168.4.8 and netmask as 255.255.255.0

Down one and update other, ping test for active.

Please capture the terminal logs and 
Host dmesg from bootup
Esp side logs using idf.py or minicom

I can try the procedure to reproduce. If I can reproduce easily, I can fix it fast

renliangli, 2024-11-26:
<details><summary>terminal logs</summary>
<p>

root@sofar:~#
root@sofar:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 7E:0A:F2:0A:6D:95
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:56 Base address:0xa000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:224 errors:0 dropped:0 overruns:0 frame:0
          TX packets:224 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:16576 (16.1 KiB)  TX bytes:16576 (16.1 KiB)

usb0      Link encap:Ethernet  HWaddr EA:98:F9:51:4A:C6
          inet addr:169.254.212.35  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::5193:b6b6:e566:1813/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:116 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:7965 (7.7 KiB)  TX bytes:3230 (3.1 KiB)

root@sofar:~#
root@sofar:~#
root@sofar:~#
root@sofar:~# killall dnsmasq
root@sofar:~#
root@sofar:~# killall dnsmasq
killall: dnsmasq: no process killed
root@sofar:~#
root@sofar:~# ps | grep dhcp
  185 root     dhcpcd: [manager] [ip4] [ip6]
  327 root     grep dhcp
root@sofar:~#
root@sofar:~# killall dhcpcd
root@sofar:~#
root@sofar:~# killall dhcpcd
killall: dhcpcd: no process killed
root@sofar:~#
root@sofar:~#
root@sofar:~# cd /
root@sofar:/# ls
app        dev        lib        media      proc       sbin       user
bin        esp_utils  lib32      mnt        root       sys        usr
boot       etc        linuxrc    opt        run        tmp        var
root@sofar:/# ./esp_utils enable_wifi
Enable heartbeat with duration 120
Heartbeat operation successful
------ ESP-Hosted FW [FG-0.0.6.0.0] ------
enable feature [1]
mac address is e4:65:b8:50:34:74
ethsta0 interface down
MAC address e4:65:b8:50:34:74 set to ethsta0 interface
ethsta0 interface up
mac address is e4:65:b8:50:34:75
ethap0 interface down
MAC address e4:65:b8:50:34:75 set to ethap0 interface
ethap0 interface up
Feature config change successful

Requested operation complete
[... 672 lines cut ...]
I (144969) slave_ctrl: HB timer started for 120 sec

I (145030) wifi:mode : sta (e4:65:b8:50:34:74)
I (145031) wifi:enable tsf
I (145032) slave_ctrl: station mode set
I (145032) slave_ctrl: Get station mac address
I (145036) slave_ctrl: mac [e4:65:b8:50:34:74]
I (146741) wifi:new:<12,2>, old:<1,1>, ap:<255,255>, sta:<12,2>, prof:1
I (147225) wifi:state: init -> auth (b0)
I (147230) wifi:state: auth -> assoc (0)
I (147239) wifi:state: assoc -> run (10)
I (147298) wifi:connected with lrl-9527, aid = 3, channel 12, 40D, bssid = 5c:de:34:53:64:eb
I (147299) wifi:security: WPA2-PSK, phy: bgn, rssi: -30
I (147302) wifi:pm start, type: 1

I (147305) slave_ctrl: Manual Wi-Fi connected, no event raised
I (147311) slave_ctrl: connected to ap SSID:'lrl-9527', password:'13428935786'
I (147319) slave_ctrl: req_connect_ap_handler:655 Set resp to Success
I (147388) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (148325) wifi:<ba-add>idx:0 (ifx:0, 5c:de:34:53:64:eb), tid:0, ssn:2, winSize:64
I (159549) slave_ctrl: Stop Heatbeat
I (159550) slave_ctrl: Stopping HB timer

</p>
</details>

mantriyogesh, 2024-11-26:
Right, please let me try myself and get back on this.

XDjackieXD, 2025-01-24:
@renliangli have you figured out what caused the throughput issues? I have almost the same values with esp_hosted_ng (about 45-50Mbit/s from host to ESP and only 1.5Mbit/s from ESP to host)