espressif/esp-hosted #498: FG mode: OTA end failed

lwj850533527, 2024-09-25:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hi @mantriyogesh
FG mode, I only send "network_adapter.bin" to esp32, OTA begin and OTA write success
But OTA end failed

Do I need to send other files
[image]

mantriyogesh, 2024-09-25:
1. please avoid opening as feature request, if you treat this as bug.
This helps us in categorisation.

2. network_adapter.bin is the only file need to be sent. no other partitions are needed or supported in ota.

3. Is it possible to flash this image on your ESP32 directly and confirm it works fine?

4. What is the host (Linux/MCU, details) and transport medium (SPI/SDIO)?

5. Can you please include git commits used at both sides?

lwj850533527, 2024-09-27:
Hi @mantriyogesh
> 3. Is it possible to flash this image on your ESP32 directly and confirm it works fine?

I  flash this image on my ESP32 directly, and confirm it works fine
```
I (30) boot: ESP-IDF 8fceee7 2nd stage bootloader                               
I (30) boot: compile time Aug 26 2024 14:02:21                                  
I (30) boot: chip revision: v0.4                                                
I (33) boot.esp32c3: SPI Speed      : 80MHz                                     
I (38) boot.esp32c3: SPI Mode       : DIO                                       
I (43) boot.esp32c3: SPI Flash Size : 4MB                                       
I (47) boot: Enabling RNG early entropy source...                               
I (53) boot: Partition Table:                                                   
I (56) boot: ## Label            Usage          Type ST Offset   Length         
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000       
I (71) boot:  1 otadata          OTA data         01 00 0000d000 00002000       
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000       
I (86) boot:  3 factory          factory app      00 00 00010000 00100000       
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000       
I (101) boot:  5 ota_1            OTA app          00 11 00210000 00100000      
I (109) boot: End of partition table                                            
I (113) boot: Defaulting to factory image                                       
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=2b620h (177696) map                                                                            
I (154) esp_image: segment 1: paddr=0003b648 vaddr=3fc97000 size=033fch ( 13308) load                                                                           
I (157) esp_image: segment 2: paddr=0003ea4c vaddr=40380000 size=015cch (  5580) load                                                                           
I (161) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a7d5ch (687452) map                                                                            
I (278) esp_image: segment 4: paddr=000e7d84 vaddr=403815cc size=159b8h ( 88504) load                                                                           
I (295) esp_image: segment 5: paddr=000fd744 vaddr=50000000 size=00020h (    32) load                                                                           
I (303) boot: Loaded app from partition at offset 0x10000                       
I (303) boot: Disabling RNG early entropy source...                             
I (316) cpu_start: Unicore app                                                  
I (316) cpu_start: Pro cpu up.                                                  
I (325) cpu_start: Pro cpu start user code                                      
I (325) cpu_start: cpu freq: 160000000 Hz                                       
I (325) cpu_start: Application information:                                     
I (328) cpu_start: Project name:     network_adapter                            
I (333) cpu_start: App version:      FG-0.0.6.0.1                               
I (339) cpu_start: Compile time:     Sep 23 2024 17:19:05                       
I (345) cpu_start: ELF file SHA256:  5c922f0f1deb3f41...                        
I (351) cpu_start: ESP-IDF:          v5.1.3-dirty                               
I (356) cpu_start: Min chip rev:     v0.3                                       
I (361) cpu_start: Max chip rev:     v1.99                                      
I (366) cpu_start: Chip rev:         v0.4   
```

> 4. What is the host (Linux/MCU, details) and transport medium (SPI/SDIO)?

host is stm32,   transport medium is SPI

> 5. Can you please include git commits used at both sides?

commits is https://github.com/espressif/esp-hosted/commit/490d2c3ed03139650d82b09f079d6fc4b1c48e49
I update the code to the latest version, It also has the same issue

Do I need to modify the content or format of “network_adapter.bin”, after compilation completed

mantriyogesh, 2024-09-27:
stm32 as in the microprocessor (esp_hosted_fg/host/stm32) or Linux (esp_hosted_fg/host/linux)?

lwj850533527, 2024-09-27:
esp_hosted_fg/host/stm32

mantriyogesh, 2024-09-27:
There might be issue on your transport that the packets are being lost silently.

Also change, https://github.com/espressif/esp-hosted/blob/250e29294a740a45dd853115a25b62f1b2cfc51b/esp_hosted_fg/host/stm32/driver/transport/spi/spi_drv.c#L305-L338

such that 

```c
static void check_and_execute_spi_transaction(void)
{
	uint8_t * txbuff = NULL;
	uint8_t is_valid_tx_buf = 0;
	GPIO_PinState gpio_handshake = GPIO_PIN_RESET;
	GPIO_PinState gpio_rx_data_ready = GPIO_PIN_RESET;
/* move here */
xSemaphoreTake(mutex_spi_trans, portMAX_DELAY);

	/* handshake line SET -> slave ready for next transaction */
	gpio_handshake = HAL_GPIO_ReadPin(GPIO_HANDSHAKE_GPIO_Port,
			GPIO_HANDSHAKE_Pin);

	/* data ready line SET -> slave wants to send something */
	gpio_rx_data_ready = HAL_GPIO_ReadPin(GPIO_DATA_READY_GPIO_Port,
			GPIO_DATA_READY_Pin);

	if (gpio_handshake == GPIO_PIN_SET) {

		/* Get next tx buffer to be sent */
		txbuff = get_tx_buffer(&is_valid_tx_buf);

		if ( (gpio_rx_data_ready == GPIO_PIN_SET) ||
		     (is_valid_tx_buf) ) {

			/* Execute transaction only if EITHER holds true-
			 * a. A valid tx buffer to be transmitted towards slave
			 * b. Slave wants to send something (Rx for host)
			 */
			//xSemaphoreTake(mutex_spi_trans, portMAX_DELAY);
			spi_trans_func[hardware_type](txbuff);
			//xSemaphoreGive(mutex_spi_trans);
		}
	}

/* move here */
xSemaphoreGive(mutex_spi_trans);

}
```

mantriyogesh, 2024-09-27:
in general, this is old code, many fixes done for MCU only branch, at 

https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host

But you would need porting efforts to move to this branch. Many such issues like one above stated are fixed in that branch.

lwj850533527, 2024-09-30:
Hi @mantriyogesh
I use crc checksum to make sure the OTA file is right. Now OTA can run and restart.

```
E (334693) slave_ctrl: checkSum[0x73bb94f]                             
I (334733) esp_image: segment 0: paddr=00110020 vaddr=3c0b0020 size=2ac00h (175104) map                                                                         
I (334753) esp_image: segment 1: paddr=0013ac28 vaddr=3fc96600 size=031c4h ( 12740)                                                                             
I (334753) esp_image: segment 2: paddr=0013ddf4 vaddr=40380000 size=02224h (  8740)                                                                             
I (334763) esp_image: segment 3: paddr=00140020 vaddr=42000020 size=a23d4h (664532) map                                                                         
I (334853) esp_image: segment 4: paddr=001e23fc vaddr=40382224 size=141d8h ( 82392)                                                                             
I (334863) esp_image: segment 5: paddr=001f65dc vaddr=50000000 size=00020h (    32)                                                                             
I (334863) esp_image: segment 0: paddr=00110020 vaddr=3c0b0020 size=2ac00h (175104) map                                                                         
I (334903) esp_image: segment 1: paddr=0013ac28 vaddr=3fc96600 size=031c4h ( 12740)                                                                             
I (334903) esp_image: segment 2: paddr=0013ddf4 vaddr=40380000 size=02224h (  8740)                                                                             
I (334913) esp_image: segment 3: paddr=00140020 vaddr=42000020 size=a23d4h (664532) map                                                                         
I (335003) esp_image: segment 4: paddr=001e23fc vaddr=40382224 size=141d8h ( 82392)                                                                             
I (335013) esp_image: segment 5: paddr=001f65dc vaddr=50000000 size=00020h (    32)                                                                             
E (335063) slave_ctrl: **** OTA updated successful, ESP32 will reboot in 5 sec ****        

                                                                     
ESP-ROM:esp32c3-api1-20210207                                                   
Build:Feb  7 2021                                                               
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)                         
Saved PC:0x40048b82                                                             
SPIWP:0xee                                                                      
mode:DIO, clock div:1                                                           
load:0x3fcd5820,len:0x1704                                                      
load:0x403cc710,len:0x968                                                       
load:0x403ce710,len:0x2f68                                                      
entry 0x403cc710 
```

But atfer restart, It  has an errro.
failed to load RF calibration data
Do I need to do any other actions
```
 I (2089) NETWORK_ADAPTER: *********************************************************************                                                            
I (2089) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5 
I (2109) NETWORK_ADAPTER:                 Transport used :: SPI only                                                                                            
I (2119) NETWORK_ADAPTER: *********************************************************************                                                                 
I (2119) NETWORK_ADAPTER: Supported features are:                               
I (2129) NETWORK_ADAPTER: - WLAN over SPI                                       
I (2129) ESP_BT: - BT/BLE                                                       
I (2139) ESP_BT:    - HCI Over SPI                                              
I (2139) ESP_BT:    - BLE only                                                  
I (2139) NETWORK_ADAPTER: capabilities: 0xe8                                    
I (2159) BLE_INIT: BT controller compile version [963cad4]                      
I (2159) phy_init: phy_version 970,1856f88,May 10 2023,17:44:12                 
W (2169) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration                                                            
I (2179) BLE_INIT: Bluetooth MAC: 80:65:99:99:ce:56
I (2179) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 80:65:99:99:ce:56             
I (2189) SPI_DRIVER: Using SPI interface  
```

mantriyogesh, 2024-09-30:
1. So, now the OTA is working every time?
2. The warning,
```
W (2169) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration
```
is only on first boot after OTA, or consecutive reboot also?

lwj850533527, 2024-09-30:
> 1. So, now the OTA is working every time?

Yes, the OTA is working every time

> 2. The warning,
> 
> ```
> W (2169) phy_init: failed to load RF calibration data (0xffffffff), falling back to full calibration
> ```
> 
> is only on first boot after OTA, or consecutive reboot also?

It is only on first boot after OTA, after reboot the warning disappear

mantriyogesh, 2024-10-02:
Possibly it is related to , https://github.com/espressif/esp-idf/issues/6462

Can you confirm from your side?
I might pick this up, but may take some time.

lwj850533527, 2024-10-09:
Hi @mantriyogesh
What impact does this warning have, what kind of abnormalities will it cause
And what actions do I need to take to confirm

mantriyogesh, 2024-10-09:
For now, it seems not very serious, considering quote from https://github.com/espressif/esp-idf/issues/6462#issuecomment-771516547

closed 2024-11-05 by lwj850533527