espressif/esp-hosted #501: Can I change the BT MAC address via esp-hosted-fg?I was used linux driver.

244219195, 2024-09-28:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I want to add device node in linux .and then .,i can set this node value to set BT mac for ESP32 -C3 .if have other way to set the BT MAC address.Please tell me .Actually, my idea is to be able to set the Bluetooth address through Linux instead of using the default address.however ，my linux kernel is low ,it can"t support python.

mantriyogesh, 2024-09-28:
Hello @244219195 , yes this needs to be added.

Plan would be to make [wifi_set_mac](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#16-ctrl_cmd_t--wifi_set_macctrl_cmd_t-req) deprecated and add up additional mode (diff name then with uint32) for Bluetooth with newer api, set_mac()

We will add this support and let you know. However, It might take some time.

For your case, you can just make small changes and re-use this API at higher layer, with hacks to handle bluetooth case, at both esp and host side code.

244219195, 2024-09-28:
Can you provide some examples?

mantriyogesh, 2024-09-28:
### Slave side hack
###### Get
https://github.com/espressif/esp-hosted/blob/250e29294a740a45dd853115a25b62f1b2cfc51b/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_control.c#L303-L316
###### Set
https://github.com/espressif/esp-hosted/blob/250e29294a740a45dd853115a25b62f1b2cfc51b/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_control.c#L1517-L1537

### Host side hack
###### Get
Add test_get_bt_mac_addr() similar to
https://github.com/espressif/esp-hosted/blob/250e29294a740a45dd853115a25b62f1b2cfc51b/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L697-L701

###### Set
https://github.com/espressif/esp-hosted/blob/250e29294a740a45dd853115a25b62f1b2cfc51b/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L732

mantriyogesh, 2024-09-28:
Above are not exactly the solution, but the code places to hack, to easily get your work done for now.

244219195, 2024-09-28:
Thanks，mantriyogesh.
Looking forward to the development of MAC address configuration in the future.