espressif/esp-hosted #502: SoftAP works if connecting via Mobile (Android/iOS) but Laptops(Windows/Linux) did not work.

mayurpatel19912009, 2024-09-30:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hello Devs,

We are currently using this setup

STM32mp133
esp32c3
interface: SPI
mode: SoftAp
type: esp-hosted-ng
Linux HostAP startup tool: Hostapd
Master commit used on both side: https://github.com/espressif/esp-hosted/commit/250e29294a740a45dd853115a25b62f1b2cfc51b

Description:
We have compiled the driver, we can see it via **ifconfig**, even station mode (connecting from esp32 to a hotspot) works fine.

Then we wanted to confirm that connecting from any device to esp **softAP** and we noticed it did not work. We were also seeinig errors on esp32 monitor side on logs.

Logs:

- Linux Host side logs: [STM32MP133_log.txt](https://github.com/user-attachments/files/17184437/STM32MP133_log.txt)
- ESP32C3 side logs: [ESP32C3_log.txt](https://github.com/user-attachments/files/17184444/ESP32C3_log.txt)
(look for the line **_E (238925) FW_MAIN: Sending data failed=12309_**  which come only connecting via Laptop)
- Hostapd config: 
```
interface=wlan0
driver=nl80211
ssid=test_ssid
hw_mode=g
channel=1
macaddr_acl=0
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase=test_ssid
rsn_pairwise=CCMP
wpa_pairwise=CCMP
ctrl_interface=/tmp/hostapd
```

error screenshot: 
[image]

Can you tell us what is missing and what can be different via Laptop?
Thank you very much in advance.

mantriyogesh, 2024-09-30:
@kapilkedawat ++

mayurpatel19912009, 2024-09-30:
@mantriyogesh,

can you tell me at least for now if its a known topic or something wrong from our side?

kapilkedawat, 2024-09-30:
@Shreyas0-7 please help to check. @mayurpatel19912009 please share hostapd logs as well. Please pass `-ddd -K` while starting hostapd.

mayurpatel19912009, 2024-09-30:
@mantriyogesh 

is there anything wrong in this on our side? Should we invest more time on this, or does it make sense to wait for your analysis?

mayurpatel19912009, 2024-10-01:
we added that flags in the service and logged it. We dont see many logs. Posting screenshot.

[image]

service running with your given flags
[image]

esp32c3 logs which i already posted  above before.
[image]

Shreyas0-7, 2024-10-01:
Hey @mayurpatel19912009 is possible for you to capture the sniffer
Here are steps, can you please try this to capture sniffer
If you are using a Windows machine, try: https://www.youtube.com/watch?v=As9BUoOMk_A
If you are using an iOS machine, try: https://youtu.be/hxtgin0hgf8?si=nRvTdL24KiKHOPIs
If you are using a Linux machine do iwconfig, and select interface which you are not using
For example your interface is wlp0s20f3, try:

ifconfig wlp0s20f3 down
iwconfig wlp0s20f3 mode monitor
ifconfig wlp0s20f3 up
iwconfig wlp0s20f3 channel 6
wireshark

Please note that a lot of WiFi cards (NIC) don't support putting them in monitor mode. So it is vital to check the compatibility of your laptop's NIC.
Hope that helps.

mayurpatel19912009, 2024-10-01:
Just some que ,

- Are you not able to reproduce this issue by yourself? 
- Connecting via Laptop works fine at your side?

It will make things fast if you can try to connect via Laptop and see the same issue.

Shreyas0-7, 2024-10-03:
Yes @mayurpatel19912009 I used the same hostapd configuration provided for test_ssid and was able to connect a linux client to the access point

iradmilo-bytelab, 2024-10-04:
Hi @Shreyas0-7 ,

could you please state commit and branch you used for host and network adapter?
What hardware setup are you using and what is the communication interface between chips?

I followed these instructions (https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/README.md#configuring-ap-with-hostapd) 
on the config @mayurpatel19912009 described in the opening question and its still the same behavior as is described there (Mobile connects but not PC).

I am configuring CONFIG_AP_SUPPORT=y for the host, do I have to setup anything additional for the network adapter firmware?

Shreyas0-7, 2024-10-05:
Hey @iradmilo-bytelab 
Hardware used c3 spi with rpi as host.
I  used `a4a5f00923b45f11d833a3631768f8f7ad1df868` commit for testing
Enabling `CONFIG_AP_SUPPORT=y` will enable ap support.
You can also do `./rpi_init.sh spi ap_support`.
If you are facing error with connection with a client we need hostapd logs as well sniffer to debug it further more

iradmilo-bytelab, 2024-10-22:
Hi @Shreyas0-7 ,

I am attaching wireshark capture file of linux PC trying to connect to AP provided by ESP32C3,
Setup is the same as in the original post. 

[ESP32_C3_AP_mode_PC_linux.pcapng.gz](https://github.com/user-attachments/files/17470895/ESP32_C3_AP_mode_PC_linux.pcapng.gz)

Shreyas0-7, 2024-10-22:
Hey @iradmilo-bytelab I can see only association request in the capture, is it possible for you to turn on the capture first and then try to connect with softap. Also please do attach hostapd logs for the same. you can use `-dd` with hostapd command to enable logs.

iradmilo-bytelab, 2024-10-22:
Hi @Shreyas0-7 , do you have a mail address where I can send you these files?

Shreyas0-7, 2024-10-22:
Hey @iradmilo-bytelab you can attach your files here

iradmilo-bytelab, 2024-10-22:
[hostapd_start_logs.gz](https://github.com/user-attachments/files/17474837/hostapd_start_logs.gz)
[hostapd_logs.gz](https://github.com/user-attachments/files/17474838/hostapd_logs.gz)
[ESPC3_AP.pcapng.gz](https://github.com/user-attachments/files/17474839/ESPC3_AP.pcapng.gz)

Shreyas0-7, 2024-10-23:
Hey @iradmilo-bytelab from the logs and capture we can observe that we are able associate a station and start 4 way handshake. But we are failing in between of 4 way handshake. To debug it further more can you please share dmesg logs as well.

iradmilo-bytelab, 2024-10-23:
I am attaching dmesg logs form host side and logs from esp32c3 side:

[Host_logs.gz](https://github.com/user-attachments/files/17487300/Host_logs.gz)
[ESP_C3_side_logs.gz](https://github.com/user-attachments/files/17487301/ESP_C3_side_logs.gz)

iradmilo-bytelab, 2024-10-23:
Hi @Shreyas0-7 ,

Thank you for the analysis.
Can you please point out the messages from the logs that indicate that connection is unstable?

Shreyas0-7, 2024-10-23:
Command timeout logs in dmesg suggest that connection might be unstable. Also you can check raw throughput to verify if connection is stable https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md.

Shreyas0-7, 2024-10-23:
Hey @iradmilo-bytelab do you have values for esp as well?

iradmilo-bytelab, 2024-10-23:
[Host_2_esp_raw_tp_host_side.gz](https://github.com/user-attachments/files/17491188/Host_2_esp_raw_tp_host_side.gz)
[Host_2_esp_raw_tp_esp_side.gz](https://github.com/user-attachments/files/17491189/Host_2_esp_raw_tp_esp_side.gz)
[Esp_2_host_raw_tp_host_side.gz](https://github.com/user-attachments/files/17491190/Esp_2_host_raw_tp_host_side.gz)
[Esp_2_host_raw_tp_esp_side.gz](https://github.com/user-attachments/files/17491191/Esp_2_host_raw_tp_esp_side.gz)

Shreyas0-7, 2024-10-24:
Hey @iradmilo-bytelab  thanks for the logs, raw throughput looks okay. I see scan issued on the interface multiple times in the logs, could you please check if there is a wpa_supplicant instance running on the host besides hostapd.

iradmilo-bytelab, 2024-10-24:
wpa_supplicant and network manager were both killed and not working at the moment of testing.

Shreyas0-7, 2024-10-25:
Hey @iradmilo-bytelab can you please check if there are some background script which may be issuing scan? Could you also share a ps -a output when issue is coming.

iradmilo-bytelab, 2024-10-25:
ps -ef
UID        PID  PPID  C STIME TTY          TIME CMD
root         1     0  0 11:15 ?        00:00:00 /sbin/init
root         2     0  0 11:15 ?        00:00:00 [kthreadd]
root         3     2  0 11:15 ?        00:00:00 [rcu_gp]
root         4     2  0 11:15 ?        00:00:00 [rcu_par_gp]
root         5     2  0 11:15 ?        00:00:00 [slub_flushwq]
root         6     2  0 11:15 ?        00:00:00 [netns]
root        10     2  0 11:15 ?        00:00:00 [mm_percpu_wq]
root        11     2  0 11:15 ?        00:00:00 [rcu_tasks_kthread]
root        12     2  0 11:15 ?        00:00:00 [ksoftirqd/0]
root        13     2  0 11:15 ?        00:00:00 [rcu_preempt]
root        14     2  0 11:15 ?        00:00:00 [kdevtmpfs]
root        15     2  0 11:15 ?        00:00:00 [inet_frag_wq]
root        22     2  0 11:15 ?        00:00:00 [oom_reaper]
root        23     2  0 11:15 ?        00:00:00 [writeback]
root        24     2  0 11:15 ?        00:00:00 [kcompactd0]
root        25     2  0 11:15 ?        00:00:00 [kblockd]
root        26     2  0 11:15 ?        00:00:00 [ata_sff]
root        27     2  0 11:15 ?        00:00:00 [edac-poller]
root        28     2  0 11:15 ?        00:00:00 [devfreq_wq]
root        29     2  0 11:15 ?        00:00:00 [watchdogd]
root        32     2  0 11:15 ?        00:00:00 [rpciod]
root        33     2  0 11:15 ?        00:00:00 [xprtiod]
root        34     2  0 11:15 ?        00:00:00 [kswapd0]
root        36     2  0 11:15 ?        00:00:00 [nfsiod]
root        38     2  0 11:15 ?        00:00:00 [optee_pcpu_noti]
root        39     2  0 11:15 ?        00:00:00 [hwrng]
root        40     2  0 11:15 ?        00:00:00 [mld]
root        41     2  0 11:15 ?        00:00:00 [ipv6_addrconf]
root        58     2  0 11:15 ?        00:00:00 [irq/43-5c004000.rtc]
root        59     2  0 11:15 ?        00:00:00 [irq/44-mmci-pl18x (cmd)]
root        61     2  0 11:15 ?        00:00:00 [mmc_complete]
root        63     2  0 11:15 ?        00:00:00 [jbd2/mmcblk1p8-8]
root        64     2  0 11:15 ?        00:00:00 [ext4-rsv-conver]
root        95     1  0 11:15 ?        00:00:00 /sbin/udevd -d
root       105     2  0 11:15 ?        00:00:00 [irq/47-4c002000.spi]
root       106     2  0 11:15 ?        00:00:00 [spi0]
root       110     2  0 11:15 ?        00:00:00 [kworker/0:3H-kblockd]
root       144     2  0 11:15 ?        00:00:00 [cfg80211]
root       148     2  0 11:15 ?        00:00:00 [ESP_SPI_WORK_QU]
message+   223     1  0 11:15 ?        00:00:14 /usr/bin/dbus-daemon --system
root       233     1  0 11:15 ?        00:00:00 sshd: /usr/sbin/sshd [listener] 0 of 10-100 startups
root       237     1  0 11:15 ?        00:00:00 /usr/libexec/bluetooth/bluetoothd
root       244     2  0 11:15 ?        00:00:00 [ESP_CMD_WORK_QU]
root       246     2  0 11:15 ?        00:00:00 [kworker/u3:0-hci0]
root       248     2  0 11:15 ?        00:00:00 [kworker/u3:2-hci0]
root       252     2  0 11:15 ?        00:00:00 [cryptd]
root       274     1  0 11:15 ?        00:00:01 /usr/sbin/ModemManager
ntp        294     1  0 11:15 ?        00:00:00 /usr/sbin/ntpd -u ntp:ntp -p /var/run/ntpd.pid -g
root       302     1  0 11:15 ?        00:00:00 /sbin/syslogd -n -O /var/log/messages
root       305     1  0 11:15 ?        00:00:00 /sbin/klogd -n
root       316     1  0 11:15 ?        00:00:00 /usr/sbin/xinetd -pidfile /var/run/xinetd.pid -stayalive
root       329     1  0 11:15 ttySTM0  00:00:00 -sh
root       330     1  0 11:15 tty1     00:00:00 /sbin/getty 38400 tty1
root       378     2  0 12:19 ?        00:00:01 [kworker/0:2-events_power_efficient]
root       385     2  0 12:42 ?        00:00:00 [kworker/u2:2-events_unbound]
root       408     2  0 12:44 ?        00:00:00 [kworker/0:0H-kblockd]
root       422     2  0 12:46 ?        00:00:00 [kworker/0:1-pm]
root       428     2  0 12:47 ?        00:00:00 [kworker/u2:0-events_unbound]
root       430     1  0 12:47 ?        00:00:00 hostapd /etc/hostapd.conf -ddd -K -B
nobody     433     1  0 12:48 ?        00:00:00 dnsmasq -C /etc/dnsmasq.conf
root       447   329 99 12:54 ttySTM0  00:00:00 ps -ef

No scanning request at the moment, connecting to esp AP still not working.

iradmilo-bytelab, 2024-10-29:
@Shreyas0-7 

1) RPI 3 - ESP32C3 SPI setup test
 
We have connected Raspberry Pi 3 Model B with esp32C3 chip using the hostapd conf and dnmasq template you provided in the README and did not get the AP mode to work. We cannot even get the AP to advertise in this RPI setup.

[esp32c3_log.txt](https://github.com/user-attachments/files/17544301/esp32c3_log.txt)
[hostapd_log.txt](https://github.com/user-attachments/files/17544302/hostapd_log.txt)

commit used: 30f4082314b6e13d869e9bdff7949fa428713337

Can you please screenshot me where you got Raspberry Pi ESP setup to work in AP mode?

Additionally: How did you get wpa_key_mgmt=SAE to work with hostapd (we are using version 2.10)?

2) STM32MP133 - ESP32C3 SPI patch test

After testing the provided patch we still are not able to connect to AP.

[Host_logs_after_patch.txt](https://github.com/user-attachments/files/17557716/Host_logs_after_patch.txt)
[Esp_logs_after_patch.txt](https://github.com/user-attachments/files/17557717/Esp_logs_after_patch.txt)

3) Stable release
@mantriyogesh When can we expect stable release of ESP32 AP mode for ng_hosted?

Our customer will soon (in month or so) go into testing of the final product which has to have WIFI chip with AP mode functional.
At the moment we are not sure if we should continue with this chip or maybe try to use some other WIFI chip.

Can we organize a call session where we could troubleshoot the issue and get the information about a stable release?

cc: @mayurpatel19912009

mantriyogesh, 2024-10-29:
@iradmilo-bytelab ,

Let me sync up internally and come back on this tomorrow..

mayurpatel19912009, 2024-10-29:
@mantriyogesh,

I request you to please post a complete Video with Setup where you have esp_hosted_ng setup running with exact commit where it's connected with any standard Linux Host (can be Raspberry Pi) where the AP is fully functional and connected with other Linux client or phone. **We want to make sure that setup is working at your side, and then we can look to our side where we are going wrong.**

So will be good if you can deliver us 
- Demo video (starting and connecting AP and connected)
- Logs from esp side
- Exact commit which you use

mantriyogesh, 2024-10-29:
Hello @mayurpatel19912009 ,

We are making a short video to showcase how we trigger the connection.
I think this would be the first thing, which can help to understand if there is any difference in setup / config or procedure.

@iradmilo-bytelab I would request you to wait up till tomorrow.
@Shreyas0-7 is making video. We will share this asap for your reference.

Shreyas0-7, 2024-10-30:
Hey @iradmilo-bytelab uploading the video with ps logs. 
Here's a link to google drive with attachments 
https://drive.google.com/file/d/1hCWqH7UuhlOrNf8ES_JBL7apczy2M7Io/view?usp=drive_link

mayurpatel19912009, 2024-10-30:
Thanks for the fast response. Our team will look into this.
before that some que which were not present in the vdo.

- Did you try any other client like andoid / ios / windows to connect?
- what about dhcp server/ dnsmasq configutation? Does it matter here? because an ip must be assigned to a client which is connecting to it.

mantriyogesh, 2024-10-30:
@Shreyas0-7  PTAL

iradmilo-bytelab, 2024-10-30:
@Shreyas0-7 
Additionally please can you provide info how did you get 'wpa_key_mgmt=SAE' hostapd conf setting to run with hostapd?

mayurpatel19912009, 2024-10-30:
Yes, as @iradmilo-bytelab mentioned we get error when we use SAE in hostapd.conf

[image]

kapilkedawat, 2024-10-30:
> Thanks for the fast response. Our team will look into this. before that some que which were not present in the vdo.
>  Did you try any other client like andoid / ios / windows to connect?

Yes, we tried android and iOS client, however we don't test using windows much.

>  what about dhcp server/ dnsmasq configutation? Does it matter here? because an ip must be assigned to a client which is connecting to it.

Yes, dhcp server is required for IP, you can also use static IP depending on your requirement and usecase.

>  which version of hostapd are you using?

We use current hostapd version (v2.11) normally, however hostapd version doesn't matter much here since most features are present from long and you can use your current version(v2.10).

>  Are you testing with kernel **4.19.118 which is a failry old kernel**. we are using Linux **6.1.28** kernel. It will be really nice if you can test it on that. because min. / max tested supported kernel i have not seen on your documentation.

Sure, we will retest with (recent kernel version + Linux laptop) again and see if any issue is observed.

> Additionally please can you provide info how did you get 'wpa_key_mgmt=SAE' hostapd conf setting to run with hostapd?

You can recompile the hostapd with SAE feature enabled. Please enable this [flag](https://w1.fi/cgit/hostap/tree/hostapd/defconfig#n172
) during recompilation.

mayurpatel19912009, 2024-10-30:
We are trying to replicate the same thing which you are doing
- hostapd 2.10 (with SAE mode enabled) + exact same as config as yours
- wpa_supplicant as connecting client + exact same config as yours

I am attaching a screenshot because thats what i can send you at the moment. Its not connecting. (left : hostapd log , right : wpa_supplicant log)
[image]

iradmilo-bytelab, 2024-10-30:
@Shreyas0-7 

With using hostapd 2.10 ( with SAE enabled) and hostapd and wpa_supplicant setup as in video example I get this result 

wpa_supplicant linux notebook side:

sudo wpa_supplicant -i wlp0s20f3 -c ~/wpa2_esp.conf                                                                                                                                                                                                                                                                                                                                                                                                               
Successfully initialized wpa_supplicant
wlp0s20f3: CTRL-EVENT-REGDOM-CHANGE init=DRIVER type=WORLD
wlp0s20f3: CTRL-EVENT-REGDOM-CHANGE init=DRIVER type=WORLD
wlp0s20f3: SME: Trying to authenticate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: CTRL-EVENT-REGDOM-CHANGE init=DRIVER type=COUNTRY alpha2=HR
wlp0s20f3: CTRL-EVENT-REGDOM-CHANGE init=DRIVER type=COUNTRY alpha2=HR

wlp0s20f3: SME: Trying to authenticate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: PMKSA-CACHE-ADDED dc:54:75:56:26:dd 0
wlp0s20f3: Trying to associate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Associated with dc:54:75:56:26:dd
wlp0s20f3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20f3: CTRL-EVENT-DISCONNECTED bssid=dc:54:75:56:26:dd reason=2
wlp0s20f3: Added BSSID dc:54:75:56:26:dd into ignore list, ignoring for 10 seconds
wlp0s20f3: Removed BSSID dc:54:75:56:26:dd from ignore list (clear)
wlp0s20f3: SME: Trying to authenticate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Trying to associate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Associated with dc:54:75:56:26:dd
wlp0s20f3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20f3: CTRL-EVENT-DISCONNECTED bssid=dc:54:75:56:26:dd reason=2
wlp0s20f3: Added BSSID dc:54:75:56:26:dd into ignore list, ignoring for 10 seconds
wlp0s20f3: Removed BSSID dc:54:75:56:26:dd from ignore list (clear)
wlp0s20f3: SME: Trying to authenticate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Trying to associate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Associated with dc:54:75:56:26:dd
wlp0s20f3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20f3: CTRL-EVENT-DISCONNECTED bssid=dc:54:75:56:26:dd reason=2
wlp0s20f3: Added BSSID dc:54:75:56:26:dd into ignore list, ignoring for 10 seconds
wlp0s20f3: Removed BSSID dc:54:75:56:26:dd from ignore list (clear)
wlp0s20f3: SME: Trying to authenticate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Trying to associate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Associated with dc:54:75:56:26:dd
wlp0s20f3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20f3: WPA: Could not verify EAPOL-Key MIC - dropping packet
wlp0s20f3: CTRL-EVENT-DISCONNECTED bssid=dc:54:75:56:26:dd reason=2
wlp0s20f3: Added BSSID dc:54:75:56:26:dd into ignore list, ignoring for 10 seconds
wlp0s20f3: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MY_SSID" auth_failures=1 duration=10 reason=CONN_FAILED
wlp0s20f3: CTRL-EVENT-SSID-REENABLED id=0 ssid="MY_SSID"
wlp0s20f3: BSSID dc:54:75:56:26:dd ignore list count incremented to 2, ignoring for 10 seconds
wlp0s20f3: Removed BSSID dc:54:75:56:26:dd from ignore list (clear)
wlp0s20f3: SME: Trying to authenticate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Trying to associate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Associated with dc:54:75:56:26:dd
wlp0s20f3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20f3: CTRL-EVENT-DISCONNECTED bssid=dc:54:75:56:26:dd reason=2
wlp0s20f3: Added BSSID dc:54:75:56:26:dd into ignore list, ignoring for 10 seconds
wlp0s20f3: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MY_SSID" auth_failures=2 duration=20 reason=CONN_FAILED
wlp0s20f3: CTRL-EVENT-SSID-REENABLED id=0 ssid="MY_SSID"
wlp0s20f3: BSSID dc:54:75:56:26:dd ignore list count incremented to 2, ignoring for 10 seconds
wlp0s20f3: Removed BSSID dc:54:75:56:26:dd from ignore list (clear)
wlp0s20f3: SME: Trying to authenticate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Trying to associate with dc:54:75:56:26:dd (SSID='MY_SSID' freq=2437 MHz)
wlp0s20f3: Associated with dc:54:75:56:26:dd
wlp0s20f3: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlp0s20f3: CTRL-EVENT-DISCONNECTED bssid=dc:54:75:56:26:dd reason=2
wlp0s20f3: Added BSSID dc:54:75:56:26:dd into ignore list, ignoring for 10 seconds
wlp0s20f3: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MY_SSID" auth_failures=3 duration=30 reason=CONN_FAILED

cat ~/wpa2_esp.conf                                                                                                                                                                                                                                                                                                                                                                                                                                                   
ctrl_interface=/var/run/wpa_supplicant
network={
        ssid="MY_SSID"
        psk="MY_PASSPHRASE"
        key_mgmt=SAE
        ieee80211w=2
}

Host side:

root@mcm-ng:~# echo $(uname -r)
6.1.28

cat /etc/hostapd.conf

interface=wlan0
driver=nl80211
ssid=MY_SSID
hw_mode=g
channel=6
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_key_mgmt=SAE
wpa_passphrase=MY_PASSPHRASE
rsn_pairwise=CCMP
ieee80211w=2

root@mcm-ng:~# cat /etc/dnsmasq.conf 
interface=wlan0          # Use the wireless interface
dhcp-range=192.168.1.2,192.168.1.100,255.255.255.0,24h  # DHCP range
domain-needed             # Don't forward short names
bogus-priv                # Never forward addresses in the non-routed address spaces
expand-hosts              # Use the expanded hostname

mayurpatel19912009, 2024-10-30:
to make sure its not in version of hostapd -> cross compiled 2.11 (same as yours)
Still the same error. There is something wrong with esp_32 host SPI Driver <> Firmware which is not letting us connect.
[image]

mantriyogesh, 2024-10-30:
Btw, @mayurpatel19912009 ,

Do you have raspberry pi same as that of the setup?

If nno, it's okay.

If yes, it would fasten to localise issue.

If you do not issue in raspberry pi and only in your Linux, I think we might need to dig deeper.

In any case we will try with latest raspberry pi (with highest possible kernel which come with raspberry pi) and let you know our findings.
cc: @Shreyas0-7

mayurpatel19912009, 2024-11-04:
No we dont have raspebrryPi here at the moment and have no plan and resources for that.
We tried to match exact tools and exact versions of tools which you are using. TLDR: we did exactly the same that you are doing in the video. But hard luck. Now only the difference is : Kernel  version (which can make an impact on driver level).

That's why we are keen to know the output of your latest kernel tests. Fingers crossed.. :)

mantriyogesh, 2024-11-04:
Sure , we are working on it, keep you posted.

Shreyas0-7, 2024-11-04:
Hey @mayurpatel19912009 @iradmilo-bytelab 
We have observed an issue with one of our clients that appears similar to what you are experiencing (we can't be certain yet, as your packet capture didn't include all the required packets). In this particular case, we receive multiple auth/assoc requests, which causes the hostapd and driver states to become unstable. Could you please try the attached patch to see if it resolves your issue?
[del_station.zip](https://github.com/user-attachments/files/17619295/del_station.zip)

iradmilo-bytelab, 2024-11-05:
@Shreyas0-7 and @mantriyogesh Have you setup a configuration with linux kernel version later than 4.16 and what version did you setup. Additionally did you try this patch on that setup?

iradmilo-bytelab, 2024-11-05:
Tried myself only the patch applied to the latest master and still not able to connect to AP

Host logs:

wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: Setup of interface done.
ctrl_iface not configured!
root@mcm-ng:~# [  832.038384] esp32_spi: decode_common_resp: [0x4] Command failed
[  832.043159] esp32_spi: cmd_scan_request: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -1
[  842.037911] esp32_spi: decode_common_resp: [0x4] Command failed
[  842.042708] esp32_spi: cmd_scan_request: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -1
[  843.566067] esp32_spi: esp_cfg80211_del_station: 
[  844.124517] esp32_spi: esp_cfg80211_del_station: 
[  856.048930] esp32_spi: decode_common_resp: [0x4] Command failed
[  856.053768] esp32_spi: cmd_scan_request: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -1
[  876.048681] esp32_spi: decode_common_resp: [0x4] Command failed
[  876.053521] esp32_spi: cmd_scan_request: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -1
[  905.068397] esp32_spi: decode_common_resp: [0x4] Command failed
[  905.073201] esp32_spi: cmd_scan_request: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -1
[  948.078848] esp32_spi: decode_common_resp: [0x4] Command failed
[  948.083631] esp32_spi: cmd_scan_request: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -1

Peripheral logs:

I (29956) supported_rates: 0x3fcbad20   01 0a 02 04 0b 16 0c 12  18 24 30 48              |.........$0H|
I (29966) ht_rates: 0x3fcbad2c   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (29976) ht_rates: 0x3fcbad3c   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (29985) vht_rates: 0x3fcbad48   00 00 00 00 00 00 00 00  00 00 00 00 00 00        |..............|
I (29995) he_rates: 0x3fcbad58   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (30005) he_rates: 0x3fcbad68   00 00 00 00 00 00 00 00  00 00 00                 |...........|
I (30015) wifi:new:<6,0>, old:<6,0>, ap:<6,1>, sta:<255,255>, prof:6
I (30084) FW_MAIN: AP station command

I (30084) FW_CMD: process_ap_station:got station add command

I (30085) mac: 0x3fcbad00   14 85 7f 23 ee d9                                 |...#..|
I (30092) FW_CMD: aid=0

I (30096) supported_rates: 0x3fcbad20   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (30106) ht_rates: 0x3fcbad2c   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (30116) ht_rates: 0x3fcbad3c   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (30125) vht_rates: 0x3fcbad48   00 00 00 00 00 00 00 00  00 00 00 00 00 00        |..............|
I (30135) he_rates: 0x3fcbad58   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (30145) he_rates: 0x3fcbad68   00 00 00 00 00 00 00 00  00 00 00                 |...........|
I (30154) FW_CMD: add_station_node_ap: not station add cmd, handle later

I (30194) FW_MAIN: AP station command

I (30194) FW_CMD: process_ap_station:got station add command

I (30194) mac: 0x3fcbad00   14 85 7f 23 ee d9                                 |...#..|
I (30202) FW_CMD: aid=0

I (30206) supported_rates: 0x3fcbad20   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (30216) ht_rates: 0x3fcbad2c   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (30226) ht_rates: 0x3fcbad3c   00 00 00 00 00 00 00 00  00 00 00 00              |............|
I (30235) vht_rates: 0x3fcbad48   00 00 00 00 00 00 00 00  00 00 00 00 00 00        |..............|
I (30245) he_rates: 0x3fcbad58   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (30255) he_rates: 0x3fcbad68   00 00 00 00 00 00 00 00  00 00 00                 |...........|
I (30264) FW_CMD: add_station_node_ap: not station add cmd, handle later

I (30424) FW_MAIN: Add key request

I (30424) FW_CMD: set_key_internal:1545

I (30424) FW_CMD: Setting PTK 

I (30426) FW_CMD: Unregistered event: 14

I (30454) FW_MAIN: AP station command

I (30454) FW_CMD: process_ap_station:got station add command

I (30454) mac: 0x3fcbad00   14 85 7f 23 ee d9                                 |...#..|
I (30462) FW_CMD: aid=0

iradmilo-bytelab, 2024-11-05:
Hi, 

thank you for providing the logs of the solution working on newer kernel version.

From my side I still cannot connect linux laptop to ESP AP, even after following all the instructions.

Commit of ESP hosted on host and peripheral is 47980497948369abaf38f60f6b05a33070f4ad50.

Kernel version is 6.1.28 (stm32mp-r1.1 mickledore)

Hostapd version is 11 built with SAE config flag. 

./hostapd -v
hostapd v2.11
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2024, Jouni Malinen <j@w1.fi> and contributors

dnsmasq conf and hostapd conf are as in README of esp/hosted_ng.

Attaching logs from host, peripheral and linux PC where wpa_supplicant with stopped NM was utilized.

[del_station_patch_test.zip](https://github.com/user-attachments/files/17632194/del_station_patch_test.zip)

mayurpatel19912009, 2024-11-05:
Hello,

we tried completely from beginning by

-  checking out repo master
- patched above patch
- built driver and and firmware both again and flashed again
- started hostapd with your config on our board (everything else off)
- tried to connect via iPhone with creds it does not work
- Tried to connect a Linux Ubuntu client with NetworkManager turned off and via wpa_supplicant.conf given by you and having support for SAE.
- and it showed us first signed of connection. 
[image]

- i tried it to stop wpa_supplicant on my client PC while leting hostapd running.
- i started wpa_supplicant again but this time it was not able to connect. (went to loop)
- I stopped hostapd also on the board and started again. and then i was able to connect again via wpa_supplicant.

**summary from this was : i have to restart hostapd every time once i am successfully connected via client to connect again.**

one more point that i found as error in hostapd logs -> 
```
nl80211: Failed to get current TX sequence for a key (link_id=-1 idx=1): -95 (Operation not supported)
nl80211: Failed to get current TX sequence for a key (link_id=-1 idx=4): -95 (Operation not supported)
```

- We treid connecting via iPhone with SSID + PASS -> Not Connected

- I am not understanding why are you using SAE. We are having many issues refgarding testing due to SAE mode. -> please try to use wpa or wpa2 from next tests so that we can do the same.

so finally, for your detailed info I am attaching a zip containing
- hostapd_test.log: hostapd logs
- wpa_supp.log: log for 1st try -> success
- wpa_supp_2ndtime.log -> not connected
- coection_test_05-11-2024.mkv -> showing all steps one by one for coonection 

[test.zip](https://github.com/user-attachments/files/17635271/test.zip)

Please let us know if these things gives some hints at your side.
- Why we are able to connect only once? Before we restart hostapd
- Why these nl80211 errors coming and what does it mean?
- Why we are not able to connect via any other client (ex. ios) . please test atleast once at your side and let us know if that also works at your end.
- Any other hint that you find from our logs?

Shreyas0-7, 2024-11-07:
Hey @mayurpatel19912009 thanks for reporting the issue. Recent issue observed was due a fix provided earlier. This issue has been resolved properly now. Please pull the latest change and check for reconnection.
Sure we can test using wpa2 also.
Don't apply previous patches and make sure the network manager is off for host
Please make sure you run `./setup.sh` in `/PATH/esp_hosted/esp_hosted_ng/esp/esp_driver` to update new firmware libraries 
If you are still facing the issue we are happy to help

iradmilo-bytelab, 2024-11-07:
Hi @Shreyas0-7 ,

we are still not able to connect to ESP AP after pulling the latest changes.
Could we arrange a session where we could go through setting up and testing the AP functionality on our board?

Additionally I am attaching logs from esp side of the moment station tries to connect to AP, not sure if they are of any help.
[esp_logs.log](https://github.com/user-attachments/files/17670485/esp_logs.log)

Shreyas0-7, 2024-11-08:
Hey @iradmilo-bytelab 
For us to debug the issue we need hostapd logs, esp logs, dmesg logs as well as sniffer capture

iradmilo-bytelab, 2024-11-08:
@Shreyas0-7 OK I will try capture the traffic, just give me more information before I do the test, should I change anything in ESP32C3 firmware menuconfig, do I need to kill network manager on the device connecting to the AP or only on the host? Can I use WPA2-PSK as key or do I need SAE for the solution to work?

Shreyas0-7, 2024-11-08:
Hey @iradmilo-bytelab 
- No you don't need to change anything for the firmware menuconfig just make sure you are trying on the latest master 
Make sure you have this commit in your git log 
commit cc1cfcc858e71e82a542863dc87588e195cf049d 
Author: Shreyas Sheth 
Date:   Thu Nov 7 12:55:28 2024 +0530

    fix(esp_hosted_ng): Softap fix delete node
- Kill network manager on host side on which you are trying to start softAP 
- You can use WPA2-PSK for the solution

iradmilo-bytelab, 2024-11-08:
@Shreyas0-7 

I get it to connect, at least from logs, however disconnection happens after around a minute.
I was connecting through Linux interface with which I normally connect to APs.
I tried connecting afterwards (not in logs) again and it does not connect at all.

Logs:
[8_11_2024_debug_connecting_2_AP.zip](https://github.com/user-attachments/files/17678258/8_11_2024_debug_connecting_2_AP.zip)

iradmilo-bytelab, 2024-11-14:
@Shreyas0-7 

Thank you for helping us get the solution working. The trick was to kill NetworkManager and wpa_supplicant.
However we need NM and wpa functional for setting up and controlling the cellular network. Can you please help us how to setup wlan interface in NetworkManager so that AP still works?

Shreyas0-7, 2024-11-15:
Hey @iradmilo-bytelab 
We have added  changes to support softap with NetworkManager.
 Please pull the new changes.
You can use this command to start softap using nmcli
`sudo nmcli dev wifi hotspot ifname wlan0 con-name MyHotspot ssid "MYSSID" password "MYPASSWORD" band bg channel 6`

iradmilo-bytelab, 2024-11-19:
Hi @Shreyas0-7 

I get the interface in NMCLI but cannot connect to the AP

HOST nmcli

> root@mcm-ng:~# nmcli
> wlan0: connected to MyHotspot
>         "wlan0"
>         wifi (unknown), DC:54:75:55:D7:58, hw, mtu 1500
>         inet4 10.42.0.1/24
>         route4 10.42.0.0/24 metric 600
>         inet6 fe80::de54:75ff:fe55:d758/64
>         route6 fe80::/64 metric 256
> 
> lo: connected (externally) to lo
>         "lo"
>         loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536
>         inet4 127.0.0.1/8
>         inet6 ::1/128
> 
> eth0: disconnected
>         "Realtek RTL8153"
>         1 connection available
>         ethernet (r8152), 00:E0:4C:68:50:B2, hw, mtu 1500
> 
> can0: unmanaged
>         "can0"
>         can (m_can_platform), hw, mtu 16
> 
> sit0: unmanaged
>         "sit0"
>         iptunnel (sit), 00:00:00:00, sw, mtu 1480
> 
> Use "nmcli device show" to get complete information about known devices and
> "nmcli connection show" to get an overview on active connection profiles.
> 
> Consult nmcli(1) and nmcli-examples(7) manual pages for complete usage details.

ESP logs

I (1533282) FW_CMD: aid=1

I (1533285) supported_rates: 0x3fcba8d4   01 0a 02 04 0b 16 0c 12  18 24 30 48              |.........$0H|
I (1533295) ht_rates: 0x3fcba8e0   2d 1a 24 00 17 ff ff 00  00 00 00 00 00 00 00 2c  |-.$............,|
I (1533305) ht_rates: 0x3fcba8f0   01 01 00 00 00 00 00 00  00 00 00 00              |............|
I (1533315) vht_rates: 0x3fcba8fc   00 00 00 00 00 00 00 00  00 00 00 00 00 00        |..............|
I (1533325) he_rates: 0x3fcba90c   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (1533335) he_rates: 0x3fcba91c   00 00 00 00 00 00 00 00  00 00 00                 |...........|
I (1533345) wifi:new:<6,0>, old:<6,0>, ap:<6,1>, sta:<255,255>, prof:6
I (1533453) FW_MAIN: disconnect request
I (1533453) FW_CMD: Disconnect request: reason [2], interface=1

I (1533453) wifi:station: 14:85:7f:23:ee:d9 leave, AID = 1, bss_flags is 3136, bss:0x3fcb8f94
I (1533461) wifi:new:<6,0>, old:<6,0>, ap:<6,1>, sta:<255,255>, prof:6
I (1544151) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as b0

I (1544187) FW_CMD: handle_wpa_ap_rx_mgmt: Got packet type as 0

I (1544213) FW_MAIN: disconnect request
I (1544213) FW_CMD: Disconnect request: reason [2], interface=1

I (1544243) FW_MAIN: AP station command
I (1544243) FW_CMD: process_ap_station:got station add command

I (1544243) mac: 0x3fcbaef8   14 85 7f 23 ee d9                                 |...#..|
I (1544252) FW_CMD: aid=1

I (1544255) supported_rates: 0x3fcbaf18   01 0a 02 04 0b 16 0c 12  18 24 30 48              |.........$0H|
I (1544265) ht_rates: 0x3fcbaf24   2d 1a 24 00 17 ff ff 00  00 00 00 00 00 00 00 2c  |-.$............,|
[... 39 lines cut ...]
I (1544611) wifi:new:<6,0>, old:<6,0>, ap:<6,1>, sta:<255,255>, prof:6
I (1544633) FW_MAIN: AP station command
I (1544633) FW_CMD: process_ap_station:got station add command

I (1544633) mac: 0x3fcbaef8   14 85 7f 23 ee d9                                 |...#..|
I (1544642) FW_CMD: aid=1

I (1544645) supported_rates: 0x3fcbaf18   01 0a 02 04 0b 16 0c 12  18 24 30 48              |.........$0H|
I (1544656) ht_rates: 0x3fcbaf24   2d 1a 24 00 17 ff ff 00  00 00 00 00 00 00 00 2c  |-.$............,|
I (1544666) ht_rates: 0x3fcbaf34   01 01 00 00 00 00 00 00  00 00 00 00              |............|
I (1544675) vht_rates: 0x3fcbaf40   00 00 00 00 00 00 00 00  00 00 00 00 00 00        |..............|
I (1544685) he_rates: 0x3fcbaf50   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
I (1544695) he_rates: 0x3fcbaf60   00 00 00 00 00 00 00 00  00 00 00                 |...........|
I (1544705) wifi:new:<6,0>, old:<6,0>, ap:<6,1>, sta:<255,255>, prof:6
I (1544813) FW_MAIN: disconnect request
I (1544813) FW_CMD: Disconnect request: reason [2], interface=1

I (1544813) wifi:station: 14:85:7f:23:ee:d9 leave, AID = 1, bss_flags is 3136, bss:0x3fcb8f94
I (1544821) wifi:new:<6,0>, old:<6,0>, ap:<6,1>, sta:<255,255>, prof:6
E (1545792) FW_MAIN: Sending data failed=12309

E (1546793) FW_MAIN: Sending data failed=12309

I (1547873) FW_MAIN: AP station command
I (1547873) FW_CMD: process_ap_station:got station add command

Shreyas0-7, 2024-11-19:
Hey @iradmilo-bytelab can you please upload complete esp logs, dmesg logs and journalctl logs

Shreyas0-7, 2024-11-21:
Hey @iradmilo-bytelab can you please attach wpa_supplicant logs as well

iradmilo-bytelab, 2024-11-21:
Hi @Shreyas0-7, these all the logs I found, where can I find wpa_supplicant logs?

[kernel.log](https://github.com/user-attachments/files/17844618/kernel.log)

Shreyas0-7, 2024-11-21:
Hey @iradmilo-bytelab you need to kill the old process which will be something like this
` /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant ` 
and then start wpa_supplicant again with this command for debug logs
`/sbin/wpa_supplicant -u -dd -O /run/wpa_supplicant`

iradmilo-bytelab, 2024-11-21:
@Shreyas0-7 
logs are form the moment of loading the kernel module & after starting wpa_supplicant with debug flags

[wpa_supplicant.log](https://github.com/user-attachments/files/17845121/wpa_supplicant.log)

[28 more replies]