espressif/esp-hosted #505: host couldn't pass build

Speakpig, 2024-10-08:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I'm using \esp-idf-v5.3\examples\wifi\iperf as a host example, it can pass build process untill I use cmds below
```
idf.py add-dependency "espressif/esp_wifi_remote"
idf.py add-dependency "espressif/esp_hosted"
```
info as follows, much thx:

[image]

Speakpig, 2024-10-08:
I've found out that if I set ESP32 to be a host target, error will occur like pic upside, but when I choose ESP32-C3 as a host target, the processing will be a suc, but actually I'm using ESP32 as host and ESP32-C3 as slave now accoding to what I've been told bofore (https://github.com/espressif/esp-hosted/issues/462#issuecomment-2303465083) better using two defferent ESP chipsets each as a host and a slave, but I've found the the tutorial document have been changed much, is it still neccesary, what should I do with my two chipest?

mantriyogesh, 2024-10-08:
Hello @Speakpig,

We are checking still why should it treat as sdmmc as 'cmake' interface.
Please spare time, we will get back on this.

@SohKamYung-Espressif PTAL..

mantriyogesh, 2024-10-08:
Assuming this build issue is only happening at host.

As I understand, if you use classic ESP32, you experience build issue at the

if(CONFIG_ESP_SDIO_HOST_INTERFACE)
    idf_component_optional_requires(PRIVATE sdmmc)
endif()

This issue not present for ESP32-C6.

Can you please help me in providing:

1. Do you wish to build for SPI or SDIO?
2. Can you please attach full build log from start in textual format?
3. What is exact ESP-IDF commit?
4. What is ESP-Hosted commit/idf_component.yml version?

Speakpig, 2024-10-08:
1.SPI
2.I upload a log, will that be helpful?
3.esp-idf-v5.3.0, if this is what u asking?
4.
[image]

Speakpig, 2024-10-08:
[build.log](https://github.com/user-attachments/files/17291793/build.log)

Speakpig, 2024-10-08:
I just found that the error happened while I'm setting the target device. not in the building process, so I just can't get to the step to do some menucofig.

mantriyogesh, 2024-10-08:
In case you are manually changing any content in `sdkconfig` or `sdkconfig.h` file, bypassing idf.py menuconfig, it is not supported. Also result is undefined.

[image]

Manual changes, such done, will override all the logic placed in KConfig.

Speakpig, 2024-10-09:
I didn't manually modify the file, I just added dependencies `espressif/esp_wifi_remote` and `espressif/esp_hosted`, and can't get into `idf.py menuconfig`.
[builid.txt](https://github.com/user-attachments/files/17300963/builid.txt)

SohKamYung-Espressif, 2024-10-09:
@Speakpig Have you disabled Native Wi-Fi on the ESP32 as documented in the troubleshooting page https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/troubleshooting.md#1-esp-host-to-evaluate-already-has-native-wi-fi :

```
Edit the ESP-IDF file components/soc/<soc>/include/soc/Kconfig.soc_caps.in and change all WIFI related configs to n. For example:

config SOC_WIFI_SUPPORTED
    bool
    # default y # original configuration
    default n

This should be done for all SOC_WIFI_xxx configs found in the file.
```

Speakpig, 2024-10-09:
Yeah, I' sure I missed that, the doc structure has been changed a lot since last time I read it, actually, I have been being concerned about this above, however, it's indeed my missing of guidance reading, I'm sry and I will try it out later, thx!

> been told bofore ([#462 (comment)](https://github.com/espressif/esp-hosted/issues/462#issuecomment-2303465083)) better using two defferent ESP chipsets each as a host and a slave, but I've found the the tutorial document have been changed much, is it still neccesary, what should I do with my two chipest?

mantriyogesh, 2024-10-09:
@Speakpig yes, the documentation is indeed changed. We tried to simplify and detail enough. Your feedback on this is valuable and welcome.

Speakpig, 2024-10-10:
If I understand correctly, the SPI FD test is best to start from 5MHz and gradually mode the actual upper limit, then I need to change this parameter in pic from 30 to 5, right?

> 4 Hardware Considerations  
> 4.1 General Considerations  
> 
> Use the lower clock frequency like 5 MHz for evaluation. Once solution verified, optimise the clock frequency in increasing steps to max possible value. To find out practical maximum SPI slave frequency for your co-processor, check IDF_PERFORMANCE_MAX_SPI_CLK_FREQ in [ESP-IDF SPI slave benchmark](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_spi/test_apps/components/spi_bench_mark/include/spi_performance.h)

[image]

mantriyogesh, 2024-10-10:
check the 'IDF_PERFORMANCE_MAX_SPI_CLK_FREQ' mentioned in [ESP-IDF SPI slave benchmark](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_spi/test_apps/components/spi_bench_mark/include/spi_performance.h).

For C3 as slave, max it can go upto [40MHz](https://github.com/espressif/esp-idf/blob/6e5a178b3120dced7fa5c29c655cc22ea182df3d/components/esp_driver_spi/test_apps/components/spi_bench_mark/include/spi_performance.h#L43). Clock is released from host, but it cannot go more than 40.

But if you use jumper cables as connection (and several other dependencies), it would saturate somewhere by ~30MHz.

**Manual KConfig change is discouraged**.

Finally, it would end here (to avoid again again again idf.py menuconfig)
https://github.com/espressif/esp-hosted/blob/32be5adb7a281eb97a64289f6a1213c6206a52d5/host/port/spi_wrapper.c#L81C35-L81C53

Once you sure what is max, you remove manual change and use 
```
idf.py menuconfig
```
and set the freq under 'SPI Clock Freq' option.

Speakpig, 2024-10-10:
OK, when I'm building the host after my menuconfig done, it ouccur sth like this, why? Before I'm using the default setting which is 
UART, and it passed the build, I've changed it to be a SPI transaction.
[build.txt](https://github.com/user-attachments/files/17328461/build.txt)

mantriyogesh, 2024-10-10:
Have you changed the sdkconfig or sdkconfig.h manually?

Can you remove
```
rm -rf build sdkconfig
```

and

```
idf.py menuconfig
```

and set values through the config GUI?

mantriyogesh, 2024-10-10:
Not related to build error, just as information,

if you wish to change to spi, exact similar transport need to be setup at slave as well.
For example, you use standard SPI (full duplex) at host, build and flash slave also with standard SPI.
If you use QSPI, use QSPI at both places and so on.

Speakpig, 2024-10-15:
[image]
I finally com to the iperf test, but what seems to be the question?

mantriyogesh, 2024-10-15:
iperf2 is only supported by default.

Speakpig, 2024-10-15:
OK,I will try iperf2, thx!

mantriyogesh, 2024-10-15:
Sorry I missed to understand, can you please assist in letting me know the host and slave ESP used?

Speakpig, 2024-10-15:
ESP32 as host, and ESP32-C3 as slave

> Sorry I missed to understand, can you please assist in letting me know the host and slave ESP used?

Speakpig, 2024-10-15:
> iperf2 is only supported by default.

I' using iperf3 on WINDOWS, shall I use iperf2?

mantriyogesh, 2024-10-15:
The IDF example integrates iperf2 minimal code. So, it needs other end running iperf2 only. Iperf3 may not work, irrespective of operating system used.

Speakpig, 2024-10-15:
[image]
It finally worked!!! So much thx! Given the result, how to judge this performance? I'm setting spi clock tp be 5MHz.

Speakpig, 2024-10-15:
[image]
And this is what it is when spi clock comes to 30MHz, just asking is it resonable?

mantriyogesh, 2024-10-15:
If you test over he air, a lot of factors come in picture.

Test first the pure transport speed.
The frequency depends upon the connections used in C6-P4. jumper cable lengths, their quality etc.

If you use PCB, C3 would cope up till 40MHz.

Test [raw transport throughput](https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/troubleshooting.md#2-raw-throughput-testing), before making wifi conclusions.

Speakpig, 2024-10-16:
> To enable the Raw Throughput Option and set Raw Throughput direction on Host, enter Menuconfig and enable Component config ---> ESP-Hosted config ---> Debug Settings ---> RawTP. Set the data transfer direction: Host to Slave, Slave to Host or Bidirectional.
How shall I set the data transfer direction, if I'will be using spi FD?

mantriyogesh, 2024-10-16:
Explained in link,

Component config ---> ESP-Hosted config ---> Debug Settings ---> RawTP. Set the data transfer direction: Host to Slave, Slave to Host or Bidirectional

You can test bidirectional

Speakpig, 2024-10-18:
[image]
Hi，this is my raw transport throughput test result, how does it look like? Thx!

mantriyogesh, 2024-10-18:
The throughput could be achieved little high. But overall both sides are symmetrical, and the transport throughput is great.

Can you please try to configure the AP to use 'auto' channel and try to reduce the interference? Alternatively, you can configure native (non hosted) wifi and compare the results.

But from hosted perspective, I see this through.

Speakpig, 2024-10-18:
> Can you please try to configure the AP to use 'auto' channel and try to reduce the interference? Alternatively, you can configure native (non hosted) wifi and compare the results.

Sry, I don't quite get it, can u be more specific?
1.first, in the raw transport throughput test, I did not connect to a wifi.
2.secondly,if shall I connect, it shall be a sta, how to configure the AP to use 'auto' ?
3."you can configure native (non hosted) wifi", which means what?

mantriyogesh, 2024-10-18:
1 & 2. 
As the transport throughput is good, but the Wi-Fi throughput is very bad (Confirm?), I suspect the radio environment is noisy.
Ways to test better environment:
a. Test in less interference like home based setup
b. Configure your AP i.e. router settings to use 'auto' channel selection. Any high interference channel is avoided by AP itself.

3. In any case, you can test the https://github.com/espressif/esp-idf/tree/master/examples/wifi/iperf on directly your host,  **without ESP-Hosted** i.e. with local Wi-Fi 'switched on' and **With ESP-Hosted** with local Wi-Fi 'switched off'.
This way, Wi-Fi performance comparison is fair.

Speakpig, 2024-10-18:
I understood，that's some work to do, but for the result I provided, how could u know as below, do u mean spi could achieve about 17M while wifi only comes out at about 7.48M?

> As the transport throughput is good, but the Wi-Fi throughput is very bad (Confirm?)

[image]
[image]

mantriyogesh, 2024-10-18:
Yes, wifi throughput should really about match the transport. 

The transport test is packet transfer from one place to other.

Also you have tested bi-directional transport test. Which would be little bit lower than single direction.

So transport I expect to bump little more say about   19-20m in single direction. Nevertheless, try to test with another AP, different environment.

In general, the test case (3) would give good amount of transparency for your test.

Speakpig, 2024-10-18:
Thanks a lot, that's very clear, I'm using the company's wifi mesh network, which is used by many employees, and the test results may be affected by the network environment

mantriyogesh, 2024-10-20:
Any updates ahead @Speakpig ?