espressif/esp-hosted #506: How to interpret src revisions on host and network-adapter

vardan, 2024-10-08:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

[   17.082465] esp32_spi: process_esp_bootup_event: Received ESP bootup event
[   17.082502] esp32_spi: process_event_esp_bootup: Bootup Event tag: 3
[   17.082517] esp32_spi: esp_validate_chipset: Chipset=ESP32-C6 ID=0d detected over SPI
[   17.082527] esp32_spi: process_event_esp_bootup: Bootup Event tag: 2
[   17.082539] esp32_spi: adjust_spi_clock: ESP Reconfigure SPI CLK to 16 MHz
[   17.082548] esp32_spi: process_event_esp_bootup: Bootup Event tag: 0
[   17.082558] esp32_spi: process_event_esp_bootup: Bootup Event tag: 1
[   17.082567] esp32_spi: process_fw_data: ESP chipset's last reset cause:
[   17.082575] esp32_spi: print_reset_reason: POWERON_RESET
[   17.082587] esp32_spi: check_esp_version: ESP Firmware version: 1.0.3
[   17.090100] esp32_spi: esp_reg_notifier: Driver init is ongoing
[   17.241949] esp32_spi: esp_cfg80211_get_tx_power:
[   17.452792] esp32_spi: init_bt: ESP Bluetooth init
[   17.453332] esp32_spi: print_capabilities: Capabilities: 0xe8. Features supported are:
[   17.453347] esp32_spi: print_capabilities:    * WLAN on SPI
[   17.453357] esp32_spi: print_capabilities:    * BT/BLE
[   17.453366] esp32_spi: print_capabilities:      - HCI over SPI
[   17.453375] esp32_spi: print_capabilities:      - BLE only

### Actual behavior (suspected bug)

[ 1517.062899] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [5],chip select [0], SPI Clock [10]
[ 1518.596709] esp32_spi: process_rx_buf: offset_rcv[0] != exp[12], drop

### Error logs or terminal output

```shell
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd44
load:0x4086e610,len:0x2de4
load:0x40875728,len:0x17d8
entry 0x4086c410
I (23) boot: ESP-IDF e7771c75-dirty 2nd stage bootloader
I (24) boot: compile time Oct  1 2024 09:51:50
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  3 factory          factory app      00 00 00010000 00100000
I (88) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (95) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (103) boot: End of partition table
I (107) boot: Defaulting to factory image
I (111) esp_image: segment 0: paddr=00010020 vaddr=420b0020 size=22098h (139416) map
I (149) esp_image: segment 1: paddr=000320c0 vaddr=40800000 size=0df58h ( 57176) load
I (163) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=a5a54h (678484) map
I (303) esp_image: segment 3: paddr=000e5a7c vaddr=4080df58 size=0f674h ( 63092) load
I (324) boot: Loaded app from partition at offset 0x10000
I (325) boot: Disabling RNG early entropy source...
I (336) cpu_start: Unicore app
I (336) cpu_start: Pro cpu up.
W (345) clk: esp_perip_clk_init() has not been implemented yet
I (352) cpu_start: Pro cpu start user code
I (352) cpu_start: cpu freq: 160000000 Hz
I (352) cpu_start: Application information:
I (355) cpu_start: Project name:     network_adapter
I (360) cpu_start: App version:      NG-1.0.3.0.0
I (366) cpu_start: Compile time:     Oct  1 2024 09:51:59
I (372) cpu_start: ELF file SHA256:  b054a6aac18cebf6...
I (378) cpu_start: ESP-IDF:          e7771c75-dirty
I (383) cpu_start: Min chip rev:     v0.0
I (388) cpu_start: Max chip rev:     v0.99
I (393) cpu_start: Chip rev:         v0.0
I (398) heap_init: Initializing. RAM available for dynamic allocation:
I (405) heap_init: At 408221D0 len 0005A440 (361 KiB): D/IRAM
I (411) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (418) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (425) spi_flash: detected chip: generic
I (429) spi_flash: flash io: dio
I (433) sleep: Configure to isolate all GPIO pins in sleep state
I (440) sleep: Enable automatic switching of GPIO sleep configuration
I (447) coexist: coex firmware version: 77cd7f8
I (452) coexist: coexist rom version 5b8dcfa
I (457) app_start: Starting scheduler on CPU0
I (462) main_task: Started on CPU0
I (466) main_task: Calling app_main()
I (470) stats: *********************************************************************
I (478) stats:                 ESP-Hosted Firmware version :: NG-1.0.3.0.0
I (488) stats:                 Transport used :: SPI only
I (496) stats: *********************************************************************
I (505) FW_MAIN: Supported features are:
I (509) FW_MAIN: - WLAN over SPI
I (513) FW_BT: - BT/BLE
I (516) FW_BT:    - HCI Over SPI
I (520) FW_BT:    - BLE only
I (524) FW_MAIN: Capabilities: 0xe8
I (532) pp: pp rom version: 5b8dcfa
I (532) net80211: net80211 rom version: 5b8dcfa
I (538) wifi:wifi driver task: 4087dfcc, prio:23, stack:6656, core=0
I (544) wifi:wifi firmware version: d23b633
I (547) wifi:wifi certification version: v7.0
I (551) wifi:config NVS flash: disabled
I (555) wifi:config nano formating: disabled
I (559) wifi:mac_version:HAL_MAC_ESP32AX_761,ut_version:N
I (564) wifi:Init data frame dynamic rx buffer num: 60
I (569) wifi:Init static rx mgmt buffer num: 5
I (573) wifi:Init management short buffer num: 32
I (577) wifi:Init dynamic tx buffer num: 40
I (581) wifi:Init static tx FG buffer num: 2
I (585) wifi:Init static rx buffer size: 1700
I (590) wifi:Init static rx buffer num: 40
I (593) wifi:Init dynamic rx buffer num: 60
I (598) wifi_init: rx ba win: 32
I (601) wifi_init: tcpip mbox: 32
I (605) wifi_init: udp mbox: 6
I (609) wifi_init: tcp mbox: 6
I (612) wifi_init: tcp tx win: 5760
I (617) wifi_init: tcp rx win: 5760
I (621) wifi_init: tcp mss: 1440
I (625) wifi_init: WiFi IRAM OP enabled
I (629) wifi_init: WiFi RX IRAM OP enabled
I (634) BLE_INIT: Using main XTAL as clock source
I (644) BLE_INIT: ble controller commit:[217f1bf]
I (645) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11
I (700) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26
I (701) FW_MAIN: ESP Bluetooth MAC addr: 40-4c-ca-46-4f-7a

I (702) FW_SPI: Using SPI interface
I (706) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (716) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1726) FW_MAIN: Initial set up done
I (1726) main_task: Returned from app_main()
```

### Steps to reproduce the behavior

- Build latest fw for network adapter and kernel module
- Update both esp32_spi kernel module and chip fw

### Project release version

a4a5f00923b45f11d833a3631768f8f7ad1df868

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

OpenEmbedded Linux

### Shell

Bash

### Additional context

I am trying to bump both host and network-adapter to the latest version (a4a5f00923b45f11d833a3631768f8f7ad1df868) and there's clearly a mismatch of versions as I keep getting printouts of dropped packets. Can you guide me hot to interpret srcrevision in the kernel module description? It is clearly not a commit id it has been built on. Same for the network-adapter. Where can I see srcrevision there?
It worked for me in 61c9415c1eb619787cc151dadc708780a172a413

I am using ST stm32mp151 MPU and ESP32C6-mini chip

mantriyogesh, 2024-10-08:
1. 

> It worked for me in https://github.com/espressif/esp-hosted/commit/61c9415c1eb619787cc151dadc708780a172a413

Just to re-clarify, if you use NG with https://github.com/espressif/esp-hosted/commit/61c9415c1eb619787cc151dadc708780a172a413 at both sides, ESP and host, it works, but when you use latest master, it doesn't work. Is this correct?

2. have you integrated resetpin and it is resetting the esp **every time** the kernel module is reloaded (on every insmod after rmmod)?

3. Can you please click a photo of setup for connections?

vardan, 2024-10-08:
1. Yes, it is correct. Everything works on 61c9415c1eb619787cc151dadc708780a172a413
2. Yes, I am sure resetpin every time. Here's modprobe.conf file
```
root@stm32mp1-pixie:~# cat /etc/modprobe.d/esp32_spi.conf
options esp32_spi resetpin=95
```
I can also see esp32c6 module reloading every time kernel module is reloaded
3. It is a PCB, so no wires here

vardan, 2024-10-08:
I strongly suspect some mismatch on my side, because I keep getting this behaviour not only on latest master but any commit later than the version I am currently on.

mantriyogesh, 2024-10-08:
@kapilkedawat ++

vardan, 2024-10-09:
Sorry, I found what the problem was - we use different CS than IO10 configured by default. I completely forgot to change that before building the firmware for ESP32C6.
It works now.

closed 2024-10-09 by vardan