espressif/esp-hosted #507: Sometimes events cannot be handled in a timely manner

vrolife, 2024-10-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

FIx patch
@@ -945,14 +968,12 @@ static void esp_events_work(struct work_struct *work)
 {
        struct sk_buff *skb = NULL;
 
-       skb = skb_dequeue(&adapter.events_skb_q);
-       if (!skb)
-               return;
-
-       if (skb->data) {
-               process_internal_event(&adapter, skb);
+       while ((skb = skb_dequeue(&adapter.events_skb_q)) != NULL) {
+               if (skb->data) {
+                       process_internal_event(&adapter, skb);
+               }
+               dev_kfree_skb_any(skb);
        }
-       dev_kfree_skb_any(skb);
 }

mantriyogesh, 2024-10-13:
Can you please suggest what host you use, slave you use, git commit and the issue you faced?

It works be easy to then relate if this is the main cause or after effect that you are trying to solve.

vrolife, 2024-10-14:
My host is f1c100s+Linux 6.6, and the board is esp32s2+esp-hosted-ng. I use the esp32s2 to listen for GPIO interrupts to implement key press events, which are sent to Linux via SPI + interface_buffer_handle_t + ESP_INTERNAL_IF. When the key is pressed rapidly multiple times, the key press events accumulate in adapter.events_skb_q without being processed, because esp_events_work processes only one event at a time.

commit 30f40823 (2024-10-16): Merge branch 'fix/events_delayed' into 'master'

XG: fix(events): Process all events in loop. Fixes #507

See merge request app-frameworks/esp_hosted!529
closed 2024-10-16 by espressif-bot