espressif/esp-hosted #508: default wifi remote library (EPPP or HOSTED)

chegewara, 2024-10-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hi,
its not an issue, more like question.
Is there any reason why since `v0.0.20` this flag is not default `CONFIG_ESP_WIFI_REMOTE_LIBRARY_HOSTED`?
By default `CONFIG_ESP_WIFI_REMOTE_LIBRARY_EPPP` is enabled and it makes confusion, because after upgrading from `0.0.19` i thought that library is broken. With `EPPP` this flag cant be enabled `CONFIG_ESP_ENABLE_BT`.

Thanks

mantriyogesh, 2024-10-14:
Hello @chegewara ,

Let me give logic reasoning for the change in 
https://github.com/espressif/esp-hosted/commit/f1940f2476e51e28aae12ce472433190ea06afcc#diff-7c9693e9c5e61a1d80e09769377d47bb415ec39ca5aeaa85d13581b0105d6359L3

I know some of these might sound cryptic, but bear with me on this.
1. Solutions, `ext-conn` and `esp-wifi-remote` cannot work together 
2. ESP-Hosted should not be enabled when `ext-conn` is used, as it may conflict with each other
3. wifi-remote, can choose solutions underlying. if eppp is chosen, esp-hosted should back off.

The recent changes complicated the things and I can just think of easy solution such that:

esp-hosted could be enabled from `idf.py menuconfig` or sdkconfig.defaults (or sdkconfig.defaults.esp32p4)
instead of hard dependency

```
config ESP_HOSTED_ENABLED
	bool
        default y if ESP_WIFI_REMOTE_ENABLED && ESP_WIFI_REMOTE_LIBRARY_HOSTED
```
we can soften it like:

```
config ESP_HOSTED_ENABLED
	bool "Enable Esp-Hosted"
# N if wifi remote enabled and *not* chosen esp-hosted as solution
        default n if ESP_WIFI_REMOTE_ENABLED && !ESP_WIFI_REMOTE_LIBRARY_HOSTED
# Y if wifi remote enabled and chosen esp-hosted as solution
        default y if ESP_WIFI_REMOTE_ENABLED && ESP_WIFI_REMOTE_LIBRARY_HOSTED
# Y if wifi remote *not* enabled and esp-host (which is used in esp-extconn) is *not* used
        default y if !ESP_WIFI_REMOTE_ENABLED && ! CONFIG_ESP_HOST_WIFI_ENABLED
```

The problem is we don't have full control over other solution and need some way to get ahead , unless this is easier for users to decide.

Let me know your views on this (if you understood what I meant). It's completely unspendable if you did not understand this.
I will try to fix it today / tomorrow. Additionally, if you have your repo as such, you can give me small example , I will make sure it would work with esp-hosted (at least build part of it)

chegewara, 2024-10-14:
Oh no, dont take me wrong. I dont see the problem here and i am not using extconn component at all, in any of my projects, since the last update in that component is 7 months ago, and wifi-remote works fine.

If ext-conn and eppp are the same components, then its dependency is comming from here
```
  espressif/esp_wifi_remote:
    component_hash: dd44abf72eab81106bb33f95728cd8cfd2fa0647dcaec0c99ecbb47f929d8a2e
    dependencies:
    - name: espressif/eppp_link
```
otherwise i dont understand why you mentioned ext-conn.

Yes, since now i know that dependency in menuconfig is changed, i am using sdkconfig.defaults to enable it, but like i said, in 0.0.19 it was not necessary, so my confusion. Thats all.

I just thought i will mention the confusion here, and maybe we can make esp_hosted default, if you think its good idea.

Thanks for support, again

mantriyogesh, 2024-10-14:
Actually, dependencies should not matter as long as you have marked 'any' esp_wifi_remote to specific version or hash.

The idf_component.yml should have, ideally like this:
```
  espressif/esp_wifi_remote:
    version: "~0.4.1"
    rules:
      - if: "target in [esp32p4, esp32h2]"
  espressif/esp_hosted:
    version: "~0.0.20"
    rules:
      - if: "target in [esp32p4, esp32h2]"
```

The versions could also be put as '*'

for example,

```
  espressif/esp_wifi_remote:
    version: "*"
    rules:
      - if: "target in [esp32p4, esp32h2]"
  espressif/esp_hosted:
    version: "*"
    rules:
      - if: "target in [esp32p4, esp32h2]"
```

This would always fetch the latest of both components.
For now as workaround, you can use '*' for esp_wifi_remote and '0.0.19' for esp_hosted.

I will try to push
```
config ESP_HOSTED_ENABLED
	bool "Enable Esp-Hosted"
# N if wifi remote enabled and *not* chosen esp-hosted as solution
        default n if ESP_WIFI_REMOTE_ENABLED && !ESP_WIFI_REMOTE_LIBRARY_HOSTED
# Y if wifi remote enabled and chosen esp-hosted as solution
        default y if ESP_WIFI_REMOTE_ENABLED && ESP_WIFI_REMOTE_LIBRARY_HOSTED
# Y if wifi remote *not* enabled and esp-host (which is used in esp-extconn) is *not* used
        default y if !ESP_WIFI_REMOTE_ENABLED && ! CONFIG_ESP_HOST_WIFI_ENABLED
```

And notify you once tested fine.

> [!NOTE]
> 
> 1. You are possibly using esp_wifi_remote with `0.3.x` or `~0.3.X` in idf_component.yml.
> 2. The esp_wifi_remote is now defaulted to esp_hosted in https://github.com/espressif/esp-wifi-remote/commit/d5261869e15592a4597f2140ef82b32c463b1a2c

I really appreciate and thank you for making us notice the issues you faced.
I would request to wait for my further update (solution).

chegewara, 2024-10-14:
Currently i have this in my yaml, but i will update (thanks for suggestion)
```
dependencies:
  espressif/esp_hosted:
    version: ~0.0.12
  espressif/esp_wifi_remote:
    version: ~0.3.0
```

And this is something ive been looking for, so thanks for posting this
```
    rules:
      - if: "target in [esp32p4, esp32h2]"

mantriyogesh, 2025-01-16:
I beleive there is nothing to be done ahead. Can you please close this issue @chegewara ?