espressif/esp-hosted #510: esp32 p4 + esp32 c5 SPI function run error

skylin008, 2024-10-14:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Used latest Esp-hosted IDF v5.4-dev-3489-g6e5a178b31 . I had used Esp32-p4-Function_EV_Board and ESP32-C5-DevkitC to development, Esp32-p4 chip used SPI host function, ESP32-C5 chip used SPI slave function.
Used \examples\wifi\getting_started\station to running,Esp32 p4 host show the error information as:

**file: "./main/station_example_main.c" line 106
func: wifi_init_sta
expression: esp_wifi_init(&cfg)

abort() was called at PC 0x4ff0947b on core 0**

the esp32c5 show as error information:
**I (2048) wifi:Init management short buffer num: 32
I (2053) wifi:Init static tx buffer num: 16
I (2057) wifi:Init tx cache buffer num: 32
W (2060) wifi:esf_buf_setup_static: alloc eb fail(10)
I (2065) wifi:Deinit lldesc rx mblock:0
I (2069) wifi:Deinit lldesc rx mblock:0
I (2072) wifi:Deinit lldesc rx mblock:0
W (2077) wifi_init: Failed to unregister Rx callbacks
E (2082) wifi_init: Failed to deinit Wi-Fi driver (0x3001)
E (2088) wifi_init: Failed to deinit Wi-Fi (0x3001)
E (2093) slave_ctrl: req_wifi_init:730 failed [esp_wifi_init(&cfg)] = [257]**

### Actual behavior (suspected bug)

run the demo example show error information.

### Error logs or terminal output

```shell
slave log as follow :

I (23) boot: ESP-IDF v5.4-dev-3489-g6e5a178b31-dirty 2nd stage bootloader
I (24) boot: compile time Oct 10 2024 16:05:56
I (25) boot: chip revision: v0.0
I (28) boot: efuse block revision: v0.0
I (33) boot.esp32c5: SPI Speed      : 80MHz
I (38) boot.esp32c5: SPI Mode       : DIO
I (42) boot.esp32c5: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (71) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (78) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  3 ota_0            OTA app          00 10 00010000 00180000
I (93) boot:  4 ota_1            OTA app          00 11 00190000 00180000
I (101) boot: End of partition table
I (105) esp_image: segment 0: paddr=00010020 vaddr=420c0020 size=31b54h (203604) map
I (149) esp_image: segment 1: paddr=00041b7c vaddr=40800000 size=0e49ch ( 58524) load
I (162) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=b8600h (755200) map
I (291) esp_image: segment 3: paddr=00108628 vaddr=4080e49c size=09fe0h ( 40928) load
I (300) esp_image: segment 4: paddr=00112610 vaddr=40818480 size=03f14h ( 16148) load
I (310) boot: Loaded app from partition at offset 0x10000
I (310) boot: Disabling RNG early entropy source...
I (322) cpu_start: Unicore app
W (330) clk: esp_perip_clk_init() has not been implemented yet
I (337) cpu_start: Pro cpu start user code
I (337) cpu_start: cpu freq: 240000000 Hz
[... 216 lines cut ...]
```

### Steps to reproduce the behavior

Power on ,as show the error information.

### Project release version

(25) boot: ESP-IDF v5.4-dev-3489-g6e5a178b31-dirty 2nd stage bootloader

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

windows 10

### Shell

CMD

mantriyogesh, 2024-10-14:
Can you please confirm if you have installed latest esp-hosted from https://components.espressif.com/components/espressif/esp_hosted/versions/0.0.22

Please use latest master esp-idf if you wish to use c5 as slave.

skylin008, 2024-10-15:
Thanks @mantriyogesh. Yes, I had update to 0.0.22 , the error issue as follow:

**I (33) boot: ESP-IDF v5.4-dev-3489-g6e5a178b31-dirty 2nd stage bootloader
I (33) boot: compile time Oct 15 2024 10:18:40
I (34) boot: Multicore bootloader
I (39) boot: chip revision: v0.1
I (42) boot: efuse block revision: v0.0
I (47) boot.esp32p4: SPI Speed      : 80MHz
I (51) boot.esp32p4: SPI Mode       : DIO
I (56) boot.esp32p4: SPI Flash Size : 16MB
I (61) boot: Enabling RNG early entropy source...
I (66) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (77) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 00177000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=40060020 size=23a18h (145944) map
I (133) esp_image: segment 1: paddr=00033a40 vaddr=30100000 size=0002ch (    44) load
I (135) esp_image: segment 2: paddr=00033a74 vaddr=3010002c size=0003ch (    60) load
I (140) esp_image: segment 3: paddr=00033ab8 vaddr=4ff00000 size=0c560h ( 50528) load
I (157) esp_image: segment 4: paddr=00040020 vaddr=40000020 size=59184h (364932) map
I (209) esp_image: segment 5: paddr=000991ac vaddr=4ff0c560 size=070ech ( 28908) load
I (215) esp_image: segment 6: paddr=000a02a0 vaddr=4ff13680 size=028f8h ( 10488) load
I (222) boot: Loaded app from partition at offset 0x10000
I (222) boot: Disabling RNG early entropy source...
I (238) hex_psram: vendor id    : 0x0d (AP)
I (239) hex_psram: Latency      : 0x01 (Fixed)
I (239) hex_psram: DriveStr.    : 0x00 (25 Ohm)
I (242) hex_psram: dev id       : 0x03 (generation 4)
I (248) hex_psram: density      : 0x07 (256 Mbit)
I (253) hex_psram: good-die     : 0x06 (Pass)
I (258) hex_psram: SRF          : 0x02 (Slow Refresh)
I (264) hex_psram: BurstType    : 0x00 ( Wrap)
I (269) hex_psram: BurstLen     : 0x03 (2048 Byte)
I (275) hex_psram: BitMode      : 0x01 (X16 Mode)
I (280) hex_psram: Readlatency  : 0x04 (14 cycles@Fixed)
I (286) hex_psram: DriveStrength: 0x00 (1/1)
I (291) MSPI DQS: tuning success, best phase id is 2
I (479) MSPI DQS: tuning success, best delayline id is 11
I (480) esp_psram: Found 32MB PSRAM device
I (480) esp_psram: Speed: 200MHz
I (484) hex_psram: psram CS IO is dedicated
I (488) cpu_start: Multicore app
I (1441) esp_psram: SPI SRAM memory test OK
I (1451) cpu_start: Pro cpu start user code
I (1451) cpu_start: cpu freq: 360000000 Hz
I (1451) app_init: Application information:
I (1454) app_init: Project name:     wifi_station
I (1460) app_init: App version:      1
I (1464) app_init: Compile time:     Oct 15 2024 10:17:28
I (1470) app_init: ELF file SHA256:  86612ff7b...
I (1476) app_init: ESP-IDF:          v5.4-dev-3489-g6e5a178b31-dirty
I (1483) efuse_init: Min chip rev:     v0.1
I (1487) efuse_init: Max chip rev:     v0.99
I (1493) efuse_init: Chip rev:         v0.1
I (1497) heap_init: Initializing. RAM available for dynamic allocation:
I (1505) heap_init: At 4FF18520 len 00022AA0 (138 KiB): RAM
I (1511) heap_init: At 4FF3AFC0 len 00004BF0 (18 KiB): RAM
I (1517) heap_init: At 4FF40000 len 00060000 (384 KiB): RAM
I (1524) heap_init: At 50108080 len 00007F80 (31 KiB): RTCRAM
I (1530) heap_init: At 30100068 len 00001F98 (7 KiB): TCM
I (1536) esp_psram: Adding pool of 32768K of PSRAM memory to heap allocator
I (1544) spi_flash: detected chip: generic
I (1548) spi_flash: flash io: dio
I (1552) host_init: ESP Hosted : Host chip_ip[18]
I (1586) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120
spi_mempool_create free:33723388 min-free:33723388 lfb-def:33030144 lfb-8bit:33030144

I (1592) spi_wrapper: Transport: SPI, Mode:3 Freq:5MHz TxQ:20 RxQ:20
[... 50 lines cut ...]
func: wifi_init_sta
expression: esp_wifi_init(&cfg)

abort() was called at PC 0x4ff0947d on core 0
--- 0x4ff0947d: _esp_error_check_failed at D:/Espressif/frameworks/esp-idf-master/components/esp_system/esp_err.c:49

--- Stack dump detected
Core  0 register dump:
MEPC    : 0x4ff00cb8  RA      : 0x4ff09488  SP      : 0x4ff230a0  GP      : 0x4ff13e80
--- 0x4ff00cb8: panic_abort at D:/Espressif/frameworks/esp-idf-master/components/esp_system/panic.c:468
0x4ff09488: __ubsan_include at D:/Espressif/frameworks/esp-idf-master/components/esp_system/ubsan.c:311

TP      : 0x4ff23300  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x4ff23104  A0      : 0x4ff230cc  A1      : 0x4ff23102
A2      : 0x00000000  A3      : 0x4ff230f9  A4      : 0x00000001  A5      : 0x4ff18000
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00011880  MTVEC   : 0x4ff00003  MCAUSE  : 0x00000002  MTVAL   : 0x00000000
--- 0x4ff00003: _vector_table at ??:?

MHARTID : 0x00000000**

How to solve this issue?

mantriyogesh, 2024-10-15:
Can you please confirm, if the **slave** is also using 0.0.22?

https://components.espressif.com/components/espressif/esp_hosted/versions/0.0.22/examples/slave?language=en
```
idf.py create-project-from-example "espressif/esp_hosted^0.0.22:slave"
```

skylin008, 2024-10-15:
Yes, the slave is also using 0.0.22. Can you build the latest firmware for test, or ours build environment is other issue.

mantriyogesh, 2024-10-15:
I can test this tomorrow earliest. Let you know my test results.

mantriyogesh, 2024-10-16:
i am checking this now. let you update in while

mantriyogesh, 2024-10-16:
[c5_p4_std_spi_0.0.22_idf_6e5a17.tgz](https://github.com/user-attachments/files/17390437/c5_p4_std_spi_0.0.22_idf_6e5a17.tgz)

We don't face this on idf commit you mentioned.
The binaries I have attached for you to try, for c5 and p4.

No code change,  `idf.py menuconfig` used to  change the needed spi pins. Check readme.txt for details.

I have used iperf example by the way, so that you can enter the ssid and password which you wish to connect to.

dmartauz, 2024-10-16:
I am getting the same error while trying to use C6 of ESP32-P4-Function-EV-Board via SDIO using esp_wifi_remote 0.4.1 and esp_hosted 0.0.22:
```
I (3472) sdio_wrapper: SDIO master: Data-Lines: 4-bit Freq(KHz)[40000 KHz]
I (3472) sdio_wrapper: GPIOs: CLK[18] CMD[19] D0[14] D1[15] D2[16] D3[17] Slave_Reset[54]
I (3472) H_SDIO_DRV: Starting SDIO process rx task
I (3472) sdio_wrapper: Queues: Tx[20] Rx[20] SDIO-Rx-Mode[1]
I (3512) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (3512) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
Name:
Type: SDIO
Speed: 40.00 MHz (limit: 40.00 MHz)
Size: 0MB
CSD: ver=1, sector_size=0, capacity=0 read_bl_len=0
SCR: sd_spec=0, bus_width=0
TUPLE: DEVICE, size: 3: D9 01 FF
TUPLE: MANFID, size: 4
  MANF: 0092, CARD: 6666
TUPLE: FUNCID, size: 2: 0C 00
TUPLE: FUNCE, size: 4: 00 00 02 32
TUPLE: CONFIG, size: 5: 01 01 00 02 07
TUPLE: CFTABLE_ENTRY, size: 8
  INDX: C1, Intface: 1, Default: 1, Conf-Entry-Num: 1
  IF: 41
  FS: 30, misc: 0, mem_space: 1, irq: 1, io_space: 0, timing: 0, power: 0
  IR: 30, mask: 1,   IRQ: FF FF
  LEN: FFFF
TUPLE: END
I (3562) sdio_wrapper: Function 0 Blocksize: 512
I (3572) sdio_wrapper: Function 1 Blocksize: 512
I (3572) H_SDIO_DRV: SDIO Host operating in STREAMING MODE
I (3582) H_SDIO_DRV: generate slave intr
I (3592) transport: Received INIT event from ESP32 peripheral
I (3592) transport: EVENT: 12
I (3592) transport: EVENT: 11
I (3602) transport: capabilities: 0xd
I (3602) transport: Features supported are:
I (3612) transport:      * WLAN
I (3612) transport:        - HCI over SDIO
I (3622) transport:        - BLE only
I (3622) transport: EVENT: 13
I (3622) transport: ESP board type is : 13

I (3632) transport: Base transport is set-up

I (3632) transport: Slave chip Id[12]
I (3642) hci_stub_drv: Host BT Support: Disabled
I (3642) H_SDIO_DRV: Received INIT event
I (3662) rpc_wrap: Received Slave ESP Init
I (4712) H_API: esp_wifi_remote_connect
E (4732) rpc_rsp: Failure resp/event: possibly precondition not met
E (4732) rpc_wrap: Failed Control Response
```
C6 contains factory firmware. Does it need to be updated?

mantriyogesh, 2024-10-16:
@dmartauz , this is different query. Can you please trigger different issue? I will check by the time.

mantriyogesh, 2024-10-16:
@dmartauz , the answer is P4 0.0.22 is still compatible with on-board C6 (based on 0.0.6)

Original binaries on-board were flashed with:
[2024.05.31_hosted_c6_bins_with_ota.tgz](https://github.com/user-attachments/files/17397703/2024.05.31_hosted_c6_bins_with_ota.tgz) Ignore P4 binaries in this and only refer to C6 binaries.

Current P4 binary: [p4_0.0.22.tgz](https://github.com/user-attachments/files/17397713/p4_0.0.22.tgz)

Slightly unrelated, but good to know. Although, it would be important to note that 0.0.6 based code base need some changes if latest IDF is used. However, if original IDF that time was used, it would still build fine.

**Nevertheless, P4 latest binaries are still compatible with on-board C6 binaries.**

dmartauz, 2024-10-17:
@mantriyogesh I am testing with custom application using recent clone of esp-idf master branch. In this case do I need to update the C6?

mantriyogesh, 2024-10-17:
Despite of https://github.com/espressif/esp-hosted/issues/510#issuecomment-2417090504, I think it is better to setup slave with latest firmware changes.

I would say it is recommended but not mandatory.

dmartauz, 2024-10-17:
Ok, I will build the slave with recent esp-idf and find out how to flash it to C6 locally without esp-prog according to https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md#52-using-esp-prog

In the meantime, could you please share some minimal application code for the P4 which tests C6's WiFi over SDIO, e.g. performs connection to an AP?

mantriyogesh, 2024-10-17:
I think it is better to open new issue. this issue was created by @skylin008 , so lets us try to keep it clean.

dmartauz, 2024-10-17:
@mantriyogesh you are right, I created new request here: https://github.com/espressif/esp-hosted/issues/513

skylin008, 2024-10-18:
@mantriyogesh Thank you for your supported. From your  firmware ,the same issue show. Can you had  tested the firmware in ESP32P4 and  esp32c5 board?

mantriyogesh, 2024-10-18:
I tested it myself, I did not observe any issues.
Can you please make sure if the binaries I sent are flashed?

mantriyogesh, 2024-10-19:
I would wait ahead.

mantriyogesh, 2024-10-22:
I had reproduced the issue. The problem is memory.

Easy workaround: If you are not using Bluetooth, disable using `idf.py menuconfig`.

Else need to lower wifi buffers, to conserve the iram.
I will send you today, the sdkconfig.defaults.esp32c5 changes, which should work.