espressif/esp-hosted #511: ESP32-P4 can not connect to ESP32-C6 on the Function_EV_Board v1.2 and v1.4

me-no-dev, 2024-10-15:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expect to be able to connect to the C6 over SDIO

### Actual behavior (suspected bug)

Instead I get `E (226409) transport: Not able to connect with ESP-Hosted slave device`

### Error logs or terminal output

```shell
E (226409) transport: Not able to connect with ESP-Hosted slave device
```

### Steps to reproduce the behavior

Follow your steps in https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md

### Project release version

latest

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

MacOS

### Operating system version

15.0.1

### Shell

Bash

### Additional context

I made sure that the versions of both the P4 host and C6 slave are the same. Logs on both devices do not show any errors, other than the one above. C6 shows zero packets and is properly reset by the P4 host. Tested on my two EV boards v1.2 and v1.4. This is for adding support for P4 to the Arduino project

me-no-dev, 2024-10-15:
Pin configuration on both host and slave have been verified to be consistent with schematic and documentation

me-no-dev, 2024-10-15:
Forgot to mention that everything is built with the latest `release/v5.3` branch of ESP-IDF

mantriyogesh, 2024-10-15:
We need following info for understanding:

Sdkconfig both sides, and complete textual logs,
Hosted registry component version used, any changes you have done in code or pins to be used.

The complete flashing steps and logs would be helpful.

me-no-dev, 2024-10-15:
`espressif/esp_hosted (0.0.22)` no changes have been done. Pins are as defaults, which have been checked to match the P4 EV board schematics.
[C6-flash-and-monitor.txt](https://github.com/user-attachments/files/17375119/C6-flash-and-monitor.txt)
[sdkconfig.slave.txt](https://github.com/user-attachments/files/17375126/sdkconfig.slave.txt)
[sdkconfig.p4.txt](https://github.com/user-attachments/files/17375166/sdkconfig.p4.txt)
[P4-flash-and-monitor.txt](https://github.com/user-attachments/files/17375214/P4-flash-and-monitor.txt)

me-no-dev, 2024-10-15:
this happens when the P4 reboots the C6, after which the same boot log repeats. This is what I meant by `repeats boot`

me-no-dev, 2024-10-15:
C6 boots correctly, the P4 attempts connect, timeouts and resets the C6 (using it's IO54)

mantriyogesh, 2024-10-15:
I can test this on ESP32-P4-Function-EV-Board v1.2 earliest by tomorrow. Let you know if I can reproduce with your sdkconfig.

Also, if works fine let me share the logs and observations.

me-no-dev, 2024-10-15:
Sure thing. I'm waiting for @igrr to give me known working binaries to at least figure out on which side is the problem. There is nothing specific I can add from our side. It's a basic WiFi scan firmware that runs on all other chips (that have WiFi).

me-no-dev, 2024-10-15:
Another unrelated issue I see is that always after the same number of unsuccessful attempts, the host reboots because of the following error:
```
E (1117493) transport: Failed to get ESP_Hosted slave transport up
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4003a48a
file: "./managed_components/espressif__esp_hosted/host/api/esp_hosted_api.c" line 183
func: esp_wifi_remote_init
expression: transport_drv_reconfigure()

abort() was called at PC 0x4ff0734f on core 1

```

mantriyogesh, 2024-10-15:
[c6_p4_sdio_flashing_0.0.22_idf_dbf5934b.tgz](https://github.com/user-attachments/files/17376780/c6_p4_sdio_flashing_0.0.22_idf_dbf5934b.tgz)

@me-no-dev , PFA C6 - SDIO - P4 tested binaries.

me-no-dev, 2024-10-15:
@mantriyogesh the problem seems to be in my p4 bin. I see that you compile with IDF 5.4. Can you try with 5.3? We can not yet update to 5.4 in Arduino

mantriyogesh, 2024-10-15:
Any specific idf commit in 5.3? or v5.3 in https://github.com/espressif/esp-idf/tags should be fine?

me-no-dev, 2024-10-15:
`v5.3.1-395-g707d097b017` is the version printed. The C6 firmware I compiled is working fine with your P4 bin. Only the P4 binary is wrong on our end

mantriyogesh, 2024-10-15:
I observed, by default esp_wifi_remote getting checked-out as 
```
[1/10] cmd_system (*) (/Users/yogesh/code/idf6/examples/system/console/advanced/components/cmd_system)
[2/10] esp-qa/ping-cmd (0.0.1)
[3/10] esp-qa/wifi-cmd (0.1.8)
[4/10] espressif/eppp_link (0.2.0)
[5/10] espressif/esp_hosted (0.0.22)
[6/10] espressif/esp_serial_slave_link (1.1.0)
[7/10] espressif/esp_wifi_remote (0.3.0)
[8/10] espressif/iperf (0.1.3)
[9/10] espressif/iperf-cmd (0.1.3)
[10/10] idf (5.3.1)
```

When I change, main/idf_component.yml so that  
```
  espressif/esp_wifi_remote:
    version: "~0.4.1"
```

It picked correct esp_wifi_remote (0.4.x needed for use hosted by default, else manually need to update from Wifi Remote menu)
```
[1/10] cmd_system (*) (/Users/yogesh/code/idf6/examples/system/console/advanced/components/cmd_system)
[2/10] esp-qa/ping-cmd (0.0.1)
[3/10] esp-qa/wifi-cmd (0.1.8)
[4/10] espressif/eppp_link (0.2.0)
[5/10] espressif/esp_hosted (0.0.22)
[6/10] espressif/esp_serial_slave_link (1.1.0)
[7/10] espressif/esp_wifi_remote (0.4.1)
[8/10] espressif/iperf (0.1.3)
[9/10] espressif/iperf-cmd (0.1.3)
[10/10] idf (5.3.1)
```

```diff
- [7/10] espressif/esp_wifi_remote (0.3.0)
+ [7/10] espressif/esp_wifi_remote (0.4.1)
```

[c6_p4_bins_2.tgz](https://github.com/user-attachments/files/17377798/c6_p4_bins_2.tgz)

me-no-dev, 2024-10-15:
it's `espressif/esp_wifi_remote (0.4.1)` on our end too

mantriyogesh, 2024-10-15:
https://github.com/espressif/esp-hosted/issues/511#issuecomment-2413750873

Binaries should work anyway, I built through release/v5.3

me-no-dev, 2024-10-15:
what commit is that, because I can not check it out. is it public 5.3 or gitlab?

me-no-dev, 2024-10-15:
I also see a very different output when starting hosted that I do not see on our end:
```
[0;32mI (383) host_init: ESP Hosted : Host chip_ip[18][0m
[0;32mI (411) H_API: ESP-Hosted starting. Hosted_Tasks: prio:23, stack: 5120 RPC_task_stack: 5120[0m
sdio_mempool_create free:172624 min-free:172624 lfb-def:131072 lfb-8bit:131072

[0;32mI (417) gpio: GPIO[18]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (426) gpio: GPIO[19]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (435) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (444) gpio: GPIO[15]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (454) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (463) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (472) H_API: ** add_esp_wifi_remote_channels **[0m
[0;32mI (478) transport: Add ESP-Hosted channel IF[1]: S[0] Tx[0x4000caee] Rx[0x4001a5ec][0m
[0;32mI (486) transport: Add ESP-Hosted channel IF[2]: S[0] Tx[0x4000ca34] Rx[0x4001a5ec][0m
```

me-no-dev, 2024-10-15:
Is it possible that we need to call some hosted API to properly init and not just call WiFi APIs?

mantriyogesh, 2024-10-15:
esp_hosted_host_init() is hooked through the component init. See here: 
https://github.com/espressif/esp-hosted/blob/83efce638beb0d59b641399df862b20e3aa12f36/host/esp_hosted_host_init.c#L10

This is hooked to component constructor. So do you mean you do not get to call this api automatically?

If you are using esp-idf, ideally this should have been called.

You case is interesting, let me know how can I reproduce this easily..

mantriyogesh, 2024-10-15:
```
![Screenshot 2024-10-15 at 8 51 04 PM](https://github.com/user-attachments/assets/673238f2-3171-46a4-9bd6-c3b79130bda5)

 I  ~/c/idf6   release/v5.3 *$…  examples/wifi/iperf  git remote -v  
origin	ssh://git@gitlab.espressif.cn:27227/espressif/esp-idf.git (fetch)
origin	ssh://git@gitlab.espressif.cn:27227/espressif/esp-idf.git (push)
```

This is indeed gitlab. But shouldn't matter, it was working on GitHub idf 5.3 also.

me-no-dev, 2024-10-15:
we use ESP-IDF in a slightly different way. I do expect this should have worked and indeed it's strange it did not.
in essence we build a very basic Arduino project in ESP-IDF (Arduino as component) and then extract all necessary libraries, headers, linker scripts and flags that were used by ESP-IDF to compile the project, then in Arduino we build the "Arduino component" again with the user code and link against everything extracted in the previous step. We also have some "constructor" code, but it is called in a slightly different way https://github.com/espressif/arduino-esp32/blob/idf-release/v5.3/cores/esp32/esp32-hal-misc.c#L256-L259

me-no-dev, 2024-10-15:
is there any other way that we can manually call `esp_hosted_host_init`?

mantriyogesh, 2024-10-15:
You can in app_main() 

Alternative 1) with extern, call this function.

Alternative 2) you can include header `#include esp_hosted_api.h` and call 
```
ESP_ERROR_CHECK(esp_hosted_init(NULL))
```

me-no-dev, 2024-10-15:
option 2 throws
```
In file included from /Users/ficeto/Documents/Arduino/hardware/espressif/esp32/tools/esp32-arduino-libs/esp32p4/include/espressif__esp_hosted/host/api/esp_hosted_api.h:27,
                 from /var/folders/74/n3f3k71s7lv_snk832kfgjfw0000gn/T/arduino_modified_sketch_53374/WiFiScan.ino:8:
/Users/ficeto/Documents/Arduino/hardware/espressif/esp32/tools/esp32-arduino-libs/esp32p4/include/espressif__esp_hosted/host/drivers/transport/transport_drv.h:106:6: error: variable or field 'process_priv_communication' declared void
  106 | void process_priv_communication(interface_buffer_handle_t *buf_handle);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/ficeto/Documents/Arduino/hardware/espressif/esp32/tools/esp32-arduino-libs/esp32p4/include/espressif__esp_hosted/host/drivers/transport/transport_drv.h:106:33: error: 'interface_buffer_handle_t' was not declared in this scope; did you mean 'StreamBufferHandle_t'?
  106 | void process_priv_communication(interface_buffer_handle_t *buf_handle);
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                 StreamBufferHandle_t
/Users/ficeto/Documents/Arduino/hardware/espressif/esp32/tools/esp32-arduino-libs/esp32p4/include/espressif__esp_hosted/host/drivers/transport/transport_drv.h:106:60: error: 'buf_handle' was not declared in this scope
  106 | void process_priv_communication(interface_buffer_handle_t *buf_handle);
      |                                                            ^~~~~~~~~~
/Users/ficeto/Documents/Arduino/hardware/espressif/esp32/tools/esp32-arduino-libs/esp32p4/include/espressif__esp_hosted/host/drivers/transport/transport_drv.h:129:23: error: 'interface_buffer_handle_t' was not declared in this scope; did you mean 'StreamBufferHandle_t'?
  129 | int serial_rx_handler(interface_buffer_handle_t * buf_handle);
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                       StreamBufferHandle_t
/Users/ficeto/Documents/Arduino/hardware/espressif/esp32/tools/esp32-arduino-libs/esp32p4/include/espressif__esp_hosted/host/drivers/transport/transport_drv.h:129:51: error: 'buf_handle' was not declared in this scope
  129 | int serial_rx_handler(interface_buffer_handle_t * buf_handle);
      |                                                   ^~~~~~~~~~
```

me-no-dev, 2024-10-15:
option 1 I doubt it will work, because the function is `static`

me-no-dev, 2024-10-15:
```
WiFiScan:8:8: error: 'static' specifier conflicts with 'extern'
    8 | extern static void esp_hosted_host_init(void);
      | ~~~~~~ ^~~~~~
```

me-no-dev, 2024-10-15:
this worked
```cpp
extern "C" {
esp_err_t esp_hosted_init(void *);
}
```

me-no-dev, 2024-10-15:
small nitpick. The following text is printed regardless of the log level. Can it please be changed to INFO or lower?
```
sdio_mempool_create free:34125312 min-free:34124992 lfb-def:33030132 lfb-8bit:33030132

Name: 
Type: SDIO
Speed: 40.00 MHz (limit: 40.00 MHz)
Size: 0MB
CSD: ver=1, sector_size=0, capacity=0 read_bl_len=0
SCR: sd_spec=0, bus_width=0
TUPLE: DEVICE, size: 3: D9 01 FF 
TUPLE: MANFID, size: 4
  MANF: 0092, CARD: 6666
TUPLE: FUNCID, size: 2: 0C 00 
TUPLE: FUNCE, size: 4: 00 00 02 32 
TUPLE: CONFIG, size: 5: 01 01 00 02 07 
TUPLE: CFTABLE_ENTRY, size: 8
  INDX: C1, Intface: 1, Default: 1, Conf-Entry-Num: 1
  IF: 41
  FS: 30, misc: 0, mem_space: 1, irq: 1, io_space: 0, timing: 0, power: 0
  IR: 30, mask: 1,   IRQ: FF FF
  LEN: FFFF
TUPLE: END
```

mantriyogesh, 2024-10-15:
great thanks for the feedback, will add this in. btw, is everything running fine now?

me-no-dev, 2024-10-15:
Please do let me know if you change the behavior so I can make sure that everything on our side is OK. I am fine with this as-is, because it allows me to init SDIO only if WiFi is used on the P4 in Arduino (thus allow for boards without C6 to use the pins as they wish)

mantriyogesh, 2024-10-15:
Sure. Your username is very much thoughtful and easy to remember!
I will notify once I get around this.