espressif/esp-hosted #512: esp_as_hosted 工程编译报错

HP4LEVI, 2024-10-17:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

现在我在尝试esp_as_hosted,我使用ESP32-S3为slave，ESP32-C3为hosted
slave部分已经成功烧录。
但在iperf工程编译中出现报错的现象，具体为：各类函数报错，找不到函数或者函数重复定义
请问是某个操作上的事务所导致的吗？

[982/984] Linking CXX executable iperf.elf
FAILED: iperf.elf
cmd.exe /C "cd . && C:\Users\26891\.espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc_zicsr_zifencei -nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C3=0 -Wl,--Map=C:/Users/26891/esp/workplace/iperf/build/iperf.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32c3.peripherals.ld -T esp32c3.rom.ld -T esp32c3.rom.api.ld -T esp32c3.rom.libgcc.ld -T esp32c3.rom.version.ld -T esp32c3.rom.eco3.ld -T esp32c3.rom.newlib.ld -T memory.ld -T sections.ld @CMakeFiles\iperf.elf.rsp -o iperf.elf  && cd ."
C:/Users/26891/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/26891/esp/v5.3/esp-idf/components/esp_wifi/lib/esp32c3/libnet80211.a(ieee80211_ioctl.o): in function `wifi_deinit':
(.text.wifi_deinit+0x0): multiple definition of `wifi_deinit'; esp-idf/espressif__esp_hosted/libespressif__esp_hosted.a(rpc_slave_if.c.obj):C:/Users/26891/esp/workplace/iperf/managed_components/espressif__esp_hosted/host/drivers/rpc/slaveif/rpc_slave_if.c:176: first defined here
C:/Users/26891/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/26891/esp/v5.3/esp-idf/components/esp_wifi/lib/esp32c3/libnet80211.a(ieee80211_ioctl.o): in function `wifi_set_country':
(.text.wifi_set_country+0x0): multiple definition of `wifi_set_country'; esp-idf/espressif__esp_hosted/libespressif__esp_hosted.a(rpc_slave_if.c.obj):C:/Users/26891/esp/workplace/iperf/managed_components/espressif__esp_hosted/host/drivers/rpc/slaveif/rpc_slave_if.c:314: first defined here
C:/Users/26891/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/26891/esp/v5.3/esp-idf/components/esp_wifi/lib/esp32c3/libnet80211.a(ieee80211_ioctl.o): in function `wifi_get_country':
(.text.wifi_get_country+0x0): multiple definition of `wifi_get_country'; esp-idf/espressif__esp_hosted/libespressif__esp_hosted.a(rpc_slave_if.c.obj):C:/Users/26891/esp/workplace/iperf/managed_components/espressif__esp_hosted/host/drivers/rpc/slaveif/rpc_slave_if.c:320: first defined here
C:/Users/26891/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/espressif__esp_hosted/libespressif__esp_hosted.a(esp_hosted_api.c.obj): in function `add_esp_wifi_remote_channels':
C:/Users/26891/esp/workplace/iperf/managed_components/espressif__esp_hosted/host/api/esp_hosted_api.c:73:(.text.esp_hosted_init+0x9e): undefined reference to `esp_wifi_remote_channel_rx'
C:/Users/26891/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/26891/esp/workplace/iperf/managed_components/espressif__esp_hosted/host/api/esp_hosted_api.c:73:(.text.esp_hosted_init+0xa4): undefined reference to `esp_wifi_remote_channel_rx'
C:/Users/26891/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/26891/esp/workplace/iperf/managed_components/espressif__esp_hosted/host/api/esp_hosted_api.c:87:(.text.esp_hosted_init+0xba): undefined reference to `esp_wifi_remote_channel_set'
C:/Users/26891/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/26891/esp/workplace/iperf/managed_components/espressif__esp_hosted/host/api/esp_hosted_api.c:87:(.text.esp_hosted_init+0xc4): undefined reference to `esp_wifi_remote_channel_rx'
C:/Users/26891/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld.exe: C:/Users/26891/esp/workplace/iperf/managed_components/espressif__esp_hosted/host/api/esp_hosted_api.c:92:(.text.esp_hosted_init+0xd8): undefined reference to `esp_wifi_remote_channel_set'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

mantriyogesh, 2024-10-17:
https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/troubleshooting.md#1-esp-host-to-evaluate-already-has-native-wi-fi

mantriyogesh, 2024-10-17:
ESP32-C3 as host , already natively supports Wi-Fi. Please disable native wifi support if you use C3 as host.

HP4LEVI, 2024-10-17:
抱歉，原来是因为这个。但是我可能不太清楚这个文件在什么地方，方便告知一下吗？

HP4LEVI, 2024-10-17:
[image]

mantriyogesh, 2024-10-17:
is is not in esp-hosted.

Please refer to the troubleshoot link earlier.

```
Edit the ESP-IDF file components/soc/<soc>/include/soc/Kconfig.soc_caps.in
```

HP4LEVI, 2024-10-17:
我想我知道您的意思了