espressif/esp-hosted #513: WiFi connection example for ESP32-P4-Function-EV-Board

dmartauz, 2024-10-17:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Please provide minimal example for the P4 host using C6 connected over SDIO to make a connection to an access point, possibly without any unnecessary dependencies such as esp-bsp.

### Use cases

Basic testing of esp-hosted compatibility with selected version of esp-idf.

mantriyogesh, 2024-10-17:
Hello @dmartauz ,

We had provided steps to integrate into simple IDF example, https://github.com/espressif/esp-idf/tree/master/examples/wifi/iperf. any other wifi example could be integrated with same steps.

steps : https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host?tab=readme-ov-file#4-quick-demo-with-esp32-p4-function-ev-board

mantriyogesh, 2024-10-17:
regarding earlier question, to flash c6, you really don't need esp-prog method and can rely on [OTA](https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md#51-ota-updates) if hosted slave was already flashed or pre-flashed board.

refer to

dmartauz, 2024-10-17:
I am attempting to run iperf example exactly described here: https://github.com/espressif/esp-idf/tree/master/examples/wifi/iperf
```
iperf> sta_connect "XXXX" "****"
I (5219) WIFI: Connecting to XXXX...
I (5236) H_API: esp_wifi_remote_connect
I (5255) WIFI: DONE.WIFI_CONNECT_START,OK.
iperf> I (8274) esp_wifi_remote: esp_wifi_internal_reg_rxcb: sta: 0x4004d508
I (8276) WIFI: WIFI_EVENT_STA_CONNECTED!
I (9292) esp_netif_handlers: sta ip: 192.168.1.106, mask: 255.255.255.0, gw: 192.168.1.1
I (9293) WIFI: IP_EVENT_STA_GOT_IP: Interface "sta" address: 192.168.1.106
I (9304) WIFI: - IPv4 address: 192.168.1.106,
iperf>
iperf> iperf -s -i 2
I (23700) IPERF: mode=tcp-server sip=localhost:5001, dip=0.0.0.0:5001, interval=2, time=30
I (23701) iperf: Socket created
iperf> E (33710) iperf: iperf_run_tcp_server(325): Unable to accept connection: errno 11
I (33711) iperf: iperf exit
```
It connects to an AP but then throws this error. Is it only a matter of no iperf client trying to connect? I am still running factory image on C6.

mantriyogesh, 2024-10-17:
Nothing related to esp-hosted.

esp-idf example behaviour.
if you wish can configure as you need using `idf.py menuconfig`.

[image]

dmartauz, 2024-10-17:
You are right, I increased the timeout and the error was postponed. I can even connect with windows iperf client. That means that esp-hosted is running with recent esp-idf/master clone and with factory C6 firmware of ESP32-P4-Function-EV-Board.
I think the problem is in wifi init routine implemented in my application and this is what I will focus on. I am porting my application from S3 and due to using esp_wifi_remote I was expecting it to work without any modification.

mantriyogesh, 2024-10-17:
Yes esp-hosted is writtent such a way that existing application should be as is portable. A good number of APIs of wifi are supported, some advanced use case still not, some time later. In general, it should cater to lot many APIs generally used. Any specific API which might be missing that you need, please let us know.