espressif/esp-hosted #514: esp-hosted关于蓝牙a2dp问题？

1653323560, 2024-10-17:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

请问esp-hosted方案是否支持蓝牙a2dp功能？我看到esp32与树莓派之间并没有pcm或i2s连接，那么音频信号是通过串口传输的吗？

我还有一个问题esp-hosted-ng方案是否支持Sation+softap模式？如果我需要该功能是否只能选用esp-hosted-fg方案？相比于ng方案，fg是否在传输速率上远小于ng方案？

mantriyogesh, 2024-10-17:
> Does the esp-hosted solution support the Bluetooth A2DP function?

Yes. A2DP is supported with ESP32 as slave.
There is single radio, which is time-shared between Wi-Fi & Bluetooth in ESP chipsets.
Do you intend to use Wi-Fi and Bluetooth at the same time? In those cases, Bluetooth priority (dynamically) would bump when there is packet pending. For short time, Wi-Fi throughput would be reduced. When packets drained, the priority would be auto changed. As A2DP can't bear high jitter, bluetooth on rf needs to be software priority.

> I see that there is no PCM or I2S connection between esp32 and Raspberry Pi, so is the audio signal transmitted through the serial port?

The bluetooth could take either multiplexed (Wi-Fi + Bluetooth) on same transport like SPI/SDIO. or it could take dedicated transport of UART.
On UART, it is standard HCI. on SPI/SDIO, it is vHCI (extra hosted header to multiple packet type Wifi/Bluetooth).

> I have another question: Does the esp-hosted-ng solution support the Sation+softap mode?

Currently, NG doesn't support STA+SoftAP.

> If I need this function, can I only use the esp-hosted-fg solution?

Yes. FG supports STA+SoftAP since long.

> Compared with the ng solution, is the transmission rate of fg much lower than that of the ng solution?

Transport is same. so the throughput is also just the same. Only differs is the way Wi-Fi connection is established. to know more how they differ on setting up wifi, check https://github.com/espressif/esp-hosted?tab=readme-ov-file#2-esp-hosted-fg-vs-esp-hosted-ng

1653323560, 2024-10-17:
> Yes. A2DP is supported with ESP32 as slave.
> There is single radio, which is time-shared between Wi-Fi & Bluetooth in ESP chipsets.
> Do you intend to use Wi-Fi and Bluetooth at the same time? In those [...]

那么esp-hosted-ng的优势是什么呢？是否在大多数应用中都会选择esp-hosted-fg?

mantriyogesh, 2024-10-18:
Most of the times, both are applicable, it depends on the use-case and preference.

mantriyogesh, 2024-10-19:
Please feel free to re-open if any issues you face, by the time, I would close the issue.

closed 2024-10-19 by mantriyogesh