espressif/esp-hosted #515: Can I use blue on uart and wifi on spi the same time？

Speakpig, 2024-10-18:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I've basicaly complete my test for ESP32 as host and C3 as slave for wifi use, I'd like to use blue the same time, is it possible? I've read some docs, but I'm somehow still a bit blurry, and the svg pics often apear to be available as below, I may really use much advice, thx!

[image]

mantriyogesh, 2024-10-18:
Hello @Speakpig ,

 bluetooth uart and wifi on spi is possible and is supported.

I understand you are following the doc here https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/bluetooth_design.md

I am not sure why it is not loading the graphics. It is standard svg. Can you try other browser?

Anyway, there seems little Miss in the documentation how to enable this dual mode.
You can enable the bluetooth at host using IDF.py menuconfig. Nimble runs as host software and uses uart pins to communicate standard HCI over uart.

mantriyogesh, 2024-10-18:
Can you please try idf.py menuconfig way by the time? Remember rx at host should go to tx of slave and vice versa. Also baud rate need to be exactly same.

May I know the browser and version you are using? I will try it myself why the issue is faced for graphics.

Svg makes the image zoomable without loosing the context. But if it is not loadable on some browsers we might have to fall back to pure mermaid or something.

Speakpig, 2024-10-18:
Sure, I‘m using the newest chrome.

[image]

mantriyogesh, 2024-10-18:
I have some cached doc for the time being, 
[Hosted Bluetooth Interface · Wiki · app_frameworks _ esp_hosted · GitLab.pdf](https://github.com/user-attachments/files/17432242/Hosted.Bluetooth.Interface.Wiki.app_frameworks._.esp_hosted.GitLab.pdf)

Speakpig, 2024-10-18:
Thx, that's very helpful!

mantriyogesh, 2024-10-19:
did you get idf.py menuconfig way working?

It would be extremely simple to activate the vHCI (multiplexed mode) and then afterevaluation, can configure Bluetooth on dedicated uart.

Let us know if you get stuck anywhere.

Speakpig, 2024-10-21:
I'm trying to set up blue through menuconfig now, but it seems that there is no specific guidance on this right now, since I want to use wifi on spi, blue on uart respectively but at the same time, how shall I set up menuconfig:
1.In my usage method, it is a HCI or a vHCI, and how to change it specifically?
2.this what I've done, what's more to be done?
https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/examples/bleprph_host_only_vhci/README.md

Speakpig, 2024-10-21:
On the host side, there is only one transport layer to be set, does this means, I can't get this set up?

> use wifi on spi, blue on uart respectively but at the same time

[image]

Speakpig, 2024-10-21:
On the host side, In the following setting options there is only one optional setting item which is vHCI, does this means HCI is not surpported right now?
[image]

mantriyogesh, 2024-10-21:
1. Can you first try to just use the https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/examples/bleprph_host_only_vhci/README.md as is? the config used for vHCI can be found out at https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/examples/bleprph_host_only_vhci/sdkconfig.defaults#L7-L17

```diff
 I  ~/c/idf6   release/v5.3 *$…  examples/wifi/iperf  git diff sdkconfig.defaults.esp32c3                                                                                     30.2s  Mon Oct 21 13:26:54 2024
diff --git a/examples/wifi/iperf/sdkconfig.defaults.esp32c3 b/examples/wifi/iperf/sdkconfig.defaults.esp32c3
index 7a46a91d21..6820ca2423 100644
--- a/examples/wifi/iperf/sdkconfig.defaults.esp32c3
+++ b/examples/wifi/iperf/sdkconfig.defaults.esp32c3
@@ -23,3 +23,9 @@ CONFIG_ESPTOOLPY_FLASHFREQ_80M=y

 CONFIG_LWIP_TCPIP_CORE_LOCKING=y
 CONFIG_LWIP_TCPIP_CORE_LOCKING_INPUT=y
+
+CONFIG_BT_ENABLED=y
+CONFIG_BT_CONTROLLER_DISABLED=y
+CONFIG_BT_BLUEDROID_ENABLED=n
+CONFIG_BT_NIMBLE_ENABLED=y
+CONFIG_BT_NIMBLE_TRANSPORT_UART=n
```

```
rm -rf sdkconfig buiild
idf.py set-target esp32c3
```

2. Actually, your question is right, the transport is only the main transport. basically where wifi  (or wifi and vhci bluetooth) can be run.
The doc needs to be updated , along with KConfig, to use HCI.

The standard HCI over UART is tested fine, works fine. actually, standard hci runs outside the hosted understanding and doesn't have dependency over hosted to run. what i mean, the nimble stack over uart is started, with correct rx, tx , gnd pins and correct baud rate, it should just work.

We have yet not included it in the doc or KConfig, which clearly needs to be taken into consideration and needs to be fixed. However, standard hci just would work fine. Ensure the uart pins are dedicated for uart only.

The KConfig option for nimble to use the transport is decided by config, `(Top) → Component config → Bluetooth → NimBLE Options → Host-controller Transport` . this is outside ESP-Hosted config. In ESP-Hosted KConfig, we also need to make user aware of this setting, to avoid transport confusion.

Another problem is that c3 also supports the blutooth controller. but you should only use c3 in bluetooth stack only, as you intend it to be running as some host, where bluetooth would not be supported natively and rely on the co-processor ble controller.

I understand the whole confusion. We will try to fix this asap. but in a nutshell, you can really use uart as transport, with standard hci.
but before trying hci, I would strongly suggest running the [example](https://components.espressif.com/components/espressif/esp_hosted/versions/0.0.22/examples/bleprph_host_only_vhci?language=en) as is to get idea of vHCI first.

Speakpig, 2024-10-21:
Ok,you have made your point, but it's much for a beginner, I'm now a bit confused, if I do `(Top) → Component config → Bluetooth → NimBLE Options → Host-controller Transport`, it will be presenting like this, so what mode actually am I into, HCI or vHCI, or someting out my understanding?
[image]
[image]

Speakpig, 2024-10-21:
> but before trying hci, I would strongly suggest running the [example](https://components.espressif.com/components/espressif/esp_hosted/versions/0.0.22/examples/bleprph_host_only_vhci?language=en) as is to get idea of vHCI first.

Sure, I will try vHCI first. Again, actually I'm using ESP32 as host and C3 as slave now, and I've get what's been metioned done as the [link](https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/examples/bleprph_host_only_vhci/README.md#configure-the-project), what shall I do next?

mantriyogesh, 2024-10-21:
> actually I'm using ESP32 as host and C3 as slave now

Oh, I noted now.

Do you get the nimble traces in the log?
If yes, the nimble message exchanges are already done. you can use some other ble node (phone/laptop) to scan  & connect it it?

Speakpig, 2024-10-21:
Not yet, I'm not quite clear how to make use of the example `esp_as_mcu_host/examples/bleprph_host_only_vhci`, now actually I'm still running iperf as example on the host side, how shall I build a new example from this folder? And how shall I do some Kconfig as mentioned upside?I've raised a few questions upside, maybe you can help me out please?

mantriyogesh, 2024-10-21:
no, please create different example in separate directory using
```
idf.py create-project-from-example "espressif/esp_hosted^0.0.22:bleprph_host_only_vhci"
```

once you test this, you can integrate these two. if you don't understand, we can help you to do so.

mantriyogesh, 2024-10-21:
the iperf uses esp-idf component, 'console' which is also used in the ble example, which might clash.
These things are higher layer for esp-hosted, but anyway for use experience, we can create a different example integrating these two.

the nimble is scenario specific enough, so even if we do above, it may hold zero value as it would be only specific scenario of nimble.

Speakpig, 2024-10-21:
It's getting too complicated, but sure, I will the example out first, thx.

Speakpig, 2024-10-21:
[image]
This example can't pass the build, what may have I missed?

mantriyogesh, 2024-10-21:
Can you please use latest esp-idf master?

Speakpig, 2024-10-22:
I installed ESP-IDF V5.3 through offline installation, how can I install and use ESP-IDF master at the same time? I think I knew something before, and now I forgot about it.

mantriyogesh, 2024-10-22:
I think we bugged you enough on this. Let me check this and fix this for you for vHCI.

Can you please let us know exact IDF commit you are on, and possibly provide the textual logs from start for both slave and host? Also attach your sdkconfig both sides, to assess any configuration conflicts.

We can test on Linux and Mac. Windows also should work ideally the same.

mantriyogesh, 2024-10-22:
@Speakpig ?

Speakpig, 2024-10-22:
Sry, I've been trying to checkout branch master, but I'm now confused how to make ESP-IDF v5.3 and banch master both on my windows, and switch from one to another. I'm not bugged, I'm happy to explore this out and finally make use of it in my application but it's hard for a beginner, thanks for your help.

Speakpig, 2024-10-22:
[image]

[image]

[image]

[image]

How shall I set the directory since I have got a v5.3? Will this work as I assmued? Am I doing the right thing?

Speakpig, 2024-11-19:
@mantriyogesh Hello sir, is there a master-side and slave-side instance or just a slave-side instance running Bluetooth and WIFI at the same time, can you give such an instance, or is there a plan for when such an instance will be launched in the future? I need such an instance to evaluate the feasibility of implementing my gd32f470 as a host using ESP bluetooth with wifi functionality.

mantriyogesh, 2024-11-19:
As a matter of fact , we are getting slave and host have together wifi connection.
Basically slave or host ( as per config ) would complete control the wifi. Other party would just get the ip event. Once ip received the chip starts network function. We also want to allow the host could sleep, which we demo on P4 as host (for your host, deep sleep need to handle at host side code by user).
Slave would wakeup sleeping host if need be.

For bluetooth, we wish we do this but unsure  for now, how easily it could happen, considering API changes for every stack.

mantriyogesh, 2024-11-19:
We have slave controlling the wifi fully, but in the development version. Do you wish to try it out?