espressif/esp-hosted #516: What is the means aobout "read_len" when send a cmd to esp-slave?

LuciaferKelvin, 2024-10-19:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

hello there.
I rebulid my project host recently , And I find one varriable I can`t understand at the demo code.
I am using the firmware version v0.4 in the slave.
My wonder is in that code.
`uint8_t * transport_pserial_data_handler(uint8_t* data, uint16_t data_length,
		uint8_t wait, uint32_t* pro_len)
{
	int count = 0, ret = 0;
	uint16_t buf_len = 0, read_len = 0;
	uint8_t *write_buf = NULL, *read_buf = NULL;
	struct esp_hosted_driver_handle_t* esp_hosted_driver_handle = NULL;

/*
 * TLV (Type - Length - Value) structure is as follows:
 * --------------------------------------------------------------------------------------------
 *  Endpoint Type | Endpoint Length | Endpoint Value  | Data Type | Data Length | Data Value  |
 * --------------------------------------------------------------------------------------------
 *
 *  Bytes used per field as follows:
 * --------------------------------------------------------------------------------------------
 *       1        |        2        | Endpoint length |     1     |      2      | Data length |
 * --------------------------------------------------------------------------------------------
 */
	buf_len = SIZE_OF_TYPE + SIZE_OF_LENGTH + strlen(ep_name) +
		SIZE_OF_TYPE + SIZE_OF_LENGTH + data_length;

	read_len = buf_len - data_length;

	write_buf = (uint8_t* )esp_hosted_calloc(1, buf_len);
	if (!write_buf) {
		command_log("Failed to allocate memory \n");
		return NULL;
	}

	esp_hosted_driver_handle = esp_hosted_driver_open(transport);
	if (!esp_hosted_driver_handle) {
		esp_hosted_free(write_buf);
		write_buf = NULL;
		return NULL;
	}

	count = compose_tlv(write_buf, data, data_length);
	if (!count) {
		command_log("Failed to compose TX data\n");
		goto err;
	}

	ret = esp_hosted_driver_write(esp_hosted_driver_handle, write_buf, count, &count);
	if (ret != SUCCESS) {
		command_log("Failed to write TX data\n");
		goto err;
	}

	read_buf = esp_hosted_driver_read(esp_hosted_driver_handle, read_len, wait, pro_len);
	if (! read_buf) {
		command_log("Failed to read RX data\n");
		goto err;
	} else if (!pro_len) {
		command_log("Zero protobuf data length received \n");
	}

	ret = esp_hosted_driver_close(&esp_hosted_driver_handle);
	if (ret != SUCCESS) {
		command_log("Failed to close driver interface\n");
	}

	return read_buf;
err:
	if (write_buf) {
		esp_hosted_free(write_buf);
		write_buf = NULL;
	}

	ret = esp_hosted_driver_close(&esp_hosted_driver_handle);
	if (ret != SUCCESS) {
		command_log("Failed to close driver interface\n");
	}
	return NULL;
}
`
First：
The "read_len" is calculate by the “read_len = buf_len - data_length;”.
That is right I can understand.
But I found the slave can send some response which the host not request when the slave lost connect with the route in station mode.
My host always send one cmd ,then wait for the response.
The "read_len" is provide for vefiy the response.
but if the response is not what the host resquest, 
May it caused some problems?Can I set the read_len to the Max, only let the code using the len what the host had received?Or set a vlaue which is suitable for the host memory？
Second：
The reason let me to rebuild the my project that I found the slave will always reponse the host last request instead of current request. It let the host can not get the current request`s repsonse.
This issue is quite difficult to trigger, with a probability of one in ten thousand. I am unable to reproduce the problem, but through analyzing the logs from the host controller`s log, I can identify the occurrence of this phenomenon. Therefore, I have decided to change my software design.
What I am software design now is that the host always let the esp_hosted_driver_handle open.
The always check if the slave respsonse something to the host , and send to the application the response cmd.
Control send the cmd queue, If the slave repsonse current cmd  then send the next cmd.
Wait for 15 seconds send the next cmd if the slave not repsonse current cmd.
Now it work normaly in my test for 1 days.
Does the software design could be work, Is there has any suggestions?
Thank you so much

mantriyogesh, 2024-10-20:
Hello @LuciaferKelvin ,

Thank you for using ESP-Hosted and your continued understanding.

I couldn't understand first problem exactly but the second problem was fixed in master. Master doesn't face these issues, btw.

Also for not I am thinking first problem to be after effect of second problem.

Is it possible to try to migrate to master?
It is most natural and maintainable way to support. But let me remind you, 0.4 to 0.0.5 is design change. Also, 
If not, I can get you some selective fixes if you could try them. (We generally do not support on older branches, marking this case as exception).

We had observed part of these issues, I can provide you some commits .

There are two alternatives,
1. Faster, without testing myself . I can provide the commits. For you to test and be back. 
If you get some issues with that, please let me know, then I will test myself. 
2. Considerable Slower, I test myself and get it working and provide you commits. 

But anyway, fixing this involve debugging, back and first, which involve your patience and understanding for fixing this correctly for you.

> [!Note]
>
> I am addressing second issue only right now, as root cause for first one. If you face first issue despite, let me know.

LuciaferKelvin, 2024-10-20:
hello @mantriyogesh ,
for my project , I can change the slave version ,because many of the productions was using by customer.
for First questions, I am not ask for the solution. Just need to know the "read_len" is must set by the command which send to slave?
could I set up the value by a const value, In the function "transport_pserial_data_handler" it always get the "read_len" when send some command, then it will check the response when the slave feed back, and get the min value between "read_len" and the feed back length value, it may caused any problem if I do that?
The second one, I changed my host using case. And In my test , the control is working normal . For that I only also just want to know is any possible problem may be trigger if I using the control case .

mantriyogesh, 2024-10-21:
I'm checking this, please spare some time to let me get back on this.

LuciaferKelvin, 2024-10-21:
Thank you so much

mantriyogesh, 2024-10-21:
Some code explanation, before directly going to answer:

## Code understanding
### TLV understanding (two TLVs always)
```
/*
 * TLV (Type - Length - Value) structure is as follows:
 * --------------------------------------------------------------------------------------------
 *  Endpoint Type | Endpoint Length | Endpoint Value  | Data Type | Data Length | Data Value  |
 * --------------------------------------------------------------------------------------------
 *
 *  Bytes used per field as follows:
 * --------------------------------------------------------------------------------------------
 *       1        |        2        | Endpoint length |     1     |      2      | Data length |
 * --------------------------------------------------------------------------------------------
 */
```

Basically, first TLV is constant, to let know the 'control endpoint'.
if first tlv is correct, only then it would go for second tlv in this. 
T - Tag (1 byte) (PROTO_PSER_TLV_T_**EPNAME**)
L - Length - 2 bytes
V - the endpoint string ('control' for v0.0.4)

Second TLV:
T - PROTO_PSER_TLV_T_**DATA**
L - Proto encoded length
V - Actual proto msg (of length L)

### esp_hosted_driver_read()

###### Fixed read bytes
```
read_buf = esp_hosted_driver_read(esp_hosted_driver_handle, read_len, wait, pro_len);
```

Every time, the read_len to this function passed as size:

 Endpoint Type | Endpoint Length | Endpoint Value  | Data Type | Data Length | ~~Data Value~~ |
 
as `Data Value` is to be understood after parse_tlv(). Anything before is fixed length. We read till that part here:

the code in `esp_hosted_driver_read()`,
```c
                do {
                    count = read(esp_hosted_driver_handle->file_desc,
                            (buf+total_read_len), (read_len-total_read_len));
                    if (count <= 0) {
                        perror("read: Failed to read fixed length serial data");
                        goto err;
                    }
                    total_read_len += count;
                } while (total_read_len < read_len);

                if (total_read_len != read_len) {
                    printf("%s, Expected fixed data %d vs received %d\n",__func__, read_len, total_read_len);
                    goto err;
                }
```

###### Fetch the variable `read_length` i.e. 'Data Length' above and `read_pointer` for 'Data Value'

```c
                ret = parse_tlv(buf, buf_len);
                if ((ret != SUCCESS) || !*buf_len) {
                   goto err;
                }
```
This would verify the `first tlv` correctness and fetch the `pro_len` i.e. Data Length' above.

[... 92 lines cut ...]

I am not very clear, but as long as 
```c
    read_buf = esp_hosted_driver_read(esp_hosted_driver_handle, read_len, wait, pro_len);
    if (! read_buf) {
        command_log("Failed to read RX data\n");
        goto err;
    } else if (!pro_len) {
        command_log("Zero protobuf data length received \n");
    }
```
Returns `pro_len` as the 'Data Value' in second TLV, which is proto buff length. Function returns, `read_buf` which is passed to 
```c
resp = esp_hosted_config_payload__unpack(NULL, rx_len, rx_data);
```
in caller of transport_pserial_data_handler().

As good as the following conditions are made sure, you should really be good to go:
1. No unexpected 'read failure' returned.
I might be wrong, if not understand correctly, can come in your case? Due to read more 'static' bytes than the actual read length.
What i meant, `read()` only the valid bytes. Make sure you do not block on read() as well.
2. esp_hosted_driver_read() returns the pointer to protobuf encoded buffer
3. esp_hosted_driver_read() returns the sizeof protobuf encoded buffer

In general, If you pass more number of bytes than the real size, protobuf decoder will stop after decoding the actual number of bytes from the encoded buffer. But we should always avoid passsing more than required bytes, to (1) future compatibility (2) if transport buffer was corrupted due to hidden bug, it may crash in protobuf (just saying possibility) (3) possible security vulnerabilities or attcks using this case

mantriyogesh, 2024-10-21:
Please bear with my last large response, I just wanted to ensure that the low level details and understanding is clear, and there is no confusion or assumptions.

LuciaferKelvin, 2024-10-21:
Ops， seems we have check the different version with the demo, In my version ,the esp read function is 
`uint8_t* esp_hosted_driver_read (struct esp_hosted_driver_handle_t* esp_hosted_driver_handle,
	int read_len, uint8_t wait, uint32_t* buf_len)
{
	uint16_t rx_buf_len = 0;
	uint8_t* read_buf = NULL;
	uint8_t* buf = NULL;
	int len = 0, ret = 0;

	if (!esp_hosted_driver_handle || !read_len || !buf_len) {
		printf("Invalid parameters in read\n\r");
		return NULL;
	}
	if(! readSemaphore) {
		printf("Semaphore not initialized\n\r");
		return NULL;
	}

    if (osSemaphoreAcquire(readSemaphore , wait * MILLISEC_TO_SEC ) != osOK) {
		return NULL;
	}

	if( (!serial_if_g) ||
		(!serial_if_g->fops) ||
		(!serial_if_g->fops->read)) {
		printf("serial interface refusing to read\n\r");
		return NULL;
	}

	read_buf = serial_if_g->fops->read(serial_if_g, &rx_buf_len);
	if ((! read_buf) || (!rx_buf_len)) {
		printf("serial read failed\n\r");
		return NULL;
	}
	print_hex_dump(read_buf, rx_buf_len, "Serial read data");

/*
 * Read Operation happens in two steps because total read length is unkown
 * at first read.
 *      1) Read fixed length of RX data
 *      2) Read variable length of RX data
 *
 * 1) Read fixed length of RX data :
 * Read fixed length of received data in below format:
 * ----------------------------------------------------------------------------
 *  Endpoint Type | Endpoint Length | Endpoint Value  | Data Type | Data Length
 * ----------------------------------------------------------------------------
 *
 *  Bytes used per field as follows:
 *  ---------------------------------------------------------------------------
 *      1         |       2         | Endpoint Length |     1     |     2     |
 *  ---------------------------------------------------------------------------
 *
 *  read_len = 1 + 2 + Endpoint length + 1 + 2
 */

	len = min(rx_buf_len,read_len);
	buf = (uint8_t* )esp_hosted_calloc(1, len);
	if (!buf) {
		printf("Failed to allocate memory \n\r");
		goto err;
	}

	memset(buf, 0, read_len);
	memcpy(buf, read_buf, len);

 /* parse_tlv function returns variable payload length of received data in buf_len */
	ret = parse_tlv(buf, buf_len);
	if (ret != STM_OK) {
		esp_hosted_free(buf);
		printf("Failed to parse RX data \n\r");
		goto err;
	}

	esp_hosted_free(buf);

	buf = NULL;
/*
 * 2) Read variable length of RX data:
 */
	buf = (uint8_t* )esp_hosted_calloc(1, *buf_len);
	if (!buf) {
		printf("Failed to allocate memory \n\r");
		goto err;
	}

	memcpy((buf), read_buf+read_len, *buf_len);

	esp_hosted_free(read_buf);
	read_buf = NULL;
	return buf;
err:
	if (read_buf) {
		esp_hosted_free(read_buf);
	}
	return NULL;
}

int esp_hosted_driver_close(struct esp_hosted_driver_handle_t** esp_hosted_driver_handle)
{
	if (!esp_hosted_driver_handle || !(*esp_hosted_driver_handle)) {
		printf("Invalid parameter in close \n\r");
	}
	esp_hosted_free(*esp_hosted_driver_handle);
    *esp_hosted_driver_handle = NULL;
	return STM_OK;
}`

The verfiy code is `len = min(rx_buf_len,read_len);`

mantriyogesh, 2024-10-21:
Basically, your host is stm32 based, I assume?
Please confirm?

I assumed Linux hosts in earlier responses.
The theory is still valid,  Let me cross-check again for stm32 code base.

LuciaferKelvin, 2024-10-21:
Sorry, It`s my fault. I had not declear that.
My host is stm32 based.

mantriyogesh, 2024-10-21:
for stm32, 

### Base code
process_rx_task -> serial_rx_handler(&buf_handle); 

- buf_handle.payload_len -> spi rx
  - full frame length (excludes esp-hosted header and but includes control path first TLV)
- buf_handle.payload 
  - Serial frame along with control path first TLV (esp-hosted header trimmed already) 
 
this is                    
```c
if (pdTRUE != xQueueSend(from_slave_queue,
                                &buf_handle, portMAX_DELAY))
//and
    if (serial_ll_hdl->serial_rx_callback) {
        (*serial_ll_hdl->serial_rx_callback) ();
}
```
serial_rx_callback opens the `readSemaphore` in `esp_hosted_driver_read()` (keep this aside for later reference)

```c
ret = xQueueReceive(from_slave_queue, &buf_handle, portMAX_DELAY);
```

then 
```c
 if (buf_handle.if_type == ESP_SERIAL_IF) {

            /* serial interface path */
            serial_rx_handler(&buf_handle);
}
```

i.e. 
```c
serial_ll_rx_handler()
```

Full frame is re-created (if serial data was fragmented)
```c
memcpy(serial_buf, r.data, r.len);
```

then sent to serial task
```c
if (pdTRUE != xQueueSend(serial_ll_hdl->queue,
            &new_buf_handle, portMAX_DELAY))
```

in function `serial_ll_read()`,
```c
if (pdTRUE != xQueueReceive(serial_ll_hdl->queue, &buf_handle, portMAX_DELAY))
```

In stm32 case, 
```c
uint8_t * transport_pserial_read(uint32_t *out_nbyte)
```
 is going to output the protobuf data ptr and len of it.

this calls internally, 
```c
uint8_t * serial_drv_read(struct serial_drv_handle_t *serial_drv_handle,
        uint32_t *out_nbyte)
```

In this function, 

```c
    if (osSemaphoreWait(readSemaphore, portMAX_DELAY) != osOK)
```
is now already open as hooked earlier.

call 
```c
    read_buf = serial_ll_if_g->fops->read(serial_ll_if_g, &rx_buf_len);
    if ((!read_buf) || (!rx_buf_len)) {
        printf("serial read failed\n\r");
        return NULL;
    }
```
Which is hooked to 
```c
static uint8_t * serial_ll_read(const serial_ll_handle_t * serial_ll_hdl,
                             uint16_t * rlen)
```

here, rlen and buf_handle.payload is returned.

(rlen, buf_handle.payload) is control path frame (excluding esp-hosted header, includes first TLV).

```
    init_read_len = SIZE_OF_TYPE + SIZE_OF_LENGTH + strlen(ep_name) +
        SIZE_OF_TYPE + SIZE_OF_LENGTH;
...
memcpy(buf, read_buf, init_read_len);
...
ret = parse_tlv(buf, &buf_len);
```

Similar explanation to Linux, here. two step read. first TLV parse (although, i did not see verification part, which is in linux). then second tlv parse using parse-tlv . get proto buf and proto len and return.

mantriyogesh, 2024-10-21:
this is code understanding based on master.

I find some serious differences in release/v0.4 to master:

[image]

mantriyogesh, 2024-10-21:
this also includes:
[image]

I will find the commit where it changed.

LuciaferKelvin, 2024-10-21:
Here is the question: the init_read_len is calculated by the function itself. I think this is the correct approach, rather than using parameter passing.

LuciaferKelvin, 2024-10-21:
This makes me confused. Why does the read need the request to calculate the read length?

mantriyogesh, 2024-10-21:
huh! finally I found the commit: https://github.com/espressif/esp-hosted/commit/5be0d74f#diff-af2988b03da21689936fed992b795c68d6685d2a1809e78961ab3b569b6f7945R486

https://github.com/espressif/esp-hosted/issues/516#issuecomment-2425958773 : You are absolutely correct.

That is the reason, i stated [here](https://github.com/espressif/esp-hosted/issues/516#issuecomment-2425594643): 
```
You can also omit to pass read_len to this function. It could be hardcoded value calculated in software.
```

mantriyogesh, 2024-10-21:
we will fix this.

anyway, change comparision with master is worth to do, as many issues might have fixed..

LuciaferKelvin, 2024-10-21:
Thank you for your help. I will change it according to the commit's changes. I have no further questions, so I will close the issue now.