espressif/esp-hosted #517: I (12340306) slave_ctrl: ESP32 station is not connected with AP, can't get AP config

matlam1989, 2024-10-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I use two ARM Crotex-m4 MCU main control boards and ESP32C3 as the WiFi communication module.The main control board operates in AP mode, while the slave board operates in STA mode.When I debugged the main control board, the slave connected and sent a large number of communication messages, but eventually reported a communication interruption fault and disconnected the WiFi connection.Let the host main control normally again, clear the slave fault, and connect to the AP again, but it will never connect again.

### Actual behavior (suspected bug)

When I reset the slave esp32c3 module and reconnected to the main control AP for communication, the communication became normal

### Error logs or terminal output

```shell
I (12340306) slave_ctrl: ESP32 station is not connected with AP, can't get AP config
I (12340337) slave_ctrl: Get station mac address
I (12340338) slave_ctrl: mac [60:55:f9:7d:9b:30]
```

### Steps to reproduce the behavior

1.The ESP32 AP module on the main control end disconnects SPI communication, and the slave continues to send messages for about 2 minutes after connecting.
2.The ESP32 AP module on the main control was connected to SPI communication, and the slave machine was connected to AP communication again. Despite repeated connections, it did not connect

### Project release version

master

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

RTThread RTOS

### Shell

other (details in Additional context)

### Additional context

I used RTThread RTOS and ported the ESP hosted FG driver.

mantriyogesh, 2024-10-23:
Please attach full textual logs. Also attach the sdkconfig both sides.

Did you connect resetpin and other two mandatory pins in between host and slave, as per documentation?

matlam1989, 2024-10-23:
Thanks for your reply.I followed the instructions for the SPI connection, and there was no problem with the WiFi communication under normal circumstances. When I was debugging the host, I suspected that a large amount of data in the module was not processed normally, causing the slave module to be abnormal. Even after the host resumed normal operation, the slave connected to wifi again, but the wifi could not be connected. Tomorrow，I will extract the complete log

matlam1989, 2024-10-24:
ESP-ROM:esp32c3-api1-20210207
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xb9c
load:0x403ce710,len:0x2e4c
entry 0x403cc71a
I (30) boot: ESP-IDF v5.3.1-dirty 2nd stage bootloader
I (30) boot: compile time Oct  9 2024 14:13:43
I (30) bootloader_flash: XM25QHxxC startup flow
I (37) boot: chip revision: v0.3
I (39) boot.esp32c3: SPI Speed      : 80MHz
I (44) boot.esp32c3: SPI Mode       : DIO
I (48) boot.esp32c3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 factory          factory app      00 00 00010000 00100000
I (99) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (107) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (114) boot: End of partition table
I (119) boot: Defaulting to factory image
I (123) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=28f10h (167696) map
I (158) esp_image: segment 1: paddr=00038f38 vaddr=3fc98200 size=03730h ( 14128) load
I (161) esp_image: segment 2: paddr=0003c670 vaddr=40380000 size=039a8h ( 14760) load
I (167) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=9ea50h (649808) map
I (274) esp_image: segment 4: paddr=000dea78 vaddr=403839a8 size=14670h ( 83568) load
I (297) boot: Loaded app from partition at offset 0x10000
I (297) boot: Disabling RNG early entropy source...
I (309) cpu_start: Unicore app
I (318) cpu_start: Pro cpu start user code
I (318) cpu_start: cpu freq: 160000000 Hz
I (318) app_init: Application information:
I (321) app_init: Project name:     network_adapter
I (326) app_init: App version:      FG-0.0.6.0.0
I (331) app_init: Compile time:     Oct  9 2024 14:13:01
I (337) app_init: ELF file SHA256:  ef574e690...
I (343) app_init: ESP-IDF:          v5.3.1-dirty
I (348) efuse_init: Min chip rev:     v0.3
I (353) efuse_init: Max chip rev:     v1.99 
I (358) efuse_init: Chip rev:         v0.3
I (363) heap_init: Initializing. RAM available for dynamic allocation:
I (370) heap_init: At 3FCA0630 len 0001F9D0 (126 KiB): RAM
I (376) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (383) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (390) heap_init: At 50000200 len 00001DE8 (7 KiB): RTCRAM
I (397) spi_flash: detected chip: generic
I (401) spi_flash: flash io: dio
I (405) sleep: Configure to isolate all GPIO pins in sleep state
I (411) sleep: Enable automatic switching of GPIO sleep configuration
I (419) coexist: coex firmware version: 4482466
I (446) coexist: coexist rom version 9387209
I (447) main_task: Started on CPU0
I (447) main_task: Calling app_main()
I (448) NETWORK_ADAPTER: *********************************************************************
I (458) NETWORK_ADAPTER:                 ESP-Hosted Firmware version :: FG-0.0.6.0.0
I (466) NETWORK_ADAPTER:                 Transport used :: SPI only                      
I (475) NETWORK_ADAPTER: *********************************************************************
I (484) NETWORK_ADAPTER: Supported features are:
I (489) NETWORK_ADAPTER: - WLAN over SPI
I (494) ESP_BT: - BT/BLE
I (497) ESP_BT:    - HCI Over SPI
I (501) ESP_BT:    - BLE only
I (505) NETWORK_ADAPTER: capabilities: 0xe8
[... 1959 lines cut ...]

I (342637) slave_ctrl: Details of AP no 11
I (342641) slave_ctrl: SSID                     ZDX-071077
RSSI                    -80
Channel                 6
BSSID                   5c:a0:00:d0:be:11
Auth mode               3

I (342654) slave_ctrl: Details of AP no 12
I (342657) slave_ctrl: SSID                     midea_fc_0639
RSSI                    -83
Channel                 1
BSSID                   d8:d2:61:0f:6f:88
Auth mode               0

I (342671) slave_ctrl: Details of AP no 13
I (342675) slave_ctrl: SSID                     Ruijie-s9073
RSSI                    -84
Channel                 6
BSSID                   e0:5d:54:5a:90:75
Auth mode               4

I (342702) slave_ctrl: ESP32 station is not connected with AP, can't get AP config
I (342733) slave_ctrl: Get station mac address
I (342734) slave_ctrl: mac [60:55:f9:7d:9b:30]

matlam1989, 2024-10-24:
[slaveEsp32log.txt](https://github.com/user-attachments/files/17499446/slaveEsp32log.txt)

matlam1989, 2024-10-24:
After debugging multiple times, it was found that the slave ESP32 had automatically disconnected from the WiFi connection, but the MCU did not detect it. The MCU sent another message to disconnect the WiFi, but the WiFi disconnection failed. The WiFi control state machine was still in the connected state, resulting in an abnormal situation where the WiFi was not reconnected.I haven't figured out the ESPHosted protocol stack yet. Is there any API that can automatically detect WiFi connection status?