espressif/esp-hosted #520: About BT using interface uart0 on esp32-c3

Speakpig, 2024-10-25:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

I'm noted that if I need to use BT thourgh UART, I'd better use it on UART through IO MUX. Now I'm using ESP-C3 as slave which only has one IO MUX UART which is UART0, if I need to use UART0 for BT, how can I change the UART for monitor or flash? Or there are some other solutions to this?

mantriyogesh, 2024-10-25:
there are two controllers for uart in c3. uart0 is used by default by the usb_console.
You can use uart1 with gpio matrix. Any two (or four) unused pins could be used for uart1.

Just try to avoid bootstrap pins as much as possible.

Although it is do-able through `idf.py menuconfig`, This way, you won't need to change uart0 from monitor.

Speakpig, 2024-10-25:
Thanks, so you mean I can use UART1 for BT, and no need to change UART0, right?

mantriyogesh, 2024-10-25:
yes, no need to change uart0. use uart1 on some free pins.

Speakpig, 2024-10-25:
ok, it wil do, but is it better using a UART with IO MUX than with gpio matrix for BT?

mantriyogesh, 2024-10-25:
The throughput requirements for uart and bluetooth are small, so should not be problem. try to use 921600 at least both sides.
Just a note for your reference, (on master branch solution with Linux) the vHCI solution (multiplexed transport) performed much smooth and stable when we needed Wi-Fi+BT together. On linux, we derived this conclusion for classic BT together with Wi-Fi (simultaneous). BLE packets I assume are not that urgent as that of A2DP however!

Speakpig, 2024-10-25:
Do you mean using wifi and BT with multiplexed transport is recommended even on mcu? And does multiplexed transport means using wifi and BT on a same interface at different time like spi? But what do you mean "BLE packets I assume are not that urgent as that of A2DP however!"? Thanks.

mantriyogesh, 2024-10-25:
A2DP (used for music streaming) and HFP have priority mechanisms to avoid glitches during playback, where missing packets can cause audio disruptions. However, these scenarios are unique to classic Bluetooth and don’t occur with BLE.

Though Bluetooth and Wi-Fi can operate at the same time, they share the RF interface, so only one packet can be transmitted at a time at the PHY layer. However, for the user, both Bluetooth and Wi-Fi should appear to work in parallel.

When Wi-Fi and Bluetooth operate together, Wi-Fi packets may experience brief delays to prioritize Bluetooth packets (handled automatically at the RF level). This default prioritization can be tuned to suit different customer requirements, but usually, it’s optimal to process Bluetooth packets first. Unless there’s a specific need, this prioritization should remain unchanged.

In a nutshell, Since the RF hardware is shared, transmission is always time-multiplexed.

mantriyogesh, 2024-10-25:
> Do you mean using wifi and BT with multiplexed transport is recommended even on mcu?

Yes