espressif/esp-hosted #523: ESP32 FG driver, connected through SPI,ESP32C3 as AP, iperf testing An error occurred.

244219195, 2024-10-28:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Using ESP32 FG driver, the host is ARM ported to Linux system, connected through SPI, ESP32C3 as AP, and a laptop device connected to it. During iPerf testing, it appeared that the streaming time exceeded 1000s. Stop testing, restart testing, and find that the speed has become 0.
During daily use, there may also be situations where the connection is successful but WiFi messages cannot be sent or received normally.
What logs are needed to determine whether it is an ESP32 forwarding issue, a Linux side processing issue, a Linux side ESP32 driver issue, or ESP32 memory exhaustion?
I will capture and submit it for analysis.

Analysis ideas. At the current stage, a single packet capture was performed using Wireshark to capture packets from the PC. It was found that the ping packet was sent out 2 seconds later, but the host side used tcpdump to capture it. After 60 seconds, the host received the ping packet sent by the PC and immediately returned it. The PC also received the returned package.however ，The delay of ping is too long
[image]

mantriyogesh, 2024-10-28:
In general, root cause need to be found if the link is always slow or not. Please run [transport throughput test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md) in both (Rx & Tx) directions to identify if any link issues (without involving wifi) exist or not.

43 sec RTT being reported doesn't seem practical for ping utility, ideally it should have 'timed out' in this case. so cross-check with sniffer log, is the log showing incorrect or there is really a problem.

Also, please cross-check the [porting document](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md) for some porting guidelines being followed.

I highly doubt this big RTT is possible to treat as response.
We had some recent fix [531.patch.txt.zip](https://github.com/user-attachments/files/17536972/531.patch.txt.zip) which would work on esp-hosted latest master 50f9444c89decbc2ddb15fd8a7e1538ba6d8f881 +. but, in your case this fix is needed or not, first need to be cross-check as mentioned above.

> During daily use, there may also be situations where the connection is successful but WiFi messages cannot be sent or received normally.

Can you check if the wifi was disconnected by router silently? If the connection was disconnected, host would also get events of disconnected, if C demo application was running & subscribed to disconnection event.

244219195, 2024-10-28:
The WiFi connection has not been disconnected.It is still connected, but the ping delay is very high.

mantriyogesh, 2024-10-28:
1. Please cross-check if `data-ready` interrupt and `handshake` interrupts are triggered correctly from slave to host.
2. Check the jumper cable lengths and grounds are connected. Worth to check porting guide (link in prior comment)
3. Please test raw transport throughput test first, to get to the bottom of things.
4. Can you please check the CPU loading while pinging? 'idle' column in `top` command may help, to understand current load. Also, any patterns on CPU loading, specific thread loading, any specific CPU loading?
5. Does setting static IP differ anything

Textual logs :

- ESP
  - `minicom` or `idf.py monitor` logs
- Linux:
  - `dmesg` from system boot up
  - demo app request and responses
  - For extra logging (please note extra logging may slow the packets and should not be used in production) : [spi_debug_logs_non_built.patch](https://github.com/user-attachments/files/17537151/spi_debug_logs_non_built.patch)
 
Git commits:
Share git commits used both places. We recommend using latest master code base at both sides.
Ensure the same git commit used at esp and linux.
Please share any code changes in patch form if any changes other than GPIO numbers are done either side.

244219195, 2024-10-28:
OK,I will try it.and feedback you soon.

244219195, 2024-10-29:
[image]
when i use the spi_debug_logs_non_built.patch ,ethap0 will disappear，ESP32 log show dump.
here are log for it.

244219195, 2024-10-29:
[esplog.zip](https://github.com/user-attachments/files/17553367/esplog.zip)

244219195, 2024-10-29:
I also want to lease run [transport throughput test](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md) in both.however have some thing wrong for build it .

mantriyogesh, 2024-10-29:
the debug logs are for last sort. what happened for other inputs till now?

mantriyogesh, 2024-10-29:
https://github.com/espressif/esp-hosted/issues/523#issuecomment-2440485003

https://github.com/espressif/esp-hosted/issues/523#issuecomment-2440509103

mantriyogesh, 2024-10-29:
The excessive logging may affect the normal working.
That is the reason the watchdog is triggered.

The logs could be lowered, specially, in ESP_LOG_BUFFER_HEXDUMP

But the question is that without logs, did you cross-check  other things mentioned?

The debug log is just to understand if the packets are flowing and they are correctly sent. in normal logging, we cannot have these big prints. it would drastically lower throughput ot RTT. also it may point to some possible issues, which are not there due to incorrect buffers being printed.

debug logs patch is not fix as mentioned earlier..

244219195, 2024-10-29:
1. Add ESP32 to print out ICMP ping messages immediately upon receiving them, and determine whether there is an ESP32 receiving exception based on the time interval printed out.
2. Check the mechanism of ESP32 forwarding and determine if ESP32 forwarding is abnormal
3. Check again whether ethap0 immediately receives the ping message after ESP32 forwarding is completed. Determine whether it is a processing exception on the AP side.By using the above method, investigate the relevant pathways.

By triggering, print the corresponding log and determine the problem. So add differentiated logs based on the current driver to reduce printing.

How can i do this?

244219195, 2024-10-29:
[image]

mantriyogesh, 2024-10-29:
Are you using raspberry Pi?

mantriyogesh, 2024-10-29:
If not, please check the porting guide, which mentions to port rpi_init.sh.

Additionally, I want to bring to your attention, rpi_init.sh has some functions starting with port_XXX(). These need porting.

[porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md)

mantriyogesh, 2024-10-29:
[image]

244219195, 2024-10-29:
ok ,I will check it again

244219195, 2024-10-31:
I was not use raspberry Pi.I need some time to build it .maybe it will spend time.

mantriyogesh, 2024-10-31:
I think you have already ported it. All I am asking is just to cross check if you have done anything differently.

mantriyogesh, 2025-01-09:
Hello @PzxAndHkq ,

The issue created by @244219195 .
But I am unsure if you both work together or non related.

If you happen to have similar issue but not representing author, better to create new issue and refer here. We can then understand your case in better way while keeping sanity of current issue.

PzxAndHkq, 2025-01-10:
Hi @mantriyogesh 
We are working together, so I reply in this issue.

mantriyogesh, 2025-01-10:
I think It is important to understand if it is still an issue at SPI level or not.

I have some old code, which I intend to add to make SPI robust also, notify if the packets are missedpain the transport.
You can refer to the code specially spi file both side, and 'pkt_num' change.
[465.patch.txt](https://github.com/user-attachments/files/18370427/465.patch.txt)

Worth to check if any packets are getting missed on transport.
If packets are still through, final way to find would be through logic analyser.

PzxAndHkq, 2025-01-11:
We make a mofication in the https://github.com/espressif/esp-hosted/blob/7144dd99a60f86e8b84d2f26d9a1596c75b344f2/esp_hosted_fg/host/linux/host_driver/esp32/main.c#L566
When trigger this work function, read the packet until the read interface return NULL, it seems the ping issue has been resolved, we think maybe this is the root cause.

PzxAndHkq, 2025-01-11:
We don't have a code fork, it is very simple, I upload code to you.
```
static int esp_get_packets(struct esp_adapter *adapter)
{
	struct sk_buff *skb = NULL;
        while (1)
	{
	        if (!adapter || !adapter->if_ops || !adapter->if_ops->read)
		    return -EINVAL;

	
		skb = adapter->if_ops->read(adapter);

		if (!skb)
			return -EFAULT;

		process_rx_packet(skb);
	}

	return 0;
}
```

mantriyogesh, 2025-01-11:
Actually this is not correct way for the work queue handling.

I understand that this solves your problem, can you please state that you have used `531 workqueue patch` from https://github.com/espressif/esp-hosted/issues/523#issuecomment-2440485003 ?

Alternatively, if unsure, can you please attach your full esp_spi.c with your workaround?

PzxAndHkq, 2025-01-13:
We didn't apply that patch, we develop base on commit id 8db84e42e71823e6a04ed55eef06d3ccacd8e4f6.
I will upload the `esp_spi.c` and `main.c` later.

mantriyogesh, 2025-01-13:
Continue with your workaround, I will try fix such that work queue infinite loop is not required.

mantriyogesh, 2025-01-24:
@XDjackieXD, Unsure if this relates to you as workaround: https://github.com/espressif/esp-hosted/issues/523#issuecomment-2585086946