espressif/esp-hosted #526: Bluetooth broadcast content and scan reply settings failed, returning 0x12

xiezhuangwei, 2024-10-30:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

能正常设置广播数据和扫描回复数据

### Actual behavior (suspected bug)

[image]

设置广播数据或者扫描回复数据失败

### Error logs or terminal output

```shell
linxu这边的日志：
# 
# hcitool -i hci0 cmd 0x08 0x0009 0A 09 09 49 50 43 5F 65 30 64 30
< HCI Command: ogf 0x08, ocf 0x0009, plen 11
  0A 09 09 49 50 43 5F 65 30 64 30 
> HCI Event: 0x0e plen 4
  01 09 20 12 
# 
# hcitool -i hci0 cmd 0x08 0x0008 15 02 01 06 11 ff 11 68 10 00 0c cf 89 28 13 c
4 00 00 19 00 00 00
< HCI Command: ogf 0x08, ocf 0x0008, plen 22
  15 02 01 06 11 FF 11 68 10 00 0C CF 89 28 13 C4 00 00 19 00 
  00 00 
> HCI Event: 0x0e plen 4
  01 08 20 12 
------------------------------------------

wifi端日志：
```

### Steps to reproduce the behavior

启动蓝牙，重置蓝牙，设置scan resp，设置adv数据
hciconfig hci0 up
hcitool -i hci0 cmd 0x03 0x0003
hcitool -i hci0 cmd 0x08 0x0009 0A 09 09 49 50 43 5F 65 30 64 30
hcitool -i hci0 cmd 0x08 0x0008 15 02 01 06 11 ff 11 68 10 00 0c cf 89 28 13 c4 00 00 19 00 00 00

### Project release version

最新master

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

linux

### Shell

sh

mantriyogesh, 2024-10-30:
The ESP-Hosted is responsible to transfer the bluetooth messages from controller to stack and return path.

Are you facing any kind of message drop for bluetooth in this case?
If not, this is not ESP-Hosted bug.

If this is not ESP-Hosted bug, and if it is urgent, I would suggest checking your BlueZ configuration and the commands fired are supported and cross check with BlueZ support channel https://github.com/bluez/bluez/issues

From the logs shared, I can confirm there is no issue in ESP-Hosted as such.

mantriyogesh, 2024-10-30:
Hello again,

Can you please attach all the logs and git commit version used:

1. esp monitor or minicom log
2. host dmesg log from system bootup
3. host commands run and their outputs
4. sudo btmon log (to be started before loading the kernel module for complete logs)

possibly:

6. git commits used in esp-hosted for esp and host side, also idf
7. sdkconfig used at esp

we will try re-assess if the issue is at esp-hosted. we can also try to get internal bt team help (high turnover time), possibly, if it is not esp-hosted issue, just as help and not as support.

mantriyogesh, 2024-10-30:
Just to be sure, @xiezhuangwei 

Can you reproduce this on your native laptop, if it works with this?

mantriyogesh, 2024-10-30:
The commands you have sent are responded with invalid commands (0x12) So check your full scenario first if it works on native laptop bluetooth, first.

xiezhuangwei, 2024-10-30:
[Serial-COM6-15_47_00.log](https://github.com/user-attachments/files/17567904/Serial-COM6-15_47_00.log)
[Serial-COM22-15_46_59.log](https://github.com/user-attachments/files/17567907/Serial-COM22-15_46_59.log)
Serial-COM22-15_46_59.log是主控的日志，里面包含了命令运行，dmesg
Serial-COM6-15_47_00.log是WiFi的，有打印出主控下发的数据

xiezhuangwei, 2024-10-30:
异常的
hcitool -i hci0 cmd 0x08 0x0008 15 02 01 06 11 ff 11 68 10 00 0c cf 89 28 13 c4 00 00 19 00 00 00
hcitool -i hci0 cmd 0x08 0x0009 0A 09 09 49 50 43 5F 65 30 64 30

正常可以设置的。发现不需要的数据要补0，底层wifi才没报0x12错误
hcitool -i hci0 cmd 0x08 0x0008 1e 02 01 06 11 ff 11 68 10 00 0c cf 89 28 13 c4 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
hcitool -i hci0 cmd 0x08 0x0009 1e 09 09 49 50 43 5F 65 30 64 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

mantriyogesh, 2024-10-30:
I note the difference of '0x19' in your first command. is it intentional

mantriyogesh, 2024-10-30:
## Length issue discussion

### Pure esp32-c3 testing
The bluetooth team tested with following 

```
31 #if CONFIG_EXAMPLE_EXTENDED_ADV
32 static uint8_t ext_adv_pattern_1[] = {
33     0x02, 0x01, 0x06,
34     0x11, 0XFF, 'a', 'b', 'c', 'd', 'e', 'f', 'g', 'h', 'j', 'k', 'l', 'p', 'e', 'f', 'e', 'q',
35 };
36 #endif
```
in bleprph esp-idf example with pure esp32-c3 and it is working fine.

### ESP-Hosted testing
So I had cross checked on latest master of esp-hosted FG,

It is again working fine at our place.

##### Log at Raspberry Pi
```
kysoh@raspberrypi:~ $ hciconfig
hci0:	Type: Primary  Bus: SPI
	BD Address: 40:4C:CA:4A:A8:52  ACL MTU: 517:10  SCO MTU: 0:0
	UP RUNNING
	RX bytes:288 acl:0 sco:0 events:56 errors:0
	TX bytes:303 acl:0 sco:0 commands:0 errors:0

kysoh@raspberrypi:~ $ sudo hcitool -i hci0 cmd 0x08 0x0008 1e 02 01 06 11 ff 11 68 10 00 0c cf 89 28 13 c4 00 00 19 00 00 00 00 00 00 00 00 00 00 00 00 00
< HCI Command: ogf 0x08, ocf 0x0008, plen 32
  1E 02 01 06 11 FF 11 68 10 00 0C CF 89 28 13 C4 00 00 19 00
  00 00 00 00 00 00 00 00 00 00 00 00
> HCI Event: 0x0e plen 4
  01 08 20 0C
kysoh@raspberrypi:~ $ sudo hcitool -i hci0 cmd 0x08 0x0008 15 02 01 06 11 ff 11 68 10 00 0c cf 89 28 13 c4 00 00 19 00 00 00
< HCI Command: ogf 0x08, ocf 0x0008, plen 22
  15 02 01 06 11 FF 11 68 10 00 0C CF 89 28 13 C4 00 00 19 00
  00 00
> HCI Event: 0x0e plen 4
  01 08 20 0C
kysoh@raspberrypi:~ $ hciconfig
hci0:	Type: Primary  Bus: SPI
	BD Address: 40:4C:CA:4A:A8:52  ACL MTU: 517:10  SCO MTU: 0:0
	UP RUNNING
	RX bytes:300 acl:0 sco:0 events:60 errors:0
	TX bytes:363 acl:0 sco:0 commands:0 errors:0
```

In any case, you check your commands are correct. hcitool needs specific understanding of the commands.

Further, The Bluetooth messages could also be tracked to and fro using traces at:

## Hosted debug logs

### Slave
 
 
https://github.com/espressif/esp-hosted/blob/856fd96f03846a9f43dce9487e7d77f90352884a/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_bt.c#L109
 
 
https://github.com/espressif/esp-hosted/blob/856fd96f03846a9f43dce9487e7d77f90352884a/esp_hosted_fg/esp/esp_driver/network_adapter/main/slave_bt.c#L85

### Host
 
https://github.com/espressif/esp-hosted/blob/856fd96f03846a9f43dce9487e7d77f90352884a/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c#L137
 
 
https://github.com/espressif/esp-hosted/blob/856fd96f03846a9f43dce9487e7d77f90352884a/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c#L88