espressif/esp-hosted #527: FG mode, ctrl_req_sem semaphore no post

lwj850533527, 2024-10-30:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

ctrl_req_sem semaphore get and post are matching
[image]

### Actual behavior (suspected bug)

When the SSID length I set is too long, it can return normally this time
[image]

But the next time, all command cannot get semaphore
[image]

### Error logs or terminal output

```shell
resp NULL                                                                       
[ctrl_app_send_req]1250                                                         
Failed to send control req 123                                                  
resp NULL                                                                       
null version:-0.0.0.0.0                                                         
[ctrl_app_send_req]1250                                                         
Failed to send control req 101                                                  
Failed to get wifi mac                                                          
esp32 mac init err:0xffffffff
```

### Steps to reproduce the behavior

SSID length set too long

### Project release version

0.0.6.0.0

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

Linux

### Operating system version

STM32

### Shell

ZSH

mantriyogesh, 2024-10-30:
Can you please send the git commit you have used both sides and the logs both sides?

lwj850533527, 2024-10-30:
Hi @mantriyogesh
git commit is:https://github.com/espressif/esp-hosted/commit/70de43ac9cc30ef55960d8025274df7a0fe1dfc8

the logs only 
`Failed to send control req`

I added printing and found that it return at hosted_get_semaphore(ctrl_req_sem)

mantriyogesh, 2024-10-30:
Hello @lwj850533527 , Please give us time to reproduce and fix this.

lwj850533527, 2024-11-02:
Hi @mantriyogesh
Has it been reproduced

mantriyogesh, 2024-11-02:
Hello @lwj850533527 ,

I can check this issue earliest by Tuesday.

mantriyogesh, 2024-11-04:
@lwj850533527 , Do you have esp side logs?

mantriyogesh, 2024-11-04:
I would really appreciate it if you can provide full textual logs both sides..

mantriyogesh, 2024-11-04:
I have not yet tested this, but I can confirm the race condition could be there, as you experienced in this code leg.

Can you please cross check if [ctrl_correct_sem.patch](https://github.com/user-attachments/files/17613565/ctrl_correct_sem.patch) solves your problems?

lwj850533527, 2024-11-04:
> @lwj850533527 , Do you have esp side logs?

Esp side no command received, the log is empty
So I replied before,  the valid log is only `Failed to send control req`

lwj850533527, 2024-11-04:
> I have not yet tested this, but I can confirm the race condition could be there, as you experienced in this code leg.
> 
> Can you please cross check if [ctrl_correct_sem.patch](https://github.com/user-attachments/files/17613565/ctrl_correct_sem.patch) solves your problems?

It can solve the issue

closed 2024-11-12 by lwj850533527