espressif/esp-hosted #530: esp32 NG, connect fail

tinswf, 2024-11-01:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

if the AP power on before our device，our device can connect the AP success，
but when our device power on, the AP is power off ,  
and several hours later,  the esp32 change to inactive，our device can not connect the AP

### Actual behavior (suspected bug)

[Uploading esp32 debug.txt…]()

### Steps to reproduce the behavior

refer to host debug message.txt

key info:
esp32: [0xb] Command failed

### Project release version

ESP-Hosted Firmware version :: 1.0.2   

### System architecture

ARM 32-bit (Raspberry Pi 32-bit)

### Operating system

Linux

### Operating system version

win10

### Shell

Bash

tinswf, 2024-11-01:
[esp32 debug.txt](https://github.com/user-attachments/files/17597529/esp32.debug.txt)

[host debug message.txt](https://github.com/user-attachments/files/17597531/host.debug.message.txt)

tinswf, 2024-11-01:
esp32 key debug info

[0;32mI (43989617) FW_CMD: STA disconnected [2]
[0m
[0;32mI (43989619) FW_CMD: STA Disconnect event: 2

mantriyogesh, 2024-11-01:
@kapilkedawat ++