espressif/esp-hosted #531: STM32F412ZG-NUCLEO  SDIO INIT FAILED

abing520, 2024-11-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hi, I use the stm32f412zgt6 nucleo board communication with esp32_devkitc_v4 via sdio.but stm32 sdio init failed. I don't konw why.
What I did:
1.Connect the two development boards correctly.
2.Wiring shorter than 5 cm.
3.I tried resistors of 10 and 51 ohm.
[image]
[image]
[image]

abing520, 2024-11-04:
[image]

mantriyogesh, 2024-11-04:
Could you please help with:
1. host side logs (texual logs much appreaciated)
2. Full backtrace, how it entered here
3. FreeRTOS has enough heap

abing520, 2024-11-11:
1.host side has not run into main function，there is no log 。
[image]
[image]
[flow.zip](https://github.com/user-attachments/files/17694857/flow.zip)

mantriyogesh, 2024-11-11:
Can you first try spi? It should be easier to get along.

SD we had to add the HAL layer wrapper (which is very unportable across STM32) here: 
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/stm32/driver/transport/sdio/sdio_ll.c

The problem with STM32 is that it doesn't have default SDIO only HAL. So we needed to avoid 'MMC' part of it in process of initialisation.

The SPI port doesn't have such dependencies and straight forward to setup.

We are also coming up with network stack integrated solution with SPI (in current code network is need to be hooked by user).

Always ensure you use same code commit at slave and stm32 host. code mismatch also may lead to unrecognised errors.

abing520, 2024-11-11:
1.My client uses sdio, so I don't care spi first.

abing520, 2024-11-11:
> Can you first try spi? It should be easier to get along.
>
> SD we had to add the HAL layer wrapper (which is very unportable across STM32) here: [...]

how to acoid ''MMC'' question,sorry i don't konw

abing520, 2024-11-13:
hi，can you help about question above.thank you best. because the question i‘m  so unhappy.

mantriyogesh, 2024-11-13:
If you need sdio only, you need to handle check if your stm board hal layer supports the 'sdio natively. If it does, remove sdio__ll.c and use hal sdio.

Last time we used sdio, our stm did not support sdio and only had sd mmc. So we re-wrote small hal in sdio_ll.c. you can refer it and adapt similar way for your stm.

It is problem at stm host hal drivers, not at ESP as such.

abing520, 2024-11-13:
> If you need sdio only, you need to handle check if your stm board hal layer supports the 'sdio natively. If it does, remove sdio__ll.c and use hal sdio.
>
> Last time we used sdio, our stm did not [...]

I used the same stm32 chip with you.

mantriyogesh, 2024-11-13:
Can you please find out when does it run into Error_Handler?

What is the backtrace?