espressif/esp-hosted #532: 2 SDIO data lanes in menuconfig for 1-bit SDIO on ESP32-P4 with ESP32-C6

dmartauz, 2024-11-04:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

When transport configuration is switched to SDIO 1-bit in menuconfig, pin configuration for D0 and D1 is displayed.
[image]

Is D1 really being used in this case?
What throughput is expected for 1-bit SDIO transport?
Are both WiFi and BLE currently supported with this transport? WiFi works for me, did not test BLE yet.
Offtopic: are you planning to support 802.15.4 interface of C6 with esp-hosted?

### Actual behavior (suspected bug)

Menuconfig displays only D0 pin configuration in this case.

### Steps to reproduce the behavior

Run menuconfig in a project using esp-hosted and configure transport to SDIO 1-bit.

### Project release version

0.0.22

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

CMD

mantriyogesh, 2024-11-04:
> are you planning to support 802.15.4 interface of C6 with esp-hosted?

Yes, (definitely.) But not yet planned.

> What throughput is expected for 1-bit SDIO transport

1-bit is supported in SDIO.
The throughput numbers were not noted, but you can compare similar to standard SPI full duplex mode (or a little higher).

> Is D1 really being used in this case?

Yes. D1 line is standard SDIO interrupt line and is needed. [Further learning!](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32c6/api-reference/peripherals/sdio_slave.html#connections)

If you use out-of-bound interrupt (not yet coded, but should work), you can avoid the D1/DAT1 line.
Also, if you use polling (highly non-recommended), you can avoid D1.

### Pull-ups for C6
Please **ignore** if you are using ESP32-P4-Function_EV-Board and on-board C6.
Even when Data lines unused, needs the pull-ups connected.
C6 for SDIO may need extra pull-ups: [C6 Pull-up requirements](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/sd_pullup_requirements.html).

mantriyogesh, 2024-11-04:
But I understand your confusion, I think it would be better to if we change text to clearly mention D1 as interrupt.

dmartauz, 2024-11-04:
@mantriyogesh thanks for your answer, I understand it now. Yes, explanation in menuconfig for D1 would clarify why it is presented for 1-bit mode.

mantriyogesh, 2024-11-04:
[sdio_1bit_desc.patch](https://github.com/user-attachments/files/17618347/sdio_1bit_desc.patch)

This should look better. I will push these changes later.