espressif/esp-hosted #535: get memory information Interface

lwj850533527, 2024-11-08:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Does  esp32 have Interface with get memory information?

### Actual behavior (suspected bug)

get memory information

### Error logs or terminal output

```shell
/
```

### Steps to reproduce the behavior

/

### Project release version

1.0.3.0.0

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

linux

### Shell

ZSH

mantriyogesh, 2024-11-08:
Not understood. Can you please explain in detail?

lwj850533527, 2024-11-08:
Is there any interface that can retrieve the usage of memory, the free of memory and the peak usage of memory

mantriyogesh, 2024-11-08:
Are you talking at slave side stats?

lwj850533527, 2024-11-08:
ESP32 the slave side memory information, I want get and printf at slave side.
Also  want to send to host at the next step.

mantriyogesh, 2024-11-08:
For esp side information, please rely on esp-idf APIS, https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/mem_alloc.html All those APIs would be visible. in general, slave is a contained environment and designed not to go out of memory.

Also, not sure, how would be useful for general public, host knowing slave memory etc.

mantriyogesh, 2024-11-09:
```c
static void print_mem_stats()
{
    uint32_t freeSize = esp_get_free_heap_size();
    printf("The available total size of heap:%" PRIu32 "\n", freeSize);

    printf("\tDescription\tInternal\tSPIRAM\n");
    printf("Current Free Memory\t%d\t\t%d\n",
           heap_caps_get_free_size(MALLOC_CAP_8BIT) - heap_caps_get_free_size(MALLOC_CAP_SPIRAM),
           heap_caps_get_free_size(MALLOC_CAP_SPIRAM));
    printf("Largest Free Block\t%d\t\t%d\n",
           heap_caps_get_largest_free_block(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL),
           heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM));
    printf("Min. Ever Free Size\t%d\t\t%d\n",
           heap_caps_get_minimum_free_size(MALLOC_CAP_8BIT | MALLOC_CAP_INTERNAL),
           heap_caps_get_minimum_free_size(MALLOC_CAP_SPIRAM));
}
```

Some sample function, if you need something handy. but anyway, the heap allocation is based on a lot of diff tags (check idf component heap for details).

mantriyogesh, 2024-11-09:
These are some of esp-idf functions.
Please go through the API description for details.

For now, I don’t think there is any action for us, so closing the issue. Please don't hesitate to re-open if you think still not solved.

closed 2024-11-09 by mantriyogesh