espressif/esp-hosted #536: NG mode：ESP32 restart abnormaly

lwj850533527, 2024-11-11:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

rarely

### Expected behavior

ESP32 work normaly
I want to why print regularly every hour `Rx PACKET_TYPE_EAPOL!!!!`. Is it related to the ap?
Is there any reason that may cause ESP32 to restart.

### Actual behavior (suspected bug)

esp32 restart abnormaly
1、01:53:24 start connect ap
2、01:53:30 end connect ap
3、print regularly every hour `Rx PACKET_TYPE_EAPOL!!!!`
4、after 2 days, 01:57:06 `Command[0x8] timed out` ,  it is CMD_ADD_KEY. This time esp32 should happen to the restart
5、after esp32 restarted, `[0x4] Command failed`,  it is CMD_SCAN_REQUEST. It may because of slave side no init

### Error logs or terminal output

```shell
Nov  9 01:53:24 kernel: [   23.671159] esp32_spi: cmd_auth_request: Authentication request: 0c:8e:29:c3:b1:ea 11 0 0 0
Nov  9 01:53:25 kernel: [   24.027593] esp32_spi: cmd_assoc_request: Association request: 0c:8e:29:c3:b1:ea 11 40
Nov  9 01:53:25 kernel: [   24.076208] esp32_spi: process_assoc_event: Connection status: 0
Nov  9 01:53:25 kernel: [   24.427329] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 01:53:25 kernel: [   24.456601] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 01:53:25 kernel: [   24.456694] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
Nov  9 01:53:25 udhcpc[335]: started, v1.34.1
Nov  9 01:53:25 udhcpc[335]: broadcasting discover
Nov  9 01:53:28 udhcpc[335]: broadcasting discover
Nov  9 01:53:29 udhcpc[335]: broadcasting select for 192.168.2.201, server 192.168.2.1
Nov  9 01:53:30 udhcpc[335]: lease of 192.168.2.201 obtained from 192.168.2.1, lease time 1814400
Nov  9 01:53:30 kernel: [   29.397809] esp32_spi: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.002.201
Nov  9 01:56:53 kernel: [  232.513649] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 01:56:53 kernel: [  232.513737] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
Nov  9 02:56:53 kernel: [ 3832.736640] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 02:56:53 kernel: [ 3832.736726] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
Nov  9 03:56:54 kernel: [ 7432.850740] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 03:56:54 kernel: [ 7432.850842] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
Nov  9 04:56:54 kernel: [11033.173934] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 04:56:54 kernel: [11033.174018] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
Nov  9 05:56:54 kernel: [14633.188303] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 05:56:54 kernel: [14633.188399] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
Nov  9 06:56:54 kernel: [18233.300732] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 07:56:54 kernel: [21833.514643] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 07:56:54 kernel: [21833.514728] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
Nov  9 08:56:54 kernel: [25433.637592] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 08:56:54 kernel: [25433.637677] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
Nov  9 09:56:55 kernel: [29033.860859] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 09:56:55 kernel: [29033.860951] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
Nov  9 10:56:54 kernel: [32633.974622] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 11:56:54 kernel: [36234.197373] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 11:56:54 kernel: [36234.197450] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
Nov  9 12:56:54 kernel: [39834.310609] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 13:56:54 kernel: [43434.533568] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 14:56:54 kernel: [47034.647371] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 15:56:55 kernel: [50634.870507] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 16:56:55 kernel: [54234.993355] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 17:56:55 kernel: [57835.207071] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 18:56:54 kernel: [61435.330679] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 19:56:54 kernel: [65035.483295] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 20:56:54 kernel: [68635.667146] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
Nov  9 21:56:55 kernel: [72235.791061] esp32_spi: process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[... 47 lines cut ...]

### Steps to reproduce the behavior

1、sta connect to ap
2、there are no other operations available

### Project release version

1.0.3.0.0

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

linux

### Shell

ZSH

mantriyogesh, 2024-11-11:
@kapilkedawat ++

kapilkedawat, 2024-11-11:
Hi @lwj850533527, AP may be doing periodic GTK rekeying for which you are getting a EAPOL in regular interval. Please share the logs for ESP32 when it is rebooting.

lwj850533527, 2024-11-11:
This is a problem that takes a long time to occur and rarely. 
I am unable to directly get the logs of ESP32, is there any way to send the logs for ESP32 when it is rebooting to host

kapilkedawat, 2024-11-19:
> is there any way to send the logs for ESP32 when it is rebooting to host

Unfortunately that is not possible. Is there a way for you to connect uart to chip?

lwj850533527, 2025-03-25:
https://github.com/espressif/esp-hosted/commit/f5880368fcabf36d8d21ebfb2f9346c30627a32a

Hi，@kapilkedawat
What is the reason for changing it to memory request here?
Do it need to release the memory after using it up?

Is this change related to this issue，notice，the log shows that the restart occurred after the 0x8 instruction was sent
[image]