espressif/esp-hosted #545: esp-host-fg  RTSP failed

zsy118, 2024-11-21:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The host linux nt98566 connects to the ESP32c6 through the main sdio, and the esp32c6 is used as the wifi soc. When testing the RTSP push stream, the pull stream through the vlc fails. It is ok for us to use 8189wifi rtsp for the same machine. So the problem is with the esp32c6 wifi

### Actual behavior (suspected bug)

[image]
[image]
[image]
[image]

### Steps to reproduce the behavior

command

### Project release version

esp_host_fg

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

linux kernel4.19.91

### Shell

sh

mantriyogesh, 2024-11-21:
Can you please state the procedure to reproduce?

Also please attach the logs from both sides in text format.

Esp: idf.py monitor or minicom
Host: 1. dmesg from system bootup
2. Commands fired

So in general, the connectivity is established and able to test throughput but fail on the streaming test?

zsy118, 2024-11-21:
dmesg:hd_common_init
process id 0 _isf_init 0
venc init: config tile cut if w>1280 (LL slow)
csi_platform_clk_enable....
csi_open:CSI HW reset!
vdoprc0 init: stripe LV1 = cut w>1280, GDC =  on, 2D_LUT off after cut (LL slow)
vprc: enable one-buf func at path 0, (3ndr_ref=1) (max_strp=3)
WRN:_isf_vdoprc_oport_do_new() vdoprc0.out[0]! Start single blk mode => blk_id=8                                        5fe6fc0 addr=85fe7000
WRN:_isf_vdoprc_oport_do_new() vdoprc0.out[0]! Start single blk mode-2 => blk_id                                        =8652dfc0 addr=8652e000
========== AE_CBMSG_PREVIEWSTABLE ==========
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)
SDIO1 nvt_mmc_abort_data R sts 0x452 (CMD53)

esp32 log:dx(0, stop:8)), snr:79, ampduState:wait operational
I (113260) wifi:ifidx:0, rssi:-14, nf:-93, phytype(0x3, CBW20-SGI), phymode(0x3,                                11bgn), max_rate:144, he:0, vht:0, ht:1
I (113270) wifi:(ht)max.RxAMPDULenExponent:3(65535 bytes), MMSS:5(4 us)
I (113346) wifi:connected with MyWifi, aid = 1, channel 5, BW20, bssid = e4:67:1                               e:1e:e5:00
I (113347) wifi:security: WPA2-PSK, phy:11bgn, rssi:-14, cipher(pairwise:0x3, gr                               oup:0x3), pmf:0,
I (113352) wifi:pm start, type: 1, twt_start:0

I (113356) wifi:pm start, type:1, aid:0x1, trans-BSSID:e4:67:1e:1e:e5:00, BSSID[                               5]:0x0, mbssid(max-indicator:0, index:0), he:0
I (113368) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us t                               o 307200 us
I (113376) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 1                               0, mt_time: 10000
I (113384) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:64, A-MSD                               U:0(not supported), policy:1(IMR), ssn:0(0x0)
I (113395) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:64, A-MSD                               U:0(not supported), policy:1(IMR), ssn:0(0x20)
I (113406) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:64, A-MSD                               U:0(not supported), policy:1(IMR), ssn:0(0x0)
I (113418) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (113424) slave_ctrl: Manual Wi-Fi connected, no event raised
I (113430) slave_ctrl: connected to ap SSID:'MyWifi', password:'MyWifiPass@123'
I (113430) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:64                               , batimeout:0, txa_wnd:64
I (113447) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:64                               , batimeout:0, txa_wnd:64
I (113456) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:64                               , batimeout:0, txa_wnd:64
I (113466) slave_ctrl: req_connect_ap_handler:653 Set resp to Success
I (162441) slave_ctrl: Stop Heatbeat
I (162442) slave_ctrl: Stopping HB timer
W (181480) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x10000e5, TALO:0x1e1e67e4, (s                               sn:2, win:64, cur_ssn:2), CONF:0xc0000005
I (254436) wifi:[ADDBA]TX addba request, tid:6, dialogtoken:4, bufsize:64, A-MSD                               U:0(not supported), policy:1(IMR), ssn:0(0x0)
I (254449) wifi:[ADDBA]RX addba response, status:0, tid:6/tb:0(0xa1), bufsize:64                               , batimeout:0, txa_wnd:64

mantriyogesh, 2024-11-21:
I will wait for rest of the ilogs.

zsy118, 2024-11-21:
▒?▒!▒V▒▒        I (59514) NETWORK_ADAPTER: Start Data Path
I (59806) NETWORK_ADAPTER: Initial set up done
I (59806) slave_ctrl: event ESPInit
I (59807) main_task: Returned from app_main()
I (110434) slave_ctrl: HB timer started for 20 sec

I (110495) wifi:mode : sta (40:4c:ca:55:bc:14)
I (110495) wifi:enable tsf
I (110495) slave_ctrl: station mode set
I (110496) slave_ctrl: Get station mac address
I (110500) slave_ctrl: mac [40:4c:ca:55:bc:14]
W (110505) wifi:Password length matches WPA2 standards, authmode threshold chang                               es from OPEN to WPA2
I (113217) wifi:new:<5,0>, old:<1,0>, ap:<255,255>, sta:<5,0>, prof:1, snd_ch_cf                               g:0x0
I (113218) wifi:(connect)dot11_authmode:0x3, pairwise_cipher:0x3, group_cipher:0                               x3
I (113221) wifi:state: init -> auth (0xb0)
I (113228) wifi:state: auth -> assoc (0x0)
I (113235) wifi:state: assoc -> run (0x10)
I (113235) wifi:(trc)phytype:CBW20-SGI, snr:79, maxRate:144, highestRateIdx:0
W (113240) wifi:(trc)band:2G, phymode:3, highestRateIdx:0, lowestRateIdx:11, dat                               aSchedTableSize:14
I (113248) wifi:(trc)band:2G, rate(S-MCS7, rateIdx:0), ampdu(rate:S-MCS7, schedI                               dx(0, stop:8)), snr:79, ampduState:wait operational
I (113260) wifi:ifidx:0, rssi:-14, nf:-93, phytype(0x3, CBW20-SGI), phymode(0x3,                                11bgn), max_rate:144, he:0, vht:0, ht:1
I (113270) wifi:(ht)max.RxAMPDULenExponent:3(65535 bytes), MMSS:5(4 us)
I (113346) wifi:connected with MyWifi, aid = 1, channel 5, BW20, bssid = e4:67:1                               e:1e:e5:00
I (113347) wifi:security: WPA2-PSK, phy:11bgn, rssi:-14, cipher(pairwise:0x3, gr                               oup:0x3), pmf:0,
I (113352) wifi:pm start, type: 1, twt_start:0

I (113356) wifi:pm start, type:1, aid:0x1, trans-BSSID:e4:67:1e:1e:e5:00, BSSID[                               5]:0x0, mbssid(max-indicator:0, index:0), he:0
I (113368) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us t                               o 307200 us
I (113376) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 1                               0, mt_time: 10000
I (113384) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:64, A-MSD                               U:0(not supported), policy:1(IMR), ssn:0(0x0)
I (113395) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:64, A-MSD                               U:0(not supported), policy:1(IMR), ssn:0(0x20)
I (113406) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:64, A-MSD                               U:0(not supported), policy:1(IMR), ssn:0(0x0)
I (113418) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (113424) slave_ctrl: Manual Wi-Fi connected, no event raised
I (113430) slave_ctrl: connected to ap SSID:'MyWifi', password:'MyWifiPass@123'
I (113430) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:64                               , batimeout:0, txa_wnd:64
I (113447) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:64                               , batimeout:0, txa_wnd:64
I (113456) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:64                               , batimeout:0, txa_wnd:64
I (113466) slave_ctrl: req_connect_ap_handler:653 Set resp to Success
I (162441) slave_ctrl: Stop Heatbeat
I (162442) slave_ctrl: Stopping HB timer
W (181480) wifi:<ba-add>idx:0, ifx:0, tid:0, TAHI:0x10000e5, TALO:0x1e1e67e4, (s                               sn:2, win:64, cur_ssn:2), CONF:0xc0000005
I (254436) wifi:[ADDBA]TX addba request, tid:6, dialogtoken:4, bufsize:64, A-MSD                               U:0(not supported), policy:1(IMR), ssn:0(0x0)
I (254449) wifi:[ADDBA]RX addba response, status:0, tid:6/tb:0(0xa1), bufsize:64                               , batimeout:0, txa_wnd:64
[rtsp-tcp.zip](https://github.com/user-attachments/files/17839291/rtsp-tcp.zip)

zsy118, 2024-11-21:
Test video
[e91cd81bb36471411824c74b03092b27.zip](https://github.com/user-attachments/files/17839617/e91cd81bb36471411824c74b03092b27.zip)

mantriyogesh, 2024-11-21:
Video is completely blurr, cannot make out anything apart from vlc trying something.
Can you send the steps to reproduce?

Also you have attached pcap capture. Can you please help to point to the observations or conclusions you have?
Packet number you see the problem, the ip address, who is who, etc

mantriyogesh, 2024-11-21:
There are only esp side logs attached. no host side logs attached.

Please attach as text file:
1. The procedure steps you have run, along with their outputs on terminal
2. dmesg from system bootup, till issue is observed

What is exact issue? From esp logs, it doesn't show any issue. So I fail to understand where exactly the problem is.

From pcap I suppose you are able to set up tcp connection.
Is the video started and then stopped, or it is not at all starting?
Is the problem of networking or some crash has happened at host or kernel panic or something else?

zsy118, 2024-11-21:
Network ok, host ok. I did a comparison test with the 8189 network card and the esp32c6. esp32c6 network speed problem found
esp32c6，
[image]
8189fs
[image]

mantriyogesh, 2024-11-21:
So, is it only speed? Or you face some disconnection?

zsy118, 2024-11-21:
TCP communication test: Ubuntu (192.168.0.103) as client,
arm linux used 8189 (192.168.0.100) and arm linux used esp32c6 (192.168.0.102) as servers for comparison testing

mantriyogesh, 2024-11-21:
We have some fixes done on https://gitlab.espressif.cn:6688/app-frameworks/esp_hosted_mcu (for microprocessors) to bump the throughput, but they are not synced to Linux repo yet. I can change and get back on this by end of week.

Also please note the 8089 is esp32 based processor with SDIO.
For fair comparison, I think better to use <ins>esp32 as SDIO with ESP-Hosted-FG</ins>.

Notable differences: ESP32 is dual-core , 240MHz. C6 is single code, 160MHz processor.

jinhe1314, 2024-11-21:
hi mantriyogesh.
I'm a colleague of zsy118,I don't think it's a throughput issue, the bandwidth is sufficient from the ipref test.In the pcap file to see the re-transmission of too much, not yet much data is transmitted on the stream cut off.

mantriyogesh, 2024-11-21:
Oh I see.

@jinhe1314 , are the tcp retransmissions only happening with RTSP / vlc or also observed with pure iperf as well?

jinhe1314, 2024-11-22:
yes ,only RTSP

mantriyogesh, 2024-11-22:
Okay, this is very interesting.

Obviously, the throughput is also concerning for me. but I understood that it is not a blocker for you right now. I will check this in lower priority than the RTSP..

Can you please state the reproducing steps, or some patch, which I can try to easily test on my side?
As it should be generic to try this test case, I will have better control if I reproduce it at my end.

mantriyogesh, 2024-11-29:
Any steps please?

mantriyogesh, 2024-12-04:
Hello @zsy118 , @jinhe1314 I have got the doc, 
[Build LIVE555 Streaming Media server.docx](https://github.com/user-attachments/files/18003126/Build.LIVE555.Streaming.Media.server.docx)

to reproduce the steps.

We will reproduce this behaviour on Raspberry Pi and come back to this.

Just confirming before starting to debug this problem, if the issue is still experienced.

jinhe1314, 2024-12-06:
yes

SohKamYung-Espressif, 2025-02-12:
Hello @zsy118 @jinhe1314 

I have a Raspberry Pi using the ESP32-C6 over SDIO for the network connection.

I have also build the test mediaserver on a Raspberry Pi and provided it with the `test.264` file from `http://www.live555.com/liveMedia/public/264/`. I then ran the mediaserver:

```
kysoh@raspberrypi:~/live555/live/mediaServer $ ./live555MediaServer 
LIVE555 Media Server
	version 1.13 (LIVE555 Streaming Media library version 2025.01.17).
Play streams from this server using the URL
	rtsp://10.10.2.64:8554/<filename>
where <filename> is a file present in the current directory.
Each file's type is inferred from its name suffix:
	".264" => a H.264 Video Elementary Stream file
	".265" => a H.265 Video Elementary Stream file
	".aac" => an AAC Audio (ADTS format) file
	".ac3" => an AC-3 Audio file
	".amr" => an AMR Audio file
	".dv" => a DV Video file
	".m4e" => a MPEG-4 Video Elementary Stream file
	".mkv" => a Matroska audio+video+(optional)subtitles file
	".mp3" => a MPEG-1 or 2 Audio file
	".mpg" => a MPEG-1 or 2 Program Stream (audio+video) file
	".ogg" or ".ogv" or ".opus" => an Ogg audio and/or video file
	".ts" => a MPEG Transport Stream file
		(a ".tsx" index file - if present - provides server 'trick play' support)
	".vob" => a VOB (MPEG-2 video with AC-3 audio) file
	".wav" => a WAV Audio file
	".webm" => a WebM audio(Vorbis)+video(VP8) file
See http://www.live555.com/mediaServer/ for additional documentation.
(We use port 8000 for optional RTSP-over-HTTP tunneling).)
```

I then ran VLC on my test PC and entered `rtsp://10.10.2.64:8554/test.264` as the Network Stream to play.

The stream was able to play without errors: I did not see any pause, distorted or dropped frames, etc. See the screenshot showing the VLC statistics.

[image]

I also captured the data between the Rasp Pi and the test PC in Wireshark. If you want to see the data, I can attach it as a separate comment.

zsy118, 2025-02-12:
Thank you very much. Please test it with iperf，Does the network lose packets?

mantriyogesh, 2025-02-12:
Can you please exactly describe the issue?

Is it RTSP?
is it iperf, is it TCP or UDP?

Can you please help to provide exact issue , then we can focus on test that part.

zsy118, 2025-02-12:
In the past, it was the network that dropped packets, resulting in the failure of rtsp streaming. If the iperf test does not drop the package, the rtsp test can be 100% successful.

zsy118, 2025-02-12:
I mean, it is easy to fail to start running rtsp when the network has lost packets，It doesn't mean that it will be disconnected during rtsp operation. Can you test iperf?

mantriyogesh, 2025-02-12:
1. As this is RTSP, the link bandwidth needed is much less, typically lower than 10mbps.

2. We would run below commands:
Raspberry Pi as iperf client.

- Data direction: 
```
RaspberryPi == SDIO ==> ESP --- wireless --> Router
```
- iperf commands
  - Raspberry Pi
  ```
  iperf2 -c <iperf_server_on_laptop_ip> -i 1 -t 60 -u -b 10m -B <ethsta0_ip>
  ```
  - Laptop (over the air Wi-Fi station)
  ```
  iperf2 -s -i 1  -t 500 -B <wifi_ip> -u
  ```

If not convinced with above commands, We would request the commands you seek, also protocol and the direction.
We would keep  -b within link bandwidth if is UDP.

3. We can test tomorrow morning, If some more information you require from our side, please let us know, we can gather it together and package and send it to you.

4. Also, please confirm you have tested on Raspberry Pi?
Reason : If some other issue on the SDIO, can also result in the iperf instability.

zsy118, 2025-02-13:
The new code did not run successfully on Raspberry Pi.
oudraspperrypl:/espatal Python error: Bus erro
controL/oytnorSU0Omy-venv/bin/python -/test.Py
urrent thread 0x0000007f9c4c54c0recent calfirst)File:control/python support/commands lib.py", line 1037 in test get fw_version"/esp hosted/esp hostedfq/host/linux/hostFileesp hostedcontrol/python support/py _parse/process py", line 341 in process get fw _versionFileesp hoste(esp hostedfa/host/linux/hostcontrol/python support/py parse/cmds.py,line 359 in get fw versionFilefg/host/linux/host control/python_support/ /test py", line 175 in mainesp hoste(Fileinux/host control/python support/./test.py", line 232 in <module>esp hosted/esp hosted
Extension modules: charset normalizer.md, requests.packages.charset normalizer .md,requests.packages.chardet.md(total:3)Bus errorpi

mantriyogesh, 2025-02-13:
Hello @zsy118 ,

I see issue closed recently, but unsure if your problem is resolved?

zsy118, 2025-02-14:
> ### Checklist
> * [x]  Checked the issue tracker for similar issues to ensure this is not a duplicate[x]  Read the documentation to confirm the issue is not addressed there and your configuration is set correctly[x]  Tested with the latest version to ensure the issue hasn't been fixed
> 
> ### How often does this bug occurs?
> always
> 
> ### Expected behavior
> The host linux nt98566 connects to the ESP32c6 through the main sdio, and the esp32c6 is used as the wifi soc. When testing the RTSP push stream, the pull stream through the vlc fails. It is ok for us to use 8189wifi rtsp for the same machine. So the problem is with the esp32c6 wifi
> 
> ### Actual behavior (suspected bug)
> [image] [image] <img alt="1f0eddb36b478a657df68036bcb9a68" width="841" src="https://private-user-images.githubusercontent.com/116060739/388326088-45452003-8257-445c-add0-a4a5b8973110.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzk0OTYyMzgsIm5iZiI6MTczOTQ5NTkzOCwicGF0aCI6Ii8xMTYwNjA3MzkvMzg4MzI2MDg4LTQ1NDUyMDAzLTgyNTctNDQ1Yy1hZGQwLWE0YTViODk3MzExMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIxNFQwMTE4NThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yNjY4NjA5MTM5MzA2YWI1YWE5ZTc2YmY1NDdhN2FlMDRlY2Y1NjNhMDA1NTJjZTUzNmEyZTMxYWM4NjMyZWExJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Dw2h89NSxYrbIUY-ZODdlUg5fWG4j8pbQFYzhMyBoh8"> [image]
> 
> ### Error logs or terminal output
> _No response_
> 
> ### Steps to reproduce the behavior
> command
> 
> ### Project release version
> esp_host_fg
> 
> ### System architecture
> other (details in Additional context)
> 
> ### Operating system
> Linux
> 
> ### Operating system version
> linux kernel4.19.91
> 
> ### Shell
> sh
> 
> ### Additional context
> _No response_

zsy118, 2025-02-14:
Sorry，misoperation it.   Maybe there's something wrong with my raspberry PI and tested failed，I'm going to test it on a new platform, so the results will be more accurate.

[image]

mantriyogesh, 2025-02-14:
As you already have Raspberry Pi, I think better to get Raspberry Pi test first.

Have you installed the latest slave from hosted master?
If so, can you please attach complete textual (dmesg from start for linux) and the (esp log from bootup)?

SohKamYung-Espressif, 2025-02-14:
Tested using iperf as suggested in https://github.com/espressif/esp-hosted/issues/545#issuecomment-2653796396

Results on iperf server side (on Test PC):

```
> .\iperf-2.2.1-win64.exe -s -i 1 -t 63 -u
------------------------------------------------------------
Server listening on UDP port 5001
UDP buffer size: 64.0 KByte (default)
------------------------------------------------------------
[  1] local 10.10.2.140 port 5001 connected with 10.10.2.64 port 53489
[ ID] Interval       Transfer     Bandwidth        Jitter   Lost/Total Datagrams
[  1] 0.00-1.00 sec  1.21 MBytes  10.1 Mbits/sec   1.271 ms 0/860 (0%)
[  1] 1.00-2.00 sec  1.19 MBytes  9.98 Mbits/sec   1.699 ms 0/849 (0%)
[  1] 2.00-3.00 sec  1.19 MBytes  10.0 Mbits/sec   1.654 ms 0/852 (0%)
[  1] 3.00-4.00 sec  1.20 MBytes  10.0 Mbits/sec   1.646 ms 0/853 (0%)
[  1] 4.00-5.00 sec  1.18 MBytes  9.94 Mbits/sec   2.038 ms 0/845 (0%)
[  1] 5.00-6.00 sec  1.19 MBytes  10.0 Mbits/sec   1.641 ms 0/852 (0%)
[  1] 6.00-7.00 sec  1.19 MBytes  9.96 Mbits/sec   1.526 ms 0/847 (0%)
[  1] 7.00-8.00 sec  1.20 MBytes  10.1 Mbits/sec   1.446 ms 1/857 (0.12%)
[  1] 8.00-9.00 sec  1.19 MBytes  10.0 Mbits/sec   1.450 ms 0/851 (0%)
[  1] 9.00-10.00 sec  1.18 MBytes  9.91 Mbits/sec   1.495 ms 0/843 (0%)
[  1] 10.00-11.00 sec  1.18 MBytes  9.94 Mbits/sec   2.403 ms 0/845 (0%)
[  1] 11.00-12.00 sec  1.21 MBytes  10.1 Mbits/sec   1.558 ms 0/863 (0%)
[  1] 12.00-13.00 sec  1.17 MBytes  9.80 Mbits/sec   1.924 ms 1/834 (0.12%)
[  1] 13.00-14.00 sec  1.21 MBytes  10.2 Mbits/sec   2.018 ms 0/864 (0%)
[  1] 14.00-15.00 sec  1.19 MBytes  10.0 Mbits/sec   1.538 ms 0/851 (0%)
[  1] 15.00-16.00 sec  1.20 MBytes  10.1 Mbits/sec   1.294 ms 0/856 (0%)
[  1] 16.00-17.00 sec  1.09 MBytes  9.18 Mbits/sec   1.461 ms 62/843 (7.4%)
[  1] 17.00-18.00 sec  1.12 MBytes  9.37 Mbits/sec   2.013 ms 0/797 (0%)
[  1] 18.00-19.00 sec  1.23 MBytes  10.3 Mbits/sec   1.525 ms 21/900 (2.3%)
[  1] 19.00-20.00 sec  1.09 MBytes  9.18 Mbits/sec   2.385 ms 0/781 (0%)
[  1] 20.00-21.00 sec  1.25 MBytes  10.5 Mbits/sec   1.192 ms 32/927 (3.5%)
[  1] 21.00-22.00 sec  1.19 MBytes  10.0 Mbits/sec   1.413 ms 0/850 (0%)
[  1] 22.00-23.00 sec  1.19 MBytes  10.0 Mbits/sec   1.435 ms 0/851 (0%)
[  1] 23.00-24.00 sec  1.14 MBytes  9.60 Mbits/sec   1.645 ms 0/816 (0%)
[  1] 24.00-25.00 sec  1.21 MBytes  10.1 Mbits/sec   1.713 ms 24/886 (2.7%)
[  1] 25.00-26.00 sec  1.19 MBytes  9.98 Mbits/sec   1.218 ms 0/849 (0%)
[  1] 26.00-27.00 sec  1.19 MBytes  10.0 Mbits/sec   1.502 ms 0/850 (0%)
[  1] 27.00-28.00 sec  1.16 MBytes  9.77 Mbits/sec   1.478 ms 0/831 (0%)
[  1] 28.00-29.00 sec  1.22 MBytes  10.2 Mbits/sec   1.696 ms 0/870 (0%)
[  1] 29.00-30.00 sec  1.19 MBytes  10.0 Mbits/sec   1.195 ms 0/850 (0%)
[  1] 30.00-31.00 sec  1.19 MBytes  10.0 Mbits/sec   1.474 ms 0/850 (0%)
[  1] 31.00-32.00 sec  1.13 MBytes  9.44 Mbits/sec   1.391 ms 0/803 (0%)
[  1] 32.00-33.00 sec  1.21 MBytes  10.1 Mbits/sec   1.364 ms 37/899 (4.1%)
[  1] 33.00-34.00 sec  1.18 MBytes  9.91 Mbits/sec   1.519 ms 0/843 (0%)
[  1] 34.00-35.00 sec  1.20 MBytes  10.1 Mbits/sec   1.451 ms 0/857 (0%)
[  1] 35.00-36.00 sec  1.19 MBytes  10.0 Mbits/sec   1.276 ms 0/850 (0%)
[  1] 36.00-37.00 sec  1.19 MBytes  10.0 Mbits/sec   1.480 ms 0/850 (0%)
[  1] 37.00-38.00 sec  1.19 MBytes  10.0 Mbits/sec   1.778 ms 0/850 (0%)
[  1] 38.00-39.00 sec  1.19 MBytes  10.0 Mbits/sec   1.540 ms 0/852 (0%)
[  1] 39.00-40.00 sec  1.19 MBytes  9.97 Mbits/sec   1.621 ms 0/848 (0%)
[  1] 40.00-41.00 sec  1.17 MBytes  9.85 Mbits/sec   1.324 ms 0/838 (0%)
[  1] 41.00-42.00 sec  1.21 MBytes  10.2 Mbits/sec   1.687 ms 0/864 (0%)
[  1] 42.00-43.00 sec  1.19 MBytes  10.0 Mbits/sec   1.608 ms 0/852 (0%)
[  1] 43.00-44.00 sec  1.19 MBytes  9.98 Mbits/sec   1.627 ms 0/849 (0%)
[  1] 44.00-45.00 sec  1.19 MBytes  10.0 Mbits/sec   1.459 ms 0/851 (0%)
[  1] 45.00-46.00 sec  1.18 MBytes  9.91 Mbits/sec   1.709 ms 0/843 (0%)
[  1] 46.00-47.00 sec  1.20 MBytes  10.1 Mbits/sec   1.312 ms 0/857 (0%)
[  1] 47.00-48.00 sec  1.15 MBytes  9.64 Mbits/sec   2.448 ms 0/820 (0%)
[  1] 48.00-49.00 sec  1.18 MBytes  9.90 Mbits/sec   1.539 ms 39/881 (4.4%)
[  1] 49.00-50.00 sec  1.20 MBytes  10.0 Mbits/sec   1.358 ms 0/853 (0%)
[  1] 50.00-51.00 sec  1.19 MBytes  10.0 Mbits/sec   1.244 ms 0/851 (0%)
[  1] 51.00-52.00 sec  1.19 MBytes  10.0 Mbits/sec   1.291 ms 0/850 (0%)
[  1] 52.00-53.00 sec  1.18 MBytes  9.91 Mbits/sec   1.293 ms 0/843 (0%)
[  1] 53.00-54.00 sec  1.19 MBytes  10.0 Mbits/sec   1.636 ms 0/850 (0%)
[  1] 54.00-55.00 sec  1.19 MBytes  10.0 Mbits/sec   1.449 ms 0/851 (0%)
[  1] 55.00-56.00 sec  1.19 MBytes  10.0 Mbits/sec   1.290 ms 0/850 (0%)
[  1] 56.00-57.00 sec  1.20 MBytes  10.0 Mbits/sec   1.519 ms 0/854 (0%)
[  1] 57.00-58.00 sec  1.17 MBytes  9.84 Mbits/sec   1.600 ms 0/837 (0%)
[  1] 58.00-59.00 sec  1.21 MBytes  10.1 Mbits/sec   1.528 ms 0/862 (0%)
[  1] 59.00-59.99 sec  1.18 MBytes  10.1 Mbits/sec   1.348 ms 0/843 (0%)
[  1] 0.00-59.99 sec  71.2 MBytes  9.96 Mbits/sec   1.707 ms 217/51024 (0.43%)
```

There is occasional UDP packet loss, but I think this is to be expected for UDP traffic over Wi-Fi.

I also retested VLC fetching the video from the live555MediaServer on the RaspPi and did not see any issues playing the video.

zsy118, 2025-02-14:
See the attachment for log

[pi log.txt](https://github.com/user-attachments/files/18796206/pi.log.txt)

[esp32 log.txt](https://github.com/user-attachments/files/18796210/esp32.log.txt)

zsy118, 2025-02-14:
In the past, when testing the ESP32C6 rtsp was prone to failure, the iperf data was similar to the results of your test，can it be further optimized?

SohKamYung-Espressif, 2025-02-14:
@zsy118 
In your `pi_log.txt` I see this:

```
[  351.782540] esp32_sdio: esp_handle_isr: CMD53 read/write error at 185
[  351.782614] esp32_sdio: esp_get_len_from_slave: Len from slave[1048210] exceeds max [1536]
[  351.782622] esp32_sdio: read_packet: Rx large packet: 1048210
[  351.782660] esp32_sdio: read_packet: Failed to read data: Err[-22] bytes_to_read[1048576]u
```
This indicates that your SDIO connection may not be stable as the data received was corrupted.

Can you take a photo of your setup, showing the SDIO connections between your system and the ESP32-C6?

In your `esp32_log.txt` I see this:

```
I (10653) slave_ctrl: Station disconnected, reason[201]
I (10654) slave_ctrl: Send DHCP-DNS status to Host
I (10654) slave_ctrl: Sta mode disconnect, retry[3]
I (13068) slave_ctrl: Station disconnected, reason[201]
I (13069) slave_ctrl: Send DHCP-DNS status to Host
I (13069) slave_ctrl: Sta mode disconnect, retry[4]
I (15484) slave_ctrl: Station disconnected, reason[201]
I (15484) slave_ctrl: Send DHCP-DNS status to Host
I (15484) slave_ctrl: Sta mode disconnect, retry[5]
I (17899) slave_ctrl: Station disconnected, reason[201]
I (17899) slave_ctrl: Send DHCP-DNS status to Host
I (17899) slave_ctrl: Sta mode disconnect, retry[6]
I (20314) slave_ctrl: Station disconnected, reason[201]
I (20314) slave_ctrl: Send DHCP-DNS status to Host
I (20315) slave_ctrl: Sta mode disconnect, retry[7]
I (22729) slave_ctrl: Station disconnected, reason[201]
I (22729) slave_ctrl: Send DHCP-DNS status to Host
I (22730) slave_ctrl: Sta mode disconnect, retry[8]
I (25144) slave_ctrl: Station disconnected, reason[201]
```
Reason[201], from the `ESP-IDF esp_wifi_types_generic.h` file is:

```c
    WIFI_REASON_NO_AP_FOUND                        = 201,    /**< No AP found */
```
Which indicates the ESP32-C6 cannot get a stable connection to the AP.

Is the AP located far away from the ESP32-C6, or is the wifi network very congested?

Can you try using a stand-alone AP or hotspot near the ESP32-C6.

zsy118, 2025-02-15:
According to the log, it should be the sdio connection problem, but the sdio line still fails after I have repeatedly connected it, so I plan to test the platform again，The AP device is less than 1 meter away from the platform.I'll let you know the test results when the new platform is ready
[image]

mantriyogesh, 2025-02-15:
Refrain from using jumper cables for sdio. You can use PCB. Ensure you add external pull ups on the lines, as mentioned in the documentation.

It looks like C6 - Raspberry Pi connected.

If you wish to continue using jumper cables, switch to spi.

mantriyogesh, 2025-02-15:
You need to use minimum pull up like 10 K Ohms on specific lines. We typically use 51KOhm. In case you want the schema of our board for sdio, connecting c6-sdio-raspberry-pi, I would request you to mail me.

zsy118, 2025-02-18:
hi，The new board or the problem is still the same, should not be the problem of sdio connection, sdio 10k pull up has been increased. Could you please send me your sdio architecture? My email is master@zeoyea.com.
[image]

zsy118, 2025-02-19:
It has nothing to do with the hardware ESP32-C6, and has replaced several ESP32-C6.It should be a software environment or code problem. 

[image]

mantriyogesh, 2025-02-19:
I have shared you the schematic (hardware), software code is already with you.

Once you check with your SDIO hardware schematic, let us know further.

From the image, the **left** side device **doesn't** look like Espressif manufactured device. Right side device looks legitimate.

Use right side device on your PCB.

The obvious issues for SDIO involve:

1. Using jumper cables. Big no to using jumper cables as SDIO is very sensitive to signal. Signal integrity issues start with jumper cables. Only use PCB with same length.
2. Ensure correct voltage at power and all the SDIO pins.
3. external pull-ups >10KOhm is mandatory for specific likes. check the porting guide https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md for details.
4. Pins are not correctly mapped or the host is not set up for SDIO from device tree or the configuration.
5. SDIO timing is incorrect from host to slave (then may need to tune the timing on slave)

6. Has this board worked anytime earlier or this setup is tested for the first time?

Please answer above questions, to debug further.

mantriyogesh, 2025-02-19:
If you use wires like in https://github.com/espressif/esp-hosted/issues/545#issuecomment-2660640138, DO NOT use SDIO. use SPI.

Please answer above comment to futher assess the platform set up and the environment.

zsy118, 2025-02-24:
Hi, I have made a comparison test between the esp32c6 module of Le Xin and the purchased module, and the two modules will appear bus error when run with python_demo; Using c_demo test can be connected normally, but the network speed is very poor, there is packet loss phenomenon, can not RTSP push stream.

[image]

mantriyogesh, 2025-02-24:
Thanks for the observations. Can you send us back the full code that you used along with sdkconfig and the both side logs from start?

It is C6 and raspberry pi, so we should reproduce the issue you face.

Possibly pack the binaries at C6 and Linux c binary at raspberry pi.

Kernel module would not work if the kernel version is even slightly different, so we can build with our current kernel and try again.

Jut one more input though, are you sure the raspberry pi input is sufficient powered as per its input rating?

> The network speed is poor

As you are packing the code exactly you have, we can test in our environment the set of commands that you execute.

We know how to load and test etc. but I think it would be perfect if we follow your executed commands in that sequence.

So I think either the commands or a complete textual logs for all terminals that you use from flashing to issue reproduce, would be much useful.

mantriyogesh, 2025-02-24:
What are the three wires soldered on raspberry pi 3b+?

The pins are for UART, but they are used for? Are they connected to C6 in any way?

zsy118, 2025-02-24:
These three wires are not connected to c6, and these three wires are only connected to PI's debug uart

[image]