espressif/esp-hosted #546: Is the esp-hosted support the BluFi solution?

wsun987, 2024-11-21:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

From the block diagram, the wi-fi driver and bluetooth driver are in the esp32 chip side, the tcp/ip stack and bluetooth stack in the host mcu side.
For my production, my mcu only care about the wi-fi transportation, for the bluetooth which only used to config the Wi-Fi network connection with the router, so  is it possible to deploy the BluFi related solution only in the esp32 chip side?
If yes, is it easily or need a lot of work?

thanks!

mantriyogesh, 2024-11-21:
Is it possible, yes, definitely, in fact some of our customers have it using hosted on host.

For host, you can just add up the https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/blufi at host over bluetooth stack at host?

But you need only at slave. I did not understand then how would you manage the wifi connection/disconnection from host mcu.

I am just trying to understand the full scenario, so that if something I can think of can be a help to you.

wsun987, 2024-11-21:
Thanks for your answering.

My considering as below:
First, my product is a data collection gateway which collect the local devices' data and publish to the cloud server. The Wi-Fi only need to config once after the gateway mounting, it will connect to customer's wi-fi router always. The host mcu doesn't  need to connect/disconnect the Wi-Fi router frequently.

Second, if I deploy the BluFi in host MCU side, then i need to porting the bluetooth stack and application code to MCU. I think it is a little complex. If all bluetooth related stack and codes are deployed in the slave side , i think the system will more simple and more stable, right?

wsun987, 2024-11-21:
The iphone/android phone can communicate and config the wifi router via blufi based on bluetooth connection , the host mcu(stm32) only need to operate the tcp/ip ,mqtt connection.

mantriyogesh, 2024-11-21:
For microcontroller case actually, we have some internal code, which controls the wifi completely from slave side. 

It would also store the wifi said , password in nvs, so need to repeat wifi provision once done.

Now we assume, the wifi is getting resolved using the blufi (provisioning). Once wifi is connected, it also gets the ip address and informs the host directly the {ip,nm,gw,dns} to be used.

On host, it would simply:
1. Do not start any network app unless got_ip event
2. Start network function after got up event
3. Stop network function after lost ip event

This code is already in development, but having some performance issues, to take care.
It uses ESP MCU as slave and another ESP as host MCU.

mantriyogesh, 2024-11-21:
But if you use Linux , I would need to dig code to find dependencies if any, similar to above in MCU case. And, similar code might need to be coded for Linux.

Which might take considerable time. 

To answer your question, yes said use case is little customised, but can be fulfilled with ESP hosted.

wsun987, 2024-11-22:
Yes, your thoughts are exactly same with ours. 

My host is MCU, which is a cortex m4 core, similar to the stm32f4 series.

So, as you mentioned, this solution still in development, right?   Then due to the deadline of our project, maybe will use the ESP-AT solution or something.

Thanks anyway!

mantriyogesh, 2024-11-22:
Well, it is actually customised solution for specific customer set, but not yet generic enough to fit in the public repo. that is the reason it is not yet public..

It has good amount of customizations, so to make it public, we need to mould it correctly, which taking some time.

Can you please send test mail, so that I can send you a link to the code to you.
My mail id is attached to GitHub account.