espressif/esp-hosted #548: Unloading and loading of ESP32 SPI linux kernel module behavior

iradmilo-bytelab, 2024-11-21:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

only once

### Expected behavior

I am expecting graceful termination of SPI connection between Linux host and ESP peripheral after kernel module is unloaded and graceful load of module after loading it again.

Host MPU:  STM32MP133F
WIFI/BT network adapter MCU: ESP32_C3
Comm interface: SPI
Commit: 927a34f99724bb744efaf865a7c75cc1615f59a2 

### Actual behavior (suspected bug)

Please guide me if this is expected behavior and are there any issues based on the logs?
Why is there unregistered event on ESP side (I (1930) FW_CMD: Unregistered event: 3
) and is the cleanup on linux host side complete (logs are attached below)?

[esp32_c3_peripheral.log](https://github.com/u
[linux_host.log](https://github.com/user-attachments/files/17845993/linux_host.log)
ser-attachments/files/17845992/esp32_c3_peripheral.log)

During testing there was one occasion where module would not load properly and wlan0 interface did not show up.
However I cannot replicate it at the moment. Issue appeared only once.

Note that kernel module is automatically set to load at bootup.
Module is loaded with these options:
> cat /etc/modprobe.d/esp32_spi.conf
> options esp32_spi resetpin=71

module location:

> /lib/modules/6.1.28/extra/esp32_spi.ko

and depends upon and loads these modules:

> cat /lib/modules/6.1.28/modules.dep
> extra/esp32_spi.ko: kernel/net/bluetooth/bluetooth.ko kernel/crypto/ecdh_generic.ko kernel/crypto/ecc.ko kernel/net/wireless/cfg80211.ko kernel/lib/crypto/libaes.ko

Additional question, do bluetooth and cfg80211 drivers need to be loaded and unloaded every time you load kernel module (since I am not manipulating them directly after startup at the moment)? However performing the action of unloading other module would be inconvenient to us since other programs could use them.

### Steps to reproduce the behavior

1. Create esp_hosted_ng solution with SPI connection
2. Load module at bootup
4. Unload module with: modprobe -r <kernel_object>
5. Load module with: modprobe <kernel_object>

### Project release version

927a34f99724bb744efaf865a7c75cc1615f59a2

### System architecture

ARM 32-bit (Raspberry Pi 32-bit)

### Operating system

Linux

### Operating system version

6.1.28 (Mickledore) stm32-linux

### Shell

Bash

mantriyogesh, 2024-11-21:
@kapilkedawat ++

mantriyogesh, 2025-01-14:
Apologies, we are late on this. are you still stuck at same place?

mantriyogesh, 2025-01-14:
> do bluetooth and cfg80211 drivers need to be loaded and unloaded every time you load kernel module 

@iradmilo-bytelab , no this is required one time. but as part of script, we showcase the dependency modules and unload, assuming might not be needed.

In your case, it would be better to use `sooftdep` from https://man7.org/linux/man-pages/man5/modprobe.d.5.html.

iradmilo-bytelab, 2025-01-14:
Hi @mantriyogesh, thank you for the answer.

My guess is that the issue was/is a hard dependency set between esp kernel and cfg80211 and after the cfg80211 was unloaded and loaded as an esp dependency, wpa_supplicant had to be restarted for wifi to work properly. So as a workaround I am removing esp module with rmmod which does not unload the dependencies. It is good to know that with softdep I can configure that.

closed 2025-01-14 by iradmilo-bytelab