espressif/esp-hosted #549: Very poor performance with SDIO, sporadic SDIO errors

mdeneen, 2024-11-22:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I expected reasonable throughput

### Actual behavior (suspected bug)

Throughput is about 2 megabits per second, and it stops working entirely after less than 10 megabytes have been transferred.

### Error logs or terminal output

```shell
root@imx93:~# modprobe esp32_sdio resetpin=608 clockspeed=60
root@imx93:~# wpa_supplicant -i wlan0 -B -c /data/tds/wpa_supplicant.conf 
Successfully initialized wpa_supplicant
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
nl80211: kernel reports: Registration to specific type not supported
root@imx93:~# udhcpc -i wlan0
udhcpc: started, v1.36.1
Dropped protocol specifier '.udhcpc' from 'wlan0.udhcpc'. Using 'wlan0' (ifindex=29).
udhcpc: broadcasting discover
udhcpc: broadcasting select for 192.168.2.24, server 192.168.2.1
udhcpc: lease of 192.168.2.24 obtained from 192.168.2.1, lease time 3600
RTNETLINK answers: File exists
/etc/udhcpc.d/50default: Adding DNS 8.8.8.8
Dropped protocol specifier '.udhcpc' from 'wlan0.udhcpc'. Using 'wlan0' (ifindex=29).
root@imx93:~# ip route add xxx.xxx.xxx.xxx/32 via 192.168.2.1
root@imx93:~# iperf -c xxx.xxx.xxx.xxx -p 8000
connect failed: Connection refused
root@imx93:~# iperf -c xxx.xxx.xxx.xxx -p 8000
------------------------------------------------------------
Client connecting to xxx.xxx.xxx.xxx, TCP port 8000
TCP window size: 85.0 KByte (default)
------------------------------------------------------------
[  3] local 192.168.2.24 port 45854 connected with xxx.xxx.xxx.xxx port 8000
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-11.0 sec  2.00 MBytes  1.52 Mbits/sec
[... 88 lines cut ...]
I (30113) stats: 27-28 sec       74460.00 kbits/sec
I (31113) stats: 28-29 sec       73513.28 kbits/sec
I (32113) stats: 29-30 sec       74209.06 kbits/sec
I (33113) stats: 30-31 sec       74551.25 kbits/sec
I (34113) stats: 31-32 sec       73741.41 kbits/sec
I (35113) stats: 32-33 sec       74152.03 kbits/sec
I (36113) stats: 33-34 sec       74505.62 kbits/sec
I (37113) stats: 34-35 sec       74562.66 kbits/sec
I (38113) stats: 35-36 sec       73536.09 kbits/sec
I (39113) stats: 36-37 sec       74585.47 kbits/sec
I (40113) stats: 37-38 sec       74631.09 kbits/sec
I (41113) stats: 38-39 sec       73433.44 kbits/sec
I (42113) stats: 39-40 sec       74448.59 kbits/sec
I (43113) stats: 40-41 sec       74585.47 kbits/sec
I (44113) stats: 41-42 sec       73490.47 kbits/sec
I (45113) stats: 42-43 sec       74551.25 kbits/sec
I (46113) stats: 43-44 sec       74562.66 kbits/sec
I (47113) stats: 44-45 sec       74243.28 kbits/sec
I (48113) stats: 45-46 sec       73558.91 kbits/sec
I (49113) stats: 46-47 sec       74471.41 kbits/sec
I (50113) stats: 47-48 sec       74163.44 kbits/sec

I had previously reported problems with instability in issue https://github.com/espressif/esp-hosted/issues/482, which was related to our linear regulator being insufficient.  The output current of the LDO was great, but the transient response was not.  It has been replaced with a more capable part -- the esp32c6 is no longer suffering from brownout conditions, but the performance isn't great.  This particular SoC has three SDIO interfaces and it can really move some data, so I'm quite trusting in its SDIO abilities.

I have tried many clock speeds, but lower ones don't seem any more stable than high ones.

mantriyogesh, 2024-11-22:
clockspeed=60

Exceeds the max capacity of slave SDIO for [esp32-c6](https://www.espressif.com/sites/default/files/documentation/esp32-c6_technical_reference_manual_en.pdf).
[image]

Worth to check what real frequency Linux released although.

```sh
sudo cat /sys/kernel/debug/mmcX/ios
```

mdeneen, 2024-11-22:
I hear you, but I have the same behavior with 1, 5, 10, 20, 30, 40, 50, and 60.

At 60, it does this:

clock:		60000000 Hz
actual clock:	57142857 Hz
vdd:		21 (3.3 ~ 3.4 V)
bus mode:	2 (push-pull)
chip select:	0 (don't care)
power mode:	2 (on)
bus width:	2 (4 bits)
timing spec:	2 (sd high-speed)
signal voltage:	0 (3.30 V)
driver type:	0 (driver type B)

I agree that it is out of spec, and perhaps that was the wrong example to paste in here; it just happened to be the last test that I did.

mantriyogesh, 2024-11-22:
Its better not to exceed 50.

it would be undefined behaviour.
For adjusting the timing, there is mention of [sdio adjustment](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md#242-sdio) timing in porting doc.

You can try changing. but in any case you both side traffic test makes sense.

-84 is worrysome. Can you please send me recent picture for c6 integration?

mdeneen, 2024-11-22:
> Additionally, you have captured the throughput one side, what about reverse direction?

Would these be show in the logs on the host side?  If so they looked nearly identical to the messages printed on the esp32c6 side.

> Considering -84 errors, Did you check the external pull-up register requirements from [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md)?

I will confirm

> Also, please note, the SDIO can only work over the PCB, not jumper cables.

The esp32c6 module is on the PCB, and the traces are ~30mm and length balanced.

mdeneen, 2024-11-22:
> Considering -84 errors, Did you check the external pull-up register requirements from [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md)?

Yes, we have pull-ups

mantriyogesh, 2024-11-22:
> Would these be show in the logs on the host side? If so they looked nearly identical to the messages printed on the esp32c6 side.

one direction, suppose esp to host. other direction would be host to esp.

Both need to test in individual test case. check  https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Raw_TP_Testing.md. two different test cases / runs with reloads of esp.