espressif/esp-hosted #550: Add Releases

giuliobenetti, 2024-11-26:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Maintaining esp-hosted in Buildroot would be simpler and clearer if versions are released from time to time.
This has been also asked here:
https://lore.kernel.org/all/193607b65c0.2856.ffe91ee9dc07baa0c1c4561a0e181c1b@free.fr/
and I find it very useful.

mantriyogesh, 2024-11-27:
@kapilkedawat, I think we can do this..

@giuliobenetti , will get back on this..

kapilkedawat, 2024-11-28:
@giuliobenetti yes, new release will be coming out shortly.

giuliobenetti, 2024-11-28:
> @giuliobenetti yes, new release will be coming out shortly.

Perfect, thank you!

kaidegit, 2024-12-27:
Hi,
When can we get a new release version?