espressif/esp-hosted #551: Zephyr support

andrewleech, 2024-11-27:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

Is there any known support / port of the esp-hosted "host" api to [zephyr rtos](https://github.com/zephyrproject-rtos/zephyr) ?

I already know and use esp-hosted to provide wifi and bluetooth to host MCU's running micropython and would like to use this same system for host MCU's running zephyr.

I'm not that familar with the zephyr networking implementations though I know it's got some overlay / inspiration from linux, so hoping there might be a known working implementation we could reference. It would be fantastic if this integration could eventually be included in [upstream zephyr wifi](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/wifi) and [bluetooth](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/bluetooth/hci) so everyone can take advantage of this project.

### Use cases

I would like to use esp-hosted running on a esp32-s3 or similar to provide network and bluetooth for host mcu's running zephyr, most commonly nxp mimxrt or stm32 devices in our applications.

### Alternatives

Currently the only officially supported means I have to adding wifi and ble to both micropython and zephyr is based on infineon chipsets but espressif parts are far more readily available on pre-certified modules so I'd like to move to this platform.

kaidegit, 2025-01-02:
You can refer to the this. However, it is not provided by espressif.

https://cloud.listenai.com/zephyr/esp-hosted/-/tree/master/

andrewleech, 2025-01-02:
That's great, thanks for the example!

matt-wood-ct, 2025-05-09:
This would be amazing, currently zephyr only has the ESP-AT interface for espressif parts which is a socket offload. A L2 
external wifi driver would make using ESP parts much more useful.