espressif/esp-hosted #554: ESP Module as cooprocessor with STM32 as host

NorbertLis, 2024-12-17:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

Hello, I want to use the ESP module as a Wi-Fi module in my application written for STM32. I would like to use the ESPHosted library, but I am having trouble with compilation. After adding the esp_hosted_mcu and esp_wifi_remote submodules, I encounter many references to header files such as esp_err.h, esp_log.h, esp_wifi.h, and sdkconfig.h. These are likely headers from the ESP-IDF framework.

Why can’t you separate these files so they are not required for compilation on other MCUs?

Over a year ago, I tested your library in an earlier version called ESPHosted-FG, which even included examples for STM32. That version didn’t require ESP-IDF for compilation.

Please simplify the integration process for other MCUs.

mantriyogesh, 2024-12-18:
Dedicated MCU support is added in ESP-Hosted-MCU repo.

Port for MCU is also being added at 

[Doc](https://github.com/espressif/esp-hosted/issues/522#issuecomment-2540549367)
[Code](https://github.com/espressif/esp-hosted/issues/522#issuecomment-2540549850)

mantriyogesh, 2024-12-18:
I will close this issue here now. Please evaluate that repo and let us know if issues on that repo. https://github.com/espressif/esp-hosted-mcu

closed 2024-12-18 by mantriyogesh