espressif/esp-hosted #555: bleprph_host_only_vhci idf.py menuconfig missing Component config -> ESP-Hosted config menu?

csalinasonline, 2024-12-23:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Trying to build and run esp-hosted -> example -> bleprph_host_only_vhci
On HW ESP32-P4 and using the Bluetooth Controller of the ESP32C6 on the ESP32-P4-Function-EV-Board.

### Actual behavior (suspected bug)

Cannot Build and Flash project: bleprph_host_only_vhci. 

### Error logs or terminal output

```shell
xxx@MSI:~/esp/esp-idf/examples/esp-hosted/examples/bleprph_host_only_vhci$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/xxx/esp/esp-idf/examples/esp-hosted/examples/bleprph_host_only_vhci/build
Executing "ninja all"...
[1/1] cd /home/xxx/esp/esp-idf/examples/esp-hoste...leprph_host_only_vhci/build/bootloader/bootloader.bin
Bootloader binary size 0x5960 bytes. 0x6a0 bytes (7%) free.
[2/4] Linking CXX executable bleprph_host_only_vhci.elf
FAILED: bleprph_host_only_vhci.elf
: && /home/xxx/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20240906/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f -nostartfiles -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32P4=0 -Wl,--Map=/home/xxx/esp/esp-idf/examples/esp-hosted/examples/bleprph_host_only_vhci/build/bleprph_host_only_vhci.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32p4.peripherals.ld -T esp32p4.rom.ld -T esp32p4.rom.api.ld -T esp32p4.rom.rvfp.ld -T esp32p4.rom.wdt.ld -T esp32p4.rom.systimer.ld -T esp32p4.rom.version.ld -T esp32p4.rom.newlib.ld -T memory.ld -T sections.ld -Wl,--enable-non-contiguous-regions CMakeFiles/bleprph_host_only_vhci.elf.dir/project_elf_src_esp32p4.c.obj -o bleprph_host_only_vhci.elf -L/home/xxx/esp/esp-idf/components/riscv/ld   -L/home/xxx/esp/esp-idf/components/soc/esp32p4/ld   -L/home/xxx/esp/esp-idf/components/esp_rom/esp32p4/ld   -L/home/xxx/esp/esp-idf/examples/esp-hosted/examples/bleprph_host_only_vhci/build/esp-idf/esp_system/ld esp-idf/riscv/libriscv.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_security/libesp_security.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_driver_gptimer/libesp_driver_gptimer.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/esp_driver_uart/libesp_driver_uart.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/esp_driver_usb_serial_jtag/libesp_driver_usb_serial_jtag.a  esp-idf/esp_vfs_console/libesp_vfs_console.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a  esp-idf/esp_driver_spi/libesp_driver_spi.a  esp-idf/esp_driver_mcpwm/libesp_driver_mcpwm.a  esp-idf/esp_driver_ana_cmpr/libesp_driver_ana_cmpr.a  esp-idf/esp_driver_i2s/libesp_driver_i2s.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_driver_sdmmc/libesp_driver_sdmmc.a  esp-idf/esp_driver_sdspi/libesp_driver_sdspi.a  esp-idf/esp_driver_rmt/libesp_driver_rmt.a  esp-idf/esp_driver_tsens/libesp_driver_tsens.a  esp-idf/esp_driver_sdm/libesp_driver_sdm.a  esp-idf/esp_driver_i2c/libesp_driver_i2c.a  esp-idf/esp_driver_ledc/libesp_driver_ledc.a  esp-idf/esp_driver_parlio/libesp_driver_parlio.a  esp-idf/driver/libdriver.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_driver_isp/libesp_driver_isp.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_driver_jpeg/libesp_driver_jpeg.a  esp-idf/esp_driver_ppa/libesp_driver_ppa.a  esp-idf/esp_driver_touch_sens/libesp_driver_touch_sens.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/rt/librt.a  esp-idf/spiffs/libspiffs.a  esp-idf/usb/libusb.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  -Wl,--whole-archive  -Wl,--no-whole-archive  esp-idf/espressif__esp_serial_slave_link/libespressif__esp_serial_slave_link.a  esp-idf/espressif__eppp_link/libespressif__eppp_link.a  esp-idf/espressif__esp_wifi_remote/libespressif__esp_wifi_remote.a  esp-idf/nimble_peripheral_utils/libnimble_peripheral_utils.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/app_trace/libapp_trace.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_driver_isp/libesp_driver_isp.a  esp-idf/esp_driver_jpeg/libesp_driver_jpeg.a  esp-idf/esp_driver_ppa/libesp_driver_ppa.a  esp-idf/esp_driver_touch_sens/libesp_driver_touch_sens.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/espcoredump/libespcoredump.a  -u esp_system_include_coredump_init  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  -u nvs_sec_provider_include_impl  esp-idf/rt/librt.a  esp-idf/spiffs/libspiffs.a  esp-idf/usb/libusb.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/nimble_peripheral_utils/libnimble_peripheral_utils.a  esp-idf/console/libconsole.a  esp-idf/riscv/libriscv.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_security/libesp_security.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_driver_gptimer/libesp_driver_gptimer.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/esp_driver_uart/libesp_driver_uart.a  esp-idf/esp_event/libesp_event.a  esp-idf/esp_driver_usb_serial_jtag/libesp_driver_usb_serial_jtag.a  esp-idf/esp_vfs_console/libesp_vfs_console.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a  esp-idf/esp_driver_spi/libesp_driver_spi.a  esp-idf/esp_driver_mcpwm/libesp_driver_mcpwm.a  esp-idf/esp_driver_ana_cmpr/libesp_driver_ana_cmpr.a  esp-idf/esp_driver_i2s/libesp_driver_i2s.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_dr [... cut ...]
  + Disable Nimble Options-> Host-controller Transport -> Enable Uart Transport.
+ Component config-> ESP-Hosted config. NOT OKAY. CANNOT FIND ITEM?
  + Enable Bluetooth Support-> Enable Hosted Bluetooth support
+ Build and Flash. NO
  + idf.py build

### Project release version

https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

Ubuntu 18

### Shell

Bash

SohKamYung-Espressif, 2024-12-24:
@csalinasonline Thank you for reporting this.

> Component config-> ESP-Hosted config. NOT OKAY. CANNOT FIND ITEM? 

To resolve this, follow these steps:

1. select `Component config` ---> `Wi-Fi Remote` ---> `Choose WiFi-remote implementation` ---> `ESP-HOSTED`
2. return to `Component config` level. You should now see a `ESP-Hosted config` option.
3. select `ESP-Hosted config` ---> `Bluetooth Support` ---> ` Enable Hosted Bluetooth support`

Save the configuration. It should now build as expected.

There have been changes to the Wi-Fi Remote component that caused the problem you saw. I will update the documentation to reflect this.

csalinasonline, 2024-12-24:
> @csalinasonline Thank you for reporting this.
>
>
> To resolve this, follow these steps:
>
> 1. select `Component config` ---> `Wi-Fi Remote` ---> `Choose WiFi-remote implementation` ---> `ESP-HOSTED`
> 2. [...]

@SohKamYung-Espressif Thanks for the tip. I am now able to successfully build and run the project.
But I think the C6 might not be configured correctly with P4?  Previously I have followed the steps here to create a project copy of "espressif/esp_hosted:slave" to configure C6 and program via ESP-Prog.
https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md 
via (5.2 Using ESP-Prog).

Is this still sufficient to have a working C6 and P4 using esp-hosted bleprph_host_only_vhci?
I get issues when trying to connect via iOS App LightBlue to interrogate the BLE connection.

Please see attached "idf_flash_monitor.txt" debug output and the attached images.
[idf_flash_monitor_log.txt](https://github.com/user-attachments/files/18235758/idf_flash_monitor_log.txt)
[image]
[image]
[image]
[image]

csalinasonline, 2024-12-24:
@SohKamYung-Espressif 

Specific idf menuconfig settings on P4 and C6

For idf.py menuconfig on C6 via copy of "espressif/esp_hosted:slave"

[image]

For idf.py menuconfig on C6 via https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/docs/esp32_p4_function_ev_board.md (5.2 Using ESP-Prog).

[image]

SohKamYung-Espressif, 2024-12-24:
@csalinasonline Something does not look right. You should see `nimble-bleprph` as the device.

[image]

The device name is set in `main/main.c`:

```c
    rc = ble_svc_gap_device_name_set("nimble-bleprph");
```

In https://github.com/espressif/esp-hosted/issues/555#issue-2756717421 : 

> On ubuntu 18, esp-idf 5.4:
>
>    Git pull esp-hosted inside esp-idf/example
>    Git checked out feature/esp_as_mcu_host branch

Could you try again, but check out esp-hosted into a empty directory, not into esp-idf/example ? The esp-hosted NimBLE example does not depend on being put inside the esp-idf directory.

>  Previously I have followed the steps here to create a project copy of "espressif/esp_hosted:slave" to configure C6 and program via ESP-Prog.

I also did checked out the ESP32-C6 using the ESP-Prog steps and it is still correct. 

The ESP32-C6 configuration should just work as it is as a co-processor for the P4, without any changes, after doing `idf.py set-target esp32c6` .

For your reference, here are my generated sdkconfig files for the C6 and P4 in a zip folder.

[sdkconfigs.zip](https://github.com/user-attachments/files/18236104/sdkconfigs.zip)

SohKamYung-Espressif, 2024-12-24:
@csalinasonline I did a grep of the esp-idf code and the name you see appears in two places:

```shell
> grep -rn --exclude-dir=".git" --include="*.[ch]" "ESP BLE HID2"
examples/bluetooth/esp_hid_device/main/esp_hid_device_main.c:394:    .device_name        = "ESP BLE HID2",
examples/bluetooth/esp_hid_host/main/esp_hid_gap.c:855:            ((adv_name_len > 0 && memcmp("ESP BLE HID2", adv_name, adv_name_len) == 0) ||
```

I think the code you are trying to build may not be the correct one.

As I suggest above, check out esp-hosted into an empty directory and build the esp-hosted ble example from there.

csalinasonline, 2024-12-26:
@SohKamYung-Espressif Thanks for the quick update. I will apply your updates and let you know.

SohKamYung-Espressif, 2025-01-07:
@csalinasonline Any updates? Have you managed to build the code?

csalinasonline, 2025-01-07:
Hi @SohKamYung-Espressif not yet as Im at CES this week. Next week likely I'll get back to you on more news.

csalinasonline, 2025-01-13:
@SohKamYung-Espressif 

I'm back from CES and have met some of your Espressif teammates! Great folks and Awesome new stuff!
And I have good news!
Your suggestions have worked wonderfully!

On a high level:
0) Have ESP32-P4 Function_EV_Board v1.4 on 'USB-UART' port (esp32c6 already programmed with esp-hosted from before) and connected to computer usb-c.
1) Clean computer (new MacBook)
2) Git pull esp-idf repo
3) Switch branch release/v5.4
4) Git pull esp-hosted repo
5) Switch branch feature/esp_as_mcu_host
6) Go back to esp-idf dir and run ./install.sh and . ./export.sh
7) Navigate to esp-hosted dir, then examples, then bleprph_host_only_vhci
8) Backup old sdkconfig
9) Override sdkconfig-for-p4 (your copy) to sdkconfig (my local copy)
10) idf.py  set-target esp32p4
11) idf.py menuconfig to check
12) idf.py build GOOD
13) idf.py flash monitor GOOD
14) Open iPhone App 'LightBlue' to interrogate 'nimble-bleprph'
15) Save pictures and debug.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[debug_log_011325.txt](https://github.com/user-attachments/files/18401553/debug_log_011325.txt)

SohKamYung-Espressif, 2025-01-14:
@csalinasonline Thanks for the kind words. 😄 

If everything looks okay, you can close this issue. You can, of course, open a new one if you encounter any further problems.

csalinasonline, 2025-01-14:
@SohKamYung-Espressif I sure I will have more!
Thanks for your patience and help.
Closing isssue.

closed 2025-01-14 by csalinasonline

mantriyogesh, 2025-01-15:
@SohKamYung-Espressif is there any missing default option in sdkconfig for example?

We can default the bluetooth comfigs if anything missing inexample?

SohKamYung-Espressif, 2025-01-16:
`bleprph_host_only_vhci` example has a `sdkconfig.default` that should set up the BT stack for Nimble (vHCI) and enabled BT support in ESP-Hosted.

Just tested and discovered a change in the name of a `CONFIG` setting causes the build to fail after the recent commit to enable BlueDroid. Will post a MR to fix this.