espressif/esp-hosted #557: esp32c2 hosted for NG on spi interface,  restart abnormaly when run wget

WindWard-atm, 2024-12-26:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

esp32c2 work in linux system on spi interface ,restart abnormaly when run 3 wget apps
IDF commit: e7771c75bd1dbbfb7b3c5381be7e063b197c9734
application commit : 0ee257d1c4fb0d01777692c9eec4f0b7a0f08de6

[image]

[Serial-COM82_2024-12-25_18-42-09-984.log](https://github.com/user-attachments/files/18248096/Serial-COM82_2024-12-25_18-42-09-984.log)

mantriyogesh, 2024-12-26:
This is due to C2 low memory. We need to fix this.

mantriyogesh, 2024-12-26:
@kapilkedawat ++

WindWard-atm, 2024-12-26:
> This is due to C2 low memory. We need to fix this.

waiting for your update

WindWard-atm, 2025-01-06:
> This is due to C2 low memory. We need to fix this.

any news?

mantriyogesh, 2025-01-06:
Hello @WindWard-atm ,

I apologise for the delay. Let me come back on this today.

mantriyogesh, 2025-01-06:
@kapilkedawat , were we able to test and reproduce this ?

kapilkedawat, 2025-01-06:
Hi @WindWard-atm, please apply this patch and see if you still face any crash after that. Please note that you need to clean and then rebuild the firmware for this to apply properly.

diff --git a/esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32c2 b/esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32c2
index 519e7abc..29d8aae0 100644
--- a/esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32c2
+++ b/esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32c2
@@ -30,10 +30,13 @@ CONFIG_PARTITION_TABLE_FILENAME="partitions.esp32c2.csv"
 CONFIG_ESP_SPI_RX_Q_SIZE=6
 CONFIG_ESP_SPI_TX_Q_SIZE=6
 
-CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=14
+CONFIG_ESP_WIFI_STATIC_RX_BUFFER_NUM=6
 CONFIG_ESP_WIFI_DYNAMIC_RX_BUFFER_NUM=18
 CONFIG_ESP_WIFI_DYNAMIC_TX_BUFFER_NUM=18
 CONFIG_ESP_WIFI_AMPDU_TX_ENABLED=y
-CONFIG_ESP_WIFI_TX_BA_WIN=16
+CONFIG_ESP_WIFI_TX_BA_WIN=6
 CONFIG_ESP_WIFI_AMPDU_RX_ENABLED=y
-CONFIG_ESP_WIFI_RX_BA_WIN=16
+CONFIG_ESP_WIFI_RX_BA_WIN=6

WindWard-atm, 2025-01-08:
> Hi @WindWard-atm, please apply this patch and see if you still face any crash after that. Please note that you need to clean and then rebuild the firmware for this to apply properly.
>
> ```
> diff --git [...]

I had try as above,but the result is the same
[image]

mantriyogesh, 2025-01-08:
@WindWard-atm Can you please state
1. did you do run first line or similar?
```sh
rm -rf sdkconfig build
idf.py set-target esp32c2
idf.py flash monitor
```

2. the git commit at esp and host (both sides)?
3. sdkconfig generated in esp32c2 side?

mantriyogesh, 2025-01-10:
@WindWard-atm ?

WindWard-atm, 2025-01-13:
> @WindWard-atm ?

I apologise for the delay, 
did you do run first line or similar?   yes
sdkconfig generated in esp32c2 side ?  yes
the git commit at esp and host (both sides)?  which commit ? 

after I re-remove and re-compile ,it run normal without crash, but the speed changes slowly, about 130kpbs

mantriyogesh, 2025-01-13:
C2 max speeds are limited because of small on chip memory. Do you need bluetooth? If not bluetooth could be disabled and put in sdkconfig.defaults.esp32c2.

This would release some memory for the help.

Then increase the config that you just reduced.

WindWard-atm, 2025-01-14:
it restart abnormaly again after 3 hours, before is 15 minutes
my diff:
[image]

spi speed is 30MHz