espressif/esp-hosted #559: ESP32S3 w/ iMX6ULL port host driver issue

jfelser, 2024-12-26:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Expected to interpret a boot up packet from the ESP32S3 properly.

### Actual behavior (suspected bug)

I am using esp-hosted-ng (commit 526ae2e2c6efaedad04e3a99f971c5e383eba957) and have followed the porting guide.  Both host and ESP are running at 1MHz SPI clock.  I can verify this on the oscilloscope and the clock trace looks clean.

I have added debug logs in the interrupt routines and verified that DR and HS are being appropriately triggered (have followed the steps in issue #200).  I have played with various SPI modes, and it does offset the bits by 1 when it would be appropriate to do so.  I have tried the slave 3, master 0 solution offered in issue #200 and it does not change the output.

However - the output in the RX buffer is either this, or this offset by 1:

`[  632.459338] esp32_spi: process_rx_buf: SPI RX buf: IF type = 0
[  632.465318] SPI RX buf: 9f07f3b4: 00 00 00 00 00 81 00 27 80 80 00 00 00 00 00 00  .......'........
[  632.474488] SPI RX buf: 6236b62f: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  632.483629] esp32_spi: process_rx_buf: offset_rcv[9984] != exp[12], drop`

And then it does not initialize properly as it hasn't received a boot up event.

Other things I have tried:
- Played with various pad config settings in the device tree (matching those in issue #200 as well as the defaults, and the suggested SPI mode configs from NXP).
- Played with the pull up configuration on the ESP.
- Ensured each pin is correct, and have verified that if I flip MISO and MOSI the data is all 0's (ie it is reading actual signal not just noise).
- Checked everything physical under the microscope, and with a DMM to verify no shorts etc.

Here is an example trace on MISO from the ESP side when I would expect the bootup event to occur.  I will grab the logic analyzer tomorrow if needed - however, this trace seems correct to me at first glance and it seems it does not match what is being shown in the RX buffer.  This makes me think that the issue is in the host driver (or host configuration) rather than the ESP.
[image]

Thank you for any suggestions/input!

### Steps to reproduce the behavior

Start the esp32_spi driver either with modprobe or lsmod (I adjusted the clock variables to be 1MHz with a patch - [  816.898715] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [3],chip select [0], SPI Clock [1]).

Running Linux on iMX6ULL.

### Project release version

526ae2e2c6efaedad04e3a99f971c5e383eba957

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

6.6.4

### Shell

Bash

jfelser, 2024-12-27:
Confirmed with the logic analyzer that the signals are correct (or, at least decode correctly).  Does this look reasonable to you - can we deduce that the issue must be on the host side?
[image]

mantriyogesh, 2024-12-27:
Can you please test the spi mode 1 both sides?

Also can you please share the camera photo of your setup?

How long the wires are?

mantriyogesh, 2024-12-27:
The firmware might be defaulted to spi mode 2. Have you changed it as well to use spi mode 1, similar to host?

jfelser, 2024-12-27:
Physical setup: integrated on PCB, I have an external 100k PU on CS.  30-50mm trace length, MISO is about 15mm shorter than MOSI/CLK.

SPI mode 1 (both sides) results:

Logic analyzer - looks right.
[image]

Host log.
`[   69.182879] esp32_spi: spi_dev_init: Using SPI MODE 1
[   69.189494] esp32_spi: spi_dev_init: ESP32 peripheral is registered to SPI bus [3],chip select [0], SPI Clock [1]
# 
# [   70.600765] esp32_spi: process_rx_buf: SPI RX buf: IF type = 0
[   70.606776] SPI RX buf: 7826c0c1: 00 00 00 00 00 02 01 4e 00 01 00 00 00 00 00 00  .......N........
[   70.615948] SPI RX buf: 0098e803: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[   70.625079] esp32_spi: process_rx_buf: offset_rcv[19969] != exp[12], drop`

ESP log.
`I (406) stats: *********************************************************************
I (414) stats:                 ESP-Hosted Firmware version :: NG-1.0.3.0.1                        
I (424) stats:                 Transport used :: SPI only                      
I (432) stats: *********************************************************************
I (441) FW_MAIN: Supported features are:
I (445) FW_MAIN: - WLAN over SPI
I (449) FW_MAIN: Capabilities: 0xa0
pp rom version: e7ae62f
net80211 rom version: e7ae62f
I (459) wifi:wifi driver task: 3fca4124, prio:23, stack:6656, core=0
I (465) wifi:wifi firmware version: 0abed4a
I (468) wifi:wifi certification version: v7.0
I (472) wifi:config NVS flash: disabled
I (476) wifi:config nano formating: disabled
I (480) wifi:Init data frame dynamic rx buffer num: 64
I (485) wifi:Init static rx mgmt buffer num: 5
I (489) wifi:Init management short buffer num: 32
I (493) wifi:Init dynamic tx buffer num: 64
I (497) wifi:Init static tx FG buffer num: 2
I (501) wifi:Init static rx buffer size: 1600
I (505) wifi:Init static rx buffer num: 16
I (509) wifi:Init dynamic rx buffer num: 64
I (513) wifi_init: rx ba win: 32
I (517) wifi_init: tcpip mbox: 32
I (521) wifi_init: udp mbox: 6
I (524) wifi_init: tcp mbox: 6
I (528) wifi_init: tcp tx win: 5760
I (532) wifi_init: tcp rx win: 5760
I (536) wifi_init: tcp mss: 1440
I (540) wifi_init: WiFi IRAM OP enabled
I (545) wifi_init: WiFi RX IRAM OP enabled
I (550) FW_SPI: Using SPI interface
I (554) FW_SPI: Using SPI MODE 1
I (558) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (567) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1578) FW_MAIN: Initial set up done
I (1578) main_task: Returned from app_main()
I (1630) FW_SPI: 0x3fcb14a8   03 00 00 00 1f 00 0c 00  b8 01 00 00 01 00 18 00  |................|
I (1630) FW_SPI: 0x3fcb14b8   17 00 00 00 03 01 09 02  01 01 00 01 a0 01 0c 4e  |...............N|`

jfelser, 2024-12-29:
This is resolved - it had do with issues in underlying DMA configuration.  Thank you!

closed 2024-12-29 by jfelser

mantriyogesh, 2024-12-30:
Can you please briefly explain, the changes you needed to be done, if comfortable for you?

It would be good knowledge gain for us for your platform..

jfelser, 2025-01-02:
Sure - sorry for the slow response.

I knew the SPI portion of the hardware was solid as it was passing the loopback test (ie wires routed properly, pads configured properly, etc).  I knew it was not likely a SPI mode issue, as the data was totally corrupted, not an off by 1 error.  It seemed unlikely to be a byte padding issue, etc, since the data did not match at any point.  In addition, the data was also very consistent, making it unlikely to be electrical noise or routing etc.

The logic analyzer supported this, showing that the signals were all fully correct.  I could reproduce the off by 1 type errors by changing the SPI mode, and it made no difference to what I was seeing in the driver logs.  I isolated the issue back to the spi_transfer call itself.  At that point, I was left with - the direct kernel spi_transfer call fails in a consistent way, while the spidev userspace version works.  My loopback test was using smaller chunks of data, which don't use DMA in some cases.  So I disabled DMA (direct memory access) for that node in the device tree, and it now works.

That's not a perfect solve - we want DMA to be working - but at least it has nothing to do with your driver.  Hopefully this explanation helps somebody in the future!