espressif/esp-hosted #560: channel range

obito1997-wky, 2024-12-27:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

How can the channel range supported by the ESP32 be modified?It seems that only 11 channels are supported at the moment.

### Actual behavior (suspected bug)

none

### Error logs or terminal output

```shell
none
```

### Steps to reproduce the behavior

always

### Project release version

none

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

none

### Shell

ZSH

### Additional context

none

mantriyogesh, 2024-12-27:
@kapilkedawat ++

obito1997-wky, 2025-01-06:
hello，can anyone answer my question

kapilkedawat, 2025-01-06:
@obito1997-wky what's the country set in the host system?

obito1997-wky, 2025-01-06:
I use “esp_hosted_wifi_get_country”，the country code read out is “01”

> @obito1997-wky what's the country set in the host system?

kapilkedawat, 2025-01-06:
@obito1997-wky are you using esp_hosted_fg?

obito1997-wky, 2025-01-06:
> @obito1997-wky what's the country set in the host system?

> @obito1997-wky what's the country set in the host system?

Can I set this country code in the ESP firmware somewhere?

obito1997-wky, 2025-01-06:
> @obito1997-wky are you using esp_hosted_fg?

yep

mantriyogesh, 2025-01-06:
do you mean the 'channel number' of 'softAP' ?
if so, please check API: [wifi_start_softap()](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#115-ctrl_cmd_t--wifi_start_softapctrl_cmd_t-req)

mantriyogesh, 2025-01-06:
yes, for country, there is yet no api exposed for Linux.

some reference how channel and country is related: [esp_wifi_set_country](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_wifi.html#_CPPv420esp_wifi_set_countryPK14wifi_country_t).

Need to add FG API for country, which would allow additional channels depending upon country

mantriyogesh, 2025-01-06:
@SohKamYung-Espressif , Can we please add this?

obito1997-wky, 2025-01-06:
> do you mean the 'channel number' of 'softAP' ? if so, please check API: [wifi_start_softap()](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#115-ctrl_cmd_t--wifi_start_softapctrl_cmd_t-req)

this only supports the setting of 11 channels.

mantriyogesh, 2025-01-06:
@obito1997-wky , yes. we need to add either new `API` or `argument` in the existing API for supporting country.

esp_wifi_**X**et_country() or
esp_wifi_**X**et_country_code()

would be then taken care in firmware accordingly.

If the channel requested allowed for that country, should be allowed.

For now to unblock you, you can manually set your country in the slave firmware code. else, please wait from us for the fix.

obito1997-wky, 2025-01-07:
> @obito1997-wky , yes. we need to add either new `API` or `argument` in the existing API for supporting country.
>
> esp_wifi_**X**et_country() or esp_wifi_**X**et_country_code()
>
> would be then taken [...]

ok，thank you

obito1997-wky, 2025-01-10:
> ok，thank you
where can I set  country in the slave firmware code

obito1997-wky, 2025-01-10:
Another question，call esp_hosted_wifi_set_mac returns 
error 7688-1 16:33:54.899 -- [/I] [esp wifi - rpc_rsp]Failure resp/event: possibly precondition not met   
Does esp firmware not support mac address change?

mantriyogesh, 2025-01-10:
> @obito1997-wky , yes. we need to add either new `API` or `argument` in the existing API for supporting country.
> esp_wifi_**X**et_country() or esp_wifi_**X**et_country_code()
> would be then taken care [...]

Fixed in latest master:

https://github.com/espressif/esp-hosted/commit/0a6eca02a9ddc1cee2592bbab66d17411ad2a965

mantriyogesh, 2025-01-10:
> error 7688-1 16:33:54.899 -- [/I] [esp wifi - rpc_rsp]Failure resp/event: possibly precondition not met
> Does esp firmware not support mac address change?

Please let us know:
1. Git commit both sides
2. Host and slave logs - full logs from startup - textual host name
dmesg for host, commands and response logs in terminal
4. kernel version?

mantriyogesh, 2025-01-10:
Please also have a look at [wifi_set_mac()](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#16-ctrl_cmd_t--wifi_set_macctrl_cmd_t-req).

[image]