espressif/esp-hosted #562: esp32_sdio: tx_process: not ready

Craft4Cube, 2025-01-02:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

ESP32 Hosted NG works as expected

### Actual behavior (suspected bug)

Driver does not finish init. Outputs repated:

esp32_sdio: tx_process: not ready

### Error logs or terminal output

```shell
U-Boot SPL 2024.10 (Jan 01 2025 - 22:50:37 +0100)
DRAM: 64 MiB
Trying to boot from sunxi SPI

U-Boot 2024.10 (Jan 01 2025 - 22:50:37 +0100) Allwinner Technology

CPU:   Allwinner V3s (SUN8I 1681)
Model: NetCube Systems Kumquat
DRAM:  64 MiB
Core:  52 devices, 23 uclasses, devicetree: separate
WDT:   Not starting watchdog@1c20ca0
MMC:   mmc@1c0f000: 0, mmc@1c10000: 1
Loading Environment from SPIFlash... SF: Detected w25q64cv with page size 256 Bytes, erase size 4 KiB, total 8 MiB
OK
In:    serial@1c28000
Out:   serial@1c28000
Err:   serial@1c28000
ethaddr set to: 44:B7:D0:D7:62:C5
Net:   eth0: ethernet@1c30000
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
1:	default
Retrieving file: /boot/zImage
append: root=/dev/mmcblk0p1 rootwait console=ttyS0,115200 rootfstype=ext4 panic=3 mtdparts=firmware:1M(u-boot)ro,512k(u-boot-env1),512k(u-boot-env2),-(user)
Retrieving file: /boot/allwinner/sun8i-v3s-netcube-kumquat.dtb
Kernel image @ 0x41000000 [ 0x000000 - 0x5b5408 ]
## Flattened Device Tree blob at 41800000
   Booting using the fdt blob at 0x41800000
Working FDT set to 41800000
   Loading Device Tree to 42d3d000, end 42d445d2 ... OK
Working FDT set to 42d3d000
ethaddr set to: 44:B7:D0:D7:62:C5

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 6.12.7 (lukas@lukas-hpz440workstation) (arm-buildroot-linux-gnueabihf-gcc.br_real (Buildroot 2024.11-368-g25c000b5e9-dirty) 13.3.0, GNU ld (GNU Binutils) 2.42) #1 SMP Thu Jan  2 21:02:38 CET 2025
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: NetCube Systems Kumquat
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 16 MiB at 0x41c00000 on node -1
[... 284 lines cut ...]
```

### Steps to reproduce the behavior

Build Buildroot Linux with esp_hosted, and Linux 6.12.7 on an Allwinner V3s

### Project release version

927a34f99724bb744efaf865a7c75cc1615f59a2

### System architecture

ARM 32-bit (Raspberry Pi 32-bit)

### Operating system

Linux

### Operating system version

Buildroot

### Shell

Bash

XDjackieXD, 2025-01-14:
Hi! I have the same issue running latest master (https://github.com/espressif/esp-hosted/commit/b83e0d5dfdd145e0448e3d7c3ac7e7d0b1e953c8). Sometimes it works for a minute or two and then starts spamming this, sometimes I can get it to work for longer.
Without having debugged this further yet, this "works sometimes" could point to an SDIO signal integrity issue? (while I do have it as rather short traces on a PCB it passes two connectors to the SoM I'm using).

I'm using an ESP32-C6-Mini-1 in SDIO only mode (set to "high speed") connected to a STM32MP1 SoC, SDIO Bus terminated directly at the ESP32 with 51kOhm resistors.

mantriyogesh, 2025-01-14:
@XDjackieXD what is the frequency of sdio that is in use?

mantriyogesh, 2025-01-14:
@Craft4Cube ,

I understand the print 
```
tx_process: not ready
```

Is too frequent and too irritating.
@Shreyas0-7 , can we lower this or print only when it is necessary?

```
Length not matching to firmware data size
```
Is a bigger problem.

1. Can you please share a photo clicked of your setup up?
2. Also, have you connected expected ext pull up?
3. What's the sdio clock? ESP32 or C6 can only take till 50MHz.
4. For `ESP32` SDIO, Have you applied the dat2 voltage fix from [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md)? It's related to eFuse flashing to avoid conflict in between bootstrap pin.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html

XDjackieXD, 2025-01-14:
> @XDjackieXD what is the frequency of sdio that is in use?

According to `/sys/kernel/debug/mmc2/clock` it's clocked at 50MHz.
Error counters stay at 0 on the host side:
```
#  cat /sys/kernel/debug/mmc2/err_stats
# Command Timeout Occurred:	 0
# Command CRC Errors Occurred:	 0
# Data Timeout Occurred:	 0
# Data CRC Errors Occurred:	 0
# Auto-Cmd Error Occurred:	 0
# ADMA Error Occurred:	 0
# Tuning Error Occurred:	 0
# CMDQ RED Errors:	 0
# CMDQ GCE Errors:	 0
# CMDQ ICCE Errors:	 0
# Request Timedout:	 0
# CMDQ Request Timedout:	 0
# ICE Config Errors:	 0
# Controller Timedout errors:	 0
# Unexpected IRQ errors:	 0
```

XDjackieXD, 2025-01-14:
@mantriyogesh what might be relevant: when entering this error mode, I can not transfer any data through wifi. Unloading and reloading the kernel module makes it work again - at least temporarily until it starts spamming this log line again

mantriyogesh, 2025-01-14:
FG we use-- https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L475-L491

Just that the sdio bus should not be idle or low voltage in between

mantriyogesh, 2025-01-14:
-84 is classical problem with the voltage and low pull up values, as per my experience in the past. But you have confirmed that you using  51K Ohm (>10KOhm).

are the traces with same length?
-84 is induced from detection perspective. If in between happening, worth to cross check what is happening on lines.

mantriyogesh, 2025-01-14:
Although this looks lower problem, imho.

XDjackieXD, 2025-01-14:
Thank you for the help!

> Any input voltage rating correct , any kind of 'performance' mode you can try?

The STM32MP1 is running in 3.3V HS Mode which should be correct as far as I can tell.
I'll check with a scope in the coming days.

> -84 is classical problem with the voltage and low pull up values, as per my experience in the past. But you have confirmed that you using 51K Ohm (>10KOhm).

Yes I just checked.

> are the traces with same length? -84 is induced from detection perspective. If in between happening, worth to cross check what is happening on lines.

The worst case skew between bus members is 4.6mm. That is well below the specs given by ST (10mm maximum at 50MHz).

> Although this looks lower problem, imho.

What do you mean by lower problem?

mantriyogesh, 2025-01-14:
I mean Linux sdio driver is identifying the sdio is not connected and loosing up.

Can you please check if continuous data transfer also ends in this issue or only intermittent data. I don't have inputs as such for that, just trying to understand or isolate the issue through some means.

mantriyogesh, 2025-01-14:
As an alternative, can you please reduce the sdio clock to 40MHz? Again, same, just localising when issue most likely occur.

XDjackieXD, 2025-01-14:
I started a longrunning, bidirectional iperf now. I'll report back if the issue also happens during the iperf

XDjackieXD, 2025-01-14:
Also happens while running an iperf test

mantriyogesh, 2025-01-14:
Can you please share your sdio connections as schematic as well? Just wanted to rule out any obvious issues.

Also, it is worth to add this in your sdio config:
```
broken-cd;
```

To auto detect again if in case the driver was unprobed by kernel.

mantriyogesh, 2025-01-14:
https://github.com/espressif/esp-hosted/issues/562#issuecomment-2590469977

Can you please share both sides textual logs from system bootup?
Also sdio config in place for your bus?

Craft4Cube, 2025-01-14:
> @Craft4Cube ,
>
> I understand the print
>
> ```
> tx_process: not ready
> ```
>
> Is too frequent and too irritating. @Shreyas0-7 , can we lower this or print only when it is necessary?
>
> ```
> Length not matching [...]
I will try to take one soon, however the Hardware has worked with Linux 6.8.7 already
> 2. Also, have you connected expected ext pull up?
Yes
> 3. What's the sdio clock? ESP32 or C6 can only take till 50MHz.
/sys/kernel/debug/mmc0/ios 
clock:		25000000 Hz
actual clock:	25000000 Hz
vdd:		21 (3.3 ~ 3.4 V)
bus mode:	2 (push-pull)
chip select:	0 (don't care)
power mode:	2 (on)
bus width:	2 (4 bits)
timing spec:	0 (legacy)
signal voltage:	0 (3.30 V)
driver type:	0 (driver type B)
> 4. For `ESP32` SDIO, Have you applied the dat2 voltage fix from [porting guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md)? It's related to eFuse flashing to avoid conflict in between bootstrap pin.
ESP32 is fused. ESP32 works on older kernel
> 
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html

XDjackieXD, 2025-01-14:
It seems to be stable when running at 25MHz (40MHz is not an option with my current clock configuration). So having some form of sdio signaling issue seems to be likely.

```
# cat /sys/kernel/debug/mmc2/ios 
clock:		40000000 Hz
actual clock:	25000000 Hz
vdd:		21 (3.3 ~ 3.4 V)
bus mode:	2 (push-pull)
chip select:	0 (don't care)
power mode:	2 (on)
bus width:	2 (4 bits)
timing spec:	2 (sd high-speed)
signal voltage:	0 (3.30 V)
driver type:	0 (driver type B)
```

> Can you please share your sdio connections as schematic as well? Just wanted to rule out any obvious issues.

It is just D0-D3, CMD and CLK connected from the ESP32-C6-Mini-1 to my SoM (dhcom-stm32mp15) and five 51k pull-up resistors to  3.3V for D0-D3 and CMD (no pullup on CLK).

[image]
[image]

I'm not using card detect - it is used as a "normal" GPIO to switch the enable pin of the ESP32.

> Also, it is worth to add this in your sdio config:
> 
> ```
> broken-cd;
> ```
> 
> To auto detect again if in case the driver was unprobed by kernel.

I already did that (as said above - I don't use CD for it's intended purpose).

> Can you please share both sides textual logs from system bootup? Also sdio config in place for your bus?

I'll have to "un-automate" my setup so I can capture the UART bootlog of the ESP while enabling it. Will do that tomorrow.

XDjackieXD, 2025-01-14:
> It seems to be stable when running at 25MHz (40MHz is not an option with my current clock configuration). So having some form of sdio signaling issue seems to be likely.

So a 1h long Iperf ran through without any issues with 25MHz.
I highly suspect some signal integrity issue.
(reachable speeds in my specific test setup are ~1.5Mbit/s downstream and ~4.5Mbit/s upstream when not testing bidirectionally btw. Not great but good enough for my usecase. Also don't seem to be limited by SDIO but something else)

Craft4Cube, 2025-01-14:
@mantriyogesh 

I just tried it again while capturing the output of both the ESP32 and the SoC

ESP32:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6680
load:0x40078000,len:14848
load:0x40080400,len:3792
entry 0x40080694
I (28) boot: ESP-IDF v4.4.1-584-g91dc99a3ce-dirty 2nd stage bootloader
I (28) boot: compile time 12:09:04
I (29) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 factory          factory app      00 00 00010000 00100000
I (100) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (107) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (115) boot: End of partition table
I (119) boot: Defaulting to factory image
I (124) boot_comm: chip revision: 3, min. application chip revision: 0
I (131) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=17950h ( 96592) map
I (174) esp_image: segment 1: paddr=00027978 vaddr=3ffbdb60 size=05574h ( 21876) load
I (183) esp_image: segment 2: paddr=0002cef4 vaddr=40080000 size=03124h ( 12580) load
I (189) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=88d8ch (560524) map
I (394) esp_image: segment 4: paddr=000b8db4 vaddr=40083124 size=1a43ch (107580) load
I (438) esp_image: segment 5: paddr=000d31f8 vaddr=50000000 size=00010h (    16) load
I (453) boot: Loaded app from partition at offset 0x10000
I (453) boot: Disabling RNG early entropy source...
I (465) cpu_start: Pro cpu up.
I (465) cpu_start: Starting app cpu, entry point is 0x40081288
I (0) cpu_start: App cpu up.
I (481) cpu_start: Pro cpu start user code
I (481) cpu_start: cpu freq: 240000000
I (481) cpu_start: Application information:
I (486) cpu_start: Project name:     network_adapter
I (491) cpu_start: App version:      2027593
I (496) cpu_start: Compile time:     Dec 19 2022 12:08:55
I (502) cpu_start: ELF file SHA256:  76d958924cb99ea3...
I (508) cpu_start: ESP-IDF:          v4.4.1-584-g91dc99a3ce-dirty
I (515) heap_init: Initializing. RAM available for dynamic allocation:
I (522) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (528) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (534) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (541) heap_init: At 3FFD1A40 len 0000E5C0 (57 KiB): DRAM
I (547) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (553) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (559) heap_init: At 4009D560 len 00002AA0 (10 KiB): IRAM
I (567) spi_flash: detected chip: generic
I (570) spi_flash: flash io: dio
W (574) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (589) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (602) stats: *********************************************************************
I (608) stats:                 ESP-Hosted Firmware version :: 1.0.2                        
[... 55 lines cut ...]
[506135.163146] esp32_sdio mmc0:0001:2: probe with driver esp32_sdio failed with error -22
[506135.180738] esp32_sdio: tx_process: not ready
[506135.200624] esp32_sdio: tx_process: not ready
[506135.240639] esp32_sdio: tx_process: not ready
...
[506135.780630] esp32_sdio: tx_process: not ready
[506135.800655] esp32_sdio: tx_process: not ready
[506135.820646] esp32_sdio: tx_process: not ready
[506135.936732] esp32_sdio: process_esp_bootup_event: Received ESP bootup event
[506135.948442] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 3
[506135.955029] esp32_sdio: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SDIO
[506135.962877] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 0
[506135.969405] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 4
[506135.975946] esp32_sdio: process_event_esp_bootup: Unsupported tag=4 in bootup event
[506135.983698] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 1
[506135.990224] esp32_sdio: process_fw_data: Length not matching to firmware data size
[506135.997898] esp32_sdio: process_event_esp_bootup: failed to process tag=1 in bootup event
[506136.006371] esp32_sdio: tx_process: not ready
[506136.020681] esp32_sdio: tx_process: not ready
[506136.040648] esp32_sdio: tx_process: not ready
...
[506140.240631] esp32_sdio: tx_process: not ready
[506140.260627] esp32_sdio: tx_process: not ready
[506140.281024] esp32_sdio: tx_process: not ready
```

mantriyogesh, 2025-01-15:
@XDjackieXD ,

Thank you for testing and letting us know.

I am still not sure where exact the limitation be, due to which the lower sdio clock is reliable, while higher is not.

Although, it was great help and efforts from your side to localise this issue. Good to know the current throughout are satisfying your needs.

The throughout you get, is very low. Is it in Bytes or bits per second? If it bytes, it should translate 12mbps and 36mbps.

Although our throughput is higher than this in over the air, it is dependent on many factors like interference. We are also working to add some throughput optimizations, so please stay tuned to master for updates.

Can you please share your part of schematic, related to hosted, to see the any problems? If you are not allowed to, that's okay. If allowed to, you can also mail me. I can compare with our own and let you know differences.
Meanwhile I am also checking if I am allowed to share mine, which would be easier for you, but due to chinese new year, I am expecting delays to confirm on this.

Nevertheless, appreciate your debug efforts.

Craft4Cube, 2025-01-15:
@mantriyogesh do you have any updates on my issue?

mantriyogesh, 2025-01-16:
@Craft4Cube are the ESP and host using same code commit?

Can you try this on latest master?

You would need to manually flash the ESP using IDF.y. steps are there in the documentation.
As part of flashing steps, Ensure that you do
```
rm -rf sdkconfig build/
```
to remove earlier build and config.

Craft4Cube, 2025-01-16:
@mantriyogesh I just tried it on a fresh build of master and esp same version, it seems to be working there. I guess this fixes the issue for me then. Thanks!

mantriyogesh, 2025-01-17:
Great. Thanks. Please feel free to close the issue..

closed 2025-01-17 by Craft4Cube