espressif/esp-hosted #564: wifi connect failed

helpfile, 2025-01-10:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

wifi connect ok

### Actual behavior (suspected bug)

The cpu used is arm 64-bit, the kernel version is linux-5.4, esp32 platform, sdio interface, using esp_hosted_ng, esp32 firmware (ESP-Hosted-NG_release_v1.0.2.tgz --> esp32 --> sdio_only). Currently, arm can scan and obtain the  AP information, but cannot connect to the specified AP.

### Error logs or terminal output

```shell
<6>[    9.479304] esp32_sdio: esp_init: esp_init start...
<6>[    9.680460] esp32_sdio: esp_init_interface_layer: sdio_register_driver 
<6>[    9.680625] esp32_sdio: esp_probe: esp_probe
<6>[    9.680642] esp32_sdio: esp_probe: ESP network device detected
<6>[   11.043966] esp32_sdio: init_sdio_func: Register IRQ: esp_handle_isr
<6>[   11.064235] esp32_sdio: process_esp_bootup_event: Received ESP bootup event
<6>[   11.064274] esp32_sdio: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SDIO
<7>[   11.064278] esp32_sdio: esp_probe: ESP SDIO probe completed
<4>[   11.064294] esp32_sdio: process_event_esp_bootup: Unsupported tag=4 in bootup event
<3>[   11.064311] esp32_sdio: process_fw_data: Length(7) not matching to firmware data size(12)
<6>[   11.064513] esp32_sdio: esp_init: esp_init end!
<6>[   11.066402] esp32_sdio: esp_reg_notifier: Driver init is ongoing
<7>[   11.066641] esp32_sdio: esp_cfg80211_add_iface: Updated priv[0] to ffffff80198c07c0
<7>[   11.072494] esp32_sdio: esp_cmd_work: Processing Command [0x1]
<3>[   11.075477] esp32_sdio: tx_process: state[1] not ready
<7>[   11.082748] esp32_sdio: wait_and_decode_cmd_resp: Resp for command [0x1]
<7>[   11.088427] esp32_sdio: esp_cmd_work: Processing Command [0x3]
<7>[   11.092614] esp32_sdio: wait_and_decode_cmd_resp: Resp for command [0x3]
<7>[   11.093034] esp32_sdio: esp_cfg80211_get_tx_power: 
<7>[   11.095482] esp32_sdio: esp_cmd_work: Processing Command [0xF]
<7>[   11.101437] esp32_sdio: esp_cmd_work: Busy in another cmd execution
<3>[   16.352439] esp32_sdio: wait_and_decode_cmd_resp: Command[0xF] timed out
<3>[   16.352471] esp32_sdio: cmd_get_tx_power: wait_and_decode_cmd_resp(priv, cmd_node) failure, ret: -22
<6>[   16.560439] esp32_sdio: init_bt: ESP Bluetooth init
<6>[   16.560734] esp32_sdio: print_capabilities: Capabilities: 0x1d. Features supported are:
<6>[   16.560769] esp32_sdio: print_capabilities:        * WLAN on SDIO
<6>[   16.560782] esp32_sdio: print_capabilities:        * BT/BLE
<6>[   16.560793] esp32_sdio: print_capabilities:          - HCI over SDIO
<6>[   16.560805] esp32_sdio: print_capabilities:          - BT/BLE dual mode
<6>[   16.561000] bt_tx: (____ptrval____): 03 0c 00                                         ...
<6>[   16.567696] bt_tx: (____ptrval____): 03 10 00                                         ...
<6>[   16.572605] bt_tx: (____ptrval____): 01 10 00                                         ...
<6>[   16.577664] bt_tx: (____ptrval____): 09 10 00                                         ...
<6>[   16.582661] bt_tx: (____ptrval____): 05 10 00                                         ...
<6>[   16.587572] bt_tx: (____ptrval____): 23 0c 00                                         #..
<6>[   16.592747] bt_tx: (____ptrval____): 14 0c 00                                         ...
<6>[   16.597670] bt_tx: (____ptrval____): 25 0c 00                                         %..
<6>[   16.602566] bt_tx: (____ptrval____): 38 0c 00                                         8..
<6>[   16.607578] bt_tx: (____ptrval____): 39 0c 00                                         9..
<6>[   16.612563] bt_tx: (____ptrval____): 05 0c 01 00                                      ....
<6>[   16.617559] bt_tx: (____ptrval____): 16 0c 02 00 7d                                   ....}
<6>[   16.622591] bt_tx: (____ptrval____): 02 20 00                                         . .
<6>[   16.627550] bt_tx: (____ptrval____): 03 20 00                                         . .
<6>[   16.632568] bt_tx: (____ptrval____): 1c 20 00                                         . .
<6>[   16.637586] bt_tx: (____ptrval____): 02 10 00                                         ...
<6>[   16.642590] bt_tx: (____ptrval____): 52 0c f1 00 00 00 00 00 00 00 00 00 00 00 00 00  R...............
<6>[   16.642620] bt_tx: (____ptrval____): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
<6>[   16.642637] bt_tx: (____ptrval____): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
<6>[   16.642654] bt_tx: (____ptrval____): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[... 204 lines cut ...]
```

### Steps to reproduce the behavior

wpa_supplicant -B -iwlan0 -Dnl80211 -c /tmp/wpa_supplicant.conf

### Project release version

ESP-Hosted-NG Release 1.0.2

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

Linux

### Operating system version

linux 5.4

### Shell

sh

mantriyogesh, 2025-01-10:
Hello @helpfile ,

Have you used exact same commit at slave and host?

Also can you get us camera photo of your sdio setup?

helpfile, 2025-01-10:
hi mantriyogesh,
    esp32 fw download path: https://github.com/espressif/esp-hosted/releases/tag/release/ng-v1.0.2

helpfile, 2025-01-10:
hi mantriyogesh，
esp32 fw download：https://github.com/espressif/esp-hosted/releases/tag/release/ng-v1.0.2

helpfile, 2025-01-10:
hi mantriyogesh，
   esp32 fw download ptah：   https://github.com/espressif/esp-hosted/releases/tag/release/ng-v1.0.2
   host driver download path:   https://github.com/espressif/esp-hosted.git
   in host driver file(src/include/esp_fw_version.h):
    #define PROJECT_NAME                "NG"
    #define PROJECT_VERSION_MAJOR_1     1
   #define PROJECT_VERSION_MAJOR_2     0
   #define PROJECT_VERSION_MINOR       3
   #define PROJECT_REVISION_PATCH_1    0
   #define PROJECT_REVISION_PATCH_2    3

  error info:  <3>[   11.064311] esp32_sdio: process_fw_data: Length(7) not matching to firmware data size(12)

helpfile, 2025-01-10:
hi mantriyogesh，
esp32 fw download ptah： https://github.com/espressif/esp-hosted/releases/tag/release/ng-v1.0.2
host driver download path: https://github.com/espressif/esp-hosted.git
in host driver file(src/include/esp_fw_version.h):
#define PROJECT_NAME "NG"
#define PROJECT_VERSION_MAJOR_1 1
#define PROJECT_VERSION_MAJOR_2 0
#define PROJECT_VERSION_MINOR 3
#define PROJECT_REVISION_PATCH_1 0
#define PROJECT_REVISION_PATCH_2 3

error log:   <3>[ 11.064311] esp32_sdio: process_fw_data: Length(7) not matching to firmware data size(12)
static int process_fw_data(struct fw_data *fw_p, int tag_len)
{
	if (tag_len != sizeof(struct fw_data)) {
		esp_err("Length(%d) not matching to firmware data size(%d)\n",tag_len,sizeof(struct fw_data)); // changed by helpfile
		//return -1; // changed by helpfile
		return 0;  // changed by helpfile
	}

	esp_info("ESP chipset's last reset cause:\n");
	print_reset_reason(le32_to_cpu(fw_p->last_reset_reason));

	return check_esp_version(&fw_p->version);
}
From the log information above, it can be seen that the esp32 firmware and my driver do not seem to match, because the data replied by esp32 and the data structure size on the host are different.

mantriyogesh, 2025-01-10:
Can you please try to flash the latest master on ESP and use latest master host driver and try again?

mantriyogesh, 2025-01-10:
The above way is **much recommended**. If you just wish to evaluate first, please try matching code from 1.0.2 commit at https://github.com/espressif/esp-hosted/commit/29208b6c156be06dfd90dadc6dd43acd4d98d748

mantriyogesh, 2025-02-06:
@helpfile , any update?