espressif/esp-hosted #565: esp32p4  spi 4 line mode  host function too slowly

skylin008, 2025-01-10:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Described the feature in detail and justified the reason for the request.
- [X] Provided specific use cases and examples.

### Feature description

esp32p4 spi 4 line function, used 40M clock, max speed only 1 MiB, does it can be 5 MiB reachable.

### Use cases

esp32p4 connect esp32c5 wifi module via  SPI 4 Line mode, used FTP protocol download file from FTP server via esp32c5 wifi，it only 1 MiB download speed , how to solve this issue. Thanks!  

### Alternatives

None

### Additional context

None

mantriyogesh, 2025-01-11:
@skylin008 let us know some details to understand the issue better..

skylin008, 2025-01-15:
@mantriyogesh  Thanks for you support. Yes, I used the [https](https://github.com/espressif/esp-hosted), esp32c5 for spi slave , and esp32p4 for spi master. I used the FTP protocol transfer file. Attach esp32c5 and esp32p4 sdkconfig
[Uploading sdkconfig.zip…]()

mantriyogesh, 2025-01-15:
The file you uploaded , could not complete. Can you please state if your normal functionality is working. The issue is only in download speed.

1MiB, meaning around 8-8.5mbps speed. This is limited by the clock. What is the SPI actual clock on the clock line?

Also, can you please click a photo how you connected the p4 and c5?
As you are using p4-c5, you can also use QSPI, which with almost similar number of wires (1 extra) can bump the speed a lot.