espressif/esp-hosted #566: esp32 sdio host drvier for linux 3.10.14

helpfile, 2025-01-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

I hope to compile successfully on linux-3.10.14

### Actual behavior (suspected bug)

The compilation failed. I checked the code and found that some structures and functions were not the same or did not exist in the cfg80211.h file of linux-3.10.14.

### Error logs or terminal output

```shell
warning: (USB_DWC2_DUAL_ROLE && FSL_USB2_OTG && USB_MV_OTG) selects USB_OTG which has unmet direct dependencies (USB_SUPPORT && USB && PM_RUNTIME)
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CALL    scripts/checksyscalls.sh
  CC      scripts/mod/devicetable-offsets.s
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTLD  scripts/mod/modpost
  CC [M]  drivers/net/wireless/esp32c6/esp_bt.o
In file included from drivers/net/wireless/esp32c6/esp_bt.c:10:0:
drivers/net/wireless/esp32c6/include/esp_kernel_port.h: In function 'CFG80211_RX_ASSOC_RESP':
drivers/net/wireless/esp32c6/include/esp_kernel_port.h:180:2: error: implicit declaration of function 'cfg80211_rx_assoc_resp' [-Werror=implicit-function-declaration]
  cfg80211_rx_assoc_resp(dev, bss, buf, len, uapsd_queues);
  ^
cc1: some warnings being treated as errors
make[4]: *** [scripts/Makefile.build:309: drivers/net/wireless/esp32c6/esp_bt.o] Error 1
make[3]: *** [scripts/Makefile.build:455: drivers/net/wireless/esp32c6] Error 2
make[2]: *** [scripts/Makefile.build:455: drivers/net/wireless] Error 2
make[1]: *** [scripts/Makefile.build:455: drivers/net] Error 2
make: *** [Makefile:795: drivers] Error 2
[ERROR]Camera build in kernel source drivers error
[INFO]Camera build drivers error
[ERROR]Camera build rootfs error
```

### Steps to reproduce the behavior

none

### Project release version

V1.0.3

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

linux-3.10.14

### Shell

Bash

### Additional context

We have multiple cpu platforms, and the linux versions are all below 4.19, including 3.10.14 (CPU: Ingenic T31), 4.4.94 (CPU: Ingenic T41), 4.9 (CPU: Ambarella S5L, CPU: HiSilicon HI3556), and 5.4 (Ambarella CV28). Currently, only Ambarella CV28 can be compiled and run successfully, and other CPUs have compilation errors.

mantriyogesh, 2025-01-13:
@kapilkedawat ,

Can we state exact minimal kernel version that we can support?

mantriyogesh, 2025-01-13:
@helpfile , there is also ESP-Hosted-FG, which has exact similar data path. But control path  (wifi management) is controlled using simple api calls.

because FG has lessor dependency over cfg802.11, it can also be built easily on older kernel versions.

mantriyogesh, 2025-01-14:
Related #564

helpfile, 2025-01-14:
> @kapilkedawat ,
> 
> Can we state exact minimal kernel version that we can support?
The minimum Linux kernel version we can support is 3.10.14

helpfile, 2025-01-14:
> @helpfile , there is also ESP-Hosted-FG, which has exact similar data path. But control path (wifi management) is controlled using simple api calls.
> 
> because FG has lessor dependency over cfg802.11, it can also be built easily on older kernel versions.

Does ESP-Hosted-FG support esp32-c6?
Does ESP-Hosted-FG continue to be updated?

mantriyogesh, 2025-01-14:
Yes and yes. FG doesn't support wpa supplicant.
We can try building but cannot test very old kernel but it had worked in customer cases for old kernels.