espressif/esp-hosted #567: ESP NG- Network with passphrase fails on linux openwrt host

BrelJordan, 2025-01-17:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Create an access point with a Passphrase stations can connect to.

Hostapd config from esphosted's page:
```

interface=wlan0
driver=nl80211
ssid=MY_SSID
hw_mode=g
channel=6
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_key_mgmt=SAE
wpa_passphrase=MY_PASSPHRASE
rsn_pairwise=CCMP
ieee80211w=2
```

### Actual behavior (suspected bug)

The AP is created but without any password (authmode=0 ) and devices fail to connect because the host complains no key is given by the peer.

Terminal output:
```
root@MC-MNR:~# hostapd -ddd host.conf
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: authentication OK (open system)
wlan0: STA c2:f8:0b:3f:27:86 MLME: MLME-AUTHENTICATE.indication(c2:f8:0b:3f:27:86, OPEN_SYSTEM)
wlan0: STA c2:f8:0b:3f:27:86 MLME: MLME-DELETEKEYS.request(c2:f8:0b:3f:27:86)
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: No WPA/RSN IE in association request
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: authenticated
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: authentication OK (open system)
wlan0: STA c2:f8:0b:3f:27:86 MLME: MLME-AUTHENTICATE.indication(c2:f8:0b:3f:27:86, OPEN_SYSTEM)
wlan0: STA c2:f8:0b:3f:27:86 MLME: MLME-DELETEKEYS.request(c2:f8:0b:3f:27:86)
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: No WPA/RSN IE in association request
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: authenticated
```

esp32c3:

```
I (80665071) wifi:mode : softAP (f0:f5:bd:e9:c5:fc)
I (80665075) wifi:Total power save buffer number: 20
I (80665079) wifi:Init max length of beacon: 752/752
I (80665084) wifi:Init max length of beacon: 752/752
I (80665089) FW_CMD: Station stop
I (80665092) FW_CMD: softap started and disabled station mode
I (80665114) FW_MAIN: Get MAC command
I (80665154) FW_MAIN: Get Tx power command
I (80665194) FW_MAIN: Get Tx power command
I (80665254) FW_MAIN: Set multicast mac address list
I (80665304) FW_MAIN: disconnect request
I (80665304) FW_CMD: Disconnect request: reason [2], interface=1

I (80665354) FW_MAIN: Set multicast mac address list
I (80665424) FW_MAIN: Get Tx power command
I (80665474) FW_MAIN: Set IE command
I (80665474) FW_CMD: Setting IE type=4 len=24

I (80665474) BEACON_PROBE_HEAD: 0x3fcb9c2c   11 04 00 07 4d 59 5f 53  53 49 44 01 08 82 84 8b  |....MY_SSID.....|
I (80665483) BEACON_PROBE_HEAD: 0x3fcb9c3c   96 0c 12 18 24 03 01 06                           |....$...|
I (80665544) FW_MAIN: Set AP config command
I (80665544) FW_CMD: ap config ssid=MY_SSID ssid_len=7, pass= channel=6 authmode=0 hidden=0 bi=100 cipher=0

```

### Error logs or terminal output

```shell
Terminal output:
root@MC-MNR:~# hostapd -ddd host.conf
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: authentication OK (open system)
wlan0: STA c2:f8:0b:3f:27:86 MLME: MLME-AUTHENTICATE.indication(c2:f8:0b:3f:27:86, OPEN_SYSTEM)
wlan0: STA c2:f8:0b:3f:27:86 MLME: MLME-DELETEKEYS.request(c2:f8:0b:3f:27:86)
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: No WPA/RSN IE in association request
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: authenticated
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: authentication OK (open system)
wlan0: STA c2:f8:0b:3f:27:86 MLME: MLME-AUTHENTICATE.indication(c2:f8:0b:3f:27:86, OPEN_SYSTEM)
wlan0: STA c2:f8:0b:3f:27:86 MLME: MLME-DELETEKEYS.request(c2:f8:0b:3f:27:86)
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: No WPA/RSN IE in association request
wlan0: STA c2:f8:0b:3f:27:86 IEEE 802.11: authenticated

esp32c3:

I (80665071) wifi:mode : softAP (f0:f5:bd:e9:c5:fc)
I (80665075) wifi:Total power save buffer number: 20
I (80665079) wifi:Init max length of beacon: 752/752
I (80665084) wifi:Init max length of beacon: 752/752
I (80665089) FW_CMD: Station stop
I (80665092) FW_CMD: softap started and disabled station mode
I (80665114) FW_MAIN: Get MAC command
I (80665154) FW_MAIN: Get Tx power command
I (80665194) FW_MAIN: Get Tx power command
I (80665254) FW_MAIN: Set multicast mac address list
I (80665304) FW_MAIN: disconnect request
I (80665304) FW_CMD: Disconnect request: reason [2], interface=1

I (80665354) FW_MAIN: Set multicast mac address list
I (80665424) FW_MAIN: Get Tx power command
I (80665474) FW_MAIN: Set IE command
I (80665474) FW_CMD: Setting IE type=4 len=24

I (80665474) BEACON_PROBE_HEAD: 0x3fcb9c2c   11 04 00 07 4d 59 5f 53  53 49 44 01 08 82 84 8b  |....MY_SSID.....|
I (80665483) BEACON_PROBE_HEAD: 0x3fcb9c3c   96 0c 12 18 24 03 01 06                           |....$...|
I (80665544) FW_MAIN: Set AP config command
I (80665544) FW_CMD: ap config ssid=MY_SSID ssid_len=7, pass= channel=6 authmode=0 hidden=0 bi=100 cipher=0
```

### Steps to reproduce the behavior

1 On a linux host with the esphosted-ng kmod installed run:
`hostapd -ddd host.conf`

### Project release version

commit hash: c79b7e11d027b147ee5fb5121ff49cf88c9de9ac

### System architecture

ARM 64-bit (Apple M1/M2, Raspberry Pi 4/5)

### Operating system

Linux

### Operating system version

openwrt 24

### Shell

sh

mantriyogesh, 2025-01-17:
Please attach the `full` `textual` logs as `attachment`:
- esp:
  - idf.py flash monitor from startup
- host
  - dmesg from system boot up
  - wpa supplicant or hostapd logs in debug

Also ensure that you use latest master at both esp and host.

BrelJordan, 2025-01-20:
I could not successfully run the idf script on the device itself due to limited resources. However I have the complete dmesg and the hostapd log with the latest esphosted version.

[dmesg.txt](https://github.com/user-attachments/files/18474954/dmesg.txt)
[hostapd.txt](https://github.com/user-attachments/files/18474953/hostapd.txt)

mantriyogesh, 2025-01-20:
@kapilkedawat PTAL..

mantriyogesh, 2025-01-20:
Every other command getting timed out.

I also note
```
[    6.444912] kmodloader: done loading kernel modules from /etc/modules.d/*
[    8.879422] irq 76: nobody cared (try booting with the "irqpoll" option)
[    8.880040] CPU: 0 PID: 2276 Comm: hotplug-call Tainted: G           O       6.6.36 #0
[    8.880743] Hardware name: MC Technologies MNR (DT)
[    8.881176] Call trace:
[    8.881394]  dump_backtrace+0xa0/0xe0
[    8.881728]  show_stack+0x18/0x24
[    8.882025]  dump_stack_lvl+0x48/0x60
[    8.882356]  dump_stack+0x18/0x24
[    8.882651]  __report_bad_irq+0x38/0xe0
[    8.882994]  note_interrupt+0x2c0/0x310
[    8.883334]  handle_irq_event+0x9c/0xa8
[    8.883675]  handle_edge_irq+0xb0/0x220
[    8.884015]  generic_handle_domain_irq+0x2c/0x44
[    8.884421]  rockchip_irq_demux+0x84/0x1b8
[    8.884786]  generic_handle_domain_irq+0x2c/0x44
[    8.885193]  gic_handle_irq+0x50/0x12c
[    8.885524]  call_on_irq_stack+0x24/0x4c
[    8.885871]  do_interrupt_handler+0x80/0x8c
[    8.886244]  el0_interrupt+0x44/0xb0
[    8.886564]  __el0_irq_handler_common+0x18/0x24
[    8.886963]  el0t_64_irq_handler+0x10/0x1c
[    8.887326]  el0t_64_irq+0x178/0x17c
[    8.887644] handlers:
[    8.887846] [<00000000f35dbec5>] irq_default_primary_handler threaded [<000000002fa8f156>] pca953x_irq_handler
[    8.888736] Disabling IRQ #76 
```

but unsure if this is related to ESP-Hosted as such.

kapilkedawat, 2025-01-20:
@BrelJordan are you using correct firmware(corresponding to https://github.com/espressif/esp-hosted/commit/c79b7e11d027b147ee5fb5121ff49cf88c9de9ac)? You should be getting [this](https://github.com/espressif/esp-hosted/blob/b83e0d5dfdd145e0448e3d7c3ac7e7d0b1e953c8/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c#L2037) print when station is getting authorized in firmware.

BrelJordan, 2025-01-20:
I think the problem occurs earlier during the configuration of the chip. Look at what the esp firmware says when I start hostapd with a paraphrase:

```
I (565350) FW_MAIN: Set AP config command
I (565350) FW_CMD: ap config ssid=OpenWrt ssid_len=7, pass= channel=6 authmode=0 hidden=0 bi=100 cipher=0

I (565354) wifi:flush txq
I (565356) wifi:stop sw txq
I (565359) wifi:lmac stop hw txq
I (565362) FW_CMD: tx cb unregister ret=0

I (565366) FW_CMD: tx cb register ret=0

I (565372) wifi:mode : softAP (f0:f5:bd:e9:c5:fc)
I (565376) wifi:Total power save buffer number: 20
I (565380) wifi:Init max length of beacon: 752/752
I (565385) wifi:Init max length of beacon: 752/752
I (565389) FW_CMD: softap started and disabled station mode
I (565400) FW_CMD: process_mgmt_tx: broadcast address, sending response immediately

I (565460) FW_MAIN: Add key request
I (565460) FW_CMD: set_key_internal:1665

I (565460) FW_CMD: Setting GTK [1]

I (565520) FW_MAIN: Set default key request
I (569040) FW_MAIN: Get Tx power command
I (569100) FW_MAIN: Get Tx power command
I (569140) FW_MAIN: Get Tx power command
I (569200) FW_MAIN: Get Tx power command
I (569260) FW_MAIN: Get Tx power command
I (569300) FW_MAIN: Get Tx power command
I (569340) FW_MAIN: Get Tx power command
I (569380) FW_MAIN: Get Tx power command
```

It sets the parameter authmode to 0` I (565350) FW_CMD: ap config ssid=OpenWrt ssid_len=7, pass= channel=6 **authmode=0** hidden=0 bi=100 cipher=0`. This however corresponds to an open AP as per definition in 

[esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_wifi/include/esp_wifi_types.h
](https://github.com/espressif/esp-idf/blob/e7771c75bd1dbbfb7b3c5381be7e063b197c9734/components/esp_wifi/include/esp_wifi_types.h#L74)
```
/* Strength of authmodes */
/* OPEN < WEP < WPA_PSK < OWE < WPA2_PSK = WPA_WPA2_PSK < WAPI_PSK < WPA3_PSK = WPA2_WPA3_PSK */
typedef enum {
    WIFI_AUTH_OPEN = 0,         /**< authenticate mode : open */
    WIFI_AUTH_WEP,              /**< authenticate mode : WEP */
    WIFI_AUTH_WPA_PSK,          /**< authenticate mode : WPA_PSK */
    WIFI_AUTH_WPA2_PSK,         /**< authenticate mode : WPA2_PSK */
    WIFI_AUTH_WPA_WPA2_PSK,     /**< authenticate mode : WPA_WPA2_PSK */
    WIFI_AUTH_ENTERPRISE,       /**< authenticate mode : WiFi EAP security */
    WIFI_AUTH_WPA2_ENTERPRISE = WIFI_AUTH_ENTERPRISE,  /**< authenticate mode : WiFi EAP security */
    WIFI_AUTH_WPA3_PSK,         /**< authenticate mode : WPA3_PSK */
    WIFI_AUTH_WPA2_WPA3_PSK,    /**< authenticate mode : WPA2_WPA3_PSK */
    WIFI_AUTH_WAPI_PSK,         /**< authenticate mode : WAPI_PSK */
    WIFI_AUTH_OWE,              /**< authenticate mode : OWE */
    WIFI_AUTH_WPA3_ENT_192,     /**< authenticate mode : WPA3_ENT_SUITE_B_192_BIT */
    WIFI_AUTH_MAX
} wifi_auth_mode_t;
```

BrelJordan, 2025-01-20:
The host driver is responsible for transmitting the hostapd configuration to the esp MCU. When looking at the source, it seems the auth_mode is not configured and hence has a default of 0.  In the funtion [esp_cfg80211_start_ap](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/esp_cfg80211.c#L844) authmode is not populated. From what I see it should be in info->[nl80211_auth_type](https://elixir.bootlin.com/linux/v4.6/C/ident/nl80211_auth_type). However some other variables from [esp_ap_config](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/include/adapter.h#L209) :

```
struct esp_ap_config {
    uint8_t ssid[32];
    uint8_t ssid_len;
    uint8_t channel;
    uint8_t authmode;
    uint8_t ssid_hidden;
    uint8_t max_connection;
    uint8_t pairwise_cipher;
    uint8_t pmf_cfg;
    uint8_t sae_pwe_h2e;
    uint16_t beacon_interval;
    uint16_t inactivity_timeout;
} __packed;

```
are not configured.

kapilkedawat, 2025-01-20:
Hi @BrelJordan,
There is no need to send authmode to the firmware. The entire AP mode MLME is controlled in userspace (hostapd). The firmware only sends beacons/probe responses, and for those the IEs are set by the host.

Please run a raw throughput test once so that we can confirm the transport connections.

BrelJordan, 2025-01-20:
Okay this are the outputs:

[host.txt](https://github.com/user-attachments/files/18478426/host.txt)

[esp.txt](https://github.com/user-attachments/files/18478430/esp.txt)

mantriyogesh, 2025-01-27:
@BrelJordan , Thanks for the Raw throughut from esp to host.
Please run the other direction as well, to understand other route is also good.

BrelJordan, 2025-01-27:
Hi @mantriyogesh  here are the results for host -> esp.

[host1.txt](https://github.com/user-attachments/files/18555944/host1.txt)
[esp1.txt](https://github.com/user-attachments/files/18555945/esp1.txt)

mantriyogesh, 2025-01-27:
Can you stop your logs in host, to realise if the tx and rx are having symmetrical bandwidths?

There are a lot of debug logs in linux side. later point of time, you can enable the required ones?

BrelJordan, 2025-02-12:
I was able to resolve the issue by reducing the SPI frequency to 10 MHz, which allowed me to connect to the AP. However, I am still not receiving an IP address from my host.

Using Wireshark and tcpdump, I monitored both the client interface and the ESP-Hosted kernel module’s network interface. I observed that the client sends a DHCP request, but it never reaches the ESP host’s network interface.

Has anyone encountered a similar issue? I’d appreciate any insights or tips on how to debug this further. Thanks in advance!

[esphosted.txt](https://github.com/user-attachments/files/18764276/esphosted.txt)
[hostapd.txt](https://github.com/user-attachments/files/18764277/hostapd.txt)

[text-client-ws.txt](https://github.com/user-attachments/files/18764284/text-client-ws.txt)
[text-host-ws.txt](https://github.com/user-attachments/files/18764283/text-host-ws.txt)

kapilkedawat, 2025-02-12:
Hi @BrelJordan, could you please share wireless sniffer capture?

BrelJordan, 2025-02-12:
Do you mean I should export the capture in another Format and not just as clear text? I just did as psml, I added a .txt ending because github does not allow psml files.

[host-ws.psml.txt](https://github.com/user-attachments/files/18765149/host-ws.psml.txt)

[client-ws.psml.txt](https://github.com/user-attachments/files/18765134/client-ws.psml.txt)

Yjxslhz, 2025-02-17:
I also encountered the situation that DHCP is unreachable and the AP intranet is inaccessible after turning on CCMP encryption of WPA. Only if I set the AP mode to open mode, it will be completely normal. What should I do in this situation?

BrelJordan, 2025-02-17:
Do you receive a DHCP request on the host? I don't.

kapilkedawat, 2025-02-17:
> Do you mean I should export the capture in another Format and not just as clear text? I just did as psml, I added a .txt ending because github does not allow psml files. [...]

I meant wifi sniffer capture. 

> I also encountered the situation that DHCP is unreachable and the AP intranet is inaccessible after turning on CCMP encryption of WPA. Only if I set the AP mode to open mode, it will be completely normal. What should I do in this situation?

could you also share sniffer capture and esp and host logs?

BrelJordan, 2025-02-18:
Hello I did as you asked, the host is : f0:f5:bd:e9:ae:c8 and the client the McTechno device.

[wifi-monitoring.psml.txt](https://github.com/user-attachments/files/18841181/wifi-monitoring.psml.txt)

[wifi-monitoring.txt](https://github.com/user-attachments/files/18841186/wifi-monitoring.txt)

BrelJordan, 2025-02-25:
Hello @kapilkedawat  from the wifi sniffer I see the dhcp request from the client. I activated debug Messages in the idf firmware to see what is going on. Comparing a psk2 secure and open network. For the secure Network I don't get this debug message:

**D (45322552) wifi:eb is dhcp or dns sport = 68, dport = 67**

Both logs look very similar up to this point. Please help me and compare the logs you may see the issue I just can't find:

kapilkedawat, 2025-02-27:
Hi @BrelJordan, sorry for delayed response. are you using latest driver and firmware? I just tried to use open mode and I was able to connect.

Could you do a wireless sniffer capture and share the pcap file with us to analyze(please do not export to text). Please also share host details(rpi version and linux version used).

BrelJordan, 2025-02-27:
Hello @kapilkedawat I am using **OpenWrt** with the **Linux kernel version:6.6.37**. Yes am am using the actual esp firmware and Kernel module. I have already done a WiFi sniffer you can download it from my second to last post. I just added .txt extension for the upload.

kapilkedawat, 2025-03-05:
Hi @BrelJordan we are not able to open [wifi-monitoring.txt](https://github.com/user-attachments/files/18841186/wifi-monitoring.txt) using wireshark, how did you capture it?

BrelJordan, 2025-03-05:
Could you open the .pslm file?

kapilkedawat, 2025-03-05:
Yes, but that only contains headers in most cases, also need to analyze the packets in wireshark :) .. we will need your help since this is not reproducible for us.

Yjxslhz, 2025-03-09:
After I used fg, everything worked fine. Thanks everyone.

mantriyogesh, 2025-03-10:
Hello @Yjxslhz ,

Can you please let us know if you needed any kind of modifications to get FG working?

@kapilkedawat 
What my confusion is, is there something that we are missing from NG point of view in this specific case?

kapilkedawat, 2025-03-10:
@BrelJordan @Yjxslhz could you please pull latest code and retry(both firmware and host)?

BrelJordan, 2025-03-19:
I had the time to try with the new firmware/ kernel mod and it worked. The issues is resolved with version "1.0.3.0.10". It is still unstable but atleast it connects.

closed 2025-03-19 by BrelJordan