espressif/esp-hosted #568: [Question] Which repo/branch should I fork to use?

kaidegit, 2025-02-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi, I want to port esp-hosted(fg) to RT-Thread. 
However, I found there are some repo have similar code, this repo master, this repo main and esp-hosted-mcu. So, which repo/branch should I fork to work?

mantriyogesh, 2025-02-04:
Please use https://github.com/espressif/esp-hosted-mcu for MCU host.
Please use https://github.com/espressif/esp-hosted only for Linux  (MPU) host.

mantriyogesh, 2025-02-04:
Closing this issue, please feel free to re-open, if still more related questions or add new for different problem.

closed 2025-02-04 by mantriyogesh