espressif/esp-hosted #573: esp_hosted_ng: WiFi + Touch Element stop working on first touch event

icvidal, 2025-02-18:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Use WiFi and touch element driver simultaneously

### Actual behavior (suspected bug)

When WiFi is running and a touch event is triggered (touch or slide), the WiFi stack stops responding and after a few seconds the watchdog kicks in.

### Error logs or terminal output

```shell
Backtrace: 0x42024866:0x3FC9FE70 0x42024C4C:0x3FC9FE90 0x4037B40A:0x3FC9FEC0 0x400559DD:0x3FCACBC0 0x4038179D:0x3FCACBD0 0x4037CB0E:0x3FCACBF0 0x4037CBE6:0x3FCACC10 0x4037CD68:0x3FCACC30 0x4037D334:0x3FCACC50 0x403884A9:0x3FCACC70 0x4038ABA9:0x3FCACC90 0x4038B0FA:0x3FCACCB0 0x403895F1:0x3FCACCF0 0x40381585:0x3FCACD20

E (68732) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (68732) task_wdt:  - IDLE (CPU 0)
E (68732) task_wdt: Tasks currently running:
E (68732) task_wdt: CPU 0: wifi
E (68732) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x42024866:0x3FC9FE70 0x42024C4C:0x3FC9FE90 0x4037B40A:0x3FC9FEC0 0x42024C4C:0x3FCACBC0 0x4038179D:0x3FCACBD0 0x4037CB0E:0x3FCACBF0 0x4037CBE6:0x3FCACC10 0x4037CD68:0x3FCACC30 0x4037D334:0x3FCACC50 0x403884A9:0x3FCACC70 0x4038ABA9:0x3FCACC90 0x4038B0FA:0x3FCACCB0 0x403895F1:0x3FCACCF0 0x40381585:0x3FCACD20
```

### Steps to reproduce the behavior

1. Boot ESP32
2. Load kernel module on host 
3. Connect to WiFi 
4. Touch the touch sensor

### Project release version

Latest master

### System architecture

other (details in Additional context)

### Operating system

Linux

### Operating system version

kernel 5.10

### Shell

ZSH

### Additional context

Base commit: 06ed6a88d00a8f688cc6960558d118a4bfa828aa (master)
Variant: ESP32S3
IDF version: 5.4

You can find all logs (host side and ESP32 side) as well as a simple program to trigger the issue.
The touch element code is modified only to match our sensor array's characteristics (pad size, sensitivity). 

[logs_esphostedng_20250218_1015.zip](https://github.com/user-attachments/files/18842085/logs_esphostedng_20250218_1015.zip)

[program.zip](https://github.com/user-attachments/files/18842097/program.zip)

Extra:
* If the host's kernel module is not loaded, the touch sensor works as expected. The watchdog is not triggered.
* I have also conducted a test to check if the touch sensor callback is ever called when the issue occurs, and I see the callback is never executed when WiFi is ongoing.

```
    void slider_handler(touch_slider_handle_t out_handle, touch_slider_message_t
  *out_message, void *arg)
    {
      esp_cpu_reset(1);
      esp_cpu_reset(2);
    /* Other code ... */
    }
```
I'm expecting the processor to reset and the boot messages to be printed out to the UART after touching the sensor, but instead I keep getting the "watchdog triggered" message every few seconds.
* This issue didn't occur an older version we were using (before april 2024). I can try gathering more information if required.

icvidal, 2025-02-26:
Hi @mantriyogesh,
Any idea of what could be causing the issue? Please let me know if you need more information.

mantriyogesh, 2025-02-27:
Some observations:

1. The touch slider [example](https://github.com/espressif/esp-idf/tree/v5.2.5/examples/peripherals/touch_sensor/touch_element/touch_slider) somehow shows ESP32-S2/S3 as targets.
I am saying that it should not work on ESP32, but then the default configs used may be mismatched (needs further investigation using [ESP32 TRM](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf)
1.1. There might be possible cases where the GPIOs used in the example are conflicting with ESP32's currently in use GPIOs for hosted-SPI communication
1.2. Possibility of strapping pin GPIO used for touch slider

2. The slave example is showing the backtrace, may/may not be useful. but need to check with elf file.
The log might have been captured with some serial terminal like minicom.

It would be easier, if the esp is flashed with 
```
idf.py flash monitor
```

This would automatically parse the backtrace with correct functions.

3. 
> This issue didn't occur an older version we were using (before april 2024).
Although, this is excellent hint, but by 'older version' : can you please detail, older version of what? (if available, please mention commits or version number of specific)

mantriyogesh, 2025-02-27:
[image]

All GPIOs are in conflict:

1. SPI: Handshake : GPIO2

https://github.com/espressif/esp-hosted/blob/06ed6a88d00a8f688cc6960558d118a4bfa828aa/esp_hosted_ng/esp/esp_driver/network_adapter/main/Kconfig.projbuild#L44

2. SPI: Data Ready: GPIO4 
https://github.com/espressif/esp-hosted/blob/06ed6a88d00a8f688cc6960558d118a4bfa828aa/esp_hosted_ng/esp/esp_driver/network_adapter/main/Kconfig.projbuild#L50

3. SPI standard GPIOs, 12,13,14,15

https://github.com/espressif/esp-hosted/blob/06ed6a88d00a8f688cc6960558d118a4bfa828aa/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c#L58-L62

Way out: Change All SPI pins.

A. The SPI can work on other controller. Classic ESP32 packaged with two user controllable controllers. **Connect** and use 
https://github.com/espressif/esp-hosted/blob/06ed6a88d00a8f688cc6960558d118a4bfa828aa/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c#L51-L56

Change the spi_slave_api.c accordingly, also do not forget to use connect these new GPIOs to your host as spi replacement.

B. Handshake and Data Ready can be mapped any unused GPIOs. change your sdkconfig.default.esp32 to 
```
CONFIG_ESP_SPI_GPIO_HANDSHAKE=<new_gpio_num>
CONFIG_ESP_SPI_GPIO_DATA_READY=<new_gpio_num>
```
and then at https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng/esp/esp_driver/network_adapter in your repo,
```
idf.py fullclean
idf.py flash monitor
```

Note: All these pins also could be rerouted to any other GPIOs using GPIO matrix ( point A was trying IO_MUX, but if not these pins possible, still is okay.)

Once you remap all these pins, please try the alone slider. then club both and try again.
Please let us know how it goes.

icvidal, 2025-02-27:
Hi @mantriyogesh 
Thanks for your assistance. Unfortunately I don't have that binary anymore but I'll rebuild it and run it tomorrow in the office.
Right now, I do have the symbols from an older backtrace of the same issue, hope it helps:
```
esp/esp_driver/esp-idf/components/esp_system/task_wdt/task_wdt.c:434
esp/esp_driver/esp-idf/components/esp_system/task_wdt/task_wdt.c:507
esp/esp_driver/esp-idf/components/xtensa/xtensa_vectors.S:1240
esp/esp_driver/esp-idf/components/xtensa/include/xt_utils.h:82
esp/esp_driver/esp-idf/components/esp_system/freertos_hooks.c:58
esp/esp_driver/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4353 (discriminator 1)
esp/esp_driver/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

We are using a custom board and we are defining custom pins. 
We didn't have this issue before, we were branching out from `esp-hosted` from an older commit (around 1 year old) and we decided to update to include some bug fixes. After the update, this behavior appears. I think this suggest is not related to GPIO clashing, please confirm in case something changed regarding GPIO usage.

mantriyogesh, 2025-02-28:
As i understand , you have updated your code as part of some fork. Did you use the same git commit at host kernel module and slave?

(A) Wifi handling individually and (B)slider individually is okay?

(B) you confirmed earlier individually to be okay. What about (A) only?

icvidal, 2025-02-28:
> Did you use the same git commit at host kernel module and slave?

Yes, double checked.

> (A) Wifi handling individually and (B)slider individually is okay?

Yes it is. 

> (B) you confirmed earlier individually to be okay. What about (A) only?

I'm not sure if I understand your question. I'll try to answer to my best:
In both forks (1 year ago, and current), using WiFi individually or slider individually work fine. In the old fork, they also work fine together, in the new fork, when WiFi is connected and then the slider is touched, the issue occurs.
Does this answer the question or I'm missing something?

mantriyogesh, 2025-02-28:
@kapilkedawat , Can we reproduce this from test team?
The code part is straight forward , what code we need to add for slider. from the zips.

For ESP32-S3
latest master -> slider + wifi  --> fail
some commit (before april 2024) -> slider + wifi --> pass

mantriyogesh, 2025-03-05:
I don't have slider with me, checking if other teams have if they can reproduce

stwirth, 2025-03-05:
@mantriyogesh I don't think you need a slider, just touching the pins should be sufficient.

mantriyogesh, 2025-03-05:
You meant low-high etc?
Would it be feasible for you to emulate this issue in simple software step?

We can then check the root cause why should it fail or why it passed earlier.

icvidal, 2025-03-06:
I can hook an ESP32-S3 DevC kit to a RaspberryPi 4 and see if that reproduces the issue. Is this enough or you have another idea in mind?

mantriyogesh, 2025-03-06:
Should be fine IMO. Please share the code change or zip which we would refer to reproduce.

icvidal, 2025-03-11:
I've just set up a new Raspberry Pi 4 B with an ESP32-S3 Devkit C to reproduce the issue.
For your convenience, I've just forked this repository with some minimal changes, you can find it here: [ESP-Hosted](https://github.com/icvidal/esp-hosted)

Just a few notes: 

1. For the latest Raspberry Pi OS, released on November 19, the kernel pin indexing is slightly different and requires adding 512 to the GPIO pin number. Therefore, pin 6 (reset) becomes GPIO 518, this applies also to the `HANDSHAKE_PIN` and `SPI_DATA_READY_PIN`. This is the only code change introduced besides app_main.c that combines the touch element example into the WiFi example.
2. You don't need a touch strip to connect to the touch port, even touching the pins with your fingers will trigger the issue. There is one caveat: since noise can trigger the touch library, you may need to restart the ESP32 a few times until the kernel module initializes properly (takes 2-3 attempts here).

### How to reproduce

1. In order to fully understand the issue, here’s how you can go trough it
2. Set up hardware connections
3. Clone the repository and flash ESP32S3
4. With all the hardware set up and connected, boot the ESP32S3 without loading the Linux kernel module on the Raspberry Pi. Make sure it runs, check the serial monitor and touch the pins 5, 6 and 7 on the board to trigger messages from the slider example like the following `I (2712) Touch Slider Example: Slider Calculate, position: 10`
5. Leave the board in a non conductive surface, do not touch any pins or cables to avoid parasitic effects. Run the `./rpi_init.sh` script as follows:
```./rpi_init.sh spi resetpin=518```
If the following message `esp32_spi: wait_and_decode_cmd_resp: Command[0x1] timed out` appears in dmesg, repeat the step.
6. Use the WiFi driver as usual, it just works fine.
7. Touch the slider pins (5, 6, 7), the WiFi driver will freeze, after a few seconds the following messages appear in IDF’s serial monitor
```
E (160824) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (160824) task_wdt:  - IDLE (CPU 0)
E (160824) task_wdt: Tasks currently running:
E (160824) task_wdt: CPU 0: IDLE
E (160824) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x42024886:0x3FC9FE70 0x42024C6C:0x3FC9FE90 0x4037B40A:0x3FC9FEC0 0x4037ED13:0x3FCA9410 0x42024406:0x3FCA9430 0x40382735:0x3FCA9450 0x40381585:0x3FCA9470
```

Please let us know if you require any assistance

mantriyogesh, 2025-03-11:
Awesome!

Noted the detailed steps, I will try this at my place tomorrow and let you know ahead.

mantriyogesh, 2025-03-12:
Can you please confirm if this is the exact watchdog and crash you get? -

<details>
  <summary>ESP32-S3 Slave log</summary>

<code>
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x159c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcc4
load:0x403cb700,len:0x2e94
entry 0x403c890c
I (27) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (27) boot: compile time Mar 12 2025 13:29:17
W (27) boot: Unicore bootloader
I (27) boot: chip revision: v0.1
I (30) boot: efuse block revision: v1.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (71) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  3 factory          factory app      00 00 00010000 00100000
I (85) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (91) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (98) boot: End of partition table
I (101) boot: Defaulting to factory image
I (105) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=1c2f0h (115440) map
I (132) esp_image: segment 1: paddr=0002c318 vaddr=3fc9d900 size=03d00h ( 15616) load
I (136) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=8a2fch (566012) map
I (236) esp_image: segment 3: paddr=000ba324 vaddr=3fca1600 size=00b4ch (  2892) load
I (237) esp_image: segment 4: paddr=000bae78 vaddr=40374000 size=1980ch (104460) load
I (262) esp_image: segment 5: paddr=000d468c vaddr=600fe100 size=0001ch (    28) load
I (272) boot: Loaded app from partition at offset 0x10000
I (273) boot: Disabling RNG early entropy source...
I (283) cpu_start: Unicore app
I (283) cpu_start: Single core mode
I (292) cpu_start: Pro cpu start user code
I (292) cpu_start: cpu freq: 240000000 Hz
I (292) app_init: Application information:
I (292) app_init: Project name:     network_adapter
I (297) app_init: App version:      NG-1.0.3.0.9
I (301) app_init: Compile time:     Mar 12 2025 13:29:04
I (306) app_init: ELF file SHA256:  a2d359dd3...
I (310) app_init: ESP-IDF:          v5.4-dirty
I (315) efuse_init: Min chip rev:     v0.0
I (318) efuse_init: Max chip rev:     v0.99
I (322) efuse_init: Chip rev:         v0.1
I (326) heap_init: Initializing. RAM available for dynamic allocation:
I (333) heap_init: At 3FCA5DD8 len 00043938 (270 KiB): RAM
I (338) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (343) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (348) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (354) spi_flash: detected chip: generic
I (357) spi_flash: flash io: dio
W (360) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (372) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (378) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (385) coexist: coex firmware version: 49a8cdc
I (403) coexist: coexist rom version e7ae62f
I (404) main_task: Started on CPU0
I (404) main_task: Calling app_main()
[... 135 lines cut ...]
0x42076266: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/stdio.c:94
0x420758d9: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:224
0x4207592a: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:278
0x42075f29: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fvwrite.c:251
0x4207cf87: __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:429
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:399
0x4207c951: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1766 (discriminator 1)
0x420762dd: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
0x4208a29e: esp_log_writev at /Users/yogesh/code/h_icvidal/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/log/src/os/log_write.c:34
0x40387595: esp_log_write at /Users/yogesh/code/h_icvidal/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/log/src/os/log_write.c:44
0x4203e937: wifi_log at ??:?
0x4203e3b5: wifi_mode_set at ??:?
0x4204695f: wifi_start_process at ??:?
0x42044d22: ieee80211_ioctl_process at ??:?
0x40389571: ppTask at ??:?
0x40381585: vPortTaskWrapper at /Users/yogesh/code/h_icvidal/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

E (21629) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (21629) task_wdt:  - IDLE (CPU 0)
E (21629) task_wdt: Tasks currently running:
E (21629) task_wdt: CPU 0: wifi
E (21629) task_wdt: Print CPU 0 (current core) backtrace
</code>

</details>

icvidal, 2025-03-13:
This is what I get when I follow the instructions I've shared:
```
Backtrace: 0x42024886:0x3FC9FE70 0x42024C6C:0x3FC9FE90 0x4037B40A:0x3FC9FEC0 0x4037ED13:0x3FCA9410 0x42024406:0x3FCA9430 0x40382735:0x3FCA9450 0x40381585:0x3FCA9470
```

But, on the other side. If I don't retry step 5, I end up with something very similar to what you have shown:

```
Backtrace: 0x42024886:0x3FC9FE70 0x42024C6C:0x3FC9FE90 0x4037B40A:0x3FC9FEC0 0x42007599:0x3FCAC620 0x4200844B:0x3FCAC650 0x42009B9A:0x3FCAC670 0x420768A2:0x3FCAC690 0x42009590:0x3FCAC6B0 0x42009B9A:0x3FCAC6D0 0x42076266:0x3FCAC6F0 0x420758D9:0x3FCAC710 0x4207592A:0x3FCAC730 0x42075F29:0x3FCAC750 0x4207CF87:0x3FCAC780 0x4207C951:0x3FCAC7A0 0x420762DD:0x3FCACAC0 0x4208A29E:0x3FCACAF0 0x40387595:0x3FCACB20 0x4203E937:0x3FCACB70 0x4203E3B5:0x3FCACBE0 0x4204695F:0x3FCACC30 0x42044D22:0x3FCACC50 0x40389571:0x3FCACC70 0x40381585:0x3FCACCA0
```

It seems like the timing pays a role, if it fails earlier (before wifi is initialized) it'll show the message you got. Otherwise, if it gets to correctly initialize wifi, shows the message I've shown above in this comment. I guess it depends on when the touch processing triggers, and it depends on noise (as it's not connected to a touch strip).

If you want to have more chances of reproducing the issue after a correct wifi initialization, run the `rpi_init.sh` script, and if it fails just try the following:
`sudo rmmod esp32_spi.ko; sudo insmod ./esp32_spi.ko resetpin=6 `

Attaching both logs for your convenience.
[issue_before_finishing_wifi_initialization.txt](https://github.com/user-attachments/files/19225294/issue_before_finishing_wifi_initialization.txt)
[issue_with_correct_wifi_initialization.txt](https://github.com/user-attachments/files/19225295/issue_with_correct_wifi_initialization.txt)

On your log, I can see, with timestamp 1653 you have the following, that suggests the watchdog is triggering before the wifi finishes initializing. It happens between `"wifi:mode : sta (<YOUR MAC ADDRESS>)"` and  `"wifi: enable tsf"`
```
I (1653) wifi:mode : sta (7c:df:a1:e3:60:d8E (6629) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
...
```

Is this information useful?

icvidal, 2025-03-18:
Hi guys, any updates?
could you reproduce the issue? 
Please let me know if there is anything else I can do to help you

mantriyogesh, 2025-03-18:
Sorry, got busy in some other issues, i will check this today.

the backtraces you mentioned are specific to the elf that is built at your side.

are you using idf.py flash monitor? It should automatically translate these backtrace info (99% times)

icvidal, 2025-03-18:
Thanks for taking time for this update.
I've run the tests again, this time using the IDF's monitor for your convenience, the file names are easy to associate to the previous ones. I'm also attaching the ELF binary for your convenience.

[touch_issue_20250318.zip](https://github.com/user-attachments/files/19316619/touch_issue_20250318.zip)

mantriyogesh, 2025-03-21:
I am so sorry @icvidal, and it gets problematic when we are not resolving faster as per your expectation.

I know that you are waiting for this since long, but I am pre-occupied for prior commitments.
Rest assured, I can check this on Monday (I would make sure this happens on Monday)

icvidal, 2025-03-21:
Yogesh, very much appreciated. Just let me know if you need anything from me.

mantriyogesh, 2025-03-24:
from your log, issue_with_correct_wifi_initialization_idf.txt, 

```
[1;31mE (18483) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:[0m
[1;31mE (18483) task_wdt:  - IDLE (CPU 0)[0m
[1;31mE (18483) task_wdt: Tasks currently running:[0m
[1;31mE (18483) task_wdt: CPU 0: IDLE[0m
[1;31mE (18483) task_wdt: Print CPU 0 (current core) backtrace[0m
```

Idle task itself got watchdog trigger. are you hooking any code in the IDLE task as such?

mantriyogesh, 2025-03-24:
```
[0;32mI (1619) FW_MAIN: Initial set up done[0m
[0;32mI (1620) Touch Slider Example: Touch element library installed[0m
[0;32mI (1620) Touch Slider Example: Touch slider installed[0m
[0;32mI (1622) Touch Slider Example: Touch slider created[0m
[0;32mI (1626) Touch Slider Example: Touch element library start[0m
[0;32mI (1631) main_task: Returned from app_main()[0m
[0;32mI (1636) FW_MAIN: INIT Interface command[0m
[0;32mI (1641) wifi:mode : sta (cc:ba:E (6617) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:[0m
[1;31mE (6617) task_wdt:  - IDLE (CPU 0)[0m
[1;31mE (6617) task_wdt: Tasks currently running:[0m
[1;31mE (6617) task_wdt: CPU 0: wifi[0m
[1;31mE (6617) task_wdt: Print CPU 0 (current core) backtrace[0m

Backtrace: 0x42024886:0x3FC9FE70 0x42024C6C:0x3FC9FE90 0x4037B40A:0x3FC9FEC0 0x420075A4:0x3FCAC620 0x4200844B:0x3FCAC650 0x42009B9A:0x3FCAC670 0x420768A2:0x3FCAC690 0x42009590:0x3FCAC6B0 0x42009B9A:0x3FCAC6D0 0x42076266:0x3FCAC6F0 0x420758D9:0x3FCAC710 0x4207592A:0x3FCAC730 0x42075F29:0x3FCAC750 0x4207CF87:0x3FCAC780 0x4207C951:0x3FCAC7A0 0x420762DD:0x3FCACAC0 0x4208A29E:0x3FCACAF0 0x40387595:0x3FCACB20 0x4203E937:0x3FCACB70 0x4203E3B5:0x3FCACBE0 0x4204695F:0x3FCACC30 0x42044D22:0x3FCACC50 0x40389571:0x3FCACC70 0x40381585:0x3FCACCA0
0x42024886: task_wdt_timeout_handling at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/task_wdt/task_wdt.c:434
0x42024c6c: task_wdt_isr at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/task_wdt/task_wdt.c:507
0x4037b40a: _xt_lowint1 at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x420075a4: uart_ll_get_txfifo_len at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/hal/esp32s3/include/hal/uart_ll.h:410
 (inlined by) uart_tx_char at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_driver_uart/src/uart_vfs.c:170
0x4200844b: uart_write at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_driver_uart/src/uart_vfs.c:223
0x42009b9a: esp_vfs_write at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/vfs/vfs.c:945 (discriminator 4)
0x420768a2: write at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/syscalls/syswrite.c:11 (discriminator 1)
0x42009590: console_write at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_vfs_console/vfs_console.c:71
0x42009b9a: esp_vfs_write at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/vfs/vfs.c:945 (discriminator 4)
0x42076266: __swrite at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/stdio.c:94
0x420758d9: __sflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:224
0x4207592a: _fflush_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fflush.c:278
0x42075f29: __sfvwrite_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/fvwrite.c:251
0x4207cf87: __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:429
 (inlined by) __sprint_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:399
0x4207c951: _vfprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vfprintf.c:1766 (discriminator 1)
0x420762dd: vprintf at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/stdio/vprintf.c:34
0x4208a29e: esp_log_writev at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/log/src/os/log_write.c:34
0x40387595: esp_log_write at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/log/src/os/log_write.c:44
0x4203e937: wifi_log at ??:?
0x4203e3b5: wifi_mode_set at ??:?
0x4204695f: wifi_start_process at ??:?
0x42044d22: ieee80211_ioctl_process at ??:?
0x40389571: ppTask at ??:?
0x40381585: vPortTaskWrapper at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

[1;31mE (11617) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:[0m
[1;31mE (11617) task_wdt:  - IDLE (CPU 0)[0m
[1;31mE (11617) task_wdt: Tasks currently running:[0m
[1;31mE (11617) task_wdt: CPU 0: wifi[0m
[1;31mE (11617) task_wdt: Print CPU 0 (current core) backtrace[0m

Backtrace: 0x42024886:0x3FC9FE70 0x42024C6C:0x3FC9FE90 0x4037B40A:0x3FC9FEC0 0x42024C6C:0x3FCAC620 0x4200844B:0x3FCAC650 0x42009B9A:0x3FCAC670 0x420768A2:0x3FCAC690 0x42009590:0x3FCAC6B0 0x42009B9A:0x3FCAC6D0 0x42076266:0x3FCAC6F0 0x420758D9:0x3FCAC710 0x4207592A:0x3FCAC730 0x42075F29:0x3FCAC750 0x4207CF87:0x3FCAC780 0x4207C951:0x3FCAC7A0 0x420762DD:0x3FCACAC0 0x4208A29E:0x3FCACAF0 0x40387595:0x3FCACB20 0x4203E937:0x3FCACB70 0x4203E3B5:0x3FCACBE0 0x4204695F:0x3FCACC30 0x42044D22:0x3FCACC50 0x40389571:0x3FCACC70 0x40381585:0x3FCACCA0

```

@kapilkedawat , Can we please check why the watchdog could be triggered in wifi task with some print?

mantriyogesh, 2025-03-24:
@icvidal , if you are using the callback, in general, it is better if you do not dump any messages. as this callback would be locking and blocking on a lot of things.

For now , can you please try with below change:

```diff 
I  ~/c/h/esp-hosted   *…  esp_hosted_ng/esp/esp_driver/network_adapter  git diff main/app_main.c                                                                             787ms  Mon Mar 24 11:47:59 2025
diff --git a/esp_hosted_ng/esp/esp_driver/network_adapter/main/app_main.c b/esp_hosted_ng/esp/esp_driver/network_adapter/main/app_main.c
index 8936f27..2f3eb5b 100644
--- a/esp_hosted_ng/esp/esp_driver/network_adapter/main/app_main.c
+++ b/esp_hosted_ng/esp/esp_driver/network_adapter/main/app_main.c
@@ -109,11 +109,11 @@ void slider_handler(touch_slider_handle_t out_handle, touch_slider_message_t *ou
         return;
     }
     if (out_message->event == TOUCH_SLIDER_EVT_ON_PRESS) {
-        ESP_LOGI(TAG_Slider, "Slider Press, position: %"PRIu32, out_message->position);
+        ESP_EARLY_LOGI(TAG_Slider, "Slider Press, position: %"PRIu32, out_message->position);
     } else if (out_message->event == TOUCH_SLIDER_EVT_ON_RELEASE) {
-        ESP_LOGI(TAG_Slider, "Slider Release, position: %"PRIu32, out_message->position);
+        ESP_EARLY_LOGI(TAG_Slider, "Slider Release, position: %"PRIu32, out_message->position);
     } else if (out_message->event == TOUCH_SLIDER_EVT_ON_CALCULATION) {
-        ESP_LOGI(TAG_Slider, "Slider Calculate, position: %"PRIu32, out_message->position);
+        ESP_EARLY_LOGI(TAG_Slider, "Slider Calculate, position: %"PRIu32, out_message->position);
     }
 }
 #endif
```

Despite the above change, I strongly recommend, $${\color{yellow}not\ to\ have\ any\ prints}$$ at all in $${\color{yellow}callback}$$ way.

mantriyogesh, 2025-03-24:
[gh_573_icvdal_slider_code_as_task.patch](https://github.com/user-attachments/files/19414473/gh_573_icvdal_slider_code_as_task.patch)

Please confirm if this works for you, @icvidal . I observe sparingly but still seen, 

```
E (255980) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (255980) task_wdt:  - IDLE (CPU 0)
E (255980) task_wdt: Tasks currently running:
E (255980) task_wdt: CPU 0: IDLE
E (255980) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x420249CE:0x3FC9FE70 0x42024E68:0x3FC9FE90 0x4037B40A:0x3FC9FEC0 0x42024E68:0x3FCA9410 0x4202443A:0x3FCA9430 0x40382735:0x3FCA9450 0x40381585:0x3FCA9470
--- 0x420249ce: task_wdt_timeout_handling at /Users/yogesh/code/h_icvidal/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/task_wdt/task_wdt.c:434
0x42024e68: task_wdt_isr at /Users/yogesh/code/h_icvidal/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/task_wdt/task_wdt.c:507
0x4037b40a: _xt_lowint1 at /Users/yogesh/code/h_icvidal/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x42024e68: task_wdt_isr at /Users/yogesh/code/h_icvidal/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/task_wdt/task_wdt.c:507
0x4202443a: esp_vApplicationIdleHook at /Users/yogesh/code/h_icvidal/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/freertos_hooks.c:58
0x40382735: prvIdleTask at /Users/yogesh/code/h_icvidal/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4353 (discriminator 1)
0x40381585: vPortTaskWrapper at /Users/yogesh/code/h_icvidal/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

but i would still ask you if this issue is observed yet on your end.

I am just touching any GPIO pin randomly, which is not really correct way as well.

icvidal, 2025-03-24:
Hi @mantriyogesh thanks for the suggestions. Unfortunately, the issue still reproduces.
```
E (8629) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (8629) task_wdt:  - IDLE (CPU 0)
E (8629) task_wdt: Tasks currently running:
E (8629) task_wdt: CPU 0: IDLE
E (8629) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4202483A:0x3FC9FE70 0x42024C20:0x3FC9FE90 0x4037B40A:0x3FC9FEC0 0x4037ED13:0x3FCA9410 0x420243BA:0x3FCA9430 0x40382735:0x3FCA9450 0x40381585:0x3FCA9470
--- 0x4202483a: task_wdt_timeout_handling at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/task_wdt/task_wdt.c:434
0x42024c20: task_wdt_isr at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/task_wdt/task_wdt.c:507
0x4037b40a: _xt_lowint1 at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/xtensa/xtensa_vectors.S:1240
0x4037ed13: xt_utils_wait_for_intr at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/xtensa/include/xt_utils.h:82
 (inlined by) esp_cpu_wait_for_intr at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_hw_support/cpu.c:55
0x420243ba: esp_vApplicationIdleHook at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/esp_system/freertos_hooks.c:58
0x40382735: prvIdleTask at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4353 (discriminator 1)
0x40381585: vPortTaskWrapper at /home/icv/ambient/esp_hosted_touch/esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

Is there anything else I can do to assist you?

mantriyogesh, 2025-03-25:
```diff
 I  ~/c/h/esp-hosted   *…  esp_hosted_ng/esp/esp_driver/network_adapter  git diff sdkconfig.defaults                                                                          842ms  Tue Mar 25 13:53:51 2025
diff --git a/esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig.defaults b/esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig.defaults
index a05d785..e36cd20 100644
--- a/esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig.defaults
+++ b/esp_hosted_ng/esp/esp_driver/network_adapter/sdkconfig.defaults
@@ -1,6 +1,5 @@
 CONFIG_ESP32_DEFAULT_CPU_FREQ_160=y
 CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=160
-CONFIG_FREERTOS_UNICORE=y
 CONFIG_SDIO_DAT2_DISABLED=

 # BT Configuration
```

Can you please try this?

```
<apply above patch on top>
rm -rf sdkconfig build/
idf.py full clean
idf.py flash monitor
```

icvidal, 2025-03-25:
Thanks @mantriyogesh , I can't reproduce the issue anymore after applying the patch on this test setup. The system is working as I'd expect it to work.

A couple follow-up questions:
1. Do you expect any side-effects or possible problems arising from this configuration? 
2. What are other implications of this change? (increased memory usage maybe?)
3. Do we need to have any extra consideration on our code when enabling multi core usage? Example: syncrhonization, initialization, hw resource sharing, etc
4. Shall we try to pin our tasks to the second core or let the IDF assign them by itself? (affinity settings)
5. Is the `esp_hosted_ng` code going to require a fix or, instead, is expected to use a multi-core setup for `esp_hosted_ng`? This question is aimed to know if I should merge any upcoming changes in the code base. 
 
I'll test it on our product setup and update the thread.

mantriyogesh, 2025-03-25:
> 1. Do you expect any side-effects or possible problems arising from this configuration?

Actually, we were forcing the S3 to be single core, despite dual-core, which was not correct.

> 2. What are other implications of this change? (increased memory usage maybe?)

Espressif-FreeRTOS would be running on both cores. Which is desired for S3.

> 3. Do we need to have any extra consideration on our code when enabling multi core usage? Example: syncrhonization, initialization, hw resource sharing, etc

The multi threading or synchronization etc concepts are the same irrespective of number of cores.

> 4. Shall we try to pin our tasks to the second core or let the IDF assign them by itself? (affinity settings)

You are free to plan your task the way you want. But keep in mind there might be other tasks too. In general, if you do not pin, it would run on available CPU, which should be flexible and better. In general, the Wi-Fi and Bluetooth tasks on slave are pinned to core. But application tasks are free to go as they see availability. For Wi-Fi or Bluetooth, pinning is also on very special cases (in case they are not pinned, do not bother or force them)

It is more of your way, how you wish to do. But always then keep in mind the pinning architecture you implemented, in case of debugging your application side issues. This is just informational. Just pinning tasks do not cause any harm or lead to issues as such, unless some tasks do let go of CPU at all. The idle watchdog is something to be taken serious.

> 5. Is the esp_hosted_ng code going to require a fix or, instead, is expected to use a multi-core setup for esp_hosted_ng? This question is aimed to know if I should merge any upcoming changes in the code base.

Up to you. This change is already on [master](https://github.com/espressif/esp-hosted/commit/f0d24b48ab7850c375e2d1c695637ee1a0d9d0aa#diff-71d0a4eae8f1d52f73ea2c17fed877a737e23aa017ab970ce30f5c963c064d8cL3) , as it is expected not to mandate single core, to utilise full strength of chipset. if some chipset like c2/c3 etc is single core, would auto behave single core. if esp32 or s3 chipset available, it was wrongly forced to be single core.

icvidal, 2025-03-25:
Sounds good!
I'm about to test this on our product, if the issue doesn't reproduce there, I'll close the thread.

Thanks for your answers.

mantriyogesh, 2025-03-25:
The change is important and good irrespective of your use case.

Although I could not understand why should the slider when started, not relinquish the CPU. Anyway the issue should be not observed with this config. Anyway, if you find any issues with slider, please let us know. We would try our best to answer. If not, may be https://github.com/espressif/esp-idf/issues would be better place to raise 'slider only' issues (now i don't it would no more related to ESP-Hosted after this fix)

mantriyogesh, 2025-03-25:
I appreciate your patience and readiness to adjust the solution for mimicking the issue.
Sometimes, we are stuck in other user/customer's issues, which delays the resolution.

Finally, **thanks** for letting us this issue, it actually fixed one of incorrect but very important config.

icvidal, 2025-03-26:
Hi @mantriyogesh a preliminary test shows it works on our product. I'll take a few days to test a bit more and see if the issue reproduces on other conditions.

And thanks for your comment!

mantriyogesh, 2025-03-26:
We can close the issue, so that I don't have to track this actively.
If you face issues on same leg, you can easily re-open.

Would it be fine for you?

icvidal, 2025-03-28:
Hi @mantriyogesh thanks for your patience.
We have been testing it on our product and the results were very good so far. We consider this issue closed.

We really appreciate your support on this matter, thanks a lot!

closed 2025-03-28 by icvidal