espressif/esp-hosted #574: ESP32C6 - SDIO: probe of mmc1:0001:1 failed with error -110

caipiblack, 2025-02-18:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello,

We have issues with the SDIO mode in a new board, we are using this commit `21a5e735dc3660e4d4da7a201f9481d3d69fdab2` for HOST driver AND esp32c6 firmware!

1. We start the module like this:

```
rmmod esp_hosted_ng
insmod /lib/modules/5.10.201-cip41-yocto-standard/extra/esp-hosted-ng.ko resetpin=506
```

2. We get these logs on ESP32C6 UART:

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x3048
entry 0x4086c11a
I (23) boot: ESP-IDF 67c1de1e-dirty 2nd stage bootloader
I (24) boot: compile time Feb 17 2025 14:30:03
I (24) boot: chip revision: v0.1
I (24) boot: efuse block revision: v0.3
I (28) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (35) boot.esp32c6: SPI Flash Size : 4MB
I (39) boot: Enabling RNG early entropy source...
I (44) boot: Partition Table:
I (46) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (59) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (66) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  3 factory          factory app      00 00 00010000 00100000
I (79) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (85) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (92) boot: End of partition table
I (95) boot: Defaulting to factory image
I (99) esp_image: segment 0: paddr=00010020 vaddr=420c0020 size=1d9b8h (121272) map
I (128) esp_image: segment 1: paddr=0002d9e0 vaddr=40800000 size=02638h (  9784) load
I (131) esp_image: segment 2: paddr=00030020 vaddr=42000020 size=b0948h (723272) map
I (266) esp_image: segment 3: paddr=000e0970 vaddr=40802638 size=1ca2ch (117292) load
I (291) esp_image: segment 4: paddr=000fd3a4 vaddr=4081f070 size=03c80h ( 15488) load
I (302) boot: Loaded app from partition at offset 0x10000
I (302) boot: Disabling RNG early entropy source...
I (312) cpu_start: Unicore app
I (321) cpu_start: Pro cpu start user code
I (321) cpu_start: cpu freq: 160000000 Hz
I (321) app_init: Application information:
I (321) app_init: Project name:     network_adapter
I (326) app_init: App version:      NG-1.0.3.0.9
I (330) app_init: Compile time:     Feb 17 2025 14:29:52
I (335) app_init: ELF file SHA256:  2f06b27ea...
I (339) app_init: ESP-IDF:          67c1de1e-dirty
I (344) efuse_init: Min chip rev:     v0.0
I (348) efuse_init: Max chip rev:     v0.99 
I (352) efuse_init: Chip rev:         v0.1
I (356) heap_init: Initializing. RAM available for dynamic allocation:
I (362) heap_init: At 40832120 len 0004A4F0 (297 KiB): RAM
[... 197 lines cut ...]
On my schematic there is a PU in SD1_CMD and all SD1_DATAx like porting guide (to 3.3v)

I take a look in all of this:

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/Troubleshoot.md
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md
https://github.com/espressif/esp-hosted/issues/318
https://github.com/espressif/esp-hosted/issues/342
https://github.com/espressif/esp-hosted/issues/275

This is what I see after some debugging:

1. The voltage of the SD1_CMD and all SD1_DATAx is at 2.12V (instead of 3.3V)
2. We found another issue with an ESP32 where they talk about strapping pin that put SDI data lines at 1.8V, but with the ESP32C6 it doesn't seems to have this. 
3. Anyway, when the ESP32C6 is in reset mode, the SD1_CMD and all SD1_DATAx lines stay at 2.12V. (So maybe the esp32c6 is not the chip causing the 2.12V)
4. I don't know why we don't have 3.3V and I don't know if it can be the source of the problem
5. The esp32c6 is correctly powered by 3.3V and on the top of the pullups we have 3.3V
6. The DTS is configured with 3.3V regulator
7. I try to lower the speed (25Mhz, 20Mhz and 5Mhz -> It has no effect)
8. Bonus: With a scope I see no effect on SD1_DATAx lines when the driver is started, the lines stay at 2.12V
9. The only effect is that ESP32C6 is reset so I have the logs of the C6 in UART

Did someone know what `probe of mmc1:0001:1 failed with error -110` means ? 

I will be happy if someone have any idea 👍

caipiblack, 2025-03-19:
Hello,

We found voltages issues in our board that has been fixed, but now we still have two issues related to the esp_hosted driver.

1. This trace appear sometimes but doesn't seems to be a problem as the WiFi is now working:

```
[   11.756092] ------------[ cut here ]------------
[   11.764956] renesas_sdhi_internal_dmac 11c10000.mmc: DMA addr 0x0000800051b23e40+4 overflow (mask ffffffff, bus limit 0).
[   11.803455] WARNING: CPU: 0 PID: 192 at kernel/dma/direct.h:97 dma_direct_map_sg+0x270/0x290
[   11.811884] Modules linked in: otodo_rgb_led(O) esp_hosted_ng(O)
[   11.817886] CPU: 0 PID: 192 Comm: Monitor process Tainted: G           O      5.10.201-cip41-yocto-standard #1
[   11.827846] Hardware name: Development board based on r9a07g043u11 (DT)
[   11.834435] pstate: 60c00005 (nZCv daif +PAN +UAO -TCO BTYPE=--)
[   11.840420] pc : dma_direct_map_sg+0x270/0x290
[   11.844845] lr : dma_direct_map_sg+0x270/0x290
[   11.849268] sp : ffff800011b23ac0
[   11.852568] x29: ffff800011b23ac0 x28: ffff800011b23c18 
[   11.857861] x27: 0000000000000000 x26: ffff00000ab39410 
[   11.863154] x25: 0000000000000000 x24: 0000000000000000 
[   11.868446] x23: 0000000000000002 x22: fffffdffffe00000 
[   11.873739] x21: ffff800010d56260 x20: ffff80001112d7e0 
[   11.879032] x19: 0000000000000001 x18: 0000000000000000 
[   11.884324] x17: 0000000000000000 x16: 0000000000000000 
[   11.889617] x15: 000023abbfbc0d30 x14: 00000000000002bd 
[   11.894909] x13: 00000000000002bd x12: 0000000000000000 
[   11.900202] x11: 0000000000000019 x10: 0000000000000940 
[   11.905494] x9 : ffff800011b23810 x8 : ffff00000ce8e820 
[   11.910787] x7 : ffff00001ff11d40 x6 : 00000000008eafb5 
[   11.916080] x5 : 0000000000000000 x4 : ffff00001ff08828 
[   11.921372] x3 : ffff00001ff0f748 x2 : 0000000000000000 
[   11.926665] x1 : 0000000000000000 x0 : ffff00000ce8de80 
[   11.931958] Call trace:
[   11.934396]  dma_direct_map_sg+0x270/0x290
[   11.938476]  dma_map_sg_attrs+0x44/0xac
[   11.942300]  renesas_sdhi_internal_dmac_pre_req+0x48/0xa0
[   11.947680]  mmc_io_rw_extended+0x1bc/0x2cc
[   11.951847]  sdio_io_rw_ext_helper+0x194/0x240
[   11.956273]  sdio_memcpy_fromio+0x20/0x30
[   11.960278]  esp_read_multi_byte+0x88/0xd4 [esp_hosted_ng]
[   11.965746]  esp_read_reg+0x24/0x44 [esp_hosted_ng]
[   11.970607]  monitor_process+0x84/0x160 [esp_hosted_ng]
[   11.975814]  kthread+0x154/0x160
[   11.979031]  ret_from_fork+0x10/0x38
[   11.982592] ---[ end trace 6700ab238365b7f5 ]---
```

2. We found that the driver don't start correctly if it controls the RESET pin of the ESP32C6.

Logs when the driver control the RESET pin of the ESP32C6:

```
# dmesg | grep mmc1
[    3.455690] renesas_sdhi_internal_dmac 11c10000.mmc: mmc1 base at 0x0000000011c10000, max clock rate 50 MHz
[    3.507802] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[    3.520717] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[    3.529435] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[    3.537256] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[    3.542838] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[    3.548457] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[    3.555507] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[    3.561087] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[    3.566696] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[    3.572373] mmc1: new SDIO card at address 0001
[    6.413383] esp_hosted_ng: probe of mmc1:0001:1 failed with error -110
[    6.426914] esp_hosted_ng: probe of mmc1:0001:2 failed with error -110

# dmesg | grep esp_hosted
[    6.172522] esp_hosted_ng: loading out-of-tree module taints kernel.
[    6.413383] esp_hosted_ng: probe of mmc1:0001:1 failed with error -110
[... 38 lines cut ...]
[    6.563340] esp_hosted_ng: check_esp_version: ESP-Hosted Version: NG-1.0.3.0.9
[    6.571083] esp_hosted_ng: tx_process: not ready
[    6.572537] esp_hosted_ng: probe of mmc1:0001:2 failed with error -22
[    6.584198] esp_hosted_ng: esp_reg_notifier: Driver init is ongoing
[    6.714270] esp_hosted_ng: print_capabilities: Capabilities: 0x9. Features supported are:
[    6.724785] esp_hosted_ng: print_capabilities: 	 * WLAN on SDIO
[    8.665730] esp_hosted_ng: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=\x00\x00
[   11.811884] Modules linked in: otodo_rgb_led(O) esp_hosted_ng(O)
[   11.960278]  esp_read_multi_byte+0x88/0xd4 [esp_hosted_ng]
[   11.965746]  esp_read_reg+0x24/0x44 [esp_hosted_ng]
[   11.970607]  monitor_process+0x84/0x160 [esp_hosted_ng]
[   18.810091] esp_hosted_ng: cmd_auth_request: Authentication request: 60:38:e0:cd:92:69 8 0 0 0
[   19.157059] esp_hosted_ng: cmd_assoc_request: Association request: 60:38:e0:cd:92:69 8 37
[   19.311094] esp_hosted_ng: process_assoc_event: Connection status: 0
[   24.379582] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.003.116
[   49.560421] esp_hosted_ng: process_disconnect_event: Disconnect event for ssid Linksys-DEA9 [reason:3]
[   49.597160] esp_hosted_ng: esp_inetaddr_event: Interface wlan0 Down: 0
[   49.623907] esp_hosted_ng: esp_reg_notifier: cfg80211 regulatory domain callback for 00, current=00
[   51.311583] esp_hosted_ng: cmd_auth_request: Authentication request: 60:38:e0:cd:92:69 8 0 0 0
[   51.568419] esp_hosted_ng: cmd_assoc_request: Association request: 60:38:e0:cd:92:69 8 37
[   51.674578] esp_hosted_ng: process_assoc_event: Connection status: 0
[   51.967339] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.003.116
```

Any ideas why when the driver control the RESET pin then it don't start correctly ? Is it wanted ?

mantriyogesh, 2025-03-20:
That's right ! Reset pin is needed.

The reason of reset pin is that if there is some unexpected scenario on sdio bus, then the complete communication would be lost. so whenever we are loading the driver on host, we reset the esp hardware, to clear up all sdio incomplete bus operations etc and get a fresh slave on load.

In general, you can try without reset pin, if it works for you, but right now our recommendation is to have such pin.
If you wish to verify your schematic, you can upload at https://www.espressif.com/en/contact-us/circuit-schematic-pcb-design-review so that hardware team can provide professional review.

If you wish to get our current PCB board schematic with Raspberry pi and ESP32-C6, I can send you for your own comparison. Please send me mail from my profile.

mantriyogesh, 2025-03-20:
I really appreciate the debugging efforts you have put. Also while designing I suggest PCB to hold extra traces in GPIOs, which may or may not be needed in future, to be future ready. Some features like host sleep etc, if supported by your linux, then these extra GPIOs initial in NC(Not connected) can be changed to connected. Just a suggestion.

Also please provide extra focus on getting the SDIO trace length to be minimal, as much as straight and equal for sdio. (multiple) ground planes could help a lot to mitigate the interference. Suggested pull-ups are >10KOhm. We use 51KOhm in our board.

caipiblack, 2025-03-25:
Also, Here there is a patch of what we change on the driver:
- Add the control of GPIO9: But maybe it is not usefull as ESP32c6 as weak pull up during/after reset. But I did it for debug
- Add the possibility to view the FW version of ESP32 from the userspace (as we update the chip from user space)
- Disable the bluetooth capability (mainly for our SPI board using ESP32c3, but the code is shared for the two boards)

```
diff --git a/main.c b/main.c
index 5f0a232..07bd3d2 100644
--- a/main.c
+++ b/main.c
@@ -12,6 +12,13 @@
 #include <linux/gpio.h>
 #include <linux/igmp.h>
 
+// CUSTOM (for sysfs)
+#include <linux/kobject.h>
+#include <linux/sysfs.h>
+#include <linux/fs.h>
+#include <linux/string.h>
+// End CUSTOM
+
 #include "esp.h"
 #include "esp_if.h"
 #include "esp_bt_api.h"
@@ -32,7 +39,19 @@
 #define RELEASE_VERSION PROJECT_NAME "-" STRINGIFY(PROJECT_VERSION_MAJOR_1) "." STRINGIFY(PROJECT_VERSION_MAJOR_2) "." STRINGIFY(PROJECT_VERSION_MINOR) "." STRINGIFY(PROJECT_REVISION_PATCH_1) "." STRINGIFY(PROJECT_REVISION_PATCH_2)
 
 static char *ota_file = NULL;
-static int resetpin = HOST_GPIO_PIN_INVALID;
+// CUSTOM Note: To find the good ID for the GPIO you need to use it:
+// Notation: Px_y     -> Gpio Number: (8*x + y) + 360
+static int resetpin = HOST_GPIO_PIN_INVALID;
+// CUSTOM
+static int gpio9pin = HOST_GPIO_PIN_INVALID;
+static uint8_t last_version_updated = 0;
+static uint8_t last_version_major1 = 0;
+static uint8_t last_version_major2 = 0;
+static uint8_t last_version_minor = 0;
+static uint8_t last_revision_patch_1 = 0;
+static uint8_t last_revision_patch_2 = 0;
+static struct kobject *esphosted_kobject;
+// End CUSTOM
 static u32 clockspeed = 0;
 extern u8 ap_bssid[MAC_ADDR_LEN];
 extern volatile u8 host_sleep;
@@ -45,6 +64,11 @@ char version_str[VERSION_BUFFER_SIZE];
 module_param(resetpin, int, S_IRUSR | S_IWUSR | S_IRGRP | S_IROTH);
 MODULE_PARM_DESC(resetpin, "Host's GPIO pin number which is connected to ESP32's EN to reset ESP32 device");
 
+// CUSTOM
+module_param(gpio9pin, int, S_IRUSR | S_IWUSR | S_IRGRP | S_IROTH);
+MODULE_PARM_DESC(gpio9pin, "Host's GPIO pin number which is connected to ESP32's GPIO9 to select the boot mode of ESP32 device");
+// CUSTOM END
+
 module_param(clockspeed, uint, S_IRUSR | S_IWUSR | S_IRGRP | S_IROTH);
 MODULE_PARM_DESC(clockspeed, "Hosts clock speed in MHz");
 
@@ -229,6 +253,10 @@ void print_capabilities(u32 cap)
 
 static void init_bt(struct esp_adapter *adapter)
 {
+	/* Disabling ESP_BT_SPI_SUPPORT */
+	adapter->capabilities &= ~(1 << 6); 
+	adapter->capabilities &= ~(1 << 1);
+	adapter->capabilities &= ~(1 << 2);
 
 	if ((adapter->capabilities & ESP_BT_SPI_SUPPORT) ||
 		(adapter->capabilities & ESP_BT_SDIO_SUPPORT)) {
@@ -240,6 +268,15 @@ static void init_bt(struct esp_adapter *adapter)
 
 static int check_esp_version(struct fw_version *ver)
 {
+	// CUSTOM
+	last_version_major1 = ver->major1;
+	last_version_major2 = ver->major2;
+	last_version_minor = ver->minor;
+	last_revision_patch_1 = ver->revision_patch_1;
+	last_revision_patch_2 = ver->revision_patch_2;
+	last_version_updated = 1;
+	// End CUSTOM
+
 	snprintf(version_str, VERSION_BUFFER_SIZE, "%s-%u.%u.%u.%u.%u",
 		ver->project_name, ver->major1, ver->major2, ver->minor, ver->revision_patch_1, ver->revision_patch_2);
 
@@ -1083,6 +1120,19 @@ static void esp_reset(void)
 			esp_warn("host resetpin (%d) configured is invalid GPIO\n", resetpin);
 			resetpin = HOST_GPIO_PIN_INVALID;
 		} else {
+			// CUSTOM
+			if (gpio9pin != HOST_GPIO_PIN_INVALID) {
+				if (!gpio_is_valid(gpio9pin)) {
+					esp_warn("host gpio9pin (%d) configured is invalid GPIO\n", gpio9pin);
+					gpio9pin = HOST_GPIO_PIN_INVALID;
+				} else {
+					gpio_request(gpio9pin, "sysfs");
+					gpio_direction_output(gpio9pin, true);
+					gpio_set_value(gpio9pin, 1);
+				}
+			}
+			// CUSTOM END
+		
 			gpio_request(resetpin, "sysfs");
 
 			/* HOST's resetpin set to OUTPUT, HIGH */
@@ -1094,16 +1144,45 @@ static void esp_reset(void)
 
 			/* HOST's resetpin set to INPUT */
 			gpio_direction_input(resetpin);
-
+			
 			esp_dbg("Triggering ESP reset.\n");
 		}
 	}
 }
 
+// CUSTOM
+static ssize_t esp32_firmware_version_show(struct kobject *kobj, struct kobj_attribute *attr, char *buf)
+{
+	if (last_version_updated) {
+		return sprintf(buf, "%d.%d.%d.%d.%d\n", last_version_major1, last_version_major2, last_version_minor, last_revision_patch_1, last_revision_patch_2);
+	} else {
+		return sprintf(buf, "0.0.0.0.0\n");
+	}
+}
+
+static ssize_t esp32_firmware_version_store(struct kobject *kobj, struct kobj_attribute *attr, const char *buf, size_t count)
+{
+	// If user write in the file, we do nothing. This is simply not supported.
+        return count;
+}
+
+static struct kobj_attribute esp32_firmware_version_attribute =__ATTR(esp32_firmware_version, 0660, esp32_firmware_version_show, esp32_firmware_version_store);
+// End CUSTOM
+
 static int __init esp_init(void)
 {
 	int ret = 0;
 	struct esp_adapter *adapter = NULL;
+	
+	// CUSTOM
+	esphosted_kobject = kobject_create_and_add("esp32-hosted-ng", kernel_kobj);
+        if(!esphosted_kobject)
+                return -ENOMEM;
+
+        if (sysfs_create_file(esphosted_kobject, &esp32_firmware_version_attribute.attr)) {
+                pr_debug("failed to create the 'esp32_firmware_version' file in /sys/kernel/esp32-hosted-ng\n");
+        }
+        // End CUSTOM
 
 	/* Reset ESP, Clean start ESP */
 	esp_reset();
@@ -1129,6 +1208,10 @@ static int __init esp_init(void)
 static void __exit esp_exit(void)
 {
 	uint8_t iface_idx = 0;
+	
+	// CUSTOM
+	kobject_put(esphosted_kobject);
+	// End CUSTOM
 #if TEST_RAW_TP
 	if (raw_tp_mode != 0) {
 		test_raw_tp_cleanup();
@@ -1145,6 +1228,10 @@ static void __exit esp_exit(void)
 	if (resetpin != HOST_GPIO_PIN_INVALID) {
 		gpio_free(resetpin);
 	}
+	
+	if (gpio9pin != HOST_GPIO_PIN_INVALID) {
+		gpio_free(gpio9pin);
+	}
 	debugfs_exit();
 }
 MODULE_LICENSE("GPL");
```

mantriyogesh, 2025-03-25:
> Previously with ESP32C3 in SPI mode we had no issues, but now with ESP32C6 + SDIO it's terrible

The SPI mode is always simple as there is no such PCB and stricter requirements. Although SPI also has limit in max throughput. If you use ESP32-C6 instead of ESP32-C3 for SPI, you would also feel it simple as well.

Please note SDIO such strict requirements are $${\color{orange}not\ ESP-Hosted specific requirement}$$. These requirements are for platform to be used correct. Noisy transport would lead to incorrect results. Also please note that the current issues you face are platform porting issues. Getting platform up and ready is not in the scope of ESP-Hosted as such, in fact it is $${\color{orange}pre-requisite}$$ for getting ESP-Hosted run over it. We understand that getting platform could be tricky some times. That is only reason we $${\color{yellow}help}$$ such issues raised.

It is always better to get the transport reliable and stable first, then get the ESP-hosted run over it.

mantriyogesh, 2025-03-25:
From your logs specially `probe of mmc1:0001:1 failed with error -110` it means the SDIO signal has integrity issues.

Be it ESP-Hosted or any other kernel driver for that sake would face the same set of issues.

mantriyogesh, 2025-03-25:
Please understand that platform porting and the Linux-based SDIO driver, including the device tree of your SoC, are within your control. Since we do not have expertise in your specific Linux platform, we are unable to provide direct guidance in that area.

As mentioned in [this comment](https://github.com/espressif/esp-hosted/issues/574#issuecomment-2750573480), software modifications should generally not be necessary. The driver is open source, allowing you to modify and extend it as needed. However, please note that we may not be able to provide support for custom modifications.

If the hardware is functioning correctly, there should be no inherent reason for the software to fail, based on our experience with other customers. Our recommendations are based on our past work with SDIO as a transport, but you are free to debug and customize as per your expertise.

In summary, ensuring that the transport is correctly configured is key to ESP-Hosted working as expected. If SDIO presents challenges, you might also consider trying SPI transport for the ESP32-C6, as it is generally simpler and involves fewer complexities compared to SDIO.

mantriyogesh, 2025-03-25:
> The driver (in SDIO mode) seems to start only the first time! So if we stop/resart the driver then it doesn't restart, we just get the two logs with 'probe of mmc1:0001:1 failed with error -110'

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md#14-dt-overlay 

[image]

mantriyogesh, 2025-03-25:
>  I try to setup KGDB and such stuff to debug the kernel

This is too far and it is not needed yet, IMHO.

mantriyogesh, 2025-03-25:
> If the driver control the reset, then the driver doesn't seems to start properly. (timing issue ? need a delay somewhere ?)

The issue description is helpful, but having logs would be even more valuable. Logs provide technical insights and allow us to understand the issue or scenario more clearly.

If you can provide a detailed issue description along with logs from both the host and ESP (preferably textual logs, including those from system boot-up), it would help us analyse the problem more effectively. This would also avoid the need to recapture the logs later.

caipiblack, 2025-03-25:
Hello,

I come back with some logs.

The logs present 4 situations/test cases:
1. The board is powered ON and the driver is started **without controling resetpin**: OK
2. The board is powered ON and the driver is started **without controling resetpin**, then it is re-started **without controling resetpin**: NOK
3. The board is powered ON and the driver is started **and controling resetpin**: NOK
4. The board is powered ON and the driver is started **without controling resetpin**, then we restart the board using `reboot -f` (brutal restart but keep power in), then when we start the driver **without controling resetpin**: NOK

## Informations about the test

This is the commands used to start/stop the driver:

```
modprobe esp_hosted_ng
modprobe -r esp_hosted_ng
```

When configured, the resetpin is configured like this:

```
echo "options esp_hosted_ng resetpin=506" > /etc/modprobe.d/esp_hosted_ng.conf
depmod -a
```

So when the driver is started with modprobe it takes the settings from this file.

## Tests results

1. When the 'resetpin' is controlled the startup of the driver directly fails, but the ESP32C6 is correctly resetted (as we can see the logs from bootloader):
```
# ESP32C6
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x3048
entry 0x4086c11a
....

# Host
[   83.346060] esp_hosted_ng: probe of mmc1:0001:1 failed with error -110
[   83.358055] esp_hosted_ng: probe of mmc1:0001:2 failed with error -110
```
2. When the 'resetpin' is not controlled, the startup of the driver works BUT if we have to stop/restart it without controlling the resetpin, then we can enter in "big troubles"

I think the main issue here is to understand **why when we control the resetpin the driver fails to start**.

- Is there a known timing issue ?

Another questions:

- What is the best way to update the firmware of the ESP32C6 ? Currently we stop the driver at runtime, then use GPIO to enter in bootloader and then we flash the ESP32C6 using espressif libs.
- Can you confirm that you don't have issues if you stop/restart the driver ?
- Do you have `non-removable` in your DTS ?

## Appendices

### Appendice 1

In the logs you will see some error with DMA: The message doesn't seems to be important:

1. It also appear when the driver works.
2. It seems to be "common" issue on RZG2UL: https://community.renesas.com/rz/f/rz-forum/18439/rzg2l-sdio-dma-issue

### Appendice 2

I am not sure that these messages are always to be considered like errors:

```
[   83.346060] esp_hosted_ng: probe of mmc1:0001:1 failed with error -110
[   83.358055] esp_hosted_ng: probe of mmc1:0001:2 failed with error -110
```

Because:
- It appear in many issues, even when working
- It also appear in the docs: [esp_hosted_fg/docs/Linux_based_host/SDIO_setup.md](esp_hosted_fg/docs/Linux_based_host/SDIO_setup.md) (but this is the doc of the **fg** instead of **ng**)

### Appendice 3

I am on the same commit: `21a5e735dc3660e4d4da7a201f9481d3d69fdab2` for host and esp32c6 firmware.

### Appendice 4

Our DTS looks like this currently:

```
&sdhi1 {
	pinctrl-names = "default";
	pinctrl-0 = <&sdio_esp32_pins>;
	bus-width = <4>;
	keep-power-in-suspend;
	non-removable;
	vmmc-supply = <&reg_3p3v>;
	vqmmc-supply = <&reg_1p8v>;
		
	broken-cd;

	max-frequency = <50000000>;
	status = "okay";
	
	esp32: esp32@1 {
		compatible = "espressif,esp_sdio";
		reg = <1>;
	};
};
```

broken-cd was already set previously, but after reading the kernel sources, I found that broken-cd is NOT parsed if `non-removable` is configured.

AND if I don't set `non-removable` then the driver will simply fails to start with NO LOG in host side (I mean, we also don't have the two lines `probe of mmc1:0001:1 failed with error`)

I also suspect something here

The documentation says:

```
Card detection:
If no property below is supplied, host native card detect is used.
Only one of the properties in this section should be supplied:
  - broken-cd: There is no card detection available; polling must be used.
  - cd-gpios: Specify GPIOs for card detection, see gpio binding
  - non-removable: non-removable slot (like eMMC); assume always present.
```

So my opinion is that the WiFi module is always present and non-removable should be set (what we do). But in the case where we need to stop/restart the driver, can it be a problem ?

In any cases, without `non-removable` it simply doesn't work with no logs when driver is started. (host + esp)

### Log files

[situation_1_esp32c6.txt](https://github.com/user-attachments/files/19451744/situation_1_esp32c6.txt)

[situation_1_host.txt](https://github.com/user-attachments/files/19451746/situation_1_host.txt)

[situation_2_esp32c6.txt](https://github.com/user-attachments/files/19451748/situation_2_esp32c6.txt)

[situation_2_host.txt](https://github.com/user-attachments/files/19451749/situation_2_host.txt)

[situation_3_esp32c6.txt](https://github.com/user-attachments/files/19451752/situation_3_esp32c6.txt)

[situation_3_host.txt](https://github.com/user-attachments/files/19451755/situation_3_host.txt)

[situation_4_esp32c6.txt](https://github.com/user-attachments/files/19451758/situation_4_esp32c6.txt)

[situation_4_host.txt](https://github.com/user-attachments/files/19451760/situation_4_host.txt)

Thanks for help

mantriyogesh, 2025-03-25:
I will take a look on these tomorrow morning,but anyway,

'without controling resetpin' doesn't make any sense. 

The resetpin is mandatory. On kernel module load or reload the reset pin should reset the slave. This is the reason it is called 'slave reset' gpio. It is not optional to reset the slave or not.

If the driver for some reason reloaded or crashed it would let the sdio trigger probe . The kernel sdio driver should be able to probe if such reset of slave (for example say slave had crashed and rebooted. Just stating a scenario for understanding). To get the driver auto probed is important. This driver probing is done by base Linux sdio driver. 'broken-cd' is not ESP32 kernel driver property , but the base sdio Linux driver property. If the base Linux driver doesn't identify the sdio being say unplugged and plugged, the higher level ESP kernel module would not have any control.

This is also same in raspberry pi, that is the reason we add `poll_once=off`. This is equivalent of `broken-cd` in raspberry pi. The driver should identify the sdio driver unplugged and re-plugged.
```
dtoverlay=sdio,poll_once=off
dtoverlay=disable-bt
```

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md#22-sdio-configuration -> Raspberry-Pi Setup

Anyway it is very late here, I can check the logs tomorrow morning.

mantriyogesh, 2025-03-25:
There is card detection pin also available and is set to be optional. If incase you wish to use it and you base Linux driver supports.

Check https://github.com/espressif/esp-hosted/blob/d0f54358be25974cd629747d7fd7f66e80968d1c/esp_hosted_ng/esp/esp_driver/network_adapter/main/app_main.c#L659

mantriyogesh, 2025-03-25:
The ESP should be hot pluggable. If the ESP for reason reset, while going down, Linux based sdio driver auto calls 'reset' registered:
https://github.com/espressif/esp-hosted/blob/d0f54358be25974cd629747d7fd7f66e80968d1c/esp_hosted_ng/host/sdio/esp_sdio.c#L927

And once slave again sets sdio slave driver exposed, Linux based sdio driver should auto detect and call 'probe' registered automatically:
https://github.com/espressif/esp-hosted/blob/d0f54358be25974cd629747d7fd7f66e80968d1c/esp_hosted_ng/host/sdio/esp_sdio.c#L926

These things should work reliably. For our raspberry pi case it works smoothly. For your sdio driver also,  this should happen.

If base driver fails to do this then either base Linux driver has limited support and you might need to check other ways like card detection above.

mantriyogesh, 2025-03-26:
Situation 1)

Crash is in the SDHCI driver of renesas. 

```
[   79.734072]  dma_direct_map_sg+0x268/0x2a8
[   79.738157]  dma_map_sg_attrs+0xb8/0xc4
[   79.741986]  renesas_sdhi_internal_dmac_map+0x54/0xb8
[   79.747022]  renesas_sdhi_internal_dmac_pre_req+0x24/0x34
[   79.752406]  mmc_io_rw_extended+0x208/0x2d8
[   79.756577]  sdio_io_rw_ext_helper+0x164/0x1f0
[   79.761007]  sdio_memcpy_fromio+0x20/0x2c
[   79.765032]  esp_read_multi_byte+0xb0/0xc4 [esp_hosted_ng]
[   79.770505]  esp_read_reg+0x24/0x3c [esp_hosted_ng]
[   79.775371]  monitor_process+0x94/0x148 [esp_hosted_ng]
[   79.780585]  kthread+0x160/0x164
[   79.783806]  ret_from_fork+0x10/0x30
```

However, It looks like you have enabled 'CONFIG_ENABLE_MONITOR_PROCESS'. Please do not use this thread.

This leg is not tested in recent times. anyway, Makefile by default doesn't enable this check:
https://github.com/espressif/esp-hosted/blob/d0f54358be25974cd629747d7fd7f66e80968d1c/esp_hosted_ng/host/Makefile#L1

Do not enable `CONFIG_ENABLE_MONITOR_PROCESS`. We will try to remove in upcoming changes.

mantriyogesh, 2025-03-26:
Scenario 2)

Again the crash in the sdhci. because monitor process and actual writer compete? or something else? The crash is in base linux driver of sdio. possibly because DMA not configured correctly or something else? unsure.

[image]

mantriyogesh, 2025-03-26:
## Reset pin

> When the 'resetpin' is not controlled, the startup of the driver works BUT if we have to stop/restart it without controlling the resetpin, then we can enter in "big troubles"

Reset pin is mandatory. The timing and noise level on SDIO traces vary per design. You are having the schematic of ours, made by experts in espressif. May be comparison would help.

## SDIO Timing
The timing of SDIO is crucial to be the same at host and slave. $${\color{yellow} \text{what timing is used at host through device tree.}}$$ The slave timing could be adjusted at 
https://github.com/espressif/esp-hosted/blob/d0f54358be25974cd629747d7fd7f66e80968d1c/esp_hosted_ng/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L119-L126

you can change from
```diff
--         .timing             = SDIO_SLAVE_TIMING_NSEND_PSAMPLE,
++         .timing             = SDIO_SLAVE_TIMING_PSEND_PSAMPLE,
```

In general, we suggest to use same sdio timing at both places, but some times, these may vary due to the sdio trace lengths.

By the way, this timing adjustments also briefed as part of porting guide [here](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md#242-sdio).

mantriyogesh, 2025-03-26:
[image]

mantriyogesh, 2025-03-26:
Porting plays a crucial role in this process, as no two Linux architectures and their base drivers are exactly the same.

In some cases, the same SDIO device tree configuration name may be interpreted differently across different Linux systems.

While we aim to maintain default settings in a way that ensures compatibility across multiple Linux platforms, variations can still exist.

Since porting the transport layer is a prerequisite, we kindly ask for understanding when raising questions. Issues related to porting or base drivers should be considered carefully rather than solely attributing non-functionality to the ESP-Hosted driver.

mantriyogesh, 2025-03-26:
Scenario 3)

## Hot pluggable

We understand esp software can crash in rare cases. therefore , we treat it as hot pluggable. afterall, it is not SDMMC driver and SDIO (communication) driver. the SD MMC could be treated as `non-removable` . SDIO should not be confused. it should $${\color{yellow} \text{not}}$$ be `non-removable`

## -110 on Function 1

Any error of -110 on Function 2 should be ignored as it is unused.
```
[   83.346060] esp_hosted_ng: probe of mmc1:0001:1 failed with error -110
```
Please be informed, this is $${\color{yellow} \text{TIMEOUT}}$$ case by the SDIO communication. The error code cross-reference: https://elixir.bootlin.com/u-boot/v2024.10/source/include/linux/errno.h#L118

If logs like these happen, $${\color{yellow} \text{The probing is already failed or timed out by the ESP kernel driver could operate}}$$ hence and control given back to the Linux base SDIO driver.

In our observations, -110 is the main outcome when the SDIO experience a lot of noise. I am no expert of SDIO. But from experience, I deduce that external pull-ups or proper ground planes were not introduced in the PCB. Repeating myself, you have our reference SDIO schematic with you, to compare yourself. If you want your schematic to be reviewed by Espressif, you can also do so [here](https://github.com/espressif/esp-hosted/issues/574#issuecomment-2739154356)

Any case, if the -110 or time out happened, the slave still exposes the SDIO slave driver, which is retry to detect by your base Linux SDIO driver would get detected in second time.

Reasons for -110 could be:
1. insufficient pull-ups
2. pull-ups not applied on expected lines
3. Power supply on SDIO pins when connected is < 3v3
4. High trace lengths
5. unequal trace lengths for sdio lines
6. base linux sdio driver itself busy
7. SDIO [timing mismatch](https://github.com/espressif/esp-hosted/issues/574#issuecomment-2753271526) in between host and slave.

$${\color{yellow} \text{So even if you experience -110 one time, there is almost 100% chance that Linux would identify again and probe and then again control would be passed to esp32 kernel module, to probe. }}$$

## Polling of SDIO and probing

Despite the one failed attempt the base SDIO driver can still retrigger detection if 
```
broken-cd;
```

is supported by your SDIO driver.

You can also $${\color{yellow} \text{optionally}}$$ enable [card detection](https://github.com/espressif/esp-hosted/issues/574#issuecomment-2751940003) if in case broken-cd is not supported or failed.

scenario 4)
```
[  120.705907] Call trace:
[  120.708350]  dma_direct_map_sg+0x268/0x2a8
[  120.712436]  dma_map_sg_attrs+0xb8/0xc4
[  120.716266]  renesas_sdhi_internal_dmac_map+0x54/0xb8
[  120.721301]  renesas_sdhi_internal_dmac_pre_req+0x24/0x34
[  120.726685]  mmc_io_rw_extended+0x208/0x2d8
[  120.730856]  sdio_io_rw_ext_helper+0x164/0x1f0
[  120.735286]  sdio_memcpy_fromio+0x20/0x2c
[  120.739311]  esp_read_multi_byte+0xb0/0xc4 [esp_hosted_ng]
[  120.744783]  esp_read_reg+0x24/0x3c [esp_hosted_ng]
[  120.749649]  monitor_process+0x94/0x148 [esp_hosted_ng]
[  120.754862]  kthread+0x160/0x164
[  120.758084]  ret_from_fork+0x10/0x30

```
-> https://github.com/espressif/esp-hosted/issues/574#issuecomment-2753240894

```
[  120.789777] esp_hosted_ng: tx_process: not ready
[  120.813788] esp_hosted_ng: tx_process: not ready
[  120.837827] esp_hosted_ng: tx_process: not ready
[  120.861788] esp_hosted_ng: tx_process: not ready
[  120.885804] esp_hosted_ng: tx_process: not ready
[  120.909788] esp_hosted_ng: tx_process: not ready
[  120.933786] esp_hosted_ng: tx_process: not ready
[  120.957785] esp_hosted_ng: tx_process: not ready
[  120.981821] esp_hosted_ng: tx_process: not ready
[  121.005788] esp_hosted_ng: tx_process: not ready
[  121.029820] esp_hosted_ng: tx_process: not ready
...
```

These logs are annoying, no doubt about it. I have reduced the log level so that these would not pop-up with default settings.

[image]

This would take a day or two to get merged. However, this log simply means that the SDIO driver is not able to progress ahead and probing was successful but still we did not receive expected boot up event from slave. monitor_process keeping busy should never be used. We have removed this code already in [FG SDIO host code](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c). We will remove this part from NG code as well.

bijucdas, 2025-03-26:
Have you checked SD interrupts? It could be you got lot of IRQS and the dma is failing.

mantriyogesh, 2025-03-27:
@caipiblack have you got rid of the monitor_process() already? `monitor_process` code is already $${\color{yellow} \text{removed}}$$ from latest master.

Possibly this code is linked to @bijucdas comment above..

caipiblack, 2025-03-27:
> [@caipiblack](https://github.com/caipiblack) have you got rid of the monitor_process() already? `monitor_process` code is already removed from latest master.
> 
> Possibly this code is linked to [@bijucdas](https://github.com/bijucdas) comment above..

Hello,

Yes, Thanks,  if I disable the monitor_process, then the DMA error disapear.

Then I passed my day debuging the 'reset issue' with @bijucdas  on a renesas discussion.

1. The `non-removable` presence or not in the DTS was a big debate as the ESP32C6 is always present (like an emmc) so it "should be" set in the DTS. But if I understand, your opinion is to remove `non-removable` from the DTS ?  (currently If I remove it, then it never works, this is another thing to investigate so)

2. But we focus all the day on the reset issue, i'm going to post the experiments with the result just after, but I'm going to continue debuging this today

caipiblack, 2025-03-27:
# Descriptions of the test

The objective is to see the difference between these two cases:
- When the driver (esp_hosted_ng) control the reset pin of ESP32C6 on startup
- When the driver (esp_hosted_ng) don't control the reset pin of ESP32C6 on startup

We are using the same DTS during the test:

```
/* ESP32-C6-MINI-1-N4 - WiFi - SDIO */
&sdhi1 {
	pinctrl-names = "default";
	pinctrl-0 = <&sdio_esp32_pins>;
	bus-width = <4>;
	keep-power-in-suspend;
	non-removable;
	vmmc-supply = <&reg_3p3v>;
	vqmmc-supply = <&reg_1p8v>;
		
	broken-cd;

	max-frequency = <50000000>;
	status = "okay";
	
	esp32: esp32@1 {
		compatible = "espressif,esp_sdio";
		reg = <1>;
	};
};
```

## When the driver (esp_hosted_ng) control the reset

1. After power OFF and power ON of the board, I do these commands:

```
root@board:~# cat /proc/interrupts | grep 11c10000.mmc
 95:        198     GICv3 138 Level     11c10000.mmc
 96:          0     GICv3 139 Level     11c10000.mmc
 
root@board:~# cat /proc/interrupts | grep 11c10000.mmc
 95:        198     GICv3 138 Level     11c10000.mmc
 96:          0     GICv3 139 Level     11c10000.mmc
 
root@board:~# cat /sys/kernel/debug/mmc1/ios
clock:		25000000 Hz
actual clock:	100000000 Hz
vdd:		21 (3.3 ~ 3.4 V)
bus mode:	2 (push-pull)
chip select:	0 (don't care)
power mode:	2 (on)
bus width:	2 (4 bits)
timing spec:	0 (legacy)
signal voltage:	0 (3.30 V)
driver type:	0 (driver type B)

root@board:~# cat /sys/devices/platform/soc/11c10000.mmc/mmc_host/mmc
1/mmc1\:0001/mmc1\:0001\:1/uevent 
OF_NAME=esp32
OF_FULLNAME=/soc/mmc@11c10000/esp32@1
OF_COMPATIBLE_0=espressif,esp_sdio
OF_COMPATIBLE_N=1
SDIO_CLASS=00
SDIO_ID=0092:6666
SDIO_REVISION=0.0
MODALIAS=sdio:c00v0092d6666
```

2. Use minicom on another SSH to display the logs of ESP32C6 (`/dev/ttySC2`):

[... 331 lines cut ...]
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1280  Metric:1
          RX packets:0 errors:0 dropped:1 overruns:0 frame:0
          TX packets:26 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500 
          RX bytes:0 (0.0 B)  TX bytes:3278 (3.2 KiB)
```

7. At this point I get put interface up and restart wpa_supplicant, then it successfully connects to AP:
```
root@board:~# systemctl restart wpa_supplicant
root@board:~# 
[  277.667619] esp_hosted_ng: cmd_auth_request: Authentication request: 60:38:e0:cd:92:69 8 0 0 0
[  277.921866] esp_hosted_ng: cmd_assoc_request: Association request: 60:38:e0:cd:92:69 8 37
[  278.015925] esp_hosted_ng: process_assoc_event: Connection status: 0
[  278.219537] IPv4: martian source 192.168.3.116 from 192.168.3.1, on dev wlan0
[  278.226772] ll header: 00000000: 9c 9e 6e 7a 04 48 60 38 e0 cd 92 68 08 00
[  278.239310] IPv4: martian source 192.168.3.116 from 192.168.3.1, on dev wlan0
[  278.246573] ll header: 00000000: 9c 9e 6e 7a 04 48 60 38 e0 cd 92 68 08 00
[  278.284819] esp_hosted_ng: esp_inetaddr_event: NETDEV_UP interface wlan0 ip changed to  192.168.003.116
```
At this point the driver is started succefully, I have internet connection

# Results

- This test confirm that when the driver control the RESET pin of the ESP32, then 'something happens' that makes troubles.

caipiblack, 2025-03-27:
# Descriptions of the test

14:14 < biju> Test1: use sysfs GPIO to make card in reset state and do modprobe
14:15 < biju> Test2: use sysfs GPIO ti make the card out of reset and then do modprobe
14:19 < biju> set "resetpin=-1 for both tests

## test1 : Set RESET line to LOW and modprobe

1. After power OFF and power ON of the board, I do these commands:

```
root@board:~# echo "options esp_hosted_ng resetpin=-1" > /etc/modprobe.d/esp_hosted_ng.conf
root@board:~# depmod -a
root@board:~# cat /etc/modprobe.d/esp_hosted_ng.conf
```
2. Set the reset pin with sysfs to LOW

```
echo 506 > /sys/class/gpio/export
echo out > /sys/class/gpio/P18_2/direction
echo 0 > /sys/class/gpio/P18_2/value
```

3. Get the logs of ESP32C6 using minicom (should be empty as ESP32 is reseted):

```
```

Logs are empty

4. Start the driver:

```
root@board:~# modprobe esp_hosted_ng
[  179.594958] esp_hosted_ng: probe of mmc1:0001:1 failed with error -110
[  179.601769] esp_hosted_ng: probe of mmc1:0001:2 failed with error -110
```

## test2 : Set RESET line to HIGH (after being low) and modprobe

1. After power OFF and power ON of the board, I do these commands:

```
root@board:~# echo "options esp_hosted_ng resetpin=-1" > /etc/modprobe.d/esp_hosted_ng.conf
root@board:~# depmod -a
root@board:~# cat /etc/modprobe.d/esp_hosted_ng.conf
```
2. Set the reset pin with sysfs to LOW

```
echo 506 > /sys/class/gpio/export
echo out > /sys/class/gpio/P18_2/direction
echo 0 > /sys/class/gpio/P18_2/value
sleep 0.5
echo 1 > /sys/class/gpio/P18_2/value
```

3. Get the logs of ESP32C6 using minicom (after reset):

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x3048
entry 0x4086c11a
I (23) boot: ESP-IDF 67c1de1e-dirty 2nd stage bootloader
[... 91 lines cut ...]
I (724) FW_MAIN: ESP Bluetooth MAC addr: 9c-9e-6e-7a- 4-4a

I (725) FW_SDIO_SLAVE: Using SDIO interface
I (728) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
```

4. Start the driver:

```
root@board:~# modprobe esp_hosted_ng
[  179.854111] esp_hosted_ng: probe of mmc1:0001:1 failed with error -110
[  179.860892] esp_hosted_ng: probe of mmc1:0001:2 failed with error -110
```

5. Get the logs of ESP32C6 using minicom (after modprobe):

```
```

Logs are empty

# Results

- The problem is not related to the fact that the driver control the reset pin of ESP32, but in general if the ESP32 is reseted then it makes troubles
- Also we have checked that after reset the ESP32 is starting and the RESET line stay HIGH

caipiblack, 2025-03-27:
# Descriptions of the test

Put logs in the `esp_hosted_ng` driver AND on the linux kernel side.

# Results

In `esp_hosted_ng`:

1. The function `static int __init esp_init(void)` is called when I call `modprobe esp_hosted_ng` and it succeed, (return 0)
2. During this step, it calls `sdio_register_driver()` that also succeed. This function is implemented in [drivers/mmc/core/sdio_bus.c](https://github.com/torvalds/linux/blob/1a9239bb4253f9076b5b4b2a1a4e8d7defd77a95/drivers/mmc/core/sdio_bus.c#L272)
3. This is after this function call that we see the two messages:
```
[  428.660605] esp_hosted_ng: probe of mmc1:0001:1 failed with error -110
[  428.667830] esp_hosted_ng: probe of mmc1:0001:2 failed with error -110
```
4. The `-110` error code comes when [drivers/mmc/core/sdio_bus.c:sdio_bus_probe()](https://github.com/torvalds/linux/blob/1a9239bb4253f9076b5b4b2a1a4e8d7defd77a95/drivers/mmc/core/sdio_bus.c#L153) calls [drivers/mmc/core/sdio_bus.c:sdio_set_block_size(func, 0)](https://github.com/torvalds/linux/blob/1a9239bb4253f9076b5b4b2a1a4e8d7defd77a95/drivers/mmc/core/sdio_bus.c#L188) 
5. The `-110` error because we try to RW on SDIO in one of the functions of [drivers/mmc/core/sdio_bus.c:sdio_set_block_size()](https://github.com/torvalds/linux/blob/1a9239bb4253f9076b5b4b2a1a4e8d7defd77a95/drivers/mmc/core/sdio_io.c#L159) and it seems to have no responses

caipiblack, 2025-03-27:
> This is the final step. We are very far yet. I hope you understand what I mean. Once everything else is fine, only then wlan0 would come up. It would never be there if your communication is not done complete both ways.

I know.

> As you do modprobe, i safely assume that you have already integrated kernel module along with kernel.

Yes

> Considering these options placed, when you modprobe, does the ESP reboot? Your minicom log would show the full log from bootup again if esp is restarted. Please confirm if resetpin does reset the esp or not.

- When the esp_hosted_mod module control the reset (resetpin=506): yes the RESET pin is pulled LOW and HIGH and then stay HIGH (checked with scope)
- When I do it from the sysfs, yes the RESET pin is pulled LOW and HIGH and then stay HIGH (checked with scope)
- Yes the ESP32 is reseted and restarted.
- The logs with minicom show the "full restart" (also bootloader), because the esp32 restart.

> unless you follow complete documentation steps, you would not automatically get wlan0 up in this case. please check https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng#3-get-started again (only when you already got ahead of Start Data Path.

- Yes I know how to configure the WiFi interface. 
- Remember: When the driver don't control the reset (which I know it's not a solution and that is why I am debuging it) I arrive to get working the WiFi, get an ip by dhcp, access to the internet through this interface, so ping 8.8.8.8).
- I know that not controlling the reset **is not a solution** due to many factors (like if we need to restart the driver), **so that's why I am debugging it**.

> The code should automatically handle this at

Yes, I know. The test was just to check if we control the reset from outside the driver, did we have the same result. The response is yes.

So, if my board is started (linux started) then we restart the esp32 (playing with reset line from sysfs or esp_hosted_ng resetpin) then the drivers (sdio_io.c in kernel) can't communicate with it.

> As stated earlier, please allow your base SDIO driver to retry and re-enumerate if it failed first time. broken-cd does work for this.

> sdio is not emmc. It should always be hot pluggable.
> If it is set to non-removable, sdio linux base driver would even bother to retry to probe.

There are many issues in my setup:
1. If ESP32 is reset then the driver can't communicate with it (it's not a delay issues, adding delay after reset don't help)
2. I have 'non-removable' in DTS and if I remove it then the driver can't communicate with the ESP32 in anycases (with or without reset)

I have the two things to investigate and I put the priority on the first problem. Then I will debug the "non-removable" issue, to understand WHY , if I remove it, then it NEVER works

Thanks for help.

------

Also I have identified which function call fails in [drivers/mmc/core/sdio_bus.c:sdio_set_block_size()](https://github.com/torvalds/linux/blob/1a9239bb4253f9076b5b4b2a1a4e8d7defd77a95/drivers/mmc/core/sdio_io.c#L159):

```
	ret = mmc_io_rw_direct(func->card, 1, 0,
		SDIO_FBR_BASE(func->num) + SDIO_FBR_BLKSIZE,
		blksz & 0xff, NULL);
```

Now it's scope time.

mantriyogesh, 2025-03-27:
Once you encountered -110, and base driver doesn't re enumerate, the slave would just wait without doing any useful. 

I think you understand all the techniques now. 
Yes score would actually help, Good that you have scope with you. So is it logic analyser?

Yes I did not understand yet why (1) is causing problems. As you have done a lot of trials, would it be too much if I ask you to evaluate the FG latest master driver on both sides? Just in case if it works we would have some conclusion.

If it fails same way , the logic analyser is better way to understand why it is failing exactly

mantriyogesh, 2025-03-27:
For reloading (and resting), you can simply 
``
sudo rmmod esp_hosted_ng
sudo modprobe esp_hosted_ng
```

To save time. Just suggestion..

mantriyogesh, 2025-03-27:
I missed in our discussion, have you also checked the voltage levels on the sdio lines?

caipiblack, 2025-03-28:
Hello,

Yes we have a scope, it's not a logic analyser, we have to decode each packets by hands to understand, it's really time consuming. 

The scope can decode spi but it sucks so we do it "by hand"

I'm going to post the investigation of yesterday, but today it's going to continue!

caipiblack, 2025-03-28:
# Description

The idea is to capture with a scope the CLK, CMD and DATA0 lines of SDIO during the kernel bootup and then when the esp_hosted_ng driver is started (`modprobe esp_hosted_ng`).

Then the idea is to extract AND decode each SDIO packets to have a clear sequence of "what is exchanged" in each situations.

Just to remember, we have two issues:
1. If we **restart** the driver (with or without reseting the ESP32) **it never works**, we think it can be related to the presence of 'non-removable' stuff.
2. If we start the driver without reseting the ESP32 after power ON (and full boot) it works (internet access through the WIFI interface). But if the ESP32 is reseted (after bootup) (by the esp_hosted_ng module or from sysfs) then it wont work.

Currently we are debugging the SECOND issue, so the idea is to analyse the traces of the TWO situations (with/without reset)!

## Note

For the tests, as we introduce wires to our board, we need to reduce the speed of SDIO lines: 

```
echo "1000000" > /sys/kernel/debug/mmc1/clock
```

## SDIO Packets capture when the reset IS NOT CONTROLLED

1. During the Kernel Bootup sequence (so before probing esp_hosted_ng) there are A LOT of data exchanged in the SDIO bus between HOST<>CARD, these packets has not been analysed YET

2. Once the system is bootup, we execute the command `modprobe esp_hosted_ng` and this is what we see on SDIO lines:

```
# CMD Line: 0 1 110100 1 000 0 0 00000000100010000 0 00000000 1011111 1
host -> card : IO_RW_DIRECT, Write on Common I/O area (CIA), Reg 0x110 (I/O block size for Function 1 - Byte[0]), Data: 0
# CMD Line: 0 0 110100 0000000000000000 00010000 00000000 0011011 1
card --> host: IO_RW_DIRECT, Response Flag: [CRC: NoError, IllegalCommand: NoError, IOCurrentState: CMD, Error: NoError, FunctionNumber: NoError, OutOfRange: NoError], Data that has been written: 0x00

...80us later...

# CMD Line: 0 1 110100 1 000 0 0 00000000100010001 0 00000010 1011011 1
host -> card : IO_RW_DIRECT, Write on Common I/O area (CIA), Reg 0x111 (I/O block size for Function 1 - Byte[1]), Data: 2
# CMD Line: 0 0 110100 0000000000000000 00010000 00000010 0001001 1
card --> host: IO_RW_DIRECT Response, Response Flag: [CRC: NoError, IllegalCommand: NoError, IOCurrentState: CMD, Error: NoError, FunctionNumber: NoError, OutOfRange: NoError], Data that has been written: 0x02

...60ms later...

# CMD Line: 0 1 110100 0 000 0 0 00000000000000010 0 00000000 1000100 1
host -> card : IO_RW_DIRECT, Read on Common I/O area (CIA), Reg 0x02 (Card Common Control Registers (CCCR) (I/O Enable))
# CMD Line: 0 0 110100 0000000000000000 00010000 00000000 0011011 1
card --> host: IO_RW_DIRECT Response, Response Flag: [CRC: NoError, IllegalCommand: NoError, IOCurrentState: CMD, Error: NoError, FunctionNumber: NoError, OutOfRange: NoError], Data Read: 0x00

...90us later...

# CMD Line: 0 1 110100 1 000 0 0 00000000000000010 0 00000010 1001101 1
host -> card : IO_RW_DIRECT, Write on Common I/O area (CIA), Reg 0x02 (Card Common Control Registers (CCCR) (I/O Enable)), Data: 0x02 (enable IOE1)
# CMD Line: 0 0 110100 0000000000000000 00010000 00000010 0001001 1
card --> host: IO_RW_DIRECT, Response Flag: [CRC: NoError, IllegalCommand: NoError, IOCurrentState: CMD, Error: NoError, FunctionNumber: NoError, OutOfRange: NoError], Data that has been written: 0x02

...60us later...

# CMD Line: 0 1 110100 0 000 0 0 00000000000000011 0 00000011 1111111 1
# WARNING: Data field contains 0x3, on a READ request it should be 0 according to specifications
host -> card : IO_RW_DIRECT, Read on Common I/O area (CIA), Reg 0x03 (Card Common Control Registers (CCCR) (I/O Ready))
# CMD Line: 0 0 110100 0000000000000000 00010000 00000010 0001001 1
card --> host: IO_RW_DIRECT, Response Flag: [CRC: NoError, IllegalCommand: NoError, IOCurrentState: CMD, Error: NoError, FunctionNumber: NoError, OutOfRange: NoError], Data Read: 0x02

...60us later...

# CMD Line: 0 1 110100 0 000 0 0 00000000000000100 0 00000000 0110000 1
host -> card : IO_RW_DIRECT, Read on Common I/O area (CIA), Reg 0x04 (Card Common Control Registers (CCCR) (Int Enable)), Data: 0x00
# CMD Line: 0 0 110100 0000000000000000 00010000 00000000 0011011 1
card --> host: IO_RW_DIRECT, Response Flag: [CRC: NoError, IllegalCommand: NoError, IOCurrentState: CMD, Error: NoError, FunctionNumber: NoError, OutOfRange: NoError], Data Read: 0x00

...60us later...

[... 52 lines cut ...]
**I am not saying that the problem is in the ESP32**.

At this point, what we think is that all **the messages exchanged during the kernel bootup** (that we don't analysed yet because it is REALLY HUGE) **are really needed for the ESP32C6 to work** AND when we reset the ESP32C6 he loose some of the configuration set by this "huge exchanges".

So at this point:
1. It should be nice to know if on a system where the reset works, this "huge messages" are only exchanged during the bootup sequence OR if the system retransmit all these messages once the ESP32C6 is restarted.
2. I am going to analyse the content of these huge messages (or only the begining..), but it's going to be really difficult, because there are really a LOT of messages exchanged during this phase, and this is not emited by esp_hosted_ng driver but instead some low level driver.

@bijucdas : Do you have an idea about which driver is doing the "massive exchanges" during the bootup sequence ? (that doesn't appear in my sequence diagrams or analyse as I don't decode these packets yet). And did you know [... cut ...]
Just to give an idea about how a packet looks like when zoomed:

[image]

BLUE: CLK
Yellow: CMD
Pink: DATA0 but only has data in the last requests (the request from esp_hosted_ng driver) but we don't really analyse the content of this.

Just to give an idea about "huge packets" exchanged during the kernel bootup sequence:

[image]

This capture is really unzoomed, so it looks trashy. But it's just to give an idea of this sequence that we want to analyse now.. Each small lines contains many packets once zoomed in..

caipiblack, 2025-03-31:
Hello,

I think I understand what happens.

1. According to the specifications (`PartE1_SDIO_Simplified_Specification_Ver3.00.pdf`), an SDIO device SHOULD NOT respond to any request until it receive the CMD0/CMD5
2. On my board, these commands are only send when the kernel starts. This is why when I start the driver for the 'first time' it works
3. When we reset the ESP32, then as we don't transmit the CMD0/CMD5 then it doesn't respond to any other request. That's why the probe in SDIO Bus driver fails (it tries to set the block size)

I think it is not the responsability to esp_hosted_ng to transmit these commands, because it can't access the SDIO bus has the SDIO Bus driver is failing when trying to set the BlockSize to 0.

So it should be something low level, possibly related to non-removable / broken-cd stuff.

But without non-removable, there is NO DATA or CLOCK on SDIO lines (at kernel bootup or after), so the esp32 is simply no detected

So I have to find out, why 'non-removable' or 'broken-cd'  is broken

caipiblack, 2025-03-31:
Finally I have tested to configure the SD1_CD pin of RZG2UL as function 1 (SD1_CD).

Then, as this pin was connected to nothing,  I had to play with pull up/down to simulate a CD signal:

```
devmem2 0x11031C80 w 0x00010000
devmem2 0x11031C80 w 0x00020000
```

Like this, I am now able to stop/restart the ESP32/driver and re-trigger a SDIO detection (playing with CD signal).

So I think something is broken or not implemented around the broken-cd (auto detection) of SDIO devices in RZG2UL SDIO drivers or the kernel (Yocto Hardknott) (or misconfigured).

At this point, we can say the problem is solved, on our next design we will connect the CD pin of RZG2UL to ESP32C6.

It should be nice if someone add a note about CD pin in [setup guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md)