espressif/esp-hosted #575: esp-hosted-ng问题

VegardCheng, 2025-02-26:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

请问esp-hosted-ng支持的linux内核版本是？支持树莓派最新的系统吗？测试使用的是哪个版本的OS？

mantriyogesh, 2025-02-26:
Yes, latest Raspberry Pi OS is supported with ESP-Hosted.
Let us know if you face any issues.

VegardCheng, 2025-03-17:
[image]
请问这样是装好了吗？但是找不到wlan0，还需要设置哪里？

mantriyogesh, 2025-03-17:
Check the dmesg in the linux.

The sdio should be enabled as per documentation.
1. Ensure that the $${\color{orange}external\ pull-ups}$$ are added with $${\color{yellow}expected\ values}$$ and on  $${\color{yellow}expected\ lines}$$
2. Do  $${\color{orange}not\ use\ jumper\ wires}$$ to connect SDIO lines. Always design and use PCB for SDIO. SDIO becomes a lot noisy and hard when using jumper wires.
3. [Troubleshoot guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) for some common issues
4. $${\color{yellow}\/boot\/config.txt}$$ or $${\color{yellow}\/boot\/firmware\/config.txt}$$ is correctly configured as per documentation
5. If you follow (1) & (2), send us camera photo showing the connections.
6. If you wish to use jumper cables, strictly use SPI only.

VegardCheng, 2025-03-18:
[image]
我用的是ESP-WROOM-32模块，执行sudo bash rpi_init.sh sdio后显示上图的错误

VegardCheng, 2025-03-18:
[image]
还有这里应该没错吧

mantriyogesh, 2025-03-18:
1. Send us the camera photo of connections
2. Try to erase the flash at ESP32. As you are using ESP32, please check the extra steps of $${\color{yellow}efuse\ burning}$$ in addition to the $${\color{yellow}external\ pull-up\ (>=10kOhms)}$$  on D0-D3, CMD lines. PCB only. $${\color{yellow}Strictly\ No\ jumper\ cables\ For\ SDIO}$$ 

eFuse burning and external pull-ups explained [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html)

VegardCheng, 2025-03-18:
[image]

[image]
我做了这样一块底板连接树莓派CM4 IO
使用的是CM4（带EMMC，没有WIFI）

[image]

擦除重新烧录还是一样

mantriyogesh, 2025-03-18:
PCB looks fine. https://pinout.xyz/pinout/sdio.

Although this is not the current issue, I see that the input power looks to be shared in between Raspberri Pi and ESP.
Please check the minimal power rating for your Raspberry Pi and ensure the input voltage and current ratings are supported by both adapter and the power cable used.

If you are mounting the esp dev kit, ensure as many grounds are short with raspberry pi ground. Can you power the esp individually?

If ESP can be disconnected from raspberry pi (take it out from pcb) and flash individually, still this error observed?

VegardCheng, 2025-03-18:
[image]
单独上电显示的是这些信息

mantriyogesh, 2025-03-18:
Please ensure you use $${\color{yellow}latest\ master}$$ at $${\color{orange}both}$$ esp and host both sides. $${\color{orange}Strictly\ exact\ same\ commit.}$$
If the esp is able to boot individually but not when connected, some issue in the connections might be there.

Check if raspberry pi pin 31 (reset/EN) pin is connected correctly and not floating.

I also note $${\color{red}Brownout\ detector\ was\ triggered}$$ in ESP log. This circuitry is not equipped with sufficient power.
Power ESP and Raspberry Pi individually.

$${\color{yellow}The\ input\ power\ rating}$$ for $${\color{yellow}both}$$ devices must be abided. Otherwise, you would crash or extremely low performance on the SDIO lines.

Also check $${\color{yellow}power\ available\ on\ SDIO}$$ lines after connection to raspberry pi. If it is less than 3v3, it would not work.

$${\color{yellow}eFuse}$$ burning might be required at ESP32. [eFuse burning](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html)

VegardCheng, 2025-03-18:
windows有配置eFuse的软件吗？

mantriyogesh, 2025-03-18:
You do not need extra software. ESP-IDF would be already available at your side. use espfuse.py from idf as mentionedin the document., https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#conflicts-between-bootstrap-and-sdio-on-dat2

VegardCheng, 2025-03-19:
[image]
执行espefuse.py set_flash_voltage 3.3V后就一直没反应

VegardCheng, 2025-03-19:
[image]
这样是设置好了？

VegardCheng, 2025-03-19:
[image]
多了一句set up done，这是初始化好了？

VegardCheng, 2025-03-19:
[image]

[image]
不知道还有哪里有问题

mantriyogesh, 2025-03-19:
> 多了一句set up done，这是初始化好了？

Means slave SDIO init is done and init event is queued to SDIO slave driver. Now SDIO host it would wait for sdio transfer.

Can you send the below things?
Please do not attach pictures. Please attach the text logs only.

1. /boot/firmware/config.txt
2. host dmesg log from system bootup. Need to see sdio enumeration done or not
3.Are you using slave and host both from master? Same commit need to be used at slave and host for building and flashing/loading

mantriyogesh, 2025-03-19:
3. Did you verify the voltage at all the SDIO pins ?
4. Are there still brownout issues detected at slave?

VegardCheng, 2025-03-19:
> 4. 您是否验证了所有 SDIO 引脚上的电压？
> 5. 在 slave 上是否仍检测到掉电问题？

那些上拉的IO都没问题了

mantriyogesh, 2025-03-19:
If the voltages are fine, then we can fetch the textual full logs at linux.
Although current issue seem to be on linux front, it would be better if we fetch `full logs` from boot up at esp and host.
```
git log --oneline |head -3
```
captured both sides.

it clearly seen that sdio hardware is detected and probed at linux fine. but post probing the linux host is not sending the first event to slave (unsure, can check only in the logs)

seems stuck at https://github.com/espressif/esp-hosted/blob/513bdfa93f5dda86d828767e5e55def718443911/esp_hosted_ng/host/sdio/esp_sdio.c#L574

because the first H->S event did not reach slave yet?
https://github.com/espressif/esp-hosted/blob/513bdfa93f5dda86d828767e5e55def718443911/esp_hosted_ng/host/sdio/esp_sdio.c#L825

Upon receiving this message, slave would print:

https://github.com/espressif/esp-hosted/blob/513bdfa93f5dda86d828767e5e55def718443911/esp_hosted_ng/esp/esp_driver/network_adapter/main/app_main.c#L611

If this message is not printed at slave, still the sdio hardware needs to be checked again.

VegardCheng, 2025-03-19:
[image]

mantriyogesh, 2025-03-19:
both sides

mantriyogesh, 2025-03-19:
https://github.com/espressif/esp-hosted/issues/575#issuecomment-2735257335

https://github.com/espressif/esp-hosted/issues/575#issuecomment-2735259797

mantriyogesh, 2025-03-19:
unless full textual logs, we would not understand the issue.

VegardCheng, 2025-03-19:
[messages.txt](https://github.com/user-attachments/files/19335772/messages.txt)

VegardCheng, 2025-03-19:
[新建文本文档 (2).txt](https://github.com/user-attachments/files/19336058/2.txt)
我截取了最后一部分

mantriyogesh, 2025-03-19:
From log,
```
Mar 19 11:49:34 raspberrypi kernel: [ 1503.741919] mmc1: card 0001 removed
Mar 19 11:49:34 raspberrypi kernel: [ 1503.792575] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
Mar 19 11:49:34 raspberrypi kernel: [ 1503.800235] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
Mar 19 11:49:34 raspberrypi kernel: [ 1503.803802] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
Mar 19 11:49:34 raspberrypi kernel: [ 1503.856185] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
Mar 19 11:49:34 raspberrypi kernel: [ 1503.865924] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
Mar 19 11:49:34 raspberrypi kernel: [ 1503.870407] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
Mar 19 11:49:34 raspberrypi kernel: [ 1503.928377] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
Mar 19 11:49:34 raspberrypi kernel: [ 1503.942712] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
Mar 19 11:49:34 raspberrypi kernel: [ 1503.948906] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
Mar 19 11:49:34 raspberrypi kernel: [ 1504.014071] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
Mar 19 11:49:35 raspberrypi kernel: [ 1505.123593] mmc1: new SDIO card at address 0001
```

=> SDIO enumerated by linux SDIO driver.

```
Mar 19 11:49:35 raspberrypi kernel: [ 1505.123914] esp32_sdio: esp_probe: ESP network device detected
```

=> ESP kernel module sdio driver getting probed.

```
Mar 19 11:49:35 raspberrypi kernel: [ 1505.124075] esp32_sdio: get_firmware_data: Rx Pre ====== 0
Mar 19 11:49:35 raspberrypi kernel: [ 1505.124084] esp32_sdio: get_firmware_data: Rx Pos ======  38
Mar 19 11:49:35 raspberrypi kernel: [ 1505.124127] esp32_sdio: get_firmware_data: Tx Pre ======  0
Mar 19 11:49:35 raspberrypi kernel: [ 1505.124133] esp32_sdio: get_firmware_data: Tx Pos ======  10
Mar 19 11:49:35 raspberrypi kernel: [ 1505.124549] esp32_sdio: probe of mmc1:0001:2 failed with error -22
Mar 19 14:09:44 raspberrypi kernel: [ 9914.176366] esp32_sdio: probe of mmc1:0001:1 failed with error -110
Mar 19 14:09:44 raspberrypi kernel: [ 9914.176451] esp32_sdio: probe of mmc1:0001:2 failed with error -110
Mar 19 14:09:44 raspberrypi kernel: [ 9914.300455] mmc1: card 0001 removed
```

=> `mmc1:0001:2 failed with error -22` is $${\color{green}expected}$$ as we don't use function 2.
But `mmc1:0001:1 failed with error -110` is a $${\color{red}problem}$$. This is typically seen when you have some SDIO level connection issues. Check your external pull-ups are in force. Somehow the pull-ups getting low while in operation?

```
Mar 19 14:09:45 raspberrypi kernel: [ 9915.596509] sdio_read_cis: 5 callbacks suppressed
Mar 19 14:09:45 raspberrypi kernel: [ 9915.596534] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
Mar 19 14:09:45 raspberrypi kernel: [ 9915.604432] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
Mar 19 14:09:45 raspberrypi kernel: [ 9915.608456] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
Mar 19 14:09:45 raspberrypi kernel: [ 9915.616234] mmc1: new SDIO card at address 0001
Mar 19 14:09:45 raspberrypi kernel: [ 9915.623400] esp32_sdio: esp_probe: ESP network device detected
Mar 19 14:09:45 raspberrypi kernel: [ 9915.623699] esp32_sdio: get_firmware_data: Rx Pre ====== 0
Mar 19 14:09:45 raspberrypi kernel: [ 9915.623762] esp32_sdio: get_firmware_data: Rx Pos ======  0
Mar 19 14:09:45 raspberrypi kernel: [ 9915.623825] esp32_sdio: get_firmware_data: Tx Pre ======  0
Mar 19 14:09:45 raspberrypi kernel: [ 9915.623837] esp32_sdio: get_firmware_data: Tx Pos ======  10
Mar 19 14:09:45 raspberrypi kernel: [ 9915.629641] esp32_sdio: probe of mmc1:0001:2 failed with error -22
```

=> Re-enumerated by Linux  (remove earlier and poll the sdio again and probe again)

```
Mar 19 14:09:46 raspberrypi kernel: [ 9916.404564] esp32_sdio: process_esp_bootup_event: Received ESP bootup event
Mar 19 14:09:46 raspberrypi kernel: [ 9916.404577] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 3
Mar 19 14:09:46 raspberrypi kernel: [ 9916.404584] esp32_sdio: esp_validate_chipset: Chipset=ESP32 ID=00 detected over SDIO
Mar 19 14:09:46 raspberrypi kernel: [ 9916.404591] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 0
Mar 19 14:09:46 raspberrypi kernel: [ 9916.404597] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 4
Mar 19 14:09:46 raspberrypi kernel: [ 9916.404602] esp32_sdio: process_event_esp_bootup: Unsupported tag=4 in bootup event
Mar 19 14:09:46 raspberrypi kernel: [ 9916.404608] esp32_sdio: process_event_esp_bootup: Bootup Event tag: 1
```
From `Unsupported tag=4`,  Why https://github.com/espressif/esp-hosted/blob/d5542de1992ee1aa30c0f84bc17526d8ed05213d/esp_hosted_ng/host/include/adapter.h#L115

is received? This is never set anywhere. $${\color{yello}Possibly\ your\ SDIO\ trace\ length\ is\ high\ or\ there\ is\ high interference\ in\ lines}$$.

One way to is to lower the frequency.
$${\color{yellow}What\ is\ the\ frequency\ used?}$$

```
sudo cat /sys/kernel/debug/mmc1/ios
```

Use lower frequency like 10 Mhz

https://github.com/espressif/esp-hosted/blob/d5542de1992ee1aa30c0f84bc17526d8ed05213d/esp_hosted_ng/host/main.c#L36

to 
```
static u32 clockspeed = 10;
```

**reload the driver** and check again
```
sudo cat /sys/kernel/debug/mmc1/ios
```

Q. $${\color{yellow}Are\ you\ using\ different\ SDIO\ timings\ at\ slave\ and\ host}$$
https://github.com/espressif/esp-hosted/blob/d5542de1992ee1aa30c0f84bc17526d8ed05213d/esp_hosted_ng/esp/esp_driver/network_adapter/main/sdio_slave_api.c#L125

VegardCheng, 2025-03-19:
修改完已经从25MHz变成10MHz了

 .timing             = SDIO_SLAVE_TIMING_NSEND_PSAMPLE,
这里没改过

还是没有wlan

mantriyogesh, 2025-03-19:
Do not change anything at slave.

> 还是没有wlan

Cannot unless your SDIO is stable and working.

Even your log 
```
Mar 18 13:54:46 raspberrypi kernel: [  179.562243] esp32_sdio: esp_handle_isr: CMD53 read/write error at 107
```

Confirms that $${\color{yellow}slave\ SDIO\ is\ not\ contactable}$$.
$${\color{orange}Check\ your\ inout\ power,\ hardware\ voltage\ levels\ and\ pull\-up\ levels}$$

Getting the base platform up is pre-requisite.
On your set up, SDIO is not stable. The problem is in the base hardware yet. Issue not reached yet to ESP-Hosted kernel module. ESP-Hosted or any other sdio driver cannot work unless the base linux base sdio driver is stable.

VegardCheng, 2025-09-01:
终于有时间继续弄下这个SDIO了
现在能识别网卡了，初始化成功后设备经常不断扫描，导致很难连接网络

连接成功后也经常断连

[image]

VegardCheng, 2025-09-01:
初始化成功后我手动复位下就好很多，可以扫描网络，然后就会自动连接，也比较稳定

mantriyogesh, 2025-09-01:
@VegardCheng, thank you for confirming that the driver is functioning as expected.

Could you please let us know if any remaining issues that you face currently?

Kindly attach the complete logs from the ESP and host (dmesg) during system boot-up, along with a description of the problem. These logs will help us identify the issue more quickly. Please attach text-based logs rather than images.