espressif/esp-hosted #576: insmod: ERROR: could not insert module esp32_spi.ko: Unknown error 517

HemQL, 2025-02-28:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Running rpi_init.sh from esp_hosted_ng/host with argument spi, the host should be setup to interface with my ESP32S3 devkit wired over SPI per the configuration guide.

### Actual behavior (suspected bug)

Running rpi_init.sh from esp_hosted_ng/host with argument spi, the make completes, albeit with warnings regarding jobserver not available and using -j1. However, insmod returns Unknown error 517 when trying to insert esp32_spi.ko.

### Error logs or terminal output

```shell
heming@heming:~/esp-hosted/esp_hosted_ng/host $ sudo bash rpi_init.sh spi
Building for spi protocol
rm: cannot remove 'spidev_disabler.dtbo': No such file or directory
rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/6.6.62+rpt-rpi-v7/build M=/home/heming/esp-hosted/esp_hosted_ng/host clean
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/linux-headers-6.6.62+rpt-rpi-v7'
make[1]: Leaving directory '/usr/src/linux-headers-6.6.62+rpt-rpi-v7'
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/6.6.62+rpt-rpi-v7/build M=/home/heming/esp-hosted/esp_hosted_ng/host modules
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[1]: Entering directory '/usr/src/linux-headers-6.6.62+rpt-rpi-v7'
  CC [M]  /home/heming/esp-hosted/esp_hosted_ng/host/spi/esp_spi.o
  CC [M]  /home/heming/esp-hosted/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/heming/esp-hosted/esp_hosted_ng/host/main.o
  CC [M]  /home/heming/esp-hosted/esp_hosted_ng/host/esp_cmd.o
  CC [M]  /home/heming/esp-hosted/esp_hosted_ng/host/esp_utils.o
  CC [M]  /home/heming/esp-hosted/esp_hosted_ng/host/esp_cfg80211.o
  CC [M]  /home/heming/esp-hosted/esp_hosted_ng/host/esp_stats.o
  CC [M]  /home/heming/esp-hosted/esp_hosted_ng/host/esp_debugfs.o
  CC [M]  /home/heming/esp-hosted/esp_hosted_ng/host/esp_log.o
  LD [M]  /home/heming/esp-hosted/esp_hosted_ng/host/esp32_spi.o
  MODPOST /home/heming/esp-hosted/esp_hosted_ng/host/Module.symvers
  CC [M]  /home/heming/esp-hosted/esp_hosted_ng/host/esp32_spi.mod.o
  LD [M]  /home/heming/esp-hosted/esp_hosted_ng/host/esp32_spi.ko
make[1]: Leaving directory '/usr/src/linux-headers-6.6.62+rpt-rpi-v7'
insmod: ERROR: could not insert module esp32_spi.ko: Unknown error 517
```

### Steps to reproduce the behavior

Followed sections 1.1 and 2.1 of the esp_hosted_ng setup guide. https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md

### Project release version

latest

### System architecture

ARM 32-bit (Raspberry Pi 32-bit)

### Operating system

Linux

### Operating system version

Raspberry Pi OS Lite

### Shell

Bash

### Additional context

Hardware is a Raspberry Pi Zero 2W. Also tried changing ARCH to armv7l, no difference in result.

mantriyogesh, 2025-02-28:
Hello @HemQL ,

I understand the pain.
Can you please attach the logs in textual format at both side, ESP and raspberry pi?

ESP: idf.py flash monitor log from start of flashing
Host: dmesg log from system bootup 

Also please capture a photo of your setup to understand the physical connections. This would help us understand the setup you have used.

HemQL, 2025-02-28:
[esp_hosted_log_dmesg.txt](https://github.com/user-attachments/files/19029787/esp_hosted_log_dmesg.txt)
[esp_hosted_log_idfpy.txt](https://github.com/user-attachments/files/19029786/esp_hosted_log_idfpy.txt)

[image]

mantriyogesh, 2025-02-28:
How did you come up with 517 gpio?

HemQL, 2025-02-28:
Thank you, I did miss that. However, the error persists. I do see this on my gpio after trying resetpin=518 and resetpin=521.

heming@heming:~/esp-hosted/esp_hosted_ng/host $ cat < /sys/kernel/debug/gpio
gpiochip0: GPIOs 512-565, parent: platform/3f200000.gpio, pinctrl-bcm2835:
 gpio-512 (ID_SDA              )
 gpio-513 (ID_SCL              )
 gpio-514 (GPIO2               )
 gpio-515 (GPIO3               )
 gpio-516 (GPIO4               )
 gpio-517 (GPIO5               )
 gpio-518 (GPIO6               |sysfs               ) in  hi
 gpio-519 (GPIO7               |spi0 CS1            ) out hi ACTIVE LOW
 gpio-520 (GPIO8               |spi0 CS0            ) out hi ACTIVE LOW
 gpio-521 (GPIO9               |sysfs               ) in  lo
 gpio-522 (GPIO10              )
 gpio-523 (GPIO11              )
 gpio-524 (GPIO12              )
 gpio-525 (GPIO13              )
 gpio-526 (GPIO14              )
 gpio-527 (GPIO15              )
 gpio-528 (GPIO16              )
 gpio-529 (GPIO17              )
 gpio-530 (GPIO18              )
 gpio-531 (GPIO19              )
 gpio-532 (GPIO20              )
 gpio-533 (GPIO21              )
 gpio-534 (GPIO22              )
 gpio-535 (GPIO23              )
 gpio-536 (GPIO24              )
 gpio-537 (GPIO25              )
 gpio-538 (GPIO26              )
 gpio-539 (GPIO27              )
 gpio-540 (HDMI_HPD_N          |hpd                 ) in  lo ACTIVE LOW
 gpio-541 (STATUS_LED_N        |ACT                 ) out lo ACTIVE LOW
 gpio-542 (CTS0                )
 gpio-543 (RTS0                )
 gpio-544 (TXD0                )
 gpio-545 (RXD0                )
 gpio-546 (SD1_CLK             )
 gpio-547 (SD1_CMD             )
 gpio-548 (SD1_DATA0           )
 gpio-549 (SD1_DATA1           )
 gpio-550 (SD1_DATA2           )
 gpio-551 (SD1_DATA3           )
 gpio-552 (CAM_GPIO1           |cam1_regulator      ) out lo
 gpio-553 (WL_ON               )
 gpio-554 (BT_ON               )
 gpio-555 (WIFI_CLK            )
 gpio-556 (SDA0                )
 gpio-557 (SCL0                )
 gpio-558 (SMPS_SCL            )
 gpio-559 (SMPS_SDA            )
 gpio-560 (SD_CLK_R            )
 gpio-561 (SD_CMD_R            )
 gpio-562 (SD_DATA0_R          )
 gpio-563 (SD_DATA1_R          )
 gpio-564 (SD_DATA2_R          )
 gpio-565 (SD_DATA3_R          )

mantriyogesh, 2025-02-28:
Use exact pin numbers and exact gpios to be connected, if you are unsure. Otherwise you would be using some gpio, which is unconnected towards the esp.

HemQL, 2025-02-28:
I just realized the error is failure to obtain the handshake pin, not the reset pin. How do I define a pin for that?

mantriyogesh, 2025-02-28:
Check the porting guide, and the link I shared. 
Depending upon the raspberry pi os used, you should stick to the pins in software and connect the gpio correctly. 

If you wish to change on your own then ensure you do it correctly, else you would trap into issues liked this.

HemQL, 2025-02-28:
The pin definitions in the header file were changed, and the module inserted properly. 

Thank you for the help!

closed 2025-02-28 by HemQL