espressif/esp-hosted #577: Performance Issue at 50MHz and Initialization Failure at 75MHz in SPI Mode

nickfox-taterli, 2025-03-01:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

## Description

I am testing the SPI mode in my project hosted at [rp2350-esp-hosted](https://github.com/nickfox-taterli/rp2350-esp-hosted). The current performance at 50MHz is significantly lower than expected, and initialization fails completely at 75MHz.

According to the `esp_hosted_fg` tests, the performance should reach around 17Mbps. However, my implementation using `iperf2` and the `lwip` protocol stack shows much lower throughput (~5Mbps). Additionally, the ESP32-C3 fails to initialize at 75MHz, and I am unsure of the root cause.

## Details

- **Performance at 50MHz**: The SPI mode works at 50MHz, but the performance is much lower than expected.
- **Initialization Failure at 75MHz**: The ESP32-C3 cannot be initialized at 75MHz.
- **Environment**: 
  - The ESP32-C3-WROOM is placed close to the router to ensure a strong signal.
  - Using the `esp_as_mcu_host` branch, which is stable except for the performance and initialization issues.
- **Protocol Stack**: Implemented `iperf2` and `lwip` on the host side.

## Expected Behavior

- At 50MHz, the performance should be closer to the 17Mbps benchmark observed in `esp_hosted_fg`.
- At 75MHz, the ESP32-C3 should initialize successfully.

## Steps to Reproduce

1. Clone the repository: `https://github.com/nickfox-taterli/rp2350-esp-hosted`.
2. Test performance at **ANY FREQ** using `iperf2` and `lwip`.

## Additional Information

- **Hardware Setup**: ESP32-C3 module placed close to the router.
- **Performance Metrics**: Measured using `iperf2`.

## Questions

1. Are there any known performance limitations in SPI mode at 50MHz for the `esp_as_mcu_host` branch?
2. What could be causing the initialization failure at 75MHz? Are there specific timing or configuration requirements for higher frequencies?
3. Are there any recommended optimizations to improve SPI mode performance at 50MHz?

Any insights or suggestions would be greatly appreciated. Thank you!

mantriyogesh, 2025-03-01:
Do not use this branch. Use new MCU dedicated repo.

This high frequency at slave is not supported. Check full documentation at https://github.com/espressif/esp-hosted-mcu

Note the -mcu in the link. Please check the full documentation in this repo. Limit max SPI clock to 40MHZ. suggest to use 10mhz first, get it working and then step wise increase. 

Higher freq >20-30Mhz) needs dedicated PCB. Jumper wires become noisy with higher freq.

If you want to increase the throughout, you can think of using spi half duplex QSPI. Please note the frame structure (already implemented QSPI frame for ESP as host and ESP as slave).

Please feel free to close this issue and raise if any issue occur on the repo mentioned.

mantriyogesh, 2025-03-01:
We also have ported non ESP as host for full duplex spi using stm32 MCU. Let us know if you are interested in this.

nickfox-taterli, 2025-03-01:
> We also have ported non ESP as host for full duplex spi using stm32 MCU. Let us know if you are interested in this.

Can you provide a sample project on STM32? I soldered it with jumpers and there is about 10% chance of a checksum error at 50MHz. It seems that the frequency is too high, I might consider using QSPI.

mantriyogesh, 2025-03-01:
STM32 SPI Full Duplex port (little older commit) is available to download at https://we.tl/t-S3IFlb8qdt. It would expire in two days.

We will rebase and merge later, just send you for your reference. Please note, we have yet not ported other transports for STM32. also this code of stm32 contains lwip as submodule.  We tried to make this repo independednt of esp-idf for host side adaption.

Note:
1. Use spi full duplex only
2. Ensure slave and host must be flashed flashed from same commit.
3. Code is send as good gesture, any issue found we can try to resolve with lower priority than that of the public repo issues, as the environment needs full explicit change.
4. Repo has documentation changes which you can refer. although your mcu is not exactly stm32, you might need some changes around build system.

Also please note, on public repo at https://github.com/espressif/esp-hosted-mcu, We have sdio, spi full-duplex, Spi half duplex (dual and quad spi), uart transport options available. Tested with both, esp as host and slave. Public repo also has port layer which can be hooked for non esp mcus (in similar way we did for stm32 in above case)

closed 2025-03-04 by nickfox-taterli